Yurik72/ESPHap #1: Error Compiling

seamaster101, 2020-03-28:
hi, I was able to successfully use ESPHap with ESP32, but I have problem compiling Sonoff examples for esp8266. I did select esp8266 under the board selection. the error is that it does not see esp_system.h . as far as I know it comes as a core with esp8266 boards installation package on Arduino IDE. what could be going wrong?
*****************************
/Users/XXXX/Documents/Arduino/libraries/ESPHap-master/aa.c -o /dev/null
In file included from /Users/XXXX/Documents/Arduino/libraries/wolfSSL/wolfssl/wolfcrypt/settings.h:213:0,
                 from /Users/XXXX/Documents/Arduino/libraries/wolfSSL/wolfssl.h:2,
                 from /Users/XXXX/Documents/Arduino/libraries/ESPHap-master/aa.c:2:
/Users/Jordans/Documents/Arduino/libraries/wolfSSL/user_settings.h:38:24: fatal error: esp_system.h: No such file or directory
 #include <esp_system.h>
                        ^
compilation terminated. 
*****************************

Yurik72, 2020-03-29:
Hi , first be sure that you are using latest version of library, secondary please have a look instruction especially for esp8266 #3

From the attached archive https://github.com/Yurik72/ESPHap/tree/master/wolfssl unpack wolfSSL_3_13_0.rar into the same folder of the Arduino libraries. Better name it "wolfsll"

this library works with esp8266 only with this patched version of wolfssl, as well this patched version works well with esp32
If you are following that , compilation must be without error , please confirm

seamaster101, 2020-03-29:
I followed the everything you recommended:
Downloaded the wolfSSL_3_13_0.rar
Deleted previous wolfs from Arduino/library folder. Unpacked the wolfSSL_3_13_0.rar to Arduino/library folder under the name wolfssl. Then I opened the raw settings.h on GitHub, and copied ant pasted the content overwriting the content of the settings.h in the wolfssl library . Did the same with user_settings.h.
Then started Arduino and tried to compile and I get error again

Yurik72, 2020-03-30:
I will change a documentation. wolfSSL_3_13_0.rar  contains everything correctly
DO NOT replace any file including setting.h

seamaster101, 2020-03-30:
Hi,
Thank you for responding to my issue.
So still no success! 
I noticed that your library shows under examples -> incompatible 
Why that could be?
This time I removed the wolfssl and reinstalled from the version that you recommended. 
Then I removed and redownloaded your library and reinstalled. 
Still the examples were under Examples->incompatible.
Open the EspHapLed826 and I tried to compile as it is.
Got this issue:
error: 'SPIFFS' was not declared in this scope
      if (!SPIFFS.begin()) {…

Noticed that there is no declaration for FS.h under the esp8266 declarations and I added 
There:

#ifdef ESP32
#include <SPIFFS.h>
#endif
#include <ESP8266WiFi.h>
//#include <WiFi.h>
#include <ESP8266mDNS.h>
#include <FS.h>.  //this is the one that I added
#include "coredecls.h"

I tried to compile again this time did not complained about SPIFFS but there was another error:

/Users/Jordans/Documents/Arduino/libraries/ESPHap-master/homeintegration.c:88:1: error: unknown type name 'os_timer_t'
 LOCAL os_timer_t identtimer;
 ^

Why did I have to add #include <FS.h>
And why is this next error happening?

Cheers!

Yurik72, 2020-03-31:
Ok, there are 3 issues
1. Incompatible, fixed by adding proper architechture into the library.properties file. Any how this is not any impact on compilation

2.  <FS.H>  Probably you are right but in my enviropment it's compiled without error

3, /Users/Jordans/Documents/Arduino/libraries/ESPHap-master/homeintegration.c:88:1: error: unknown type name 'os_timer_t'
 LOCAL os_timer_t identtimer;
There is NO error on my enviropment.
To be able to resolve this, please let me know 
- your arduino version
- your version of ESP8266 board package (Arduino - >Tools-> Boards -> Board manager)

In my case it's
Arduino 1.8.12
ESP8266  version 2.6.3

seamaster101, 2020-03-31:
Hi Yurik72,
Problem is finally fixed! You can close the issue. Thank you for your help.
After reading carefully all the documentation you wrote, I had my doubts that something is wrong in my environment. 
I uninstalled ESP8266 core from Arduino IDE, restarted Arduino IDE, and then installed ESP8266 core it again. Did the same with esp32 core. 
Now it compiles correct! 
Thank you for your help!
By the way is it possible to send to Homekit  DHT11 sensor if I add the code for that? Where do I change from control to just reporting temperature and humidity?
Can you help with this? Should I open somehow feature request?
Cheers!
J

Yurik72, 2020-04-01:
Hi I will close an issue, 
Your request for DHT11:
Library supports itself this functionality  and I found that some people already did that
https://github.com/SamAkada/ESPHap/tree/master/examples/ENV_HAP

I will appreciate if you will do good example and put them into my examples folder

closed 2020-04-01 by Yurik72