Yurik72/ESPHap #2: new code for temperature/humidity with DHT11

seamaster101, 2020-04-02:
Hi,
I looked at the code for DHT12 (I2C) that you recommend 
but I'm using DHT11 and I had to modify it. 
one of the disadvantages of DHT11 is that the SW ie reading from the detain and the sensor is very slow.
I converted the code but experience problems with reading the sensor when combined with the rest of the code. I would presume it is because of interrupts that interfere with the reading of the sensor. Do you think there could be something  done about that?

the other issue is that I get only temperature  but im defining 2 services, temperature and humidity
is my code doing it correct?
here is the entire code:
 
`
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
//#include <FS.h>
//#include <SPIFFS.h>
#define STATIC_IP_ADDR
///
#include "DHT.h"
#ifdef STATIC_IP_ADDR
IPAddress staticIP(192,168,0,91);
IPAddress gateway(192,168,0,1);
IPAddress subnet(255,255,255,0);
#endif

#include "coredecls.h"
#define DHTTYPE DHT11
#define DHTPIN  2
DHT dht(DHTPIN, DHTTYPE, 12); // 11 works fine for ESP8266

const char* ssid = "SHAW-62CD10";
const char* password = "251169025293";
float hum;
float temp;

unsigned long previousMillis = 0;        // will store last temp was read
const long interval = 4000;   

extern "C"{
#include "homeintegration.h"
}
#include "homekitintegrationcpp.h"

String pair_file_name="/pair.dat";

//**********************
void init_hap_storage(){

  Serial.print("init_hap_storage");
    
  File fsDAT=SPIFFS.open(pair_file_name, "r");
 if(!fsDAT){
   Serial.println("Failed to read pair.dat");
   return;
 }
  int size=hap_get_storage_size_ex();
  char* buf=new char[size];
  memset(buf,0xff,size);
  int readed=fsDAT.readBytes(buf,size);
  Serial.print("Readed bytes ->");
  Serial.println(readed);
  hap_init_storage_ex(buf,size);
  fsDAT.close();
  delete []buf;

}
//*****************

//****************
void storage_changed(char * szstorage,int bufsize){

  SPIFFS.remove(pair_file_name);
  File fsDAT=SPIFFS.open(pair_file_name, "w+");
  if(!fsDAT){
    Serial.println("Failed to open pair.dat");
    return;
  }
  fsDAT.write((uint8_t*)szstorage, bufsize);

  fsDAT.close();
}
//********************
homekit_service_t* hapservice={0};
homekit_service_t* temperature;
homekit_service_t* humidity;

//////////////////////////////////
void setup() {
  disable_extra4k_at_link_time();
  Serial.begin(115200);
    delay(3000);
Format();

   
     if (!SPIFFS.begin()) {
      Serial.print("SPIFFS Mount failed");
     }

    Serial.print("Attempting connection to: ");
    Serial.println(ssid);

    WiFi.mode(WIFI_STA);
    WiFi.begin((char*)ssid, (char*)password);

#ifdef STATIC_IP_ADDR
    WiFi.config(staticIP, gateway, subnet);
#endif
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
        WiFi.begin(ssid, password);
    }
    Serial.println("");
    Serial.println("WiFi connected!");
 #ifdef STATIC_IP_ADDR
    Serial.print("Static IP address: ");
#else
Serial.print("Dinamic IP address: ");
#endif
    
    Serial.println(WiFi.localIP());

/// now will setup homekit device

    //this is for custom storaage usage
    // In given example we are using \pair.dat   file in our spiffs system
    //see implementation below
    Serial.print("Free heap: ");
    Serial.println(system_get_free_heap_size());

  
    init_hap_storage();
  
    set_callback_storage_change(storage_changed);

    /// We will use for this example only one accessory (possible to use a several on the same esp)
    //Our accessory type is light bulb , apple interface will proper show that
    hap_setbase_accessorytype(homekit_accessory_category_thermostat);
    /// init base properties
    hap_initbase_accessory_service("Environment Sensor","SeaMaster","258521521","DHT11_ESP8266","1.0");
   //we will add only one light bulb service and keep pointer for nest using
    
    
    temperature = hap_add_temperature_service("Temperature");
    humidity =   hap_add_humidity_service("Humidity");
     
   //and finally init HAP
   
    hap_init_homekit_server();   
  
}

////////////////////////////////

void loop() {

   hap_homekit_loop();
delay(10);    
    unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    // save the last time you read the sensor
    previousMillis = currentMillis;

    // Reading temperature for humidity takes about 250 milliseconds!
    // Sensor readings may also be up to 2 seconds 'old' (it's a very slow sensor)
    hum = dht.readHumidity();          // Read humidity (percent)
    temp = dht.readTemperature(false);     // Read temperature as Fahrenheit
   Serial.println(temp); // Check if any reads failed and exit early (to try again).
    if (isnan(hum) || isnan(temp)) {
      Serial.println("Failed to read from DHT sensor!");
      return;
    }
  }
     homekit_characteristic_t * ch= homekit_service_characteristic_by_type(temperature,  HOMEKIT_CHARACTERISTIC_CURRENT_TEMPERATURE);
     ch->value.float_value=temp;
     homekit_characteristic_notify(ch,ch->value);
     delay(10);
    homekit_characteristic_t * ch2= homekit_service_characteristic_by_type(humidity, HOMEKIT_CHARACTERISTIC_CURRENT_RELATIVE_HUMIDITY);
     ch2->value.float_value=hum;
     homekit_characteristic_notify(ch2,ch2->value);
  
  delay(10);
}

void Format(){
  // Next lines have to be done ONLY ONCE!!!!!When SPIFFS is formatted ONCE you can comment these lines out!!
  Serial.println("Please wait 30 secs for SPIFFS to be formatted");
  SPIFFS.format();
  Serial.println("Spiffs formatted");
}`

Yurik72, 2020-04-03:
Hi , I can check but could you please describe exact of the problem
1. Pairing OK ?
2. Device is visible on Apple ?
3. Data is available on Apple ?

Give me a hint please

seamaster101, 2020-04-03:
Hi,
1. paring is OK most of the times
2. device is visible on iPhone
3. device shows only temperature on the main icon. Once I manage to do it so it was showing temporary and humidity as it should be, but then I lost it. Maybe I changed something in the code but I don’t know exactly what caused it.  
4. when you open the device on the I phone it indicates that has temperature and humidity
5. the temperature is always 0. The humidity is stuck on “updating ...”

It is kind of hard to explain. I can send you some screenshots, but don’t know how to reach you with pictures 
Cheers!

> On Apr 2, 2020, at 23:59, Yurik72 <notifications@github.com> wrote:
> 

Hi , I can check but could you please describe exact of the problem
1. Pairing OK ?
2. Device is visible on Apple ?
3. Data is available on Apple ?

Give me a hint please

Yurik72, 2020-04-04:
Ok, seems I will be able to help, 
Could you please check first if

> homekit_service_t* temperature; 
> > homekit_service_t* humidity; 

Variables are not NULL, when you sending data to Apple, then I will give you a hint how to create two devices/ icons on Apple

Yurik72, 2020-04-06:
Step by step,
Zero values to Apple:
In your sketch at the begin please define
#define HOMEKIT_SHORT_APPLE_UUIDS

EspHap library for ESP8266 uses short UUIDs, but sketch still used long, therefore code 
 HOMEKIT_CHARACTERISTIC_CURRENT_TEMPERATURE);
 ch->value.float_value=temp;
 homekit_characteristic_notify(ch,ch->value);
is not works and characteristic not found

Yurik72, 2020-04-06:
Second , how to create two icons
first device/icons is a temperature, code 
hap_setbase_accessorytype(homekit_accessory_category_thermostat);
 hap_initbase_accessory_service("Environment Sensor","SeaMaster","258521521","DHT11_ESP8266","1.0");
 temperature = hap_add_temperature_service("Temperature");

Than you need add second device/icons and code

humidity=hap_add_hum_as_accessory(homekit_accessory_category_thermostat,"Humidity");

//This will add second device/icons only with humidity
The rest code is OK.

P.S. You should use the latest version of ESP hap

closed 2020-04-06 by Yurik72

seamaster101, 2020-04-06:
@Yurik72
Thank for responding. 
First:
I think I was using the latest version of ESP hap. D0 you see an indication in my code that I’m not?

Second:
Code
**********
    hap_setbase_accessorytype(homekit_accessory_category_thermostat);
    hap_initbase_accessory_service("Environment Sensor","SeaMaster","258521521","DHT11_ESP8266","1.0");
    temperature = hap_add_temperature_service("Temperature");
    humidity =   hap_add_humidity_service("Humidity");  
// I just added the following line as per your email
    humidity=hap_add_hum_as_accessory(homekit_accessory_category_thermostat,"Humidity");
*********    
And I get the following error:
EspHapENV8266:142:86: error: 'hap_add_hum_as_accessory' was not declared in this scope
    humidity=hap_add_hum_as_accessory(homekit_accessory_category_thermostat,"Humidity");
I tried with or without the line
humidity =   hap_add_humidity_service("Humidity");  
And still got the same error. Im not sure what I’m missing here

Third and maybe the most important one:
If I set the temp and humidity to numbers like that:
   temp = 22.2;     // Read temperature as Fahrenheit
   hum = 77.7;        // Read humidity (percent)
I can read on my phone the value
If I set it like that:
 hum = dht.readHumidity();          // Read humidity (percent)
 temp = dht.readTemperature(false);     // Read temperature as Fahrenheit
 It reads it correct only for few seconds and then start to return nan or unreal values.
In a separate sketch, without the ESP hap, this works just fine. 
As the DHT11 is a slow sensor it appears that the reading gets interrupted when using ESP hap in conjunction with it.
Is the a way to pause ESP hap interrupts or processes while reading the sensor?

seamaster101, 2020-04-07:
Please ignore "Third and maybe... "
this one was resolved by using library with interrupt for DHT11. Now reeds the temperature and humidity OK, but only have the setup screen for humidity, and does not show humidity Icon. this is back too the second issue. 
please provide example if you could for one service with 2 accessories.

there is also another thing that I don't understand:
if I enable 
format();
it logs ok on the network, but I need to reconnect to apple HomeKit on every restart as it looses the configuration data.
if I I disable the
//format();
my esp8266 will not connect to the network and it is serial is  outputing only "................."  every time I restart. I'm not sure how I suppose to use the format(); function

Yurik72, 2020-04-07:
Two icons: I have provided example yesterday
There is a new function in the library
humidity=hap_add_hum_as_accessory(homekit_accessory_category_thermostat,"Humidity");

Therefore you need to get latest version to resolve compilation error

About format, in My original example there is no function format. If you formatting SPIFFS every start , you are loosing pairing data. In my instruction i mentioned about format only when pairing not sucess.

seamaster101, 2020-04-07:
[
[image]
[image]
[image]
](url)

seamaster101, 2020-04-07:
Better but still not properly displayed:
1st image above is I get Humidity, but just the setup screen for it.

2nd image I got the temperature properly but by itself - no humidity icon and symbol

3rd image is accessory from another sketch that used HomeBridge and it is displaying temperature and humidity on one accessory and that is what I was hoping to accomplish but without the need of Homebridge. it appears we are close, but not quite there yet...

4th image is from mu "Home" screen. note that the "Humidity" icon shows always "updating".
Serial Monitor has constant stream rushing the that says something like:
`

>>> HomeKit: [Client 1073692820] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073693444] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073692820] Got characteristic 2.10 change event
>>> HomeKit: [Client 1073693444] Got characteristic 2.10 change event
>>> HomeKit: [Client 1073692820] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073693444] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073692820] Got characteristic 2.10 change event
`
etc, etc.
that continues for a while and then I get this repeating:
`!!! HomeKit: [Client 1073692820] error_in_write_data is true, abort write data
!!! HomeKit: [Client 1073692820] error_in_write_data is true, abort write data
!!! HomeKit: [Client 1073692820] error_in_write_data is true, abort write data
>>> HomeKit: [Client 1073692820] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073693444] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073692820] Got characteristic 2.10 change event
>>> HomeKit: [Client 1073693444] Got characteristic 2.10 change event
`

On the picture below notice that Default Room Humidity always show "Updating..."
maybe that has something to do with this write data error?
[
[image]
](url)

the serial monitor stream was so fast that overloaded the application and I had to forcibly quit it and reload it.

Yurik72, 2020-04-07:
About icon
It depends from the accessory type,
I have suggested - homekit_accessory_category_thermostat
humidity=hap_add_hum_as_accessory(homekit_accessory_category_thermostat,"Humidity");

Probably you need
humidity=hap_add_hum_as_accessory(homekit_accessory_category_humidifier ,"Humidity");

As well you can edit icon on Apple to define category
Full enumerable is 
typedef enum {
		homekit_accessory_category_other = 1,
		homekit_accessory_category_bridge = 2,
		homekit_accessory_category_fan = 3,
		homekit_accessory_category_garage = 4,
		homekit_accessory_category_lightbulb = 5,
		homekit_accessory_category_door_lock = 6,
		homekit_accessory_category_outlet = 7,
		homekit_accessory_category_switch = 8,
		homekit_accessory_category_thermostat = 9,
		homekit_accessory_category_sensor = 10,
		homekit_accessory_category_security_system = 11,
		homekit_accessory_category_door = 12,
		homekit_accessory_category_window = 13,
		homekit_accessory_category_window_covering = 14,
		homekit_accessory_category_programmable_switch = 15,
		homekit_accessory_category_range_extender = 16,
		homekit_accessory_category_ip_camera = 17,
		homekit_accessory_category_video_door_bell = 18,
		homekit_accessory_category_air_purifier = 19,
		homekit_accessory_category_heater = 20,
		homekit_accessory_category_air_conditioner = 21,
		homekit_accessory_category_humidifier = 22,
		homekit_accessory_category_dehumidifier = 23,
		homekit_accessory_category_apple_tv = 24,
		homekit_accessory_category_speaker = 26,
		homekit_accessory_category_airport = 27,
		homekit_accessory_category_sprinkler = 28,
		homekit_accessory_category_faucet = 29,
		homekit_accessory_category_shower_head = 30,
		homekit_accessory_category_television = 31,
		homekit_accessory_category_target_controller = 32,
	} homekit_accessory_category_t;

"Updating.." means apple did  not receive characheristic, i need a full logs to help
In my case i have tested Apple within a few devices without problem. Not Temperthure + Humidity combination, but definetelly question is not about combination, something else If you will fall again, probably to help i need your full sketch ,

seamaster101, 2020-04-08:
Thank you for following up!
here is my code:
`
//ESP8266 with DHT11  

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
//#include <FS.h>
//#include <SPIFFS.h>
#define STATIC_IP_ADDR

///---DHT11-----///
#include "PietteTech_DHT.h"  // Uncommend if building using CLI
#define DHTTYPE  DHT11       // Sensor type DHT11/21/22/AM2301/AM2302
#define DHTPIN   2           // Digital pin for communications
#define PIN   0           // Digital pin for re-format

//declaration
void dht_wrapper(); // must be declared before the lib initialization
///---DHT11 stop-----///

#ifdef STATIC_IP_ADDR
IPAddress staticIP(192,168,0,91);
IPAddress gateway(192,168,0,1);
IPAddress subnet(255,255,255,0);
#endif
#define HOMEKIT_SHORT_APPLE_UUIDS
#include "coredecls.h"

const char* ssid = "XXXXXXXX";
const char* password = "XXXXXXXX";
float hum;
float temp;

const long userdelay = 3000;            //delay for user input when tarting to format the SPIFFS
unsigned long previousMillis = 0; 
unsigned long prevMillis = 0;
const long interval = 15000;  
unsigned long currentMillis = 0; 
unsigned long currMillis = 0; 

extern "C"{
#include "homeintegration.h"
}
#include "homekitintegrationcpp.h"

String pair_file_name="/pair.dat";

///---DHT11-----///
// Lib instantiate
PietteTech_DHT DHT(DHTPIN, DHTTYPE, dht_wrapper);
int n;      // counter
///---DHT11 stop-----///

//**********************
void init_hap_storage(){
  Serial.print("init_hap_storage");
  File fsDAT=SPIFFS.open(pair_file_name, "r");
  if(!fsDAT){
   Serial.println("Failed to read pair.dat");
  return;
}
  int size=hap_get_storage_size_ex();
  char* buf=new char[size];
  memset(buf,0xff,size);
  int readed=fsDAT.readBytes(buf,size);
  Serial.print("Readed bytes ->");
  Serial.println(readed);
  hap_init_storage_ex(buf,size);
  fsDAT.close();
[... 156 lines cut ...]
  
  hap_homekit_loop(); //what does this exactly do?????
  delay(10);
 
  homekit_characteristic_t * ch1= homekit_service_characteristic_by_type(temperature, HOMEKIT_CHARACTERISTIC_CURRENT_TEMPERATURE);
  ch1->value.float_value=temp;
  homekit_characteristic_notify(ch1,ch1->value);
  delay(10);

  homekit_characteristic_t * ch2= homekit_service_characteristic_by_type(humidity, HOMEKIT_CHARACTERISTIC_CURRENT_RELATIVE_HUMIDITY);
  ch2->value.float_value=hum;
  homekit_characteristic_notify(ch2,ch2->value);
  delay(10);
 
///--- end of the main loop ---///
}

///--- Formating function ---///
void Format(){
  // Next lines have to be done ONLY ONCE!!!!!When SPIFFS is formatted ONCE you can comment these lines out!!
  Serial.println("Please wait 30 secs for SPIFFS to be formatted");
  SPIFFS.format();
  Serial.println("Spiffs formatted");
}
`

seamaster101, 2020-04-08:
`
`
//ESP8266 with DHT11  

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
//#include <FS.h>
//#include <SPIFFS.h>
#define STATIC_IP_ADDR

///---DHT11-----///
#include "PietteTech_DHT.h"  // Uncommend if building using CLI
#define DHTTYPE  DHT11       // Sensor type DHT11/21/22/AM2301/AM2302
#define DHTPIN   2           // Digital pin for communications
#define PIN   0           // Digital pin for re-format

//declaration
void dht_wrapper(); // must be declared before the lib initialization
///---DHT11 stop-----///

#ifdef STATIC_IP_ADDR
IPAddress staticIP(192,168,0,91);
IPAddress gateway(192,168,0,1);
IPAddress subnet(255,255,255,0);
#endif
#define HOMEKIT_SHORT_APPLE_UUIDS
#include "coredecls.h"

const char* ssid = "XXXXXXXX";
const char* password = "XXXXXXXX";
float hum;
float temp;

const long userdelay = 3000;            //delay for user input when tarting to format the SPIFFS
unsigned long previousMillis = 0; 
unsigned long prevMillis = 0;
const long interval = 15000;  
unsigned long currentMillis = 0; 
unsigned long currMillis = 0; 

extern "C"{
#include "homeintegration.h"
}
#include "homekitintegrationcpp.h"

String pair_file_name="/pair.dat";

///---DHT11-----///
// Lib instantiate
PietteTech_DHT DHT(DHTPIN, DHTTYPE, dht_wrapper);
int n;      // counter
///---DHT11 stop-----///

//**********************
void init_hap_storage(){
  Serial.print("init_hap_storage");
  File fsDAT=SPIFFS.open(pair_file_name, "r");
  if(!fsDAT){
   Serial.println("Failed to read pair.dat");
  return;
}
  int size=hap_get_storage_size_ex();
  char* buf=new char[size];
  memset(buf,0xff,size);
  int readed=fsDAT.readBytes(buf,size);
  Serial.print("Readed bytes ->");
  Serial.println(readed);
  hap_init_storage_ex(buf,size);
  fsDAT.close();
  delete []buf;
[... 156 lines cut ...]
  hap_homekit_loop(); //what does this exactly do?????
  delay(10);
 
  homekit_characteristic_t * ch1= homekit_service_characteristic_by_type(temperature, HOMEKIT_CHARACTERISTIC_CURRENT_TEMPERATURE);
  ch1->value.float_value=temp;
  homekit_characteristic_notify(ch1,ch1->value);
  delay(10);

  homekit_characteristic_t * ch2= homekit_service_characteristic_by_type(humidity, HOMEKIT_CHARACTERISTIC_CURRENT_RELATIVE_HUMIDITY);
  ch2->value.float_value=hum;
  homekit_characteristic_notify(ch2,ch2->value);
  delay(10);
 
///--- end of the main loop ---///
}

///--- Formating function ---///
void Format(){
  // Next lines have to be done ONLY ONCE!!!!!When SPIFFS is formatted ONCE you can comment these lines out!!
  Serial.println("Please wait 30 secs for SPIFFS to be formatted");
  SPIFFS.format();
  Serial.println("Spiffs formatted");
}
`
`

seamaster101, 2020-04-09:
Hi,
Any luck looking through my code to see why displaying humidity is not correct? 

After playing around for a while, I could not make it show properly even if it is the only service/accessory in the sketch and even if I hardcoded the hum var to let's say 55% . It always shows "Updating..." even when it reads the correct  data from the sensor.  yet it shows the proper number in the setup screen. Could it be a problem with this specific CHARACTERISTIC somewhere in the library, or I'm dong it wrong all together?

oxbown, 2020-04-17:
#7  I think I´ve got the bug that makes humidity sensors showing the incorrect icon and "updating" status...

Yurik72, 2020-04-17:
Yes, I found a mistake, #7 one of the problem, 
Second problem:  primary property must be set to true for Humidity and Temperature sensors

And finally , if you made pairing and after that added new accessory in the code. Better to clear all pairing again..

I have tested my device with BME280 and it shows temp and humidity perfectly