Yurik72/ESPHap #3: Add fan control.

Jeroen207, 2020-04-03:
I really admire the work! How can you add a fan to the accessory?

Yurik72, 2020-04-04:
You can find PWM control on light bulb Sonoff, library allows to do that, Fan is similar and controlled by PWM , you want exact examples ? or you want help how to control fan by esp ?

Jeroen207, 2020-04-04:
I figured it out! I added this to the homeintegration: 

homekit_service_t* hap_add_fan_service(const char* szname,hap_callback cb,void* context){

	homekit_service_t*service=NEW_HOMEKIT_SERVICE(FAN2, .primary = true, .characteristics=(homekit_characteristic_t*[]) {
	            NEW_HOMEKIT_CHARACTERISTIC(NAME, szname),
		    NEW_HOMEKIT_CHARACTERISTIC(ROTATION_SPEED, 0, .min_value = (float[]) { 0.0 }, .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(cb, .context=context),
),

Yurik72, 2020-04-06:
Definetelly,  Fan service now is a part of library

closed 2020-04-06 by Yurik72

Jeroen207, 2020-04-06:
Awesome! you make a mistake, you defined the fan as a: MOTION_SENSOR
`homekit_service_t* hap_new_fan_service(const char* szname, hap_callback cb, void* context) {
	return NEW_HOMEKIT_SERVICE(**MOTION_SENSOR**`

It should be: 

`homekit_service_t* hap_new_fan_service(const char* szname, hap_callback cb, void* context) {
	return NEW_HOMEKIT_SERVICE(**FAN2**`

Yurik72, 2020-04-07:
Ok, thanks, i did correct that.  usually i tested, but this time was copy paste