Yurik72/ESPHap #5: Serial Monitor overflowing with "Got characteristic X.XX change event"

seamaster101, 2020-04-09:
I'm getting serial monitor overflowing with the following data every time I open the application on my iPhone:
************
>>> HomeKit: [Client 1073688948] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073688948] Got characteristic 1.13 change event
>>> HomeKit: [Client 1073688948] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073688948] Got characteristic 1.13 change event
>>> HomeKit: [Client 1073688948] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073688948] Got characteristic 1.13 change event
>>> HomeKit: [Client 1073688948] Got characteristic 1.10 change event
************
and that repeats for long time even I I close the application (maybe still displays what's in the buffer?)
then after some time there the following error:
************
 !!! HomeKit: [Client 1073697900] error_in_write_data is true, abort write data
!!! HomeKit: [Client 1073697900] error_in_write_data is true, abort write data
!!! HomeKit: [Client 1073697900] error_in_write_data is true, abort write data
>>> HomeKit: [Client 1073697900] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073697900] Got characteristic 1.13 change event
!!! HomeKit: [Client 1073697900] error_in_write_data is true, abort write data
!!! HomeKit: [Client 1073697900] error_in_write_data is true, abort write data
!!! HomeKit: [Client 1073697900] error_in_write_data is true, abort write data
>>> HomeKit: [Client 1073697900] Got characteristic 1.10 change event
>>> HomeKit: [Client 1073697900] Got characteristic 1.13 change event
***********

you have my sketch in [#2.](https://github.com/Yurik72/ESPHap/issues/2)
it happens even if I remove the second sensor and use only temperature sensor
the application appears to work OK, but there is definitely something going wrong for that much chatter on the serial monitor...
Please help

Yurik72, 2020-04-11:
I ordered DHT11 for testing, it's on the way to me,
In my experience of usage BME280 (my preferbale device)  I have sent notification to Apple only if difference between new value and old value
For instance 
HOMEKIT_CHARACTERISTIC_CURRENT_TEMPERATURE);
if(ch1->value.float_value!=temp}   ///validate if temp is changed
ch1->value.float_value=temp;
homekit_characteristic_notify(ch1,ch1->value);
delay(10);
}

The same for humidity

You can see my usage in 
https://github.com/Yurik72/ESPHomeController

It litle bit complex

seamaster101, 2020-04-12:
Yes, this helped! No more chatter on the serial monitor.
You can close that one too. It works perfectly well now with DHT11!

closed 2020-04-13 by Yurik72

Yurik72, 2020-04-13:
Hi, please send me the latest version of your sketch, i will put into examples as well
The problem was that is too fast updating apple... After applying my solution, Apple home is not updating such frequently, but potentially it can be. The final solution is to increase a measurement time or/and compare values with some delta before sending to Apple. I suppose is enough for such purpose . It's only "user" monitor and not so important to see values with couple of second of delay or  with decreasing of accuracy.. Any how i continue working on library to prevent this "user" usage  in such manner and increasing stability on the core

seamaster101, 2020-04-15:
Here it is:
***********************************
//ESP8266 with DHT11  

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
//#include <FS.h>
//#include <SPIFFS.h>
#define STATIC_IP_ADDR

///---DHT11-----///
#include "PietteTech_DHT.h"  // Uncommend if building using CLI
#define DHTTYPE  DHT11       // Sensor type DHT11/21/22/AM2301/AM2302
#define DHTPIN   2           // Digital pin for communications
#define PIN   0           // Digital pin for re-format

//declaration
void dht_wrapper(); // must be declared before the lib initialization
///---DHT11 stop-----///

#ifdef STATIC_IP_ADDR
IPAddress staticIP(192,168,0,91);
IPAddress gateway(192,168,0,1);
IPAddress subnet(255,255,255,0);
#endif
#define HOMEKIT_SHORT_APPLE_UUIDS
#include "coredecls.h"

const char* ssid = "SHAW-62CD10";
const char* password = "251169025293";
float hum;
float temp;

const long userdelay = 3000;            //delay for user input when tarting to format the SPIFFS
unsigned long previousMillis = 0; 
unsigned long prevMillis = 0;
const long interval = 15000;  
unsigned long currentMillis = 0; 
unsigned long currMillis = 0; 

extern "C"{
#include "homeintegration.h"
}
#include "homekitintegrationcpp.h"

//homekit_service_t* hapservice={0};

homekit_service_t* temperature;
homekit_service_t* humidity; 

String pair_file_name="/pair.dat";

///---DHT11-----///
// Lib instantiate
PietteTech_DHT DHT(DHTPIN, DHTTYPE, dht_wrapper);
int n;      // counter
///---DHT11 stop-----///

//**********************
void init_hap_storage(){
  Serial.print("init_hap_storage");
  File fsDAT=SPIFFS.open(pair_file_name, "r");
  if(!fsDAT){
   Serial.println("Failed to read pair.dat");
  return;
}
  int size=hap_get_storage_size_ex();
  char* buf=new char[size];
  memset(buf,0xff,size);
  int readed=fsDAT.readBytes(buf,size);
[... 166 lines cut ...]

  homekit_characteristic_t * ch2= homekit_service_characteristic_by_type(humidity, HOMEKIT_CHARACTERISTIC_CURRENT_RELATIVE_HUMIDITY);
  //ch2->value.float_value=hum;
  if(ch2->value.float_value != hum){ ///validate if temp is changed
     ch2->value.float_value = hum;
     homekit_characteristic_notify(ch2,ch2->value);
     delay(10);
    } 

hap_homekit_loop(); //what does this exactly do?????
///--- end of the main loop ---///
}

///--- Formating function ---///
void Format(){
  // Next lines have to be done ONLY ONCE!!!!!When SPIFFS is formatted ONCE you can comment these lines out!!
  Serial.println("Please wait few seconds for SPIFFS to be formatted");
  SPIFFS.format();
  Serial.println("SPIFFS formatted");
  delay(1000);
}

***********************************
Спасиба, всево харошева,
Jordan