Yurik72/ESPHap #10: EspHapLed8266 example creates new Led on every reboot

jonasborneland, 2020-04-18:
Every time I reboot I have to re-add a new led? And the old parring is still there, just with a "offline" text.

Here is my log after rebooting the NodeMCU Board:

```
19:42:40.497 -> rl $⸮⸮| ⸮$⸮|⸮d⸮b|⸮⸮⸮⸮⸮r⸮b⸮#⸮p~⸮N⸮lon⸮⸮⸮#p⸮$sl{lp⸮o⸮⸮l⸮⸮bo⸮|l⸮l⸮p⸮⸮no⸮$⸮⸮ $`⸮onl or⸮⸮⸮N⸮$`rےo⸮d`⸮{⸮p⸮l⸮r⸮ ⸮⸮o⸮Borneland
19:42:41.155 -> .
19:42:41.155 -> IP address: 
19:42:41.155 -> 10.0.1.23
19:42:41.155 -> Free heap: 37704
19:42:41.155 -> init_hap_storageReaded bytes ->>>> HomeKit: init_storage_ex size 0x1c1
19:42:41.155 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
19:42:41.194 -> >>> Home Integration: hap_services added chararacteristic  1: 25 
19:42:41.194 -> >>> Home Integration: homekit_is_paired 0
19:42:41.194 -> >>> HomeKit: Update the CPU to run at 160MHz
19:42:41.194 -> >>> HomeKit: Free heap: 35992
19:42:41.194 -> >>> HomeKit: Starting server
19:42:41.194 -> >>> HomeKit: Formatting flash at 0x200000
19:42:41.261 -> >>> HomeKit: storage init 1
19:42:41.261 -> >>> HomeKit: if r
19:42:41.261 -> >>> HomeKit: generating id
19:42:41.261 -> >>> HomeKit: Generated new accessory ID: BE:B1:BF:E1:7F:4B
19:42:41.261 -> >>> HomeKit: Generated new accessory ID: 1073689556
19:42:41.261 -> >>> HomeKit:  key generated
19:42:41.365 -> >>> HomeKit:  key saved
19:42:41.398 -> >>> HomeKit: Generated new accessory key
19:42:41.473 -> >>> HomeKit: find pairings
19:42:41.473 -> >>> HomeKit: Preinit pairing context
19:42:41.473 -> >>> HomeKit: Using user-specified password: 111-11-111
19:42:41.511 -> >>> HomeKit: Free heap: 33200
19:42:47.524 -> >>> HomeKit: Calculating public key len=384
19:42:50.641 -> >>> HomeKit: Calculating public key result 0
19:42:50.641 -> >>> HomeKit: arduino_homekit_preinit success
19:42:50.641 -> >>> HomeKit: Configuring mDNS
19:42:50.675 -> >>> HomeKit: MDNS.begin: ES-7259, IP: 10.0.1.23
19:42:50.675 -> >>> HomeKit: Init server over
19:43:08.613 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 4)
19:43:08.613 -> >>> HomeKit: Got new client connection: local 10.0.1.23:5556, remote 10.0.1.11:54131
19:43:08.613 -> >>> HomeKit: [Client 1073692148] Got 158 incomming data, encrypted is 
19:43:08.613 -> >>> HomeKit: [Client 1073692148] Pair Verify Step 1/2
19:43:08.966 -> >>> HomeKit: Free heap: 28528
19:43:09.003 -> >>> HomeKit: [Client 1073692148] Disconnected!
19:43:09.003 -> >>> HomeKit: Free heap: 29328
19:43:09.003 -> >>> HomeKit: [Client 1073692148] Closing client connection
19:43:11.610 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 4)
19:43:11.610 -> >>> HomeKit: Got new client connection: local 10.0.1.23:5556, remote 10.0.1.11:54132
19:43:11.643 -> >>> HomeKit: [Client 1073693948] Got 158 incomming data, encrypted is 
19:43:11.643 -> >>> HomeKit: [Client 1073693948] Pair Verify Step 1/2
19:43:11.962 -> >>> HomeKit: Free heap: 28528
19:43:14.277 -> >>> HomeKit: [Client 1073693948] Disconnected!
19:43:14.277 -> >>> HomeKit: Free heap: 29608
19:43:14.277 -> >>> HomeKit: [Client 1073693948] Closing client connection
19:43:16.659 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 4)
19:43:16.659 -> >>> HomeKit: Got new client connection: local 10.0.1.23:5556, remote 10.0.1.11:54134
19:43:16.693 -> >>> HomeKit: [Client 1073694732] Got 158 incomming data, encrypted is 
19:43:16.693 -> >>> HomeKit: [Client 1073694732] Pair Verify Step 1/2
19:43:16.998 -> >>> HomeKit: Free heap: 28720
19:43:17.202 -> >>> HomeKit: [Client 1073694732] Disconnected!
19:43:17.202 -> >>> HomeKit: Free heap: 29328
19:43:17.202 -> >>> HomeKit: [Client 1073694732] Closing client connection
19:43:19.870 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 4)
19:43:19.870 -> >>> HomeKit: Got new client connection: local 10.0.1.23:5556, remote 10.0.1.7:49250
19:43:19.870 -> >>> HomeKit: [Client 1073692092] Got 158 incomming data, encrypted is 
19:43:19.870 -> >>> HomeKit: [Client 1073692092] Pair Verify Step 1/2
19:43:20.216 -> >>> HomeKit: Free heap: 28528
19:43:20.216 -> >>> HomeKit: [Client 1073692092] Disconnected!
19:43:20.216 -> >>> HomeKit: Free heap: 29328
19:43:20.216 -> >>> HomeKit: [Client 1073692092] Closing client connection
```

oxbown, 2020-04-18:
There is a mistake in "void init_hap_storage" and "storage_changed", use this ones:

```c
void init_hap_storage(){ //Not touch!
    Serial.print("init_hap_storage");   
    File fsDAT=SPIFFS.open(pair_file_name, "r");
      if(!fsDAT){
      Serial.println("Failed to read pair.dat");
      return;
       }
    int size = hap_get_storage_size_ex();
    char* buf=new char[size];
    memset(buf,0xff,size);
    int readed=fsDAT.readBytes(buf,size);
    Serial.print("Readed bytes ->");
    Serial.println(readed);
    hap_init_storage_ex(buf,size);
    fsDAT.close();
    delete []buf;
  }

void storage_changed(char * szstorage,int size){ //Not touch!

  SPIFFS.remove(pair_file_name);
  File fsDAT=SPIFFS.open(pair_file_name, "w+");
  if(!fsDAT){
    Serial.println("Failed to open pair.dat");
    return;
  }
  fsDAT.write((uint8_t*)szstorage,size);

  fsDAT.close();
  }
```

jonasborneland, 2020-04-18:
@oxbown Thanks! I'll try it

jonasborneland, 2020-04-18:
@oxbown Works perfect! Thanks!
@Yurik72 Will you update the example and close this issue? 🙌

Yurik72, 2020-04-19:
Updated, thanks for found that

closed 2020-04-19 by Yurik72