Yurik72/ESPHap #11: ESP8266_DHT11.ino empty content

jonasborneland, 2020-04-18:
The ESPHap/examples/ESP8266_DHT11/ESP8266_DHT11.ino is empty

jonasborneland, 2020-04-19:
@Yurik72 Will you add the example? Instead of the empty file? 😊

Yurik72, 2020-04-19:
Yes, I have decided to prepare next generation of examples with file browser and OTA.
Therefore it will take  one-two  day more, due to the Easter :)

oxbown, 2020-04-19:
This is my version for DHT22 sensor.
Also features, multi threads, OTA, wifi-reconnection.
[Wemos_Homekit_DHT.zip](https://github.com/Yurik72/ESPHap/files/4499972/Wemos_Homekit_DHT.zip)

jonasborneland, 2020-04-19:
@Yurik72 Awesome! Looking forward to that!
@oxbown Thanks!!

Yurik72, 2020-04-20:
While I'm working on example, you can look as well this one 
https://www.instructables.com/id/ESP32-Weather-Station-Solar-Powered/

There are additional libraies, and in my home i have all devices with that, due to one code sketch and supporting

Yurik72, 2020-04-20:
As promised 
https://github.com/Yurik72/ESPHap/tree/master/examples/EspHap_DHT11

There is universal sketch for ESP32, ESP8266

I will appreciate if you check this with ESP8266, ESP32 fully worked, ESP8266 not tested with DHT11 hardware

closed 2020-04-20 by Yurik72

jonasborneland, 2020-04-20:
@Yurik72 Thanks!!! I’ll test it as soon as possible!

jonasborneland, 2020-04-20:
@Yurik72 You forgot to delete your SSID and Pass.
The example works really well! But Homekit is rounding up or down to closes 0.5 ?
Every thing else works perfect on esp8266. I also tried to restart the board and I testet the OTA too