Yurik72/ESPHap #9: fatal error: SPIFFS.h: No such file or directory

jonasborneland, 2020-04-18:
I'm really excited about trying this out! but I'm getting an error.
I get this error when trying to compile EspHapLed:
`EspHapLed:4:20: fatal error: SPIFFS.h: No such file or directory
 #include <SPIFFS.h>
                    ^
compilation terminated.
exit status 1
SPIFFS.h: No such file or directory`

I'm using a NodeMCU board
[image]

Do you know what's wrong? Thanks!

oxbown, 2020-04-18:
Try with -->.  #include "FS.h"
For esp8266 this the right memory manager!

jonasborneland, 2020-04-18:
@oxbown Thanks! But then I get these errors:
`/Users/jonasborneland/Documents/Arduino/jonas programs/EspHapLed/EspHapLed.ino: In function 'void setup()':
EspHapLed:25:28: error: no matching function for call to 'fs::FS::begin(bool)'
      if (!SPIFFS.begin(true)) {
                            ^
/Users/jonasborneland/Documents/Arduino/jonas programs/EspHapLed/EspHapLed.ino:25:28: note: candidate is:
In file included from /Users/jonasborneland/Documents/Arduino/jonas programs/EspHapLed/EspHapLed.ino:3:0:
/Users/jonasborneland/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/FS.h:205:10: note: bool fs::FS::begin()
     bool begin();
          ^
/Users/jonasborneland/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/FS.h:205:10: note:   candidate expects 0 arguments, 1 provided
EspHapLed:33:30: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
     WiFi.begin(ssid, password);
                              ^
In file included from /Users/jonasborneland/Documents/Arduino/jonas programs/EspHapLed/EspHapLed.ino:2:0:
/Applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFi.h:79:9: error:   initializing argument 1 of 'int WiFiClass::begin(char*, const char*)' [-fpermissive]
     int begin(char* ssid, const char *passphrase);
         ^
EspHapLed:66:29: error: 'hap_init_homekit_server' was not declared in this scope
     hap_init_homekit_server();
                             ^
exit status 1
no matching function for call to 'fs::FS::begin(bool)'`

Is the EspHapLed sketch suitable for esp8266 or is there another example I could use for a simple build in led test ?

jonasborneland, 2020-04-18:
ARhhh  my bad 🙈 just saw the "EspHapLed 8266"... What a big mistake from me... Sorry! Thanks for the help!

closed 2020-04-18 by jonasborneland