Yurik72/ESPHap #13: Need a function for restart or stop the HAP server.

oxbown, 2020-04-19:
I´m not able to make HAP working after a network reconnection. I try to start the server without stopping it but makes the system not responding.

Yurik72, 2020-04-20:
First could you please describe a case which reconection you means ? For instance if you are lost Wifi you are trying to connect again ?
On which platform ? ESP32 or ESP8266 (code is different)

I will do that, but in my personal experience especially on ESP8266 due to the heap limitation, after
first start  heap is "quite" fragmented :( therefore i used different approach : completelly restart device.  For instance for thermostat is not critical and completelly solves a problem
Please think about this approach

oxbown, 2020-04-20:
Hi!
I use an ESP8266 (Wemos D1).
Let´s explain the situation.  I have the ESP8266 connected to the Wifi network and HAP running, and I restart the wifi router the ESP disconnects and tries to reconnect to the same wifi.
When the Wifi returns, ESP reconnects, but sometimes HAP not respond and in other cases the board restarts. 
The restart situation is Ok for sensors like temperature/humidity/motion but for lights, the restart may cause the light turns off. This is the reason why I want to stop HAP, I want to be able to start HAP when the board is correctly connected to the wifi network.

I think that is better to split HAP and hardware management, for example, lights and switches handling, etc...

Thank you!

Yurik72, 2020-04-20:
Ok, clear I will check with ESP8266 first, what is better handling :)

Yurik72, 2020-04-21:
Please check 
hap_restart_server();

I did as much as possible minimum to works it stable. My test is Ok.

closed 2020-04-21 by Yurik72

oxbown, 2020-04-21:
> Please check
> hap_restart_server();
> 
> I did as much as possible minimum to works it stable. My test is Ok.

Hi!! I don't see the commit where the hap_restart_server() is implemented.

Yurik72, 2020-04-21:
it was on the way:) sorry
check now