Yurik72/ESPHap #14: Using HTTPClient in sketch gives error

jonasborneland, 2020-04-23:
Hi

Im trying to add a url POST action to my sketch. But as this line:
HTTPClient http; 

Gives me this error:
`/Users/jonasborneland/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /Users/jonasborneland/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/lib/libaxtls.a(crypto_misc.o): in function `base64_decode':
xtensa-lx106-elfhome/earle/src/axtls-8266/crypto/crypto_misc.c:327: multiple definition of `base64_decode'; libraries/ESPHap-master/base64.c.o:/Users/jonasborneland/Documents/Arduino/libraries/ESPHap-master/base64.c:64: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).`

I have included it:
#include <ESP8266HTTPClient.h>

Yurik72, 2020-04-24:
Hi,  I confrinm that is library conflicts between EspHap and ESP8266HTTPClient.h
At this moment my suggestin is use another HTTP client 
I have checked my own simplified implementation and you can see it from eample
https://github.com/Yurik72/ESPHap/tree/master/examples/EspHap_Thermostat

HTTPSimpleClient.h
HTTPSimpleClient.cpp

I have used this client as well for ESP32, due to other conflicts

if success i will close an issue

jonasborneland, 2020-04-24:
@Yurik72 Thanks!! I will try this out tonight

jonasborneland, 2020-04-24:
@Yurik72 Im having trouble using "HTTPSimpleClient" - Do you have a link for some documentation on it? or any example on how to make a url POST :)

Yurik72, 2020-04-25:
Hi, in very short, should works following example
 HTTPSimpleClient http;
 http.begin(uri); //Specify the URL
int httpCode = http.POST(postString);  specify post string

jonasborneland, 2020-04-25:
My board keeps rebooting and the log looks like this:
`11:52:05.561 -> ctx: cont
11:52:05.561 -> sp: 3fff13c0 end: 3fff1690 offset: 01a0
11:52:05.561 -> 3fff1560:  3ffe8400 fffffffe 3fff1580 40217be9  
11:52:05.561 -> 3fff1570:  3ffe84e8 fffffffe 3fff0568 40217a4b  
11:52:05.561 -> 3fff1580:  00000000 00220000 80ff43dc 80ff42f4  
11:52:05.594 -> 3fff1590:  00000000 00002710 00001388 40222500  
11:52:05.594 -> 3fff15a0:  3ffe9d00 80000000 0a0d0a00 402241ed  
11:52:05.594 -> 3fff15b0:  80fe9d00 00000000 00000000 ffffffff  
11:52:05.594 -> 3fff15c0:  00000000 40231250 00000000 00001388  
11:52:05.594 -> 3fff15d0:  40220cac 00000000 3fff3ffc 3fff4300  
11:52:05.594 -> 3fff15e0:  0037003f 80ff0568 3fff3ee4 0022002f  
11:52:05.594 -> 3fff15f0:  00ffdad0 6d657400 3f003d70 802210d0  
11:52:05.594 -> 3fff1600:  74736500 3fff0500 80ff0568 69686300  
11:52:05.627 -> 3fff1610:  64695f70 80ff003d 3fff3e00 0037003f  
11:52:05.627 -> 3fff1620:  806e6100 3fff4000 000d000f 80482000  
11:52:05.627 -> 3fff1630:  5f697000 3d79656b 80ff3f00 3fff4374  
11:52:05.627 -> 3fff1640:  0037003f 006e6100 3ffe84dc 8021fa44  
11:52:05.627 -> 3fff1650:  706d6500 40231400 8001000a 3fff16d8  
11:52:05.627 -> 3fff1660:  3fffdad0 00000000 3ffe84dc 40217ab7  
11:52:05.627 -> 3fff1670:  3fffdad0 00000000 3fff1698 402224e4  
11:52:05.665 -> 3fff1680:  feefeffe feefeffe 3ffe8878 40100d75  
11:52:05.665 -> <<<stack<<<`

Heres my function. Called like readDHT();
`void sendSql(){
  WiFiClient client; 
  const int httpPort = 80; 
  if (!client.connect(host, httpPort)) 
  { 
    Serial.println("connection failed"); 
    return; 
  } 
  Serial.print("Requesting URL: "); 
  Serial.println(url); //Post Data 
  String postData = String("api_key=") + String(apiKeyValue) + String("&chip_id=") + String(sensorName) + String("&temp=") + String(DeviceData.temp) + String("&hum=") + String(DeviceData.hum);
  String address = host + url; 
  HTTPSimpleClient http;
  http.begin(address); 
 // http.addHeader("Content-Type", "application/x-www-form-urlencoded"); 
  auto httpCode = http.POST(postData); 
  Serial.println(httpCode); //Print HTTP return code 
  String payload = http.getString(); 
  Serial.println(payload); //Print request response payload 
  http.end(); //Close connection Serial.println(); 
  Serial.println("closing connection");
}`

Yurik72, 2020-04-25:
No clear exactly where crash happens in your code,
But at least i see a potential problem
You don't need to use    WiFiClient,to check  you can simple start with http client. If host is not reachable 
http.begin(address)  returns false.
And pls add more output to see where crash happens

Are you trying to connect to thingspeak ? I'm going to finish my example as well in a few days
for the thingspeak

Yurik72, 2020-04-25:
I have updated example for Thermostat
Please have a look function sendToThingspeak()

If you really need POST,   keep everything as in my function just replace 

 int httpCode = http.GET();  with 
 int httpCode http.POST(postString);

You need as well to correct URL to extract get request and put POST request to post string

Example works fine with Thingspeak

jonasborneland, 2020-04-25:
@Yurik72 Thanks for answering back! I’ll do some more tryouts and if it still crashes I will post some extended information and logs here. Thanks for the big help and effort you make in answering and developing this repository

jonasborneland, 2020-04-25:
@Yurik72 Looks like it is almost working!
This is to put data into my own web servers SQL.

I get an 200 response, but the data is not inserted. I testet the setup with the API calling APP postman, and every thing works fine on the website with a form request...

I think the code is missing a header, like this:
` http.addHeader("Content-Type", "application/x-www-form-urlencoded"); `
How can I add this?... And is there a way to get the response of the webpage instead of only the response code? 🤫

Yurik72, 2020-04-26:
Ok, I see the problem, library HttpSimpleClient client has to be slightly changed for POST
I will do that and reply

jonasborneland, 2020-04-26:
@Yurik72 Thanks!!

Yurik72, 2020-04-26:
Done  !
Usage:
To add header use  http.addHeader(const String& name, const String& value)
example as you mentioned 👍 
http.addHeader("Content-Type", "application/x-www-form-urlencoded");

To get string response use like that
 int httpCode = http.GET();
String response=http.getString();
P.S. I didn't check response yet, if not work , don't blame me, I will correct:)

jonasborneland, 2020-04-26:
@Yurik72 It's really close now! 👍 
The response works perfect.

But it's not received as a POST message in PHP.
I have this line of PHP code:
`var_dump($_POST["api_key"]);`
Which should write the value of "api_key". But in arduino IDEs console, I get "NULL".:
```
16:35:40.356 -> sendToThingspeak start
16:35:40.356 -> http://www.mysite.com/sensor/post/
16:35:40.356 -> http begin start connection www.mysite.com
16:35:40.426 -> http finish connection
16:35:40.460 -> postString api_key=tPmAT5Ab3j7F9&chip_id=14221474&temp=20.00&hum=50.00
16:35:40.460 -> http code returns 200
16:35:40.460 -> http code response NULL
16:35:40.460 -> Wrong API Key provided.

```

My code in arduino IDE:
```
void sendToThingspeak(){
   Serial.println("sendToThingspeak start");
    String url="http://www.mysite.com/sensor/post/";
    
    HTTPSimpleClient http;
  
     String postString="api_key="+String(apiKeyValue);
     postString+="&chip_id="+String(sensorName);
     postString+="&temp="+String(DeviceData.temp);
     postString+="&hum="+String(DeviceData.hum);
     Serial.println(url);
     
     if(!http.begin(url)){
       Serial.println("Failed to connect to "+url );
    }
    http.addHeader("Content-Type", "application/x-www-form-urlencoded");  
    //int httpCode = http.GET();
    int httpcode = http.POST(postString);
    String response=http.getString();
    Serial.println("postString "+String(postString) );
   Serial.println("http code returns "+String(httpcode) );
   Serial.println("http code response "+String(response) );   
}
```

And I know it is working form Postman:
[image]

Yurik72, 2020-04-26:
Your code looks OK, As i mentioned I didn't tested POST
Ok i will prepare debug version, to produce a log or you can do that by yourself by adding console output into HttpSimpleClient.cpp

Methods 
getString
WriteToStream
writeToStreamDataBlock

Should be traced, and after that you will have really simple class to handle HTTP request :)

jonasborneland, 2020-04-26:
@Yurik72 Thanks for the always quickness response! I’ll try it as soon as I can 👍

Yurik72, 2020-04-27:
Hi , I have tested POST request and in my case it's works fine
If you still have a problem, get the latest version of  HTTPSimpleClient.cpp

Uncomment 
//#define DEBUG_HTTPCLIENT

and send me a log

jonasborneland, 2020-04-27:
@Yurik72 Thanks for you big effort in helping me! I’ll test it when I’m home toninght

jonasborneland, 2020-04-27:
@Yurik72 I did some tests now.

My website supports both http and https - so I tried both.

With http, the it looks like this:

jonasborneland, 2020-04-27:
@Yurik72  I did some tests now.

My website supports both http and https - so I tried both.

With http, the it looks like this:
```
20:18:34.191 -> [HTTP] Begin http://www.borneland.com/sensor/post/
20:18:34.191 -> http begin start connection www.borneland.com
```
With https:
```
20:27:54.898 -> https://www.borneland.com/sensor/post/
20:27:54.898 -> [HTTP] Begin https://www.borneland.com/sensor/post/
```
So it looks like http cuts off the "/sensor/post/" part of the url.

But with https I get a new error
`Write to stream -> not connected`

My logs for http:
```
20:18:34.153 -> Web server is NOT SET, waiting for pairing
20:18:34.191 -> sendToThingspeak start
20:18:34.191 -> http://www.borneland.com/sensor/post/
20:18:34.191 -> [HTTP] Begin http://www.borneland.com/sensor/post/
20:18:34.191 -> http begin start connection www.borneland.com
20:18:34.191 -> [HTTP] Start connection www.borneland.com80
20:18:34.225 -> http finish connection
20:18:34.262 -> postString api_key=tPmAT5Ab3j7F9&chip_id=14661474&temp=20.00&hum=50.00
20:18:34.262 -> http code returns 200
20:18:34.262 -> http code response NULL
20:18:34.262 -> Wrong API Key provided.
```

My logs for https:
```
20:27:54.898 -> Web server is NOT SET, waiting for pairing
20:27:54.898 -> sendToThingspeak start
20:27:54.898 -> https://www.borneland.com/sensor/post/
20:27:54.898 -> [HTTP] Begin https://www.borneland.com/sensor/post/
20:27:54.932 -> http begin start connection www.borneland.com
20:27:54.932 -> [HTTP] Start connection www.borneland.com443
20:27:54.966 -> WiFi Client not able to connect
20:27:55.001 -> Failed to connect to https://www.borneland.com/sensor/post/
20:27:55.001 -> Write to stream -> not connected
20:27:55.001 -> postString api_key=tPmAT5Ab3j7F9&chip_id=14661474&temp=20.00&hum=50.00
20:27:55.001 -> http code returns -2
20:27:55.001 -> http code response 
```

Yurik72, 2020-04-28:
Hi 
"I'm also not sure if I should have "?" sign before the first part of my postString, like:
"?api_key=..."" 
For the post http you don't need "?" this is delimeter for get, when string is part of URL

About Https,  maybe problem is port number , by default https uses 443 , which port opened on your server ?

I will try to find tohight a time for your site

jonasborneland, 2020-04-28:
I contacted my host. 443 is the correct one

jonasborneland, 2020-04-28:
I have tried both http and https with the header.

HTTP:
```
22:24:47.061 -> sendToThingspeak start
22:24:47.061 -> http://www.borneland.com/sensor/post/
22:24:47.061 -> [HTTP] Begin http://www.borneland.com/sensor/post/
22:24:47.061 -> http begin start connection www.borneland.com
22:24:47.096 -> [HTTP] Start connection www.borneland.com80
22:24:47.131 -> http finish connection
22:24:47.270 -> postString ?api_key=tPmAT5Ab3j7F9&chip_id=14661474&temp=20.00&hum=50.00
22:24:47.270 -> http code returns 200
22:24:47.270 -> http code response NULL
22:24:47.303 -> Wrong API Key provided.
```

HTTPS:
```
23:02:47.628 -> sendToThingspeak start
23:02:47.628 -> https://www.borneland.com/sensor/post/
23:02:47.628 -> [HTTP] Begin https://www.borneland.com/sensor/post/
23:02:47.628 -> http begin start connection www.borneland.com
23:02:47.628 -> [HTTP] Start connection www.borneland.com443
23:02:47.663 -> WiFi Client not able to connect
23:02:47.663 -> Failed to connect to https://www.borneland.com/sensor/post/
23:02:47.663 -> Write to stream -> not connected
23:02:47.663 -> postString ?api_key=tPmAT5Ab3j7F9&chip_id=14661474&temp=20.00&hum=50.00
23:02:47.663 -> http code returns -2
23:02:47.663 -> http code response 

```

Yurik72, 2020-04-29:
I have checked
1. You don't need to add "?" to Post string
2. Https is not working on ESP8266, but works on ESp32
3.   HTTP returns NULL\r\n Wrong API Key provided. , therefore you see on the log two lines
So final issue is getting working this on ESP8266 bwith https

My code
void sendToThingspeak(){
   Serial.println("sendToThingspeak start");
    String url="https://www.borneland.com/sensor/post/";
    
    HTTPSimpleClient http;
     String apiKeyValue="tPmAT5Ab3j7F9";
     String sensorName="14661474";
     String postString="api_key="+String(apiKeyValue);
     postString+="&chip_id="+String(sensorName);
     postString+="&temp="+String(DeviceData.temp);
     postString+="&hum="+String(DeviceData.hum);
     Serial.println(url);
     
     if(!http.begin(url)){
       Serial.println("Failed to connect to "+url );
    }
    http.addHeader("Content-Type", "application/x-www-form-urlencoded");  
    int httpcode = http.POST(postString);
    String response=http.getString();
    Serial.println("postString "+String(postString) );
   Serial.println("http code returns "+String(httpcode) );
   Serial.println("http code response "+String(response) );   
}

jonasborneland, 2020-04-29:
I've change my website script to return request type and request body.
My php code is:
```
var_dump($_SERVER["REQUEST_METHOD"]);
echo "---post content---";
var_dump($_POST);
```
The expected response:
```
string(4) "POST"
---post content---array(4) {
  ["api_key"]=>
  string(13) "tPmAT5Ab3j7F9"
  ["chip_id"]=>
  string(8) "14661474"
  ["temp"]=>
  string(5) "23.70"
  ["hum"]=>
  string(5) "50.00"
}
```

The actual response from arduino serial monitor:
```
http code response string(4) "POST"
21:10:10.091 -> ---post content---array(0) {
21:10:10.091 -> }
```

It prints the request type "POST" - that is correct and great.
But the content is empty. It looks like there isn't being sent anything in the body.

I searched through google to find answers why the body would be empty. Many people said the Content-Length should be correct length - but I checked the value and it looked fine.

Do you have a clue why the body is empty?

Yurik72, 2020-05-05:
Hi , sorry for delays, do you still has a problem , i'll have a time to help

jonasborneland, 2020-05-05:
@Yurik72 Thank you so much. Yes I’m still stock at above description, but no worries if you don’t have any solutions yet 😊 I can’t do it as a GET, it don’t absolutely have to be as POST, but would be nice