Yurik72/ESPHap #15: Accessory not showing up for pairing any longer?

seamaster101, 2020-04-25:
@Yurik72 I'm experiencing strange problem. So is friend of mine @mlanziner who is trying the same thing on different computer. The problem is that after successful compiling of the new example, the accessory does not show up as accessory available for pairing. Now even my old code that I sent you before for DHT11 is behaving the same. 
On very rare occasions I see the accessory showing up or pairing but then it does not complete the paring process successfully on the iPhone. On the terminal window it shows pairing is successful, but is usually much shorter on messages that what I've seen before when it use to actually do pair successfully. 
I tried formating the entire flash, and formating the SPIFS as well. No luck.
Here is the log from my serial terminal after reboot and one of the rare cases that it showed for pairing but actually failed pairing on the iPhone. your example code has not been changed, except I added format for SPIFFS and change the pin for DHT11 to GPIO2
`ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
Please wait few seconds for SPIFFS to be formatted
SPIFFS formatted
Grapevine
..
IP address: 
192.168.1.107
Free heap: 36336
init_hap_storageFailed to read pair.dat
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 11 
>>> Home Integration: add_hum as accessory , next accessory 2
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Update the CPU to run at 160MHz
>>> HomeKit: Free heap: 33424
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x200000
>>> HomeKit: storage init 1
>>> HomeKit: if r
>>> HomeKit: generating id
>>> HomeKit: Generated new accessory ID: 82:A8:A3:40:34:B5
>>> HomeKit: Generated new accessory ID: 1073688596
>>> HomeKit:  key generated
>>> HomeKit:  key saved
>>> HomeKit: Generated new accessory key
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 30800
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: MDNS.begin: ES-F118, IP: 192.168.1.107
>>> HomeKit: Init server over
Web server is NOT SET, waiting for pairing
Temp21.90  Hum:62.00
Temp24.00  Hum:44.00
Temp24.10  Hum:43.00
>>> HomeKit: WiFi connected, ip: 192.168.1.107, mask: 255.255.255.0, gw: 192.168.1.254
>>> HomeKit: Configuring mDNS
>>> HomeKit: MDNS.restart: ES-F118, IP: 192.168.1.107
>>> HomeKit: MDNS.begin: ES-F118, IP: 192.168.1.107
Temp24.20  Hum:42.00
Temp24.30  Hum:41.00
Temp24.40  Hum:40.00
>>> HomeKit: WiFiServer receives a new client (current 0, max 4)
>>> HomeKit: Got new client connection: local 192.168.1.107:5556, remote 192.168.1.104:56576
>>> HomeKit: [Client 1073696140] Got 125 incomming data, encrypted is 
>>> HomeKit: [Client 1073696140] Pair Setup Step 1/3
>>> HomeKit: [Client 1073696140] Got 578 incomming data, encrypted is 
>>> HomeKit: [Client 1073696140] Pair Setup Step 2/3
>>> HomeKit: [Client 1073696140] Computing SRP shared secret
>>> HomeKit: [Client 1073696140] Verifying peer's proof
>>> HomeKit: [Client 1073696140] Generating own proof
Temp24.50  Hum:40.00
>>> HomeKit: [Client 1073696140] Got 280 incomming data, encrypted is 
>>> HomeKit: [Client 1073696140] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 31C042E2-1767-4A74-8CF9-AF735720357B
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: Configuring mDNS
>>> HomeKit: MDNS.restart: ES-F118, IP: 192.168.1.107
>>> HomeKit: MDNS.begin: ES-F118, IP: 192.168.1.107
>>> HomeKit: [Client 1073696140] Successfully paired
>>> HomeKit: [Client 1073696140] Disconnected!
>>> HomeKit: Free heap: 29656
>>> HomeKit: [Client 1073696140] Closing client connection
Temp24.90  Hum:39.00
Temp25.00  Hum:39.00
Temp25.20  Hum:38.00
Temp25.40  Hum:38.00
Temp25.60  Hum:37.00
Temp25.80  Hum:37.00

`

and here is an example of what ussulay shows on the serial terminal screen after reboot, but nothing chows as available to pair in home kit:
`
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
Please wait few seconds for SPIFFS to be formatted
SPIFFS formatted
Grapevine
.....
IP address: 
192.168.1.107
Free heap: 36336
init_hap_storageFailed to read pair.dat
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 11 
>>> Home Integration: add_hum as accessory , next accessory 2
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Update the CPU to run at 160MHz
>>> HomeKit: Free heap: 33424
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x200000
>>> HomeKit: storage init 1
>>> HomeKit: if r
>>> HomeKit: generating id
>>> HomeKit: Generated new accessory ID: 0F:57:B3:66:17:F2
>>> HomeKit: Generated new accessory ID: 1073688596
>>> HomeKit:  key generated
>>> HomeKit:  key saved
>>> HomeKit: Generated new accessory key
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 30800
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Init server over
Web server is NOT SET, waiting for pairing
Temp26.30  Hum:50.00
Temp26.00  Hum:42.00
>>> HomeKit: WiFi connected, ip: 192.168.1.107, mask: 255.255.255.0, gw: 192.168.1.254
>>> HomeKit: Configuring mDNS
>>> HomeKit: MDNS.begin: ES-66DD, IP: 192.168.1.107
Temp25.80  Hum:39.00
Temp25.80  Hum:38.00
Temp25.80  Hum:38.00
Temp25.80  Hum:38.00
Temp25.70  Hum:37.00
Temp25.80  Hum:38.00
Temp25.70  Hum:38.00
`
My buddy's problem is exactly the same as mine, except I don't think he've seen the accessory available for pairing on rare occasion. I wonder if some update screw the working of the mDNS or something else.
Any ideas?
Cheers!

Yurik72, 2020-04-26:
Hi, from the your description i clearly see a problem with mDNS because
1. You mentioned that device is not always appears for the pairing or with delay
2. Pairing log shows that pairing done for the 50% , after first 3 step, device should switch mDNS and apple should send Pairing verification and get accessories requests, by using new mDNS record, which is not happens :(

Therefore i can propose several ways/steps to find out this problem, At least i'm pretty sure that problem is not on the library code, because a lot of guys did that successfully, so

1. You mentioned about problem when compiling on other computer. please check his environment  
 as i mentioned on read.me it should be
ESP32 board library version 1.0.4
ESP8266 board library version 2.6.3
Arduino version 1.8.12

2. Install mDNS browser like this  https://chrome.google.com/webstore/detail/mdns-browser/kipighjpklofchgbdgclfaoccdlghidp
And you should see mDNS record appears quite immediately after device switched on (in both mode pairing and not pairing). Control this process due to the pairing. 

3. Check your Wifi hardware, this is continuation of step 2. In my experience with some routers it's happens, the problem can be solved by upgrading new router firmware or simple reboot

4. Check exactly this device in other Home Wifi enviropment

After did that please reply i will see what we can do else

Yurik72, 2021-01-30:
the same #37

closed 2021-01-30 by Yurik72