Yurik72/ESPHap #17: Two "HomeKit accessories" on one ESP32

Jeroen207, 2020-06-01:
Is there a way to make two switches for example, but as an separate device? And maybe with an different code as well?

Yurik72, 2020-06-01:
The library hasn't contsraints for that, Hovewer for the switch there are not public methods for switch. :(
I will change the library today-tommorow and will close an issue
Handling two different codes not a problem. You will have two charactestics and each callback will send characheristic which changed, you can implement logic differently

Jeroen207, 2020-06-01:
You're great! Could you maybe please also include an example code for this? That would be great!

Yurik72, 2020-06-02:
I have changed library, you need to get the latest version
Example code below, based on the [EspHap_Switch](https://github.com/Yurik72/ESPHap/tree/master/examples/EspHap_Switch)
P.S. I ahve checked this code for compilation but not on real device, suppose it should work

```c
/// BASIC CONFIGURATION 

#define ENABLE_WIFI_MANAGER  // if we want to have built-in wifi configuration
                             // Otherwise direct connect ssid and pwd will be used
                             // for Wifi manager need extra library //https://github.com/tzapu/WiFiManager

#define ENABLE_WEB_SERVER    //if we want to have built in web server /site
#define ENABLE_OTA  //if Over the air update need  , ENABLE_WEB_SERVER must be defined first
#include <Arduino.h>

#ifdef ESP32
#include <SPIFFS.h>
#endif

#ifdef ESP8266
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include "coredecls.h"
#endif

#ifdef ENABLE_WEB_SERVER
#ifdef ESP8266
#include <ESP8266WebServer.h>
 ESP8266WebServer server(80);
#endif

#ifdef ESP32
#include <WebServer.h>
WebServer server(80);
#endif
#endif 

#if defined(ESP32) && defined(ENABLE_OTA)
#include <Update.h>
#endif

#ifdef ENABLE_WEB_SERVER
#include "spiffs_webserver.h"
bool isWebserver_started=false;
#endif 

const int relay_gpio=4;
const int relay_gpio_2=5;

#ifdef ENABLE_WIFI_MANAGER
#include <WiFiManager.h>        //https://github.com/tzapu/WiFiManager
#endif

const char* HOSTNAME="ES";
const char* ssid     = "ssid";
const char* password = "pwd";

extern "C"{
#include "homeintegration.h"
}
#ifdef ESP8266
#include "homekitintegrationcpp.h"
#endif

homekit_service_t* hapservice={0};
homekit_service_t* hapservice_2={0};

String pair_file_name="/pair.dat";

void switch_callback(homekit_characteristic_t *ch, homekit_value_t value, void *context);
[... 188 lines cut ...]
        
        if(ch->value.bool_value!=val){  //wil notify only if different
          ch->value.bool_value=val;
          homekit_characteristic_notify(ch,ch->value);
        }
    }
  }
}

void switch_callback(homekit_characteristic_t *ch, homekit_value_t value, void *context) {
    Serial.println("switch_callback");
 set_switch(ch->value.bool_value,*((int*)context));
}
#ifdef ENABLE_WIFI_MANAGER
void startwifimanager() {
  WiFiManager wifiManager;

  if (!wifiManager.autoConnect(HOSTNAME, NULL)) {
      ESP.restart();
      delay(1000);
   }
}
#endif

```

closed 2020-06-02 by Yurik72