Yurik72/ESPHap #18: Add to the homeintegration.c & second setup accessory for one ESP

Jeroen207, 2020-06-02:
Hi,

Is it possible that you could add this to the homeintegration.c? I've programmed this piece of code for my ceiling fan, by adding this code you can edit the rotation speed and direction. The min_step can be removed, since this is what I personally want. If you try to add a fan it will not work by default. 

Another thing, I try to figure out how I'm able to use the same ESP(32) to run two homekit "devices". Meaning that when I have an fan but I need to add it as a new device to the Home app. I have a couple of fan's but I don't want to share my fan with other people, it seems better solution to make an different setup so that an other person can add it to their Home app. Of course, using a second ESP board is an solution but it is a struggle. 
`
homekit_service_t* hap_new_fan_service(const char* szname, hap_callback cb, void* context) {
	return NEW_HOMEKIT_SERVICE(FAN2, .characteristics = (homekit_characteristic_t*[]) {
		NEW_HOMEKIT_CHARACTERISTIC(NAME, szname),
		NEW_HOMEKIT_CHARACTERISTIC(ROTATION_DIRECTION, 0, .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(cb, .context=context)),
		NEW_HOMEKIT_CHARACTERISTIC(ACTIVE, 0, .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(cb, .context=context),
),
		NEW_HOMEKIT_CHARACTERISTIC(ROTATION_SPEED, 0,  .min_step = (float[]) {16.666}, .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(cb, .context=context)

),
NULL
	});
}
`

Yurik72, 2020-06-03:
Hi i will check your suggestion with a fan. Could you please explain what is not working by default ?

Secondary, different setup devices on the same board: I just check documentation and I think that is possible for theory, I mean for the Apple it will be the same IP and MAC, hovewer different Ports, 
So, device will introduce different mDNS records, with the same IP, but different Ports and unique name and Apple should work with that
In my library everything should be duplicated. mDNS, pairing, listening. I'd say instead that code is quite accurace it will take some efforts  ,not small:)

Let's first finish wit the Fan amd second i will think, but not promises at the moment :)

Jeroen207, 2020-06-04:
Hi! By default you can't do anything with te fan controls. Because the characteristics: "ACTIVE" is not added.

Yurik72, 2020-06-05:
I have changed the library, Yoa are right FAN2 service required characteristic ACTIVE.
Currently, I have implemented FAN service and modified FAN
As well added examples for FAN, it works on my ESP32 board

closed 2020-06-05 by Yurik72