Yurik72/ESPHap #19: Constant pairing with a new device

bfaliszek, 2020-08-17:
For devices with HimeKit, the ability to pair with a new device is available all the time. Is it possible to turn on this option in this library? Is pairing possible only when no keys are stored in the memory?

Yurik72, 2020-08-18:
I'd say that is not true.  Library is designed definetelly to handle that situation. As soon as firts pairing is done, those information are saved. In my examples, I have used pairing.dat file in the SPIFFS file system.  Second start, device is knows  about pairing and do not iniates this process again.
So, for correct answers and investigation I need your code attached or at leats logs

bfaliszek, 2020-08-19:
You're right, after first pairing is done, the information is saved in pairing.dat file in the SPIFFS file system. Second start, device is knows about pairing and do not initiate this process again.

I would like to know if it is possible to leave the process active when you already have data saved in the pairing.dat file.

Yurik72, 2020-08-19:
I do not understand a case for that request. If device is paired it works as paired with Apple, otherwise it's waiting for  pairing..
If you want, you can remove pairing.dat and device will start pairing process again.
The background is mdns communication. One of the mdns keys (according to the apple docs)  is paired or not, then Apple behaviours according that.
In theory may be possible to have two mdns on the differents ports. But currently it's out of scope for this library

bfaliszek, 2020-08-21:
I want to be able to pair the ESP with a new device while it is already paired with another.
In this case, the data stored in pairing.dat would be overwritten by the new ones.

closed 2021-01-28 by bfaliszek