Yurik72/ESPHap #21: Pairing does not persist after reboot

JohnWickham, 2020-09-20:
I tried using the EspHapLed8266 example sketch with an ESP8266-12-F board. It compiles and uploads fine, pairing works, and accessory control is mostly great.

After rebooting the ESP8266 however, it forgets the pairing data entirely; HomeKit reports it as unresponsive, and it starts advertising an entirely different accessory for pairing.

Is this a known issue?

Yurik72, 2020-09-20:
Hi  definetelly something going wrong. Pairs information should be store under spiffs file system FIle name is "/pair.dat"
Probably you haven't formatted toyur spiffs file system or other problem  like wrong partition scheme
What you can check , collect the logs from  the method void init_hap_storage() and attach to the issue
Try to use another example [EspHapAdvancedLed](https://github.com/Yurik72/ESPHap/tree/master/examples/EspHapAdvancedLed) , this is provides web interface to file system. and you can see at least that file exists, As well yo can attach to the issue your logs.

I will not close an issue until your reply , but definetelyy this is is not, due to thousands devices working well

JohnWickham, 2020-09-20:
Thank you for your reply! Your hint about trying the EspHapAdvancedLed sample put me on the right track to solving this.

I was confused at first because I didn't see EspHapAdvancedLed in the list of example sketches included with ESPHap. Eventually I found out it's because I installed the library from Arduino’s Library Manager. It seems that the Library Manager installs an outdated version of the library that doesn’t include the AdvancedLed example, and may have some bugs related to the original issue.

Now after downloading the library from this repo’s releases page and installing it manually, it works perfectly. Thank you very much for your work on this project!

closed 2020-09-20 by JohnWickham