Yurik72/ESPHap #26: Can't compile code examples which include hapfilestorage.hpp

Steilwand, 2020-11-10:
Hi, 
thanks for your great work, really appreciated!
One problem though: 
whenever one of your examples includes hapfilestorage\hapfilestorage.hpp or hapweb\hap_webserver.hpp (like in EspHapAdvancedLed.ino), I get 

`Arduino: 1.8.13 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"
...
Alternatives for hapfilestorage\hapfilestorage.hpp: []
EspHapAdvancedDimmableLed:16:45: fatal error: hapfilestorage\hapfilestorage.hpp: No such file or directory
ResolveLibrary(hapfilestorage\hapfilestorage.hpp)
compilation terminated.
  -> candidates: []
Bibliothek SPIFFS in Version 1.0 im Ordner: /Users/bob/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPIFFS  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: /Users/bob/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FS  wird verwendet
Bibliothek ESPHap in Version 1.0 im Ordner: /Users/bob/arduino/libraries/ESPHap  wird verwendet
exit status 1
hapfilestorage\hapfilestorage.hpp: No such file or directory
`

[image]

I checked out the ESPHap master and installed it properly within my library folder (working on macOS):

[image]

I am really clueless here, any help would be great.

Yurik72, 2020-11-11:
Hi, probably issue is related to compilation on the macOs. It's not reproducable on my Windows
Therefore my few suggestions: (i do not have currently macOs for properly change a library)
1. Clone library directly from github into the Arduino Library folders (I suppose you did that already)
2. Try to changhe #include <hapfilestorage\hapfilestorage.hpp>
to #include <hapfilestorage/hapfilestorage.hpp>
3. If (2) not help Try directly include those file with full path  by changing 

#include <hapfilestorage\hapfilestorage.hpp>
to 
#include " /Users/bob/arduino/libraries/ESPHap/hapfilestorage/hapfilestorage.hpp"
yor case

If any helps. give me a hint or create pull request, i will do library changes

ESPdude, 2020-11-11:
As per https://gcc.gnu.org/onlinedocs/cpp/Include-Syntax.html#Include-Syntax, for maximum compatibility the forward slash ( / ) should be used within #include, so it should be

`#include <hapfilestorage/hapfilestorage.hpp>`

Yurik72, 2020-11-11:
> As per https://gcc.gnu.org/onlinedocs/cpp/Include-Syntax.html#Include-Syntax, for maximum compatibility the forward slash ( / ) should be used within #include, so it should be
> 
> `#include <hapfilestorage/hapfilestorage.hpp>`

If that solved problem on mcOs ?

ESPdude, 2020-11-11:
Yes, this solves the problem with that example.

Yurik72, 2020-11-12:
merged from pull request

closed 2020-11-12 by Yurik72