Yurik72/ESPHap #32: Thermostat

dn1984, 2020-11-25:
Hello. Sorry for my bad English. I want to ask about your library. It's easy to make a light bulb or temperature indicator turn off. But I can't figure out how to implement a full-fledged thermostat for switching (heating, cooling), temperature control and the like?? I add hap_setbase_accessorytype(homekit_accessory_category_thermostat); Then I read the description of characteristic.h /**
Defines that the accessory contains a thermostat.

Required Characteristics:
- CURRENT_HEATING_COOLING_STATE
- TARGET_HEATING_COOLING_STATE
- CURRENT_TEMPERATURE
- TARGET_TEMPERATURE
- TEMPERATURE_DISPLAY_UNITS

Optional Characteristics:
- NAME
- CURRENT_RELATIVE_HUMIDITY
- TARGET_RELATIVE_HUMIDITY
- COOLING_THRESHOLD_TEMPERATURE
- HEATING_THRESHOLD_TEMPERATURE
*/ 
But I can't figure out how to add all the Required Characteristics so that the iphone can see and show them? 

I add hapservice = hap_add_switch_service("OnOffThermostat", led_callback, (void*)&led_gpio);
hapservice = hap_add_temperature_service ("Temperature"); But how do I add a field to set the target temperature value? Or maybe I'm not doing it right?

In the example folder, there is an example of a thermostat, but it just displays the temperature and humidity sensors, this is not a full-fledged thermostat.

Yurik72, 2020-11-25:
Hello, You are right there is not thermostat, there is inly sensor
But i have creared another example with library changes
https://github.com/Yurik72/ESPHap/tree/master/examples/EspHap_ThermostatControl

Unfortunatelly , due to the time Without testing a logic, could you please test and work on that. I will appreciate a pull request if neccessary

closed 2020-11-25 by Yurik72

dn1984, 2020-11-26:
Hello. I checked your example for the thermostat. Set const char* thing_api_key= "11112222"; when you try to connect the thermostat via iphone, an error appears on the phone screen (the Accessory does not meet the requirements), after 1 attempt, the accessory is no longer visible to the iphone, you have to do erase_flash and download the program again. At the time of connection, the log shows Verification successful, secure session established, but the phone can't display the accessory. Can you help me figure it out? I attached the serial monitor log.

12:35:05.553 -> TLT-GRUP
12:35:06.168 -> ..
12:35:06.686 -> IP address: 
12:35:06.686 -> 192.168.10.20
12:35:06.686 -> Free heap: 204704
12:35:06.686 -> init_hap_storage>>> HomeKit: init_storage_ex size 0x581
12:35:06.686 -> >>> Home Integration: hap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
12:35:06.686 -> >>> Home Integration: hap_services added chararacteristic  1: 00000011-0000-1000-8000-0026BB765291 
12:35:06.686 -> >>> Home Integration: hap_services added chararacteristic  2: 00000011-0000-1000-8000-0026BB765291 
12:35:06.686 -> >>> Home Integration: hap_services added chararacteristic  3: 00000035-0000-1000-8000-0026BB765291 
12:35:06.733 -> >>> Home Integration: hap_services added chararacteristic  4: 00000036-0000-1000-8000-0026BB765291 
12:35:06.733 -> >>> Home Integration: hap_services added chararacteristic  5: 0000000F-0000-1000-8000-0026BB765291 
12:35:06.733 -> >>> Home Integration: hap_services added chararacteristic  6: 00000033-0000-1000-8000-0026BB765291 
12:35:06.733 -> >>> Home Integration: hap_services added chararacteristic  7: 0000000D-0000-1000-8000-0026BB765291 
12:35:06.733 -> >>> Home Integration: hap_services added chararacteristic  8: 00000012-0000-1000-8000-0026BB765291 
12:35:06.733 -> >>> Home Integration: hap_services added chararacteristic  9: 00000010-0000-1000-8000-0026BB765291 
12:35:06.779 -> >>> Home Integration: homekit_is_paired 0
12:35:06.779 -> >>> HomeKit: Starting server
12:35:06.779 -> >>> HomeKit: Using existing accessory ID: FE:A3:29:15:7A:DC
12:35:06.779 -> >>> HomeKit: Configuring mDNS
12:35:07.247 -> Setting web server
12:35:07.388 -> Web site http://192.168.10.20
12:35:07.388 -> File system http://192.168.10.20/browse
12:35:07.388 -> Update http://192.168.10.20/update
12:35:07.481 -> Tempnan  Hum:nan
12:35:07.481 -> Set default temp 20
12:35:07.481 -> Set default hum 50
12:35:07.481 -> sendToThingspeak start
12:35:07.481 -> https://www.borneland.com/sensor/post/
12:35:07.481 -> [HTTP] Begin https://www.borneland.com/sensor/post/
12:35:07.481 -> http begin start connection www.borneland.com
12:35:07.481 -> [HTTP] Start connection www.borneland.com443
12:35:08.847 -> http finish connection
12:35:09.034 -> writeToStream
12:35:09.034 -> processing _transferEncoding:0
12:35:09.034 -> postString api_key=tPmAT5Ab3j7F9&chip_id=14661474&temp=20.00&hum=50.00
12:35:09.034 -> http code returns 200
12:35:09.034 -> http code response string(4) "POST"
12:35:09.034 -> ---post content---array(0) {
12:35:09.034 -> }
12:35:09.034 -> 
12:35:12.514 -> Tempnan  Hum:nan
12:35:12.514 -> Set default temp 20
12:35:12.514 -> Set default hum 50
12:35:17.584 -> Tempnan  Hum:nan
12:35:17.584 -> Set default temp 20
12:35:17.584 -> Set default hum 50
12:35:31.465 -> >>> HomeKit: Got new client connection: 57
12:35:31.465 -> >>> HomeKit: [Client 57] Pair Setup 
12:35:31.465 -> >>> HomeKit: [Client 57] Pair Setup Step 1/3
12:35:31.465 -> >>> HomeKit: [Client 57] new pairing content
12:35:31.465 -> >>> HomeKit: started crypto
12:35:31.465 -> >>> HomeKit: Free heap: 172172
12:35:32.776 -> Tempnan  Hum:nan
12:35:32.776 -> Set default temp 20
12:35:32.776 -> Set default hum 50
12:35:32.962 -> >>> HomeKit: Calculating public key len=384
12:35:33.755 -> >>> HomeKit: Calculating public key result 0
12:35:33.755 -> >>> HomeKit: [Client 57] send_tlv_response
12:35:33.755 -> >>> HomeKit: [Client 57] send_tlv_response done
12:35:36.567 -> >>> HomeKit: [Client 57] Pair Setup 
12:35:36.567 -> >>> HomeKit: [Client 57] Pair Setup Step 2/3
12:35:37.841 -> Tempnan  Hum:nan
12:35:37.841 -> Set default temp 20
12:35:37.841 -> Set default hum 50
12:35:39.210 -> >>> HomeKit: [Client 57] Pair Setup 
12:35:39.210 -> >>> HomeKit: [Client 57] Pair Setup Step 3/3
12:35:39.775 -> >>> HomeKit: Added pairing with F4BA4965-2501-4223-984D-95742773E51D
12:35:39.822 -> >>> HomeKit: Configuring mDNS
12:35:39.822 -> >>> HomeKit: [Client 57] Successfully paired
12:35:39.869 -> >>> HomeKit: [Client 57] Closing client connection
12:35:39.869 -> >>> HomeKit: Got new client connection: 58
12:35:39.869 -> >>> HomeKit: [Client 58] Pair Verify Step 1/2
12:35:39.962 -> >>> HomeKit: [Client 58] Pair Verify Step 2/2
12:35:39.962 -> >>> HomeKit: [Client 58] Found pairing with F4BA4965-2501-4223-984D-95742773E51D
12:35:40.008 -> >>> HomeKit: [Client 58] Verification successful, secure session established
12:35:40.008 -> >>> HomeKit: [Client 58] Get Accessories
12:35:40.101 -> >>> HomeKit: [Client 58] Closing client connection
12:35:42.928 -> Tempnan  Hum:nan
12:35:42.928 -> Set default temp 20
12:35:42.928 -> Set default hum 50

Yurik72, 2020-11-26:
Hi, sorry for my mistakes, As i mentioned there is no time to test, hovewer I carefully check and correct library. You can see last changes done by the commit https://github.com/Yurik72/ESPHap/commit/35efcc7a1236195483135d69a025cab2ab48ac61

Hint, you don't need to erase a flash. See my instruction. you just need to delete file pair.dat, using internal browser and device will be ready to pair again