Yurik72/ESPHap #36: Compilation error of the LED Example

emanuelelaface, 2020-12-26:
I am using a OLIMEX ESP32-EVB. With Arduino IDE 1.8.13 and the latest ESPHap.
If I compile an Arduino led blinking example it works perfectly, the same if I compile the WIFI scan example I can scan the networks at home and have a list. I tried also the the ESP32 ChipID example and I can get my chip ID on the serial console, so I think the board is up and running without issues.

But when I run the EspHapLed I get several errors about cJSON and other components with multiple definitions.
I think this is due to the use of esp32 1.0.5-rc4 instead of the stable. That is due to the fact that the previous release does not run with the latest MacOS installation, so I guess that from the next release this error will be produced also by the stable. Is there something that I can do to avoid this error?

```
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_GetErrorPtr':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/json/cJSON/cJSON.c:78: multiple definition of `cJSON_GetErrorPtr'
libraries/ESPHap/cJSON.c.o:/Users/ema/Documents/Arduino/libraries/ESPHap/cJSON.c:72: first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_Version':
cJSON.c:(.text.cJSON_Version+0x0): multiple definition of `cJSON_Version'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_Version+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_InitHooks':
cJSON.c:(.text.cJSON_InitHooks+0x0): multiple definition of `cJSON_InitHooks'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_InitHooks+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_Delete':
cJSON.c:(.text.cJSON_Delete+0x0): multiple definition of `cJSON_Delete'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_Delete+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_SetNumberHelper':
cJSON.c:(.text.cJSON_SetNumberHelper+0x0): multiple definition of `cJSON_SetNumberHelper'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_SetNumberHelper+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_ParseWithOpts':
cJSON.c:(.text.cJSON_ParseWithOpts+0x0): multiple definition of `cJSON_ParseWithOpts'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_ParseWithOpts+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_Parse':
cJSON.c:(.text.cJSON_Parse+0x0): multiple definition of `cJSON_Parse'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_Parse+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_Print':
cJSON.c:(.text.cJSON_Print+0x0): multiple definition of `cJSON_Print'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_Print+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_PrintUnformatted':
cJSON.c:(.text.cJSON_PrintUnformatted+0x0): multiple definition of `cJSON_PrintUnformatted'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_PrintUnformatted+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_PrintBuffered':
cJSON.c:(.text.cJSON_PrintBuffered+0x0): multiple definition of `cJSON_PrintBuffered'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_PrintBuffered+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_PrintPreallocated':
cJSON.c:(.text.cJSON_PrintPreallocated+0x0): multiple definition of `cJSON_PrintPreallocated'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_PrintPreallocated+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_GetArraySize':
cJSON.c:(.text.cJSON_GetArraySize+0x0): multiple definition of `cJSON_GetArraySize'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_GetArraySize+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_GetArrayItem':
Multiple libraries were found for "WiFi.h"
 Used: /Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
cJSON.c:(.text.cJSON_GetArrayItem+0x0): multiple definition of `cJSON_GetArrayItem'
 Not used: /Users/ema/Documents/Arduino/libraries/WiFiNINA
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_GetArrayItem+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_GetObjectItem':
cJSON.c:(.text.cJSON_GetObjectItem+0x0): multiple definition of `cJSON_GetObjectItem'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_GetObjectItem+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_GetObjectItemCaseSensitive':
cJSON.c:(.text.cJSON_GetObjectItemCaseSensitive+0x0): multiple definition of `cJSON_GetObjectItemCaseSensitive'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_GetObjectItemCaseSensitive+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_HasObjectItem':
cJSON.c:(.text.cJSON_HasObjectItem+0x0): multiple definition of `cJSON_HasObjectItem'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_HasObjectItem+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_AddItemToArray':
cJSON.c:(.text.cJSON_AddItemToArray+0x0): multiple definition of `cJSON_AddItemToArray'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_AddItemToArray+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_AddItemToObject':
cJSON.c:(.text.cJSON_AddItemToObject+0x0): multiple definition of `cJSON_AddItemToObject'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_AddItemToObject+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_AddItemToObjectCS':
cJSON.c:(.text.cJSON_AddItemToObjectCS+0x0): multiple definition of `cJSON_AddItemToObjectCS'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_AddItemToObjectCS+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_AddItemReferenceToArray':
cJSON.c:(.text.cJSON_AddItemReferenceToArray+0x0): multiple definition of `cJSON_AddItemReferenceToArray'
[... 103 lines cut ...]
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_IsObject':
cJSON.c:(.text.cJSON_IsObject+0x0): multiple definition of `cJSON_IsObject'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_IsObject+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_IsRaw':
cJSON.c:(.text.cJSON_IsRaw+0x0): multiple definition of `cJSON_IsRaw'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_IsRaw+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_Compare':
cJSON.c:(.text.cJSON_Compare+0x0): multiple definition of `cJSON_Compare'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_Compare+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_malloc':
cJSON.c:(.text.cJSON_malloc+0x0): multiple definition of `cJSON_malloc'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_malloc+0x0): first defined here
[... cut ...]
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_ReplaceItemInObject+0x0): first defined here
/Users/ema/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_ReplaceItemInObjectCaseSensitive':
cJSON.c:(.text.cJSON_ReplaceItemInObjectCaseSensitive+0x0): multiple definition of `cJSON_ReplaceItemInObjectCaseSensitive'
libraries/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_ReplaceItemInObjectCaseSensitive+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board OLIMEX ESP32-EVB.
```

emanuelelaface, 2020-12-26:
Ok I find a workaround (maybe useful for other users with MacOS Big Sure for the moment) I removed the file `cJSON.c` from `libraries/ESPHap` because anyway there is the same set of definitions in the library of ESP32, so it is redundant.
In this way at least the LED example compiles and works perfectly. I think this conflict should be solved in a more elegant way (a check on the release of the lib?) but for the moment I can work with it.
Thanks for the amazing library, it is very useful.

varuncornelio, 2022-03-23:
> Ok I find a workaround (maybe useful for other users with MacOS Big Sure for the moment) I removed the file `cJSON.c` from `libraries/ESPHap` because anyway there is the same set of definitions in [...]

I've been trying to fix this error from 3 hours and after tons of useless comments, this is what helped me... Thank you!!!