Yurik72/ESPHap #37: Not able to pair with Homekit under iOS 14.4.

Technofrikus, 2020-12-29:
Hello,

this library is exactly what I was looking for. Easily create some lights to integrate into homekit without an additional bridge etc. Thanks a lot for this!

I tried several of the example sketches but were never able to successfully pair with HomeKit. I used my iPhone 12 mini with the first public beta of iOS 14.4. 
**Just now I used an old iPad with 12.4. installed and pairing worked without problem. So maybe they changed something?**

This is the Output on Serial:

`16:33:34.400 -> >>> HomeKit: Init server over
16:38:22.055 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 4)
16:38:22.055 -> >>> HomeKit: Got new client connection: local 192.168.177.135:5556, remote 192.168.177.172:56389
16:38:22.089 -> >>> HomeKit: [Client 1073698628] Got 129 incomming data, encrypted is 
16:38:22.089 -> >>> HomeKit: [Client 1073698628] Pair Setup Step 1/3
16:38:49.304 -> >>> HomeKit: [Client 1073698628] Got 582 incomming data, encrypted is 
16:38:49.338 -> >>> HomeKit: [Client 1073698628] Pair Setup Step 2/3
16:38:49.338 -> >>> HomeKit: [Client 1073698628] Computing SRP shared secret
16:39:02.471 -> >>> HomeKit: [Client 1073698628] Verifying peer's proof
16:39:02.471 -> >>> HomeKit: [Client 1073698628] Generating own proof
16:39:02.576 -> >>> HomeKit: [Client 1073698628] Got 284 incomming data, encrypted is 
16:39:02.576 -> >>> HomeKit: [Client 1073698628] Pair Setup Step 3/3
16:39:03.508 -> >>> HomeKit: Added pairing with 41E4BE6F-D3F6-4F78-AF13-8BF51D5F472F
16:39:03.579 -> >>> HomeKit: Free saved_preinit_pairing_context
16:39:03.579 -> >>> HomeKit: Configuring mDNS
16:39:03.579 -> >>> HomeKit: MDNS.restart: EspLed-59F3, IP: 192.168.177.135
16:39:03.579 -> >>> HomeKit: MDNS.begin: EspLed-59F3, IP: 192.168.177.135
16:39:03.616 -> >>> HomeKit: [Client 1073698628] Successfully paired
16:39:03.687 -> >>> HomeKit: [Client 1073698628] Disconnected!
16:39:03.687 -> >>> HomeKit: Free heap: 26472
16:39:03.720 -> >>> HomeKit: [Client 1073698628] Closing client connection
16:39:03.720 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 4)
16:39:03.720 -> >>> HomeKit: Got new client connection: local 192.168.177.135:5556, remote 192.168.177.172:56390
16:39:03.720 -> >>> HomeKit: [Client 1073698628] Got 162 incomming data, encrypted is 
16:39:03.754 -> >>> HomeKit: [Client 1073698628] Pair Verify Step 1/2
16:39:04.370 -> >>> HomeKit: Free heap: 25472
16:39:04.438 -> >>> HomeKit: [Client 1073698628] Got 251 incomming data, encrypted is 
16:39:04.438 -> >>> HomeKit: [Client 1073698628] Pair Verify Step 2/2
16:39:04.438 -> >>> HomeKit: [Client 1073698628] Found pairing with 41E4BE6F-D3F6-4F78-AF13-8BF51D5F472F
16:39:05.975 -> >>> HomeKit: [Client 1073698628] Verification successful, secure session established
16:39:05.975 -> >>> HomeKit: Free heap: 25280
16:39:06.041 -> >>> HomeKit: [Client 1073698628] Disconnected!
16:39:06.041 -> >>> HomeKit: Free heap: 26280
16:39:06.041 -> >>> HomeKit: [Client 1073698628] Closing client connection`

On the iOS side: I enter the pasword (all 1s) then it starts to pair but never finishes. I waited quite a bit, but nothing.
I have deleted the pair.dat through the webserver and then tried again several times, but it never finishes.

The QR-Code is not generated properly. After uploading the sketch the setup.html is 0 KB. After the failed pairing the QR-code is generated, but when I use it, Homekit tells me I have to reset the device, because it is already paired. But it is not visible in Homekit.

Setup: LOLin v3 NodeMCU ESP8622
MacOS Catalina 10.14.6

Yurik72, 2020-12-29:
Hello, 
1. I worked with the latest version 14.4 and do not have a problem. As my understanding you are using ESP8266 (not ESP8622). 
   So i mentioned in the comment that on ESP8266 could a problem with pairing process. In my experiense i had up to the 3 attempts. Problem somewhere in communication and i could not find a clear way how to finally fix it. Therefore my sugestion is try  again : delete pair.dat and pair, after that (pairing) device will work properly and stable
I see your logs attached and would say that is not a problem of 14.4. If device is not accepted by Apple it's happened on the step  Any how using a beta version I think not good idea. But if you paired on the old iPad you should see device i  the new iphone and it should works :)

2. QR  code I understand from your message is generated,. But defintelly if device is paired it will not pair again. Because it's communicate to apple (via mDNS protocol) about pairing. So, deleteting pair.dat will help, but there is not a difference how to pair. process is the same

3. Send me your sketch or give a hint which one you are using.
P.S. I'm using in most cases ESP32, the price is comparable but it is better and modern chip

Technofrikus, 2020-12-29:
1. Yes, sorry, that was a typo, I use a ESP83266. I read about that problem, that's why I tried several times. But it never worked for me :-/ Now that it is paired, it works fine everywhere, also with 14.4 :) 
2. The QR-Code is generated, but too late. Before pairing it is not there (0 kb), only after failed pairing it is generated, when it is no use anymore.
3. I use the AdvancedDimmableLED sketch from the examples (but also tried the simpleLED, AdvancedLED and switch). I only changed the Pin # for now. Wanted to get this part working before changing more. I want to use a 1W or 10W LED with this, not sure yet.
4. Noted, will buy the ESP32 next time. But I have 4x ESP8266 here already I would like to use :) But with the workaround of the old iPad I am happy. Just wanted to let you know, in case. Thanks again!

Yurik72, 2020-12-29:
Evening,
1. Ok, seems that is well known problems, but if it worked finally, I'm happy
2. I will check, thinks that is bug
3. Ok, thanks
4. Yes, shure it will works after complicated pairing process

eford321, 2021-01-21:
I am seeing the same pairing issue with iOS 14.3 and ESP8266 (Wemos mini). Pairing does not complete on the phone, but appears to complete on the ESP8266. Using wolfSSL 3.13.0 and typing in the 111-11-111 code.
Here's the serial output using the example "EspHapAdvancedLed". (I had the same result with EspHapLed8266 example):

>>> HomeKit: init_storage_ex size 0x1c1
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Free heap: 29648
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x200000
>>> HomeKit: storage init 1
>>> HomeKit: if r
>>> HomeKit: generating id
>>> HomeKit: Generated new accessory ID: 33:3C:B0:37:B9:4F
>>> HomeKit: Generated new accessory ID: 1073695636
>>> HomeKit:  key generated
>>> HomeKit:  key saved
>>> HomeKit: Generated new accessory key
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 27024
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: MDNS.begin: EspLed-D654, IP: 192.168.1.26
>>> HomeKit: Init server over
>>> HomeKit: WiFiServer receives a new client (current 0, max 4)
>>> HomeKit: Got new client connection: local 192.168.1.26:5556, remote 192.168.1.15:60550
>>> HomeKit: [Client 1073700140] Got 129 incomming data, encrypted is 
>>> HomeKit: [Client 1073700140] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700140] Got 536 incomming data, encrypted is 
>>> HomeKit: [Client 1073700140] Got 46 incomming data, encrypted is 
>>> HomeKit: [Client 1073700140] Pair Setup Step 2/3
>>> HomeKit: [Client 1073700140] Computing SRP shared secret
>>> HomeKit: [Client 1073700140] Verifying peer's proof
>>> HomeKit: [Client 1073700140] Generating own proof
>>> HomeKit: [Client 1073700140] Got 284 incomming data, encrypted is 
>>> HomeKit: [Client 1073700140] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: Configuring mDNS
>>> HomeKit: MDNS.restart: EspLed-D654, IP: 192.168.1.26
>>> HomeKit: MDNS.begin: EspLed-D654, IP: 192.168.1.26
>>> HomeKit: [Client 1073700140] Successfully paired
>>> HomeKit: [Client 1073700140] Disconnected!
>>> HomeKit: Free heap: 25352
>>> HomeKit: [Client 1073700140] Closing client connection
>>> HomeKit: WiFiServer receives a new client (current 0, max 4)
>>> HomeKit: Got new client connection: local 192.168.1.26:5556, remote 192.168.1.15:60551
>>> HomeKit: [Client 1073700140] Got 162 incomming data, encrypted is 
>>> HomeKit: [Client 1073700140] Pair Verify Step 1/2
>>> HomeKit: Free heap: 24352
>>> HomeKit: [Client 1073700140] Got 251 incomming data, encrypted is 
>>> HomeKit: [Client 1073700140] Pair Verify Step 2/2
>>> HomeKit: [Client 1073700140] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
>>> HomeKit: [Client 1073700140] Verification successful, secure session established
>>> HomeKit: Free heap: 24160
>>> HomeKit: [Client 1073700140] Disconnected!
>>> HomeKit: Free heap: 25048
>>> HomeKit: [Client 1073700140] Closing client connection

after deleting pair.dat 2 times it worked. 

Here's the serial output when it worked:
>>> HomeKit: init_storage_ex size 0x1c1
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
[... 110 lines cut ...]
>>> HomeKit: [Client 1073703068] Get Characteristics
>>> HomeKit: WiFiServer receives a new client (current 4, max 4)
>>> HomeKit: No more room for client connections (max 4)
>>> HomeKit: WiFiServer receives a new client (current 4, max 4)
>>> HomeKit: No more room for client connections (max 4)
>>> HomeKit: [Client 1073698980] Got 194 incomming data, encrypted is 
>>> HomeKit: [Client 1073698980] Update Characteristics
>>> HomeKit: [Client 1073698980] Got 192 incomming data, encrypted is 
>>> HomeKit: [Client 1073698980] Update Characteristics
led_callback
set_led
notify hap
>>> HomeKit: [Client 1073698980] Got 192 incomming data, encrypted is 
>>> HomeKit: [Client 1073698980] Update Characteristics
led_callback
set_led
notify hap
>>> HomeKit: [Client 1073698980] Got 192 incomming data, encrypted is 
>>> HomeKit: [Client 1073698980] Update Characteristics
led_callback
set_led
notify hap

Please let me know if I can provide any more information.
Thanks.

Yurik72, 2021-01-21:
Hi eford321, 
this is i would say wellknown issue in the pairing process, see above. Next few pairing will finished sucessfully
I know, device is finish pairing, but connectio closed
HomeKit: [Client 1073700140] Disconnected!

I'm working on that, how to improve. but now on ESP8266 only a way  to start pairing again

eford321, 2021-01-29:
This appears to be a connection timing problem with mDNS.  The mDNS process is restarted and creates a race condition at the time apple is trying to connect to the paired device. The mDNS host is not resolvable until the race condition ends. Apple doesn't appear to try the connection again.

See this log with MDNS debug turned on:

>>> HomeKit: [Client 1073698484] Got 581 incomming data, encrypted is 
>>> HomeKit: [Client 1073698484] Pair Setup Step 2/3
>>> HomeKit: [Client 1073698484] Computing SRP shared secret
>>> HomeKit: [Client 1073698484] Verifying peer's proof
>>> HomeKit: [Client 1073698484] Generating own proof
>>> HomeKit: [Client 1073698484] Got 283 incomming data, encrypted is 
>>> HomeKit: [Client 1073698484] Pair Setup Step 3/3
[MDNSResponder] _callProcess (109139, triggered by: 192.168.1.213)
[MDNSResponder] _parseMessage (Time: 109140 ms, heap: 24384 bytes, from 192.168.1.213(5353), to 224.0.0.251(5353))
[MDNSResponder] _parseMessage: Done (Succeeded after 4 ms, ate 0 bytes, remaining 24384)

>>> HomeKit: Added pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: Configuring mDNS
[MDNSResponder] _announce: Announcing host Blind (content 0x3)
[MDNSResponder] _announce: Announcing service Blind-2488.hap.tcp (content 240)
[MDNSResponder] _sendMDNSMessage_Multicast: Will send to '224.0.0.251'.
[MDNSResponder] _prepareMDNSMessage
[MDNSResponder] _prepareMDNSMessage: ID:0 QR:1 OP:0 AA:1 TC:0 RD:0 RA:0 R:0 QD:0 AN:6 NS:0 AR:0
[MDNSResponder] _writeMDNSAnswer_A (192.168.1.27)
[MDNSResponder] _writeMDNSAnswer_PTR_IP4 (192.168.1.27)
[MDNSResponder] _writeMDNSAnswer_PTR_TYPE
[MDNSResponder] _writeMDNSAnswer_PTR_NAME
[MDNSResponder] _writeMDNSAnswer_SRV
[MDNSResponder] _writeMDNSAnswer_TXT
>>> HomeKit: MDNS.restart: Blind-2488, IP: 192.168.1.27
[MDNSResponder] _allocUDPContext
>>> HomeKit: MDNS.begin: Blind-2488, IP: 192.168.1.27
[MDNSResponder] addService: FAILED to add 'Blind-2488.hap.tcp'!
[MDNSResponder] addService: FAILED to add 'Blind-2488.hap.tcp'!
[MDNSResponder] setDynamicServiceTxtCallback: FAILED!
[MDNSResponder] addServiceTxt: FAILED for 'md=EspHapLed'!
[MDNSResponder] addServiceTxt: FAILED for 'pv=1.0'!
[MDNSResponder] addServiceTxt: FAILED for 'id=24:88:B4:58:BF:1F'!
[MDNSResponder] addServiceTxt: FAILED for 'c#=1'!
[MDNSResponder] addServiceTxt: FAILED for 's#=1'!
[MDNSResponder] addServiceTxt: FAILED for 'sf=0'!
[MDNSResponder] addServiceTxt: FAILED for 'ff=0'!
[MDNSResponder] addServiceTxt: FAILED for 'ci=14'!
[MDNSResponder] addServiceTxt: FAILED for 'sh=nTfY+A=='!
[MDNSResponder] _announce: FAILED!
[MDNSResponder] _updateProbeStatus: Starting host probing...
>>> HomeKit: [Client 1073698484] Successfully paired
>>> HomeKit: [Client 1073698484] Disconnected!
>>> HomeKit: Free heap: 28256
>>> HomeKit: [Client 1073698484] Closing client connection
[MDNSResponder] _callProcess (109525, triggered by: 192.168.1.15)
[MDNSResponder] _parseMessage (Time: 109526 ms, heap: 29528 bytes, from 192.168.1.15(5353), to 224.0.0.251(5353))
[MDNSResponder] _readRRQuestion
[MDNSResponder] _readRRQuestion Blind-2488._hap._tcp.local Type:0x0021 Class:0x0001 Unicast
[MDNSResponder] _replyMaskForService(Blind-2488.hap.tcp): 0x80
[MDNSResponder] _parseQuery: Possible race-condition for service domain Blind-2488.hap.tcp detected while probing.
[MDNSResponder] _parseQuery: Unicast response for 192.168.1.15!
[MDNSResponder] _readRRAnswer: -empty- Type:0x0029 Class:0x05A0 TTL:4500, RDLength:18 generic 
[MDNSResponder] _parseQuery: No reply needed
[MDNSResponder] _parseQuery: UNSOLVED tiebreak-need for service domain (Blind-2488.hap.tcp)
[MDNSResponder] _parseMessage: Done (Succeeded after 56 ms, ate 0 bytes, remaining 29528)

[MDNSResponder] _callProcess (109644, triggered by: 192.168.1.15)
[MDNSResponder] _parseMessage (Time: 109645 ms, heap: 29536 bytes, from 192.168.1.168(5353), to 224.0.0.251(5353))
[MDNSResponder] _parseMessage: Done (Succeeded after 4 ms, ate 0 bytes, remaining 29536)

[MDNSResponder] _sendHostProbe (Blind, 109668)
[MDNSResponder] _sendMDNSMessage_Multicast: Will send to '224.0.0.251'.
[MDNSResponder] _prepareMDNSMessage
[MDNSResponder] _prepareMDNSMessage: ID:0 QR:0 OP:0 AA:0 TC:0 RD:0 RA:0 R:0 QD:1 AN:0 NS:1 AR:0
[... 153 lines cut ...]
[MDNSResponder] _writeMDNSAnswer_PTR_TYPE
[MDNSResponder] _writeMDNSAnswer_PTR_NAME
[MDNSResponder] _writeMDNSAnswer_SRV
[MDNSResponder] _writeMDNSAnswer_TXT
[MDNSResponder] _writeMDNSAnswer_A (192.168.1.27)
[MDNSResponder] _updateProbeStatus: Announcing service Blind-2488.hap.tcp (4)

[MDNSResponder] _announce: Announcing host Blind (content 0x3)
[MDNSResponder] _sendMDNSMessage_Multicast: Will send to '224.0.0.251'.
[MDNSResponder] _prepareMDNSMessage
[MDNSResponder] _prepareMDNSMessage: ID:0 QR:1 OP:0 AA:1 TC:0 RD:0 RA:0 R:0 QD:0 AN:2 NS:0 AR:0
[MDNSResponder] _writeMDNSAnswer_A (192.168.1.27)
[MDNSResponder] _writeMDNSAnswer_PTR_IP4 (192.168.1.27)
[MDNSResponder] _updateProbeStatus: Announcing host (5).

[MDNSResponder] _announceService: Announcing service Blind-2488.hap.tcp (content 0xF0)
[MDNSResponder] _sendMDNSMessage_Multicast: Will send to '224.0.0.251'.
[MDNSResponder] _prepareMDNSMessage
[MDNSResponder] _prepareMDNSMessage: ID:0 QR:1 OP:0 AA:1 TC:0 RD:0 RA:0 R:0 QD:0 AN:4 NS:0 AR:1
[MDNSResponder] _writeMDNSAnswer_PTR_TYPE
[MDNSResponder] _writeMDNSAnswer_PTR_NAME
[MDNSResponder] _writeMDNSAnswer_SRV
[MDNSResponder] _writeMDNSAnswer_TXT
[MDNSResponder] _writeMDNSAnswer_A (192.168.1.27)
[MDNSResponder] _updateProbeStatus: Announcing service Blind-2488.hap.tcp (5)

eford321, 2021-01-29:
Also, from repeated testing, success rate is about 25% instead of the 3rd attempt. If the race condition is resolved prior to the connection attempt the connection is successful.

Yurik72, 2021-01-29:
Thanks for the hints and attached logs ! Looks that problem is found , I will check it and think about the way of solving.

Yurik72, 2021-01-30:
Hi eford321, you really show me a right way !!!
I did small changes, to skip all mDNS reinit during the pairing and after testing 10 attempts  finished sucess from 10 attempts whicc is 100%.
Currently i see that paired flag  in mDNS record is not big problem for the apple during the paired process(maybe since ios 14) and both device paired successfull.

closed 2021-01-30 by Yurik72

eford321, 2021-01-30:
I am happy I could help!
I didn’t see any changes in the repository except for the readme. Are there other changes in your code coming. 
I am finishing up on the window covering example and will want to use the fixes you made. Thanks!

Yurik72, 2021-01-30:
Hi commit are in repo. As well tagged by v1.0.7
Really works fine, thanks and confirm

eford321, 2021-01-30:
Got it! Thanks again.

Yurik72, 2021-01-30:
Keep me informed, you really give me a right root cause