Yurik72/ESPHap #39: Compiling with platformio

lyuq-a, 2021-01-10:
Hello!

You made a great library. Everything works with Arduino IDE, but Arduino IDE is absolutely not convenient to use. I prefer to use Platformio IDE for my home automation. Some strange things happens when using this library with Platformio. Сompiler does not see some functions from homeintegration.h.

ESPHap example EspHapLed for ESP32 compiles with Platformio, but simple modifications of this example gives error while compiling with Platformio (compiles well with Arduino IDE).

Modification is the replacement of an accessory from lightbulb to switch:
`hap_setbase_accessorytype(homekit_accessory_category_switch);`
`hap_initbase_accessory_service("host", "Yurik72", "0", "EspHapLed", "1.0");`
`hapservice = hap_add_switch_service("FAN", led_callback, (void*)&led_gpio);`

This giving next error:
`src/main.cpp: In function 'void setup()':`
`src/main.cpp:152:76: error: 'hap_add_switch_service' was not declared in this scope`

Yurik72, 2021-01-14:
Could you please get the latest version, I did a few time ago some changes also related for Mac OS Catalina and give me a hint after.
Thanks

lyuq-a, 2021-01-14:
I'm using the latest version.
I've installed external GCC yesterday using this instructions:
https://discussions.apple.com/thread/8336714
It helped, now everything compiles well with Platformio.

Thanks.

Yurik72, 2021-01-15:
Ok, issue is closed

closed 2021-01-15 by Yurik72