Yurik72/ESPHap #40: Example for a Garagedoor

archiolochos, 2021-01-11:
first of all: thanks for that great piece of software! i tried some examples and all of them work really well.

when i was looking for a garagedoor-example i was not able to find one :-(
i tried to build one by myself but failed. so is there anyone out there who could show me how to get an garagedoor to wirk?

Yurik72, 2021-01-14:
Hi,  nice to hear positive feedback.
Due to limit of my time. I can add generic definition to the library for Garage door and probably you can help me to implement example of skecth. after generic definition it will be more easy.
If agreed i will notify when library will contains generic methods for Garage door

Yurik72, 2021-01-15:
Hi again
1. I've changed library within public methods for Garage door
2. I've created initial sample ,  without testing, just showing integration into the Applle home kit

So, you can concentrate to adjust a sample how it should manage garage door mechanism. 
I will appreciate a pull request with full function sample

archiolochos, 2021-01-18:
thanks for you work. i will try things with your changes and will give you some feedback.