Yurik72/ESPHap #41: Window Covering - Roller Blind example

eford321, 2021-01-21:
I am interested in converting my existing roller blinds from MQTT/Homebridge to ESPHap.  

Is it difficult to add the windowCovering type to ESPHap?  I would like to try adding my existing code for stepper control and blind calibration to one of your examples.

Yurik72, 2021-01-21:
Hi definetelly should be simple. If you provide some initial example with a pull request i wiil help you how to integrate to apple

eford321, 2021-01-21:
Thanks.
I'll work on an example following the generic garage door example you created.  
I'm new to GitHub pull requests, so I'll look into that.

Yurik72, 2021-02-02:
I have provided basi example for windows covering
Without real logic how to handle a motor by GPIOs
Appreciate for pull request

closed 2021-02-02 by Yurik72

eford321, 2021-02-02:
Thanks!  
I have a working example in my fork that I have been testing.  I will compare your example to see if I am taking the right approach. 
https://github.com/eford321/ESPHap/tree/master/examples/ESPHapWindowCovering 
I am planning to document the assembly of a complete DIY blind and was going to submit a merge/pull request when I have it fully cleaned up.
I am currently struggling with the ESP8266 and session limitations. I bumped the session limit from 4 to 8 but run out of heap. It works at 4, but I have 6 devices trying to establish connections. I hope to get the ESP8266 working in ESPHap as I already have 8 blinds in my house running on that board via MQTT/Homebridge.  
 I am ordering a few ESP32s to make sure my works on that board. The ESP32 is a better choice for anyone building a blind. In addition to better performance, it has enough IO to drive 3 stepper motors.

eford321, 2021-02-02:
The heap problem might have been from me duplicating accessory definitions. Testing with 8 sessions again to see how it behaves.  
I'm testing with sessions established from AppleTV (hub), iPhone, iPad, 2 Macs, and 2 HomePod mini's.

Yurik72, 2021-02-02:
Good evening, give me a sketch, I think I understood a potential problem and will be happy to provide conclusion/solution

eford321, 2021-02-03:
my sketch is here: https://github.com/eford321/ESPHap/blob/master/examples/ESPHapWindowCover/ESPHapWindowCover.ino

I am compiling this with "#define HOMEKIT_MAX_CLIENTS  8" in arduino_homekit_server.cpp

I am still seeing a memory leak? Also sessions begin disconnecting and reconnecting after running for about 10 minutes.

Here's the log from boot to crash:

16:39:34.615 -> *WM: [3] WiFi station enable 
16:39:34.651 -> *WM: [3] enableSTA PERSISTENT ON 
16:39:34.787 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
16:39:40.392 -> *WM: [2] Connection result: WL_CONNECTED
16:39:40.392 -> *WM: [3] lastconxresult: WL_CONNECTED
16:39:40.392 -> *WM: [1] Connect to new AP [SUCCESS] 
16:39:40.392 -> *WM: [1] Got IP Address: 
16:39:40.392 -> *WM: [1] 192.168.1.27 
16:39:40.427 -> *WM: [2] disconnect configportal 
16:39:41.413 -> *WM: [2] restoring usermode STA
16:39:41.413 -> *WM: [2] wifi status: WL_CONNECTED
16:39:41.413 -> *WM: [2] wifi mode: STA
16:39:41.413 -> *WM: [1] config portal exiting 
16:39:41.413 -> *WM: [3] unloading 
16:39:41.413 -> Free heap: 29696
16:39:57.843 -> >>> HomeKit: init_storage_ex size 0x1c1
16:39:57.843 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
16:39:57.843 -> >>> Home Integration: hap_services added chararacteristic  1: 6D 
16:39:57.843 -> >>> Home Integration: hap_services added chararacteristic  2: 7C 
16:39:57.843 -> >>> Home Integration: hap_services added chararacteristic  3: 72 
16:39:57.843 -> >>> Home Integration: homekit_is_paired 0
16:39:57.843 -> >>> HomeKit: Free heap: 26632
16:39:57.901 -> >>> HomeKit: Starting server
16:39:57.901 -> >>> HomeKit: Formatting flash at 0x200000
16:39:57.990 -> >>> HomeKit: storage init 1
16:39:57.990 -> >>> HomeKit: if r
16:39:57.990 -> >>> HomeKit: generating id
16:39:57.990 -> >>> HomeKit: Generated new accessory ID: D7:6F:93:46:03:12
16:39:57.990 -> >>> HomeKit: Generated new accessory ID: 1073693764
16:39:57.990 -> >>> HomeKit:  key generated
16:39:58.119 -> >>> HomeKit:  key saved
16:39:58.119 -> >>> HomeKit: Generated new accessory key
16:39:58.184 -> >>> HomeKit: find pairings
16:39:58.184 -> >>> HomeKit: Preinit pairing context
16:39:58.184 -> >>> HomeKit: Using user-specified password: 111-11-111
16:39:58.184 -> >>> HomeKit: Free heap: 24008
16:40:04.235 -> >>> HomeKit: Calculating public key len=384
16:40:07.328 -> >>> HomeKit: Calculating public key result 0
16:40:07.361 -> >>> HomeKit: arduino_homekit_preinit success
16:40:07.361 -> >>> HomeKit: Configuring mDNS
16:40:07.361 -> >>> HomeKit: MDNS.begin: BLND-E79A, IP: 192.168.1.27
16:40:07.361 -> >>> HomeKit: Init server over
16:40:17.563 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
16:40:17.563 -> >>> HomeKit: Got new client connection: local 192.168.1.27:5556, remote 192.168.1.15:56047
16:40:17.563 -> >>> HomeKit: [Client 1073702644] Got 127 incomming data, encrypted is 
16:40:17.563 -> >>> HomeKit: [Client 1073702644] Pair Setup Step 1/3
16:40:20.717 -> >>> HomeKit: [Client 1073702644] Got 580 incomming data, encrypted is 
16:40:20.717 -> >>> HomeKit: [Client 1073702644] Pair Setup Step 2/3
16:40:20.717 -> >>> HomeKit: [Client 1073702644] Computing SRP shared secret
16:40:33.881 -> >>> HomeKit: [Client 1073702644] Verifying peer's proof
16:40:33.881 -> >>> HomeKit: [Client 1073702644] Generating own proof
16:40:34.072 -> >>> HomeKit: [Client 1073702644] Got 282 incomming data, encrypted is 
16:40:34.072 -> >>> HomeKit: [Client 1073702644] Pair Setup Step 3/3
16:40:34.840 -> >>> HomeKit: Added pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
16:40:34.911 -> >>> HomeKit: Free saved_preinit_pairing_context
16:40:34.911 -> >>> HomeKit: [Client 1073702644] Successfully paired
16:40:35.053 -> >>> HomeKit: [Client 1073702644] Disconnected!
16:40:35.053 -> >>> HomeKit: Free heap: 25008
16:40:35.053 -> >>> HomeKit: [Client 1073702644] Closing client connection
16:40:35.260 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
16:40:35.260 -> >>> HomeKit: Got new client connection: local 192.168.1.27:5556, remote 192.168.1.15:56048
16:40:35.260 -> >>> HomeKit: [Client 1073702644] Got 160 incomming data, encrypted is 
[... 785 lines cut ...]
0x40282b4f: chip_v6_unset_chanfreq at ?? line ?
0x4023e858: _vfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x40100b13: free at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 398
0x4023f68f: _strndup_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/string/strndup_r.c line 20
0x40282b20: chip_v6_unset_chanfreq at ?? line ?
0x402444ca: vprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vprintf.c line 40
0x4010088b: umm_free_core at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 351
0x40202ff0: crypto_ed25519_verify at /Users/evan/Documents/Arduino/libraries/ESPHap/crypto.c line 476
0x4022451e: homekit_server_on_pair_verify(_client_context_t*, unsigned char const*, unsigned int) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 1916
0x4023e858: _vfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x40281ccb: chip_v6_unset_chanfreq at ?? line ?
0x40281cca: chip_v6_unset_chanfreq at ?? line ?
[... cut ...]
0x4022a4c5: __delay at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 57
0x40226f80: hap_homekit_loop() at /Users/evan/Documents/Arduino/libraries/ESPHap/homekitintegrationcpp.cpp line 18
0x40217ef5: loop at /Users/evan/Documents/Arduino/EspHap_WindowCover/EspHap_WindowCover.ino line 490
0x40229ef0: loop_wrapper() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
0x40100dcd: cont_wrapper at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/cont.S line 81

Yurik72, 2021-02-03:
Hi, looks like memory issue. But for me is strange that all connected device is trying to Finish Pair verify process. Probably problem is inside that. 
It should not happen, when device is paired once and than restarted (normal behaviour) means all connected cliens will work on Charachteristic getting and update
Any further comment appreciated

Yurik72, 2021-02-03:
Actually i see a way how to improve, there is a big memory pagenation/reallocation which i can decrease

Yurik72, 2021-02-03:
Hi again. I found a rooms ho to improve memory pagination.
Today i will finish for ESP32. For ESP8266 i will do next few days due to my current ESP8266 is broken :(
Current testing on ESP32 shows good result

eford321, 2021-02-03:
Hello. 
If you need help testing anything on the ESP8266, let me know. 

FYI, There was no interaction with the device on the log. Just paired and left idle until the ESP8266 ran out of memory to allocate.  Let me know if there is any other debug settings you would like me to log.

Yurik72, 2021-02-03:
Yes, this is due to the heap size :(
I found a way how to reuse a buffer between connection as well to reduce memory pagination
Is not yet commited. need liitle more time

Yurik72, 2021-02-04:
Hi, i did refactoring to minize memory fragmentation and usage, you can get latest version (no tag)
I have added new define, instead of
#define HOMEKIT_MAX_CLIENTS  4
#define CLIENT_CONTEXT_CACHE_SIZE 4   //-> How many context will be cached , to decrease fragmentation due creation and initialization

Will wait for the feedback..

Next couple of improvement will be done, like proper handling of  wifi lost connection, currenly it's works , but can be better

eford321, 2021-02-04:
Hello. Thanks for the update. There is a big improvement in memory usage.  After 20 minutes, free heap is around 20000. 

Is there a reason for limiting the HOMEKIT_MAX_CLIENTS to 4 instead of the 8 connections in the apple spec? Is it just for memory conservation?  From my testing with 4, my fifth device just shows a blank icon in the home app for this device.

Here's a log from the first 20+ minutes:

13:16:01.406 -> Free heap: 28960
13:16:17.797 -> >>> HomeKit: init_storage_ex size 0x1c1
13:16:17.797 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
13:16:17.797 -> >>> Home Integration: hap_services added chararacteristic  1: 6D 
13:16:17.867 -> >>> Home Integration: hap_services added chararacteristic  2: 7C 
13:16:17.867 -> >>> Home Integration: hap_services added chararacteristic  3: 72 
13:16:17.867 -> >>> Home Integration: homekit_is_paired 0
13:16:17.867 -> >>> HomeKit: Free heap: 25816
13:16:17.867 -> >>> HomeKit: Starting server
13:16:17.867 -> >>> HomeKit: Formatting flash at 0x200000
13:16:18.043 -> >>> HomeKit: storage init 1
13:16:18.043 -> >>> HomeKit: if r
13:16:18.043 -> >>> HomeKit: generating id
13:16:18.043 -> >>> HomeKit: Generated new accessory ID: 66:06:9F:10:B8:B0
13:16:18.043 -> >>> HomeKit: Generated new accessory ID: 1073689060
13:16:18.043 -> >>> HomeKit:  key generated
13:16:18.043 -> >>> HomeKit:  key saved
13:16:18.139 -> >>> HomeKit: Generated new accessory key
13:16:18.139 -> >>> HomeKit: find pairings
13:16:18.139 -> >>> HomeKit: Preinit pairing context
13:16:18.229 -> >>> HomeKit: Using user-specified password: 111-11-111
13:16:18.229 -> >>> HomeKit: Free heap: 20720
13:16:24.231 -> >>> HomeKit: Calculating public key len=384
13:16:27.344 -> >>> HomeKit: Calculating public key result 0
13:16:27.344 -> >>> HomeKit: arduino_homekit_preinit success
13:16:27.344 -> >>> HomeKit: Configuring mDNS
13:16:27.344 -> >>> HomeKit: MDNS.begin: DRB1-9A42, IP: 192.168.1.27
13:16:27.402 -> >>> HomeKit: Init server over
13:16:37.195 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 4)
13:16:37.195 -> >>> HomeKit: Got new client connection: local 192.168.1.27:5556, remote 192.168.1.15:58016
13:16:37.289 -> >>> HomeKit: [Client 1073705932] Got client content from the cache 0
13:16:37.289 -> >>> HomeKit: [Client 1073705932] new client accepted
13:16:37.289 -> >>> HomeKit: [Client 1073705932] Got 127 incomming data, encrypted is 
13:16:37.289 -> >>> HomeKit: [Client 1073705932] Pair Setup Step 1/3
13:16:40.568 -> >>> HomeKit: [Client 1073705932] Got 580 incomming data, encrypted is 
13:16:40.568 -> >>> HomeKit: [Client 1073705932] Pair Setup Step 2/3
13:16:40.568 -> >>> HomeKit: [Client 1073705932] Computing SRP shared secret
13:16:53.786 -> >>> HomeKit: [Client 1073705932] Verifying peer's proof
13:16:53.786 -> >>> HomeKit: [Client 1073705932] Generating own proof
13:16:54.100 -> >>> HomeKit: [Client 1073705932] Got 282 incomming data, encrypted is 
13:16:54.100 -> >>> HomeKit: [Client 1073705932] Pair Setup Step 3/3
13:16:54.936 -> >>> HomeKit: Added pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
13:16:55.007 -> >>> HomeKit: Free saved_preinit_pairing_context
13:16:55.007 -> >>> HomeKit: [Client 1073705932] Successfully paired
13:16:55.108 -> >>> HomeKit: [Client 1073705932] Disconnected!
13:16:55.108 -> >>> HomeKit: [Client 1073705932] Closing client connection
13:16:55.147 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 4)
13:16:55.147 -> >>> HomeKit: Got new client connection: local 192.168.1.27:5556, remote 192.168.1.15:58017
13:16:55.147 -> >>> HomeKit: [Client 1073689084] Got client content from the cache 0
13:16:55.147 -> >>> HomeKit: [Client 1073689084] new client accepted
13:16:55.147 -> >>> HomeKit: [Client 1073689084] Got 160 incomming data, encrypted is 
13:16:55.147 -> >>> HomeKit: [Client 1073689084] Pair Verify Step 1/2
13:16:55.483 -> >>> HomeKit: Free heap: 22344
13:16:55.519 -> >>> HomeKit: [Client 1073689084] Got 249 incomming data, encrypted is 
13:16:55.519 -> >>> HomeKit: [Client 1073689084] Pair Verify Step 2/2
13:16:55.519 -> >>> HomeKit: [Client 1073689084] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
13:16:56.326 -> >>> HomeKit: [Client 1073689084] Verification successful, secure session established
13:16:56.326 -> >>> HomeKit: Free heap: 22280
13:16:56.360 -> >>> HomeKit: [Client 1073689084] Got 80 incomming data, encrypted is 
13:16:56.360 -> >>> HomeKit: [Client 1073689084] Get Accessories
13:16:56.535 -> >>> HomeKit: [Client 1073689084] Got 249 incomming data, encrypted is 
13:16:56.570 -> >>> HomeKit: [Client 1073689084] Update Characteristics
13:16:56.570 -> >>> HomeKit: [Client 1073689084] Got 249 incomming data, encrypted is 
13:16:56.570 -> >>> HomeKit: [Client 1073689084] Update Characteristics
[... 300 lines cut ...]
13:51:10.589 -> >>> HomeKit: [Client 1073702724] Closing client connection
13:51:14.126 -> >>> HomeKit: WiFiServer receives a new client (current 3, max 4)
13:51:14.126 -> >>> HomeKit: Got new client connection: local 192.168.1.27:5556, remote 192.168.1.13:54174
13:51:14.126 -> >>> HomeKit: [Client 1073703956] Got client content from the cache 2
13:51:14.126 -> >>> HomeKit: [Client 1073703956] new client accepted
13:51:14.126 -> >>> HomeKit: [Client 1073703956] Got 160 incomming data, encrypted is 
13:51:14.126 -> >>> HomeKit: [Client 1073703956] Pair Verify Step 1/2
13:51:14.475 -> >>> HomeKit: Free heap: 20000
[... cut ...]
13:51:15.407 -> >>> HomeKit: No more room for client connections (max 4)
13:51:15.501 -> >>> HomeKit: WiFiServer receives a new client (current 4, max 4)
13:51:15.501 -> >>> HomeKit: No more room for client connections (max 4)
13:51:15.575 -> >>> HomeKit: WiFiServer receives a new client (current 4, max 4)
13:51:15.575 -> >>> HomeKit: No more room for client connections (max 4)
13:55:02.693 -> >>> HomeKit: WiFiServer receives a new client (current 4, max 4)
13:55:02.693 -> >>> HomeKit: No more room for client connections (max 4)

eford321, 2021-02-04:
More info: I complied again with HOMEKIT_MAX_CLIENTS = 8 to see the effect on memory.  During my test I saw a core dump after the periodic session disconnect.  Hopefully this will be of some use in determining what is happening:

Here's the log and the exception decode:
> 14:04:00.467 -> >>> HomeKit: init_storage_ex size 0x1c1
14:04:00.467 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
14:04:00.467 -> >>> Home Integration: hap_services added chararacteristic  1: 6D 
14:04:00.467 -> >>> Home Integration: hap_services added chararacteristic  2: 7C 
14:04:00.467 -> >>> Home Integration: hap_services added chararacteristic  3: 72 
14:04:00.522 -> >>> Home Integration: homekit_is_paired 0
14:04:00.522 -> >>> HomeKit: Free heap: 26312
14:04:00.522 -> >>> HomeKit: Starting server
14:04:00.522 -> >>> HomeKit: Formatting flash at 0x200000
14:04:00.668 -> >>> HomeKit: storage init 1
14:04:00.668 -> >>> HomeKit: if r
14:04:00.668 -> >>> HomeKit: generating id
14:04:00.668 -> >>> HomeKit: Generated new accessory ID: BF:3C:95:4A:09:A0
14:04:00.668 -> >>> HomeKit: Generated new accessory ID: 1073694260
14:04:00.668 -> >>> HomeKit:  key generated
14:04:00.668 -> >>> HomeKit:  key saved
14:04:00.746 -> >>> HomeKit: Generated new accessory key
14:04:00.799 -> >>> HomeKit: find pairings
14:04:00.799 -> >>> HomeKit: Preinit pairing context
14:04:00.799 -> >>> HomeKit: Using user-specified password: 111-11-111
14:04:00.853 -> >>> HomeKit: Free heap: 21216
14:04:06.883 -> >>> HomeKit: Calculating public key len=384
14:04:10.006 -> >>> HomeKit: Calculating public key result 0
14:04:10.006 -> >>> HomeKit: arduino_homekit_preinit success
14:04:10.006 -> >>> HomeKit: Configuring mDNS
14:04:10.006 -> >>> HomeKit: MDNS.begin: DRB1-72FC, IP: 192.168.1.27
14:04:10.063 -> >>> HomeKit: Init server over
14:04:46.455 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
14:04:46.455 -> >>> HomeKit: Got new client connection: local 192.168.1.27:5556, remote 192.168.1.15:58139
14:04:46.512 -> >>> HomeKit: [Client 1073702188] Got client content from the cache 0
14:04:46.512 -> >>> HomeKit: [Client 1073702188] new client accepted
14:04:46.512 -> >>> HomeKit: [Client 1073702188] Got 127 incomming data, encrypted is 
14:04:46.512 -> >>> HomeKit: [Client 1073702188] Pair Setup Step 1/3
14:04:49.993 -> >>> HomeKit: [Client 1073702188] Got 580 incomming data, encrypted is 
14:04:49.993 -> >>> HomeKit: [Client 1073702188] Pair Setup Step 2/3
14:04:49.993 -> >>> HomeKit: [Client 1073702188] Computing SRP shared secret
14:05:03.275 -> >>> HomeKit: [Client 1073702188] Verifying peer's proof
14:05:03.275 -> >>> HomeKit: [Client 1073702188] Generating own proof
14:05:03.380 -> >>> HomeKit: [Client 1073702188] Got 282 incomming data, encrypted is 
14:05:03.380 -> >>> HomeKit: [Client 1073702188] Pair Setup Step 3/3
14:05:04.216 -> >>> HomeKit: Added pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:05:04.290 -> >>> HomeKit: Free saved_preinit_pairing_context
14:05:04.290 -> >>> HomeKit: [Client 1073702188] Successfully paired
14:05:04.463 -> >>> HomeKit: [Client 1073702188] Disconnected!
14:05:04.463 -> >>> HomeKit: [Client 1073702188] Closing client connection
14:05:04.669 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
14:05:04.705 -> >>> HomeKit: Got new client connection: local 192.168.1.27:5556, remote 192.168.1.15:58143
14:05:04.705 -> >>> HomeKit: [Client 1073702188] Got client content from the cache 0
14:05:04.705 -> >>> HomeKit: [Client 1073702188] new client accepted
14:05:04.705 -> >>> HomeKit: [Client 1073702188] Got 160 incomming data, encrypted is 
14:05:04.705 -> >>> HomeKit: [Client 1073702188] Pair Verify Step 1/2
14:05:05.027 -> >>> HomeKit: Free heap: 22520
14:05:05.098 -> >>> HomeKit: [Client 1073702188] Got 249 incomming data, encrypted is 
14:05:05.098 -> >>> HomeKit: [Client 1073702188] Pair Verify Step 2/2
14:05:05.098 -> >>> HomeKit: [Client 1073702188] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:05:05.886 -> >>> HomeKit: [Client 1073702188] Verification successful, secure session established
14:05:05.886 -> >>> HomeKit: Free heap: 22456
14:05:06.131 -> >>> HomeKit: [Client 1073702188] Got 80 incomming data, encrypted is 
14:05:06.131 -> >>> HomeKit: [Client 1073702188] Get Accessories
14:05:06.371 -> >>> HomeKit: [Client 1073702188] Got 249 incomming data, encrypted is 
14:05:06.371 -> >>> HomeKit: [Client 1073702188] Update Characteristics
14:05:06.427 -> >>> HomeKit: [Client 1073702188] Got 249 incomming data, encrypted is 
14:05:06.427 -> >>> HomeKit: [Client 1073702188] Update Characteristics
14:05:06.576 -> >>> HomeKit: [Client 1073702188] Got 249 incomming data, encrypted is 
14:05:06.576 -> >>> HomeKit: [Client 1073702188] Update Characteristics
14:05:06.576 -> >>> HomeKit: [Client 1073702188] Got 97 incomming data, encrypted is 
14:05:06.576 -> >>> HomeKit: [Client 1073702188] Get Characteristics
[... 434 lines cut ...]
0x4023f2cd: snprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/snprintf.c line 118
0x40240b17: __swbuf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/wbuf.c line 82
0x4023e759: __sfputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 433
0x40223b01: client_send(_client_context_t*, unsigned char*, unsigned int) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 911
[... cut ...]
0x4022a53d: __delay at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 57
0x40226ff8: hap_homekit_loop() at /Users/evan/Documents/Arduino/libraries/ESPHap/homekitintegrationcpp.cpp line 18
0x40217eb9: loop at /Users/evan/Documents/Arduino/EspHap_WindowCover/EspHap_WindowCover.ino line 496
0x40229f68: loop_wrapper() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
0x40100dcd: cont_wrapper at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/cont.S line 81

Yurik72, 2021-02-04:
Good evening. I'l analyze a log,thanks, but definetelly is a question about playing with 2 parameters
#define HOMEKIT_MAX_CLIENTS 4
#define CLIENT_CONTEXT_CACHE_SIZE 4

There are not any reason to limiting them, reason is to find optimal and stable solution

eford321, 2021-02-04:
Got it. I'll experiment with it. 

I do see reports of disconnects on Mixiaoxiao's repository also.  https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/103 
It appears he no longer has time to support the code.....

eford321, 2021-02-05:
Testing with 
#define HOMEKIT_MAX_CLIENTS 8
#define CLIENT_CONTEXT_CACHE_SIZE 8
Heap seems to stabilize around 14000.  It depends on who is home and whether Mac's are awake or not.  I have had 8 devices connected briefly, before the disconnects happen.

Yurik72, 2021-02-06:
Ok, this is a good progress !! Next few days I will optimize cJSON library. there are still a source of huge call of malloc and realloc.

Yurik72, 2021-02-08:
Hi, I'v impemented static buffer for json parse, and looks now is quite stable and memory fragmentation is decreased

eford321, 2021-02-08:
Awesome. Nice work!  I'll load it up and do some testing.

eford321, 2021-02-08:
I am seeing intermittent callback failure. In some cases, I am getting a callback, but the data passed fails to pass a comparison for characteristic type. Other times sending a command from the Home app doesn't generate a callback.  I do get this output:
```
10:57:52.159 -> >>> HomeKit: [Client 1073701940] Got 191 incomming data, encrypted is 
10:57:52.159 -> >>> HomeKit: [Client 1073701940] Update Characteristics
```
but it is not triggering a callback.

Here is my sketch:

/*************************************************************************************************************************
* ESPHap_WindowCover v.1.0.1 - 02/01/2021
*
* This sketch connects a DIY window shade (roller blind) to Homekit using ESPHap. 
*
* It has been tested on an ESP8266 WEMOS Mini D1 clone with a ULN2003 Stepper driver board and a 28BYJ-48 stepper motor.
* It is powered by a 2.1A USB charger using the ESP8266 USB port. A 64Kb NV-FRAM breakout board is used to store blind position
* data. This avoids Flash memory wear from frequent writes of location. A normally open momentary push button is used to 
* locally control the blinds, configure the open and closed position of the blinds, or erase the homekit configuration.
*  - Pressing and releasing the button for less than 5 seconds will OPEN/CLOSE the bind
*  - Pressing and releasing the button for more than 5 but less than 15 seconds will configure the binds
*                - The blind will move toward the closed position at slow speed, press the button at the desired closed position
*                     >If the blind moves the wrong direction, change the comments on the REVERSE_STEPPER line below
*                - The blind will move toward the open position at slow speed, press the button at the desired open position
*  - Pressing and releasing the button for more than 15 seconds will Erase the Homekit and WiFi configuration
*
*There are several options you can turn on or off by adding or removing comments on the lines below:
********************************* options ********************************************************************************/
//#define REVERSE_STEPPER //add or remove comments to this line if the stepper moves the wrong direction
#define ENABLE_WIFI_MANAGER //add comments to manually enter Wifi SSIS/PASSWORD Below. Remove to have device create access point for configutation
#define ENABLE_OTA  // remove comments to allow Over the Air updates of device
#define SERIAL_OUTPUT // remove comments to output messages to serial console
/**************************** end of options ******************************************************************************/
#include <Arduino.h>

#ifdef ESP32
#include <SPIFFS.h>
#endif
#ifdef ESP8266
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include "coredecls.h"
#endif

#ifdef ENABLE_WIFI_MANAGER
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#else
const char* ssid     = "yourssid";          //Enter your WiFi credentials if you are not using WiFiManager
const char* password = "yourpassword";      //Enter your WiFi credentials if you are not using WiFiManager
#endif

const char* device_mame = "DRB1";          //Enter a unique name to identify your window cover if you have more than one  - 4 capital alpha numerics

#ifdef ENABLE_OTA
#include <ArduinoOTA.h>
#endif

// support for stepper motor(s) using AccelStepper - Awesome stepper library from AirSpace (Mike McCauley)
#include <AccelStepper.h>  //http://www.airspayce.com/mikem/arduino/AccelStepper/
#define stepper_pin1 D4 //this is the pin map of digital outputs connected to stepper motor driver
#define stepper_pin2 D3
#define stepper_pin3 D2
#define stepper_pin4 D1
#ifdef REVERSE_STEPPER
  AccelStepper stepper1(AccelStepper::FULL4WIRE, stepper_pin1, stepper_pin3, stepper_pin2, stepper_pin4); 
#else
  AccelStepper stepper1(AccelStepper::FULL4WIRE, stepper_pin4, stepper_pin2, stepper_pin3, stepper_pin1);
#endif

//support for Adafruit FRAM module to store blind position at every step https://www.adafruit.com/product/1897
#include <SPI.h> 
[... 413 lines cut ...]
    if ((millis() - statusTimer > statusInterval) && (stepperSleep == false)){  // update current location in FRAM when stepper awake
        blindCurrentPosition = stepper1.currentPosition();
        fram.writeEnable(true);
        fram.write(FramCurrentAddr, (uint8_t *) &blindCurrentPosition, sizeof(blindCurrentPosition));
        fram.writeEnable(false);
        current_position = ceil(100.0*(float)blindCurrentPosition / blindOpenPosition);  
        statusTimer = millis();
    } 
}

void loop() {
    delay(10); // allow scheduler to service WIFI
    
    #ifdef ENABLE_OTA
        ArduinoOTA.handle(); //check for OTA updates
    #endif
    
    #ifdef ESP8266
       hap_homekit_loop();
    #endif
        
    handleButton();

    handleMotion();
}

Yurik72, 2021-02-08:
Ok, I will check

Yurik72, 2021-02-09:
Hi, Based on your question
I am seeing intermittent callback failure. In some cases, I am getting a callback, but the data passed fails to pass a comparison for characteristic type. Other times sending a command from the Home app doesn't generate a callback. I do get this output:

I think if it will continue happens i need detail log for that. Looks like not any error, because errors are logged
So, if neccessary you can compile with define HOMEKIT_DEBUG file debug.h

eford321, 2021-02-09:
I recompiled with define HOMEKIT_DEBUG and now it is working perfectly. Strange. No other changes.  I will continue testing.  Maybe this will help provide information on the disconnections. 

I'll keep you posted.

Yurik72, 2021-02-10:
I have reproduced one time this issue and created #49 ,  
Please have a look and get new piece of code

eford321, 2021-02-10:
testing with the larger packet size. appears to work well. Good job finding that!

With 5 devices connected I am seeing heap at 17384.  Do you think packet fragmentation was causing excessive heap use?

I am also seeing fewer disconnects/no response now.  I will test for #define HOMEKIT_MAX_CLIENTS & #define CLIENT_CONTEXT_CACHE_SIZE to see the effect on the ESP8266.  So far 5 seems to be stable.  I think 6 might be too many. Will report back and make a pull request for changes.

Yurik72, 2021-02-10:
No, this is different story
TCP_MSS=562 is not enought for some packets and device can't send response, therefore charachteristic is not updated. There is a bug which you found

Did yuo check latest version ? I have decreased heap fragmentation by TLV values, should be more stable

eford321, 2021-02-10:
OK.  I am running with the latest version.  I am seeing period crashes.  The latest was with the the sessions idle. Crashed with a watchdog timer reset.  Happened after 4 of 5 connected devices were disconnected and tried to reconnect.  Here's the log and decode:

10:05:45.670 -> >>> HomeKit: WiFiServer receives a new client (current 5, max 8)
10:05:45.705 -> >>> HomeKit: Got new client connection: local 192.168.1.12:5556, remote 192.168.1.33:49418
10:05:45.705 -> >>> HomeKit: [Client 1073704940] Got client content from the cache 5
10:05:45.705 -> >>> HomeKit: [Client 1073704940] new client accepted
10:05:45.705 -> >>> HomeKit: [Client 1073704940] Got 160 incomming data, encrypted is 
10:05:45.705 -> >>> HomeKit: [Client 1073704940] Pair Verify Step 1/2
10:05:46.057 -> >>> HomeKit: Free heap: 17448
10:05:46.093 -> >>> HomeKit: [Client 1073704940] Got 249 incomming data, encrypted is 
10:05:46.093 -> >>> HomeKit: [Client 1073704940] Pair Verify Step 2/2
10:05:46.093 -> >>> HomeKit: [Client 1073704940] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
10:05:46.884 -> >>> HomeKit: [Client 1073704940] Verification successful, secure session established
10:05:46.884 -> >>> HomeKit: Free heap: 17384
10:05:46.884 -> >>> HomeKit: [Client 1073704940] Got 80 incomming data, encrypted is 
10:05:46.884 -> >>> HomeKit: [Client 1073704940] Get Accessories
10:05:47.000 -> >>> HomeKit: [Client 1073704940] Got 249 incomming data, encrypted is 
10:05:47.000 -> >>> HomeKit: [Client 1073704940] Update Characteristics
10:13:37.367 -> >>> HomeKit: [Client 1073701748] Got 192 incomming data, encrypted is 
10:13:37.367 -> >>> HomeKit: [Client 1073701748] Update Characteristics
10:13:37.367 -> >>> HomeKit: [Client 1073701748] Updating characteristic 1.11 with integer 100
10:13:37.367 -> >>> HomeKit: [Client 1073701748] Got characteristic 1.12 change event
10:13:37.367 -> >>> HomeKit: [Client 1073701716] Got characteristic 1.12 change event
10:13:37.367 -> >>> HomeKit: [Client 1073705076] Got characteristic 1.12 change event
10:13:37.367 -> >>> HomeKit: [Client 1073704940] Got characteristic 1.12 change event
10:13:37.367 -> >>> HomeKit: [Client 1073701716] Got characteristic 1.11 change event
10:13:37.367 -> >>> HomeKit: [Client 1073705076] Got characteristic 1.11 change event
10:13:37.367 -> >>> HomeKit: [Client 1073704940] Got characteristic 1.11 change event
10:13:48.956 -> >>> HomeKit: [Client 1073703836] Disconnected!
10:13:48.956 -> >>> HomeKit: [Client 1073703836] Closing client connection
10:13:52.912 -> >>> HomeKit: [Client 1073701748] Got characteristic 1.10 change event
10:13:52.912 -> >>> HomeKit: [Client 1073701716] Got characteristic 1.10 change event
10:13:52.912 -> >>> HomeKit: [Client 1073705076] Got characteristic 1.10 change event
10:13:52.912 -> >>> HomeKit: [Client 1073704940] Got characteristic 1.10 change event
10:13:52.953 -> >>> HomeKit: [Client 1073701748] Got characteristic 1.12 change event
10:13:52.953 -> >>> HomeKit: [Client 1073701716] Got characteristic 1.12 change event
10:13:52.953 -> >>> HomeKit: [Client 1073705076] Got characteristic 1.12 change event
10:13:52.953 -> >>> HomeKit: [Client 1073704940] Got characteristic 1.12 change event
10:14:29.673 -> >>> HomeKit: [Client 1073695340] Got 249 incomming data, encrypted is 
10:14:29.673 -> >>> HomeKit: [Client 1073695340] Update Characteristics
10:14:29.746 -> >>> HomeKit: [Client 1073695340] Got 97 incomming data, encrypted is 
10:14:29.746 -> >>> HomeKit: [Client 1073695340] Get Characteristics
10:15:11.350 -> >>> HomeKit: [Client 1073701748] Got 249 incomming data, encrypted is 
10:15:11.350 -> >>> HomeKit: [Client 1073701748] Update Characteristics
10:15:32.821 -> >>> HomeKit: [Client 1073695340] Got 252 incomming data, encrypted is 
10:15:32.821 -> >>> HomeKit: [Client 1073695340] Update Characteristics
10:16:41.404 -> >>> HomeKit: [Client 1073695340] Got 249 incomming data, encrypted is 
10:16:41.404 -> >>> HomeKit: [Client 1073695340] Update Characteristics
10:16:41.404 -> >>> HomeKit: [Client 1073701748] Got 97 incomming data, encrypted is 
10:16:41.404 -> >>> HomeKit: [Client 1073701748] Get Characteristics
10:17:11.822 -> >>> HomeKit: [Client 1073695340] Got 252 incomming data, encrypted is 
10:17:11.857 -> >>> HomeKit: [Client 1073695340] Update Characteristics
10:17:58.387 -> >>> HomeKit: [Client 1073695340] Got 249 incomming data, encrypted is 
10:17:58.387 -> >>> HomeKit: [Client 1073695340] Update Characteristics
10:17:58.426 -> >>> HomeKit: [Client 1073695340] Got 97 incomming data, encrypted is 
10:17:58.426 -> >>> HomeKit: [Client 1073695340] Get Characteristics
10:18:01.470 -> >>> HomeKit: [Client 1073695340] Got 190 incomming data, encrypted is 
10:18:01.470 -> >>> HomeKit: [Client 1073695340] Update Characteristics
10:18:01.470 -> >>> HomeKit: [Client 1073695340] Updating characteristic 1.11 with integer 0
10:18:01.535 -> >>> HomeKit: [Client 1073701748] Got characteristic 1.12 change event
10:18:01.535 -> >>> HomeKit: [Client 1073701716] Got characteristic 1.12 change event
10:18:01.535 -> >>> HomeKit: [Client 1073705076] Got characteristic 1.12 change event
10:18:01.535 -> >>> HomeKit: [Client 1073704940] Got characteristic 1.12 change event
10:18:01.535 -> >>> HomeKit: [Client 1073695340] Got characteristic 1.12 change event
10:18:01.535 -> >>> HomeKit: [Client 1073701748] Got characteristic 1.11 change event
10:18:01.535 -> >>> HomeKit: [Client 1073701716] Got characteristic 1.11 change event
[... cut ...]
0x4022a3a9: __delay at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 57
0x40226e64: hap_homekit_loop() at /Users/evan/Documents/Arduino/libraries/ESPHap/homekitintegrationcpp.cpp line 18
0x40217995: loop at /Users/evan/Documents/Arduino/EspHap_WindowCover/EspHap_WindowCover.ino line 496
0x40229dd4: loop_wrapper() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
0x40100dcd: cont_wrapper at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/cont.S line 81

eford321, 2021-02-10:
Switched IwIP Variant to "v2 Higher Bandwidth (no features)" instead of "v2 Higher Bandwidth" and it has been running fine for more than 1 hour with 5 active sessions. Only 1 disconnect in that time period.

eford321, 2021-02-10:
All devices disconnected at 1h29m and reconnected 1m20s later without a crash. Heap dropped to a low of 7240 but recovered to 15552.   It does appear heap is dropping over time. (about 2K per hour).

eford321, 2021-02-10:
ran for 2h42m before crashing.   looks like memory.

0x40207f91: fe_sq at /Users/evan/Documents/Arduino/libraries/wolfssl/fe_operations.c line 303
0x4020b62f: ge_double_scalarmult_vartime_lowmem at /Users/evan/Documents/Arduino/libraries/wolfssl/ge_low_mem.c line 562
0x40209f1b: fe_invert at /Users/evan/Documents/Arduino/libraries/wolfssl/fe_operations.c line 653
0x4020fdb8: ge_tobytes at /Users/evan/Documents/Arduino/libraries/wolfssl/ge_operations.c line 7075
0x402506cf: ieee80211_output_pbuf at ?? line ?
0x40207a9c: wc_ed25519_verify_msg at /Users/evan/Documents/Arduino/libraries/wolfssl/ed25519.c line 282
0x4027f68b: chip_v6_unset_chanfreq at ?? line ?
0x40243ead: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x4027f65c: chip_v6_unset_chanfreq at ?? line ?
0x402436a2: vprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vprintf.c line 40
0x4010088b: umm_free_core at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 351
0x40202a38: crypto_ed25519_verify at /Users/evan/Documents/Arduino/libraries/ESPHap/crypto.c line 479
0x4022439f: homekit_server_on_pair_verify(_client_context_t*, unsigned char const*, unsigned int) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 2024
0x40100224: ets_post at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x401017ec: pp_post at ?? line ?
0x40104a97: lmacRxDone at ?? line ?
0x40100224: ets_post at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x401017ec: pp_post at ?? line ?
0x40104a97: lmacRxDone at ?? line ?
0x40102383: rcReachRetryLimit at ?? line ?
0x40102a22: wDev_ProcessFiq at ?? line ?
0x40102560: rcReachRetryLimit at ?? line ?
0x40102a22: wDev_ProcessFiq at ?? line ?
0x40100224: ets_post at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010533d: ets_timer_disarm at ?? line ?
0x4027c2b1: node_remove_from_list at ?? line ?
0x40102a5c: wDev_ProcessFiq at ?? line ?
0x40100494: _write_r at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/libc_replacements.cpp line 90 (discriminator 1)
0x4023f254: __swrite at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/stdio.c line 105
0x4023d0ad: __sflush_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/fflush.c line 226
0x4023d13d: _fflush_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/fflush.c line 282
0x402406e7: __swbuf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/wbuf.c line 82
0x4023e517: __sfputc_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 404
0x4023e57d: __sfputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 433
0x4027e4e8: chip_v6_unset_chanfreq at ?? line ?
0x4023e520: __sfputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 412
0x4023e480: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 246
0x4023e704: _vfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x4023e57d: __sfputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 433
0x4027e6b5: chip_v6_unset_chanfreq at ?? line ?
0x4023e704: _vfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x4027e6b7: chip_v6_unset_chanfreq at ?? line ?
0x4027e6b6: chip_v6_unset_chanfreq at ?? line ?
0x4023e948: _vfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 667
0x40100b48: malloc at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 552
0x40100b91: realloc at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 586
0x402269f4: homekit_server_on_message_complete(http_parser*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3203
0x4020538c: http_parser_execute at /Users/evan/Documents/Arduino/libraries/ESPHap/http_parser.c line 1927 (discriminator 1)
0x40225795: homekit_client_process(_client_context_t*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3364 (discriminator 1)
0x40236edc: esp8266::MDNSImplementation::MDNSResponder::_updateProbeStatus() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Control.cpp line 1321 (discriminator 1)
0x40225b28: homekit_server_process(homekit_server_t*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3583
0x40226d3d: arduino_homekit_loop at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 4133
0x4022a3a9: __delay at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 57
0x40226e64: hap_homekit_loop() at /Users/evan/Documents/Arduino/libraries/ESPHap/homekitintegrationcpp.cpp line 18
0x40217995: loop at /Users/evan/Documents/Arduino/EspHap_WindowCover/EspHap_WindowCover.ino line 496
0x40229dd4: loop_wrapper() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
0x40100dcd: cont_wrapper at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/cont.S line 81

Yurik72, 2021-02-10:
If that with the latest commit, because I did some code workaround for that as well

Yurik72, 2021-02-11:
Hi about the crash. Looks like memory usage, did you catch free heap at this time ?
This is my next task to optimize pair verification. it still frequently uses a heap

Yurik72, 2021-02-11:
I did a few improvements to decrease fragmentation, if that helps, i will do this on regular basis

eford321, 2021-02-11:
> Hi about the crash. Looks like memory usage, did you catch free heap at this time ?
> This is my next task to optimize pair verification. it still frequently uses a heap

I believe it was around 11000 at the last output.

eford321, 2021-02-11:
I believe it was the latest. If you did multiple commits yesterday, there is a chance it wasn't the latest.

eford321, 2021-02-11:
> I did a few improvements to decrease fragmentation, if that helps, i will do this on regular basis

I'll test this and get back to you.

Yurik72, 2021-02-11:
Ok, get the latest commit, I did some mistake

I’m able to test with 3 devices and it works currently more than 3 hours

eford321, 2021-02-11:
It ran for more then 3 hours with the commit I downloaded 4 hours ago.  Still leaking memory. The disconnects (or the cause if the disconnects) seem to trigger the drop in heap.

Here's the log and crash decode if you need it.

09:21:12.054 -> *WM: [3] WiFi station enable 
09:21:12.054 -> *WM: [3] enableSTA PERSISTENT ON 
09:21:12.054 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
09:21:16.922 -> *WM: [2] Connection result: WL_CONNECTED
09:21:16.922 -> *WM: [3] lastconxresult: WL_CONNECTED
09:21:16.922 -> *WM: [1] AutoConnect: SUCCESS 
09:21:16.922 -> *WM: [1] STA IP Address: 192.168.1.12
09:21:16.922 -> *WM: [3] unloading 
09:21:16.922 -> Free heap: 29488
09:21:16.922 -> >>> HomeKit: init_storage_ex size 0x1c1
09:21:16.922 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
09:21:16.922 -> >>> Home Integration: hap_services added chararacteristic  1: 6D 
09:21:16.959 -> >>> Home Integration: hap_services added chararacteristic  2: 7C 
09:21:16.959 -> >>> Home Integration: hap_services added chararacteristic  3: 72 
09:21:16.959 -> >>> Home Integration: homekit_is_paired 1
09:21:16.959 -> >>> HomeKit: Free heap: 27368
09:21:16.959 -> >>> HomeKit: Starting server
09:21:16.959 -> >>> HomeKit: storage init 0
09:21:16.959 -> >>> HomeKit: Using existing accessory ID: 2C:C0:9C:5E:D9:A2
09:21:16.959 -> >>> HomeKit: find pairings
09:21:16.959 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup
09:21:16.992 -> >>> HomeKit: Configuring mDNS
09:21:16.992 -> >>> HomeKit: MDNS.begin: DRB1-2CC0, IP: 192.168.1.12
09:21:16.992 -> >>> HomeKit: Init server over
09:21:19.267 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
09:21:19.267 -> >>> HomeKit: Got new client connection: local 192.168.1.12:5556, remote 192.168.1.15:62982
09:21:19.267 -> >>> HomeKit: [Client 1073694188] Got client content from the cache 0
09:21:19.267 -> >>> HomeKit: [Client 1073694188] new client accepted
09:21:19.301 -> >>> HomeKit: [Client 1073694188] Got 160 incomming data, encrypted is 
09:21:19.301 -> >>> HomeKit: [Client 1073694188] Pair Verify Step 1/2
09:21:19.613 -> >>> HomeKit: Free heap: 20240
09:21:19.682 -> >>> HomeKit: [Client 1073694188] Got 249 incomming data, encrypted is 
09:21:19.682 -> >>> HomeKit: [Client 1073694188] Pair Verify Step 2/2
09:21:19.682 -> >>> HomeKit: [Client 1073694188] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
09:21:20.450 -> >>> HomeKit: [Client 1073694188] Verification successful, secure session established
09:21:20.484 -> >>> HomeKit: Free heap: 20176
09:21:20.484 -> >>> HomeKit: [Client 1073694188] Got 80 incomming data, encrypted is 
09:21:20.484 -> >>> HomeKit: [Client 1073694188] Get Accessories
09:21:52.468 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
09:21:52.468 -> >>> HomeKit: Got new client connection: local 192.168.1.12:5556, remote 192.168.1.37:53442
09:21:52.468 -> >>> HomeKit: [Client 1073694220] Got client content from the cache 1
09:21:52.468 -> >>> HomeKit: [Client 1073694220] new client accepted
09:21:52.468 -> >>> HomeKit: [Client 1073694220] Got 160 incomming data, encrypted is 
09:21:52.504 -> >>> HomeKit: [Client 1073694220] Pair Verify Step 1/2
09:21:52.823 -> >>> HomeKit: Free heap: 19872
09:21:52.856 -> >>> HomeKit: [Client 1073694220] Got 249 incomming data, encrypted is 
09:21:52.856 -> >>> HomeKit: [Client 1073694220] Pair Verify Step 2/2
09:21:52.856 -> >>> HomeKit: [Client 1073694220] Found pairing with 0F8927F8-0F84-44E7-8181-8714ED73BD51
09:21:53.649 -> >>> HomeKit: [Client 1073694220] Verification successful, secure session established
09:21:53.649 -> >>> HomeKit: Free heap: 19808
09:21:53.649 -> >>> HomeKit: [Client 1073694220] Got 80 incomming data, encrypted is 
09:21:53.682 -> >>> HomeKit: [Client 1073694220] Get Accessories
09:22:04.947 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
09:22:04.947 -> >>> HomeKit: Got new client connection: local 192.168.1.12:5556, remote 192.168.1.13:60939
09:22:04.947 -> >>> HomeKit: [Client 1073702644] Got client content from the cache 2
09:22:04.983 -> >>> HomeKit: [Client 1073702644] new client accepted
09:22:04.983 -> >>> HomeKit: [Client 1073702644] Got 160 incomming data, encrypted is 
09:22:04.983 -> >>> HomeKit: [Client 1073702644] Pair Verify Step 1/2
09:22:05.295 -> >>> HomeKit: Free heap: 19440
09:22:05.332 -> >>> HomeKit: [Client 1073702644] Got 249 incomming data, encrypted is 
09:22:05.332 -> >>> HomeKit: [Client 1073702644] Pair Verify Step 2/2
09:22:05.332 -> >>> HomeKit: [Client 1073702644] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
09:22:06.109 -> >>> HomeKit: [Client 1073702644] Verification successful, secure session established
09:22:06.109 -> >>> HomeKit: Free heap: 19376
09:22:06.145 -> >>> HomeKit: [Client 1073702644] Got 80 incomming data, encrypted is 
09:22:06.145 -> >>> HomeKit: [Client 1073702644] Get Accessories
[... 978 lines cut ...]
0x4023f32c: __sclose at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/stdio.c line 135
0x4023f254: __sread at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/stdio.c line 38
0x4023f294: __swrite at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/stdio.c line 81
0x4023f2ec: __sseek at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/stdio.c line 113
[... cut ...]
0x40234884: ClientContext::_s_recv(void*, tcp_pcb*, pbuf*, long) at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/include/ClientContext.h line 652
0x40234a9c: ClientContext::_s_poll(void*, tcp_pcb*) at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/include/ClientContext.h line 662
0x40234a50: ClientContext::_s_error(void*, long) at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/src/include/ClientContext.h line 657

`

eford321, 2021-02-11:
downloading the latest now...

eford321, 2021-02-11:
Here's a log from commit 3d69c49 with full debug.  Heap is much improved.

I am trying to understand the disconnects. Are they expected??  I am not able to determine the cause from debug.

```
13:06:07.665 -> >>> HomeKit: WiFiServer receives*WM: [2] Connection result: WL_CONNECTED
13:06:10.193 -> *WM: [3] lastconxresult: WL_CONNECTED
13:06:10.193 -> *WM: [1] AutoConnect: SUCCESS 
13:06:10.228 -> *WM: [1] STA IP Address: 192.168.1.12
13:06:10.228 -> *WM: [3] unloading 
13:06:10.228 -> Free heap: 29200
13:06:10.228 -> >>> HomeKit: init_storage_ex size 0x1c1
13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic  1: 6D 
13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic  2: 7C 
13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic  3: 72 
13:06:10.228 -> >>> Home Integration: homekit_is_paired 1
13:06:10.262 -> >>> HomeKit: Free heap: 27080
13:06:10.262 -> >>> server_new: WiFiServer begin at port: 5556
13:06:10.262 -> 
13:06:10.262 -> >>> HomeKit: Starting server
13:06:10.262 -> >>> HomeKit: storage init 0
13:06:10.262 -> >>> HomeKit: Using existing accessory ID: 2C:C0:9C:5E:D9:A2
13:06:10.262 -> >>> HomeKit: find pairings
13:06:10.262 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair `setup`
13:06:10.262 -> >>> HomeKit: Configuring mDNS
13:06:10.262 -> >>> HomeKit: MDNS.begin: DRB1-2CC0, IP: 192.168.1.12
13:06:10.295 -> >>> homekit_mdns_init: Accessory Setup ID = YK72
13:06:10.295 -> >>> HomeKit: Init server over
13:06:40.545 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
13:06:40.545 -> >>> HomeKit: Got new client connection: local 192.168.1.12:5556, remote 192.168.1.28:51215
13:06:40.545 -> >>> HomeKit: [Client 1073702516] Got client content from the cache 0
13:06:40.580 -> >>> HomeKit: [Client 1073702516] new client accepted
13:06:40.580 -> >>> homekit_client_process: [Client 1073702516] Reading 160 bytes
13:06:40.580 -> >>> HomeKit: [Client 1073702516] Got 160 incomming data, encrypted is 
13:06:40.580 -> >>> homekit_server_on_message_complete: http_parser message_complete
13:06:40.580 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
13:06:40.580 -> >>> homekit_server_on_pair_verify: Free heap: 22040
13:06:40.617 -> >>> HomeKit: [Client 1073702516] homekit_server_on_pair_verify  ⸮⸮]⸮xBMtԳ(v*ԞDS2⸮⸮=⸮⸮jm⸮⸮⸮<
13:06:40.617 -> >>> tlv_debug: Got following TLV values:
13:06:40.617 -> >>> tlv_debug: Type 6 value (1 bytes): \x01
13:06:40.617 -> >>> tlv_debug: Type 3 value (32 bytes): \x94\x88]\x98xBMt\xD4\xB3(v*\xD4\x9EDS2\xDE\x07\x95=\xFC\x08\xD4jm\xB5\x8A\x1D\xDC<
13:06:40.617 -> >>> HomeKit: [Client 1073702516] Pair Verify Step 1/2
13:06:40.617 -> >>> homekit_server_on_pair_verify: [Client 1073702516] Importing device Curve25519 public key
13:06:40.655 -> >>> homekit_server_on_pair_verify: [Client 1073702516] Generating accessory Curve25519 key
13:06:40.762 -> >>> homekit_server_on_pair_verify: [Client 1073702516] Exporting accessory Curve25519 public key
13:06:40.795 -> >>> homekit_server_on_pair_verify: [Client 1073702516] Generating Curve25519 shared secret
13:06:40.902 -> >>> homekit_server_on_pair_verify: [Client 1073702516] Generating signature
13:06:40.973 -> >>> homekit_server_on_pair_verify: [Client 1073702516] Generating proof
13:06:41.009 -> >>> homekit_server_on_pair_verify: [Client 1073702516] Encrypting response
13:06:41.009 -> >>> send_tlv_response: [Client 1073702516] Sending TLV response
13:06:41.009 -> >>> tlv_debug: Got following TLV values:
13:06:41.009 -> >>> tlv_debug: Type 6 value (1 bytes): \x02
13:06:41.009 -> >>> tlv_debug: Type 3 value (32 bytes): \xF2V\x10>j\xF7\xC1\xFCv\xF0X\xDBr\xCC'i\xE4\x0A6\xFB\xA5\xEA\xA5rm\xCE\xC1]\x13t\xE8r
13:06:41.009 -> >>> tlv_debug: Type 5 value (101 bytes): \x89\x85\xD6!\xBF\xB8\xCD\x97\xD9\x99\x99\x1F}\xBD\xA7\x9AR\xB4\xA6\xB3\x80\x0Eh \x9A)\xB3_\x94\x82\xFE\x06m\xCDI\xED\xD0\xA4?\xDE{T\xF6\xA9^\x81\x03x\x98\x0EKV\xA3Z4\xF8\xC2\xEF4\x07\xC4\xA3\xE5H\x881\x8E\x1C\xB7\x0Aq\xDB\xCD\xE1\xDB\x05}\xBB\xCC\x8CP]\x09\x0B}\x1F\x0B\xA5c\xDA-U\xEF`{;p\xA3\xB0\x06\xE8
13:06:41.043 -> >>> client_send: [Client 1073702516] send data size=244, encrypted=false
13:06:41.080 -> >>> client_send: [Client 1073702516] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 \xF2V\x10>j\xF7\xC1\xFCv\xF0X\xDBr\xCC'i\xE4\x0A6\xFB\xA5\xEA\xA5rm\xCE\xC1]\x13t\xE8r\x05e\x89\x85\xD6!\xBF\xB8\xCD\x97\xD9\x99\x99\x1F}\xBD\xA7\x9AR\xB4\xA6\xB3\x80\x0Eh \x9A)\xB3_\x94\x82\xFE\x06m\xCDI\xED\xD0\xA4?\xDE{T\xF6\xA9^\x81\x03x\x98\x0EKV\xA3Z4\xF8\xC2\xEF4\x07\xC4\xA3\xE5H\x881\x8E\x1C\xB7\x0Aq\xDB\xCD\xE1\xDB\x05}\xBB\xCC\x8CP]\x09\x0B}\x1F\x0B\xA5c\xDA-U\xEF`{;p\xA3\xB0\x06\xE8
13:06:41.137 -> >>> write: [Client 1073702516] socket.write, data_size=244, write_size=244, availableForWrite=2676, WriteError=0
13:06:41.137 -> >>> HomeKit: Free heap: 20240
13:06:41.173 -> >>> homekit_client_process: [Client 1073702516] Reading 249 bytes
13:06:41.173 -> >>> HomeKit: [Client 1073702516] Got 249 incomming data, encrypted is 
13:06:41.173 -> >>> homekit_server_on_message_complete: http_parser message_complete
13:06:41.190 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
13:06:41.190 -> >>> homekit_server_on_pair_verify: Free heap: 21752
[... cut ...]
14:23:08.565 -> >>> homekit_server_on_update_characteristics: [Client 1073704412] There were no processing errors, sending No Content response
14:23:08.565 -> >>> client_send: [Client 1073704412] send data size=27, encrypted=true
14:23:08.565 -> >>> client_send: [Client 1073704412] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
14:23:08.565 -> >>> client_send_encrypted_: [Client 1073704412] client_send_encrypted_ size=27
14:23:08.664 -> >>> write: [Client 1073704412] socket.write, data_size=45, write_size=45, availableForWrite=2875, WriteError=0

```

eford321, 2021-02-11:
Here's a crash with debug turned on & decode: 

Decode:

Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 27 results
0x40203358: binary_to_string at /Users/evan/Documents/Arduino/libraries/ESPHap/debug.c line 26
0x40203308: binary_to_string at /Users/evan/Documents/Arduino/libraries/ESPHap/debug.c line 13
0x40283f5d: chip_v6_unset_chanfreq at ?? line ?
0x4022410d: client_send(_client_context_t*, unsigned char*, unsigned int) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 933
0x402241e0: client_send_chunk(unsigned char*, unsigned int, void*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 970
0x4021712e: json_flush at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 58
0x402063a4: json_write at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 79
0x402063fb: _do_write at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 279
0x402067f2: json_string at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 302
0x40206608: json_array_end at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 171
0x40223a6e: write_characteristic_json(json_stream*, _client_context_t*, _homekit_characteristic const*, characteristic_format_t, homekit_value_t const*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 552
0x40206402: _do_write at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 280
0x402067e8: json_string at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 298
0x40225319: homekit_server_on_get_accessories(_client_context_t*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 2187 (discriminator 2)
0x40283dd4: chip_v6_unset_chanfreq at ?? line ?
0x40227ad1: homekit_server_on_message_complete(http_parser*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3249
0x40205a0c: http_parser_execute at /Users/evan/Documents/Arduino/libraries/ESPHap/http_parser.c line 1871 (discriminator 1)
0x402033ad: print_binary at /Users/evan/Documents/Arduino/libraries/ESPHap/debug.c line 41
0x40226691: homekit_client_process(_client_context_t*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3393 (discriminator 1)
0x40237f90: esp8266::MDNSImplementation::MDNSResponder::_updateProbeStatus() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Control.cpp line 1321 (discriminator 1)
0x40226a38: homekit_server_process(homekit_server_t*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3612
0x40227de9: arduino_homekit_loop at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 4162
0x4022b451: __delay at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 57
0x40227f10: hap_homekit_loop() at /Users/evan/Documents/Arduino/libraries/ESPHap/homekitintegrationcpp.cpp line 18
0x40218311: loop at /Users/evan/Documents/Arduino/EspHap_WindowCover/EspHap_WindowCover.ino line 496
0x4022ae7c: loop_wrapper() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
0x40100dcd: cont_wrapper at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/cont.S line 81

Log:
`13:06:07.665 -> >>> HomeKit: WiFiServer receives*WM: [2] Connection result: WL_CONNECTED
13:06:10.193 -> *WM: [3] lastconxresult: WL_CONNECTED
13:06:10.193 -> *WM: [1] AutoConnect: SUCCESS 
13:06:10.228 -> *WM: [1] STA IP Address: 192.168.1.12
13:06:10.228 -> *WM: [3] unloading 
13:06:10.228 -> Free heap: 29200
13:06:10.228 -> >>> HomeKit: init_storage_ex size 0x1c1
13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic  1: 6D 
13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic  2: 7C 
13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic  3: 72 
13:06:10.228 -> >>> Home Integration: homekit_is_paired 1
13:06:10.262 -> >>> HomeKit: Free heap: 27080
13:06:10.262 -> >>> server_new: WiFiServer begin at port: 5556
13:06:10.262 -> 
13:06:10.262 -> >>> HomeKit: Starting server
13:06:10.262 -> >>> HomeKit: storage init 0
13:06:10.262 -> >>> HomeKit: Using existing accessory ID: 2C:C0:9C:5E:D9:A2
13:06:10.262 -> >>> HomeKit: find pairings
13:06:10.262 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup
13:06:10.262 -> >>> HomeKit: Configuring mDNS
13:06:10.262 -> >>> HomeKit: MDNS.begin: DRB1-2CC0, IP: 192.168.1.12
13:06:10.295 -> >>> homekit_mdns_init: Accessory Setup ID = YK72
13:06:10.295 -> >>> HomeKit: Init server over
13:06:40.545 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
13:06:40.545 -> >>> HomeKit: Got new client connection: local 192.168.1.12:5556, remote 192.168.1.28:51215
13:06:40.545 -> >>> HomeKit: [Client 1073702516] Got client content from the cache 0
13:06:40.580 -> >>> HomeKit: [Client 1073702516] new client accepted
13:06:40.580 -> >>> homekit_client_process: [Client 1073702516] Reading 160 bytes
13:06:40.580 -> >>> HomeKit: [Client 1073702516] Got 160 incomming data, encrypted is 
13:06:40.580 -> >>> homekit_server_on_message_complete: http_parser message_complete
[... cut ...]
15:01:41.870 -> >>> write: [Client 1073704428] socket.write, data_size=1042, write_size=1042, availableForWrite=1753, WriteError=0
15:01:41.870 -> >>> write: [Client 1073704428] socket.write, data_size=24, write_size=24, availableForWrite=1729, WriteError=0
15:01:41.870 -> >>> client_send_chunk: [Client 1073704428] client_send_chunk, size=449, offset=5
15:01:41.870 -> >>> client_send: [Client 1073704428] send data size=456, encrypted=true
15:01:41.870 -> >>> client_send: [Client 1073704428] Sending payload: 1c1\x0D\x0A","unit":"percentage","minValue":0,"maxValue

eford321, 2021-02-11:
Continued.....
`15:15:54.457 -> >>> tlv_debug: Type 6 value (1 bytes): \x04
15:15:54.457 -> >>> client_send: [Client 1073704004] send data size=105, encrypted=false
15:15:54.457 -> >>> client_send: [Client 1073704004] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 3\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x04
15:15:54.457 -> >>> write: [Client 1073704004] socket.write, data_size=105, write_size=105, availableForWrite=2815, WriteError=0
15:15:54.457 -> >>> HomeKit: [Client 1073704004] Verification successful, secure session established
15:15:54.457 -> >>> HomeKit: Free heap: 12816
15:15:54.457 -> >>> homekit_client_process: [Client 1073711212] Reading 249 bytes
15:15:54.457 -> >>> HomeKit: [Client 1073711212] Got 249 incomming data, encrypted is 
15:15:54.457 -> >>> homekit_server_on_message_complete: http_parser message_complete
15:15:54.457 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
15:15:54.457 -> >>> homekit_server_on_pair_verify: Free heap: 14304
15:15:54.457 -> >>> HomeKit: [Client 1073711212] homekit_server_on_pair_verify x1⸮⸮⸮
15:15:54.457 -> >>> tlv_debug: Got following TLV values:
15:15:54.457 -> >>> tlv_debug: Type 5 value (120 bytes): 1\x8D\xAB\xF3\x00\x0CB\xD1\x1F\x1D\x1C\xCC\x04\xE2\x8C\xAD\xEF\xDF{a\x0C\x98\xBA\xF2B\x08\x89\x88V\x83\xC6\x93\xB3\xB1Th\xF5\xEF\x8D4\xAF\xD6\xB6\xD8G2o\x87\xEDx\x94\x18\x15N\x1C{rU\xBD\xF4\x14\xAF\xDE\xB8E\xBCN\xB3\x01\xB3uu\x1A<\xDA|n\xF2\xD4\x83\xA8\x16\xA5\xDD\x94\xDA\xADS\xDFp4\xF5\xA3\x10\x9BVE#\xDF\xB2\x19\x86\xA8\x99\xE3\xAC\xC6}\x97\x8E4\x93\xE6k\x94"\xAD\x0A\x00\xE1
15:15:54.457 -> >>> tlv_debug: Type 6 value (1 bytes): \x03
15:15:54.457 -> >>> HomeKit: [Client 1073711212] Pair Verify Step 2/2
15:15:54.457 -> >>> homekit_server_on_pair_verify: [Client 1073711212] Decrypting payload
15:15:54.457 -> >>> homekit_server_on_pair_verify: [Client 1073711212] Searching pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
15:15:54.457 -> >>> HomeKit: [Client 1073711212] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
15:15:54.457 -> >>> homekit_server_on_pair_verify: [Client 1073711212] Verifying device signature
15:15:54.457 -> >>> send_tlv_response: [Client 1073711212] Sending TLV response
15:15:54.457 -> >>> tlv_debug: Got following TLV values:
15:15:54.457 -> >>> tlv_debug: Type 6 value (1 bytes): \x04
15:15:54.457 -> >>> client_send: [Client 1073711212] send data size=105, encrypted=false
15:15:54.457 -> >>> client_send: [Client 1073711212] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 3\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x04
15:15:54.457 -> >>> write: [Client 1073711212] socket.write, data_size=105, write_size=105, availableForWrite=2815, WriteError=0
15:15:54.457 -> >>> HomeKit: [Client 1073711212] Verification successful, secure session established
15:15:54.457 -> >>> HomeKit: Free heap: 12728
15:15:54.457 -> >>> homekit_client_process: [Client 1073704004] Reading 80 bytes
15:15:54.457 -> >>> HomeKit: [Client 1073704004] Got 80 incomming data, encrypted is 
15:15:54.457 -> >>> homekit_client_process: [Client 1073704004] Decrypted 62 bytes, available 0
15:15:54.457 -> Decrypted data (62 bytes): "GET /accessories HTTP/1.1\x0D\x0AHost: DRB1-2CC0._hap._tcp.local\x0D\x0A\x0D\x0A"
15:15:54.457 -> >>> homekit_server_on_message_complete: http_parser message_complete
15:15:54.457 -> >>> HomeKit: [Client 1073704004] Get Accessories
15:15:54.457 -> >>> homekit_server_on_get_accessories: Free heap: 15104
15:15:54.457 -> >>> client_send: [Client 1073704004] send data size=107, encrypted=true
15:15:54.457 -> >>> client_send: [Client 1073704004] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
15:15:54.457 -> >>> client_send_encrypted_: [Client 1073704004] client_send_encrypted_ size=107
15:15:54.457 -> >>> write: [Client 1073704004] socket.write, data_size=125, write_size=125, availableForWrite=2795, WriteError=0
15:15:54.457 -> >>> homekit_server_on_get_accessories: [Client 1073704004] Get Accessories, start send json body
15:15:54.457 -> >>> client_send_chunk: [Client 1073704004] client_send_chunk, size=1023, offset=5
15:15:54.457 -> >>> client_send: [Client 1073704004] send data size=1030, encrypted=true
15:15:54.457 -> >>> client_send: [Client 1073704004] Sending payload: 3ff\x0D\x0A{"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":2,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"DRB1"},{"aid":1,"iid":3,"type":"20","perms":["pr"],"description":"Manufacturer","format":"string","value":"Yurik72"},{"aid":1,"iid":4,"type":"30","perms":["pr"],"description":"Serial Number","format":"string","value":"0"},{"aid":1,"iid":5,"type":"21","perms":["pr"],"description":"Model","format":"string","value":"EspHapWindowCovering"},{"aid":1,"iid":6,"type":"52","perms":["pr"],"description":"Firmware Revision","format":"string","value":"1.0"},{"aid":1,"iid":7,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":8,"type":"8C","hidden":false,"primary":true,"characteristics":[{"aid":1,"iid":9,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"WINDOW_BLIND"},{"aid":1,"iid":10,"type":"6D","perms":["pr","ev"],"ev":false,"description":"Current Position","format":"uint8\x0D\ [... cut ...]
15:55:04.120 -> 3fff2830:  feefeffe feefeffe 3ffe88a0 40100dcd  
15:55:04.153 -> <<<stack<<<
15:55:04.153 -> 
15:55:04.153 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:55:04.153 -> 
15:55:04.153 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
15:55:04.153 -> 
15:55:04.153 -> wdt reset
15:55:04.153 -> load 0x4010f000, len 3584, room 16 
15:55:04.153 -> tail 0
15:55:04.153 -> chksum 0xb0
15:55:04.153 -> csum 0xb0
15:55:04.188 -> v2843a5ac
15:55:04.188 -> ~ld
15:55:05.252 -> `

Yurik72, 2021-02-12:
Yes, this is a memory.

I will continue to work on that. Looks like this a problem with standard library WiFiClient

When is stop and release them, not all memory released

 

From: eford321 <notifications@github.com> 
Sent: Thursday, February 11, 2021 11:01 PM
To: Yurik72/ESPHap <ESPHap@noreply.github.com>
Cc: Yurik72 <yurik.kovalenko@gmail.com>; State change <state_change@noreply.github.com>
Subject: Re: [Yurik72/ESPHap] Window Covering - Roller Blind example (#41)

 

Here's a crash with debug turned on & decode:

Decode:

Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 27 results
0x40203358: binary_to_string at /Users/evan/Documents/Arduino/libraries/ESPHap/debug.c line 26
0x40203308: binary_to_string at /Users/evan/Documents/Arduino/libraries/ESPHap/debug.c line 13
0x40283f5d: chip_v6_unset_chanfreq at ?? line ?
0x4022410d: client_send(_client_context_t*, unsigned char*, unsigned int) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 933
0x402241e0: client_send_chunk(unsigned char*, unsigned int, void*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 970
0x4021712e: json_flush at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 58
0x402063a4: json_write at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 79
0x402063fb: _do_write at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 279
0x402067f2: json_string at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 302
0x40206608: json_array_end at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 171
0x40223a6e: write_characteristic_json(json_stream*, _client_context_t*, _homekit_characteristic const*, characteristic_format_t, homekit_value_t const*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 552
0x40206402: _do_write at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 280
0x402067e8: json_string at /Users/evan/Documents/Arduino/libraries/ESPHap/json.c line 298
0x40225319: homekit_server_on_get_accessories(_client_context_t*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 2187 (discriminator 2)
0x40283dd4: chip_v6_unset_chanfreq at ?? line ?
0x40227ad1: homekit_server_on_message_complete(http_parser*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3249
0x40205a0c: http_parser_execute at /Users/evan/Documents/Arduino/libraries/ESPHap/http_parser.c line 1871 (discriminator 1)
0x402033ad: print_binary at /Users/evan/Documents/Arduino/libraries/ESPHap/debug.c line 41
0x40226691: homekit_client_process(_client_context_t*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3393 (discriminator 1)
0x40237f90: esp8266::MDNSImplementation::MDNSResponder::_updateProbeStatus() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Control.cpp line 1321 (discriminator 1)
0x40226a38: homekit_server_process(homekit_server_t*) at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 3612
0x40227de9: arduino_homekit_loop at /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp line 4162
0x4022b451: __delay at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 57
0x40227f10: hap_homekit_loop() at /Users/evan/Documents/Arduino/libraries/ESPHap/homekitintegrationcpp.cpp line 18
0x40218311: loop at /Users/evan/Documents/Arduino/EspHap_WindowCover/EspHap_WindowCover.ino line 496
0x4022ae7c: loop_wrapper() at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
0x40100dcd: cont_wrapper at /Users/evan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/cont.S line 81

Log:
13:06:07.665 -> >>> HomeKit: WiFiServer receives*WM: [2] Connection result: WL_CONNECTED 13:06:10.193 -> *WM: [3] lastconxresult: WL_CONNECTED 13:06:10.193 -> *WM: [1] AutoConnect: SUCCESS 13:06:10.228 -> *WM: [1] STA IP Address: 192.168.1.12 13:06:10.228 -> *WM: [3] unloading 13:06:10.228 -> Free heap: 29200 13:06:10.228 -> >>> HomeKit: init_storage_ex size 0x1c1 13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic 0: 23 13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic 1: 6D 13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic 2: 7C 13:06:10.228 -> >>> Home Integration: hap_services added chararacteristic 3: 72 13:06:10.228 -> >>> Home Integration: homekit_is_paired 1 13:06:10.262 -> >>> HomeKit: Free heap: 27080 13:06:10.262 -> >>> server_new: WiFiServer begin at port: 5556 13:06:10.262 -> 13:06:10.262 -> >>> HomeKit: Starting server 13:06:10.262 -> >>> HomeKit: storage init 0 13:06:10.262 -> >>> HomeKit: Using existing accessory ID: 2C:C0:9C:5E:D9:A2 13:06:10.262 -> >>> HomeKit: find pairings 13:06:10.262 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup 13:06:10.262 -> >>> HomeKit: Configuring mDNS 13:06:10.262 -> >>> HomeKit: MDNS.begin: DRB1-2CC0, IP: 192.168.1.12 13:06:10.295 -> >>> homekit_mdns_init: Accessory Setup ID = YK72 13:06:10.295 -> >>> HomeKit: Init server over 13:06:40.545 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8) 13:06:40.545 -> >>> HomeKit: Got new client connection: local 192 [... cut ...] CC0._hap._tcp.local\x0D\x0AContent-Length: 108\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"ev":true},{"aid":1,"iid":11,"ev":true},{"aid":1,"iid":12,"ev":true}]}" 13:31:31.729 -> >>> homekit_server_on_message_complete: http_parser message_complete 13:31:31.729 -> >>> HomeKit: [Client 1073704140] Update Characteristics 13:31:31.729 -> >>> homekit_server_on_update_characteristics: Free heap: 20224 13:31:31.729 -> >>> homekit_server_on_update_chara

eford321, 2021-02-12:
snippet of log before crash with commit e59cab2

`12:57:40.575 -> >>> homekit_client_process: [Client 1073707076] Reading 80 bytes
12:57:40.575 -> >>> HomeKit: [Client 1073707076] Got 80 incomming data, encrypted is 
12:57:40.575 -> >>> homekit_client_process: [Client 1073707076] Decrypted 62 bytes, available 0
12:57:40.575 -> Decrypted data (62 bytes): "GET /accessories HTTP/1.1\x0D\x0AHost: DRB1-2CC0._hap._tcp.local\x0D\x0A\x0D\x0A"
12:57:40.618 -> >>> homekit_server_on_message_complete: http_parser message_complete
12:57:40.618 -> >>> HomeKit: [Client 1073707076] Get Accessories
12:57:40.618 -> >>> homekit_server_on_get_accessories: Free heap: 5968
12:57:40.618 -> >>> client_send: [Client 1073707076] send data size=107, encrypted=true
12:57:40.618 -> >>> client_send: [Client 1073707076] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
12:57:40.618 -> >>> client_send_encrypted_: [Client 1073707076] client_send_encrypted_ size=107
12:57:40.656 -> >>> write: [Client 1073707076] socket.write, data_size=125, write_size=125, availableForWrite=2920, WriteError=0
12:57:40.656 -> >>> homekit_server_on_get_accessories: [Client 1073707076] Get Accessories, start send json body
12:57:40.656 -> >>> client_send_chunk: [Client 1073707076] client_send_chunk, size=1023, offset=5
12:57:40.698 -> >>> client_send: [Client 1073707076] send data size=1030, encrypted=true
12:57:40.698 -> >>> client_send: [Client 1073707076] Sending payload: 3ff\x0D\x0A{"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":2,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"DRB1"},{"aid":1,"iid":3,"type":"20","perms":["pr"],"description":"Manufacturer","format":"string","value":"Yurik72"},{"aid":1,"iid":4,"type":"30","perms":["pr"],"description":"Serial Number","format":"string","value":"0"},{"aid":1,"iid":5,"type":"21","perms":["pr"],"description":"Model","format":"string","value":"EspHapWindowCovering"},{"aid":1,"iid":6,"type":"52","perms":["pr"],"description":"Firmware Revision","format":"string","value":"1.0"},{"aid":1,"iid":7,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":8,"type":"8C","hidden":false,"primary":true,"characteristics":[{"aid":1,"iid":9,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"WINDOW_BLIND"},{"aid":1,"iid":10,"type":"6D","perms":["pr","ev"],"ev":false,"description":"Current Position","format":"uint8\x0D\x0A
12:57:40.767 -> >>> client_send_encrypted_: [Client 1073707076] client_send_encrypted_ size=1030
12:57:40.807 -> >>> write: [Client 1073707076] socket.write, data_size=1042, write_size=1042, availableForWrite=2920, WriteError=0
12:57:40.807 -> >>> write: [Client 1073707076] socket.write, data_size=24, write_size=24, availableForWrite=2920, WriteError=0
12:57:40.807 -> >>> client_send_chunk: [Client 1073707076] client_send_chunk, size=445, offset=5
12:57:40.840 -> >>> client_send: [Client 1073707076] send data size=452, encrypted=true
12:57:40.840 -> >>> client_send: [Client 1073707076] Sending payload: 1bd\x0D\x0A","unit":"percentage","minValue":0,"maxValue":100,"minStep":1,"value":0},{"aid":1,"iid":11,"type":"7C","perms":["pr","pw","ev"],"ev":false,"description":"Target Position","format":"uint8","unit":"percentage","minValue":0,"maxValue":100,"minStep":1,"value":0},{"aid":1,"iid":12,"type":"72","perms":["pr","ev"],"ev":false,"description":"Position State","format":"uint8","minValue":0,"maxValue":2,"minStep":1,"valid-values":[0,1,2],"value":2}]}]}]}\x0D\x0A
12:57:40.880 -> >>> client_send_encrypted_: [Client 1073707076] client_send_encrypted_ size=452
12:57:40.922 -> >>> write: [Client 1073707076] socket.write, data_size=470, write_size=470, availableForWrite=2920, WriteError=0
12:57:40.922 -> >>> client_send_chunk: [Client 1073707076] client_send_chunk, size=0, offset=3
12:57:40.922 -> >>> client_send: [Client 1073707076] send data size=5, encrypted=true
12:57:40.922 -> >>> client_send: [Client 1073707076] Sending payload: 0\x0D\x0A\x0D\x0A
12:57:40.922 -> >>> client_send_encrypted_: [Client 1073707076] client_send_encrypted_ size=5
12:57:40.957 -> >>> write: [Client 1073707076] socket.write, data_size=23, write_size=23, availableForWrite=2920, WriteError=0
12:59:05.497 -> >>> HomeKit: [Client 1073707076] Disconnected!
12:59:05.497 -> >>> HomeKit: [Client 1073707076] Closing client connection
12:59:05.497 -> >>> homekit_server_close_client: [Client 1073707076] The sockect is stoped
12:59:05.497 -> >>> HomeKit: [Client 0] client content free
13:00:00.482 -> >>> HomeKit: [Client 1073713580] Disconnected!
13:00:00.482 -> >>> HomeKit: [Client 1073713580] Closing client connection
13:00:00.482 -> >>> homekit_server_close_client: [Client 1073713580] The sockect is stoped
13:00:00.482 -> >>> HomeKit: [Client 0] client content free
13:00:44.408 -> >>> HomeKit: [Client 1073705652] Disconnected!
13:00:44.408 -> >>> HomeKit: [Client 1073705652] Closing client connection
13:00:44.451 -> >>> homekit_server_close_client: [Client 1073705652] The sockect is stoped
13:00:44.451 -> >>> HomeKit: [Client 0] client content free
13:00:44.675 -> >>> HomeKit: [Client 1073702516] Disconnected!
13:00:44.675 -> >>> HomeKit: [Client 1073702516] Closing client connection
13:00:44.675 -> >>> homekit_server_close_client: [Client 1073702516] The sockect is stoped
[... cut ...] "},{"aid":1,"iid":7,"type":"14","perms":["pw"],"description":"Identify","format":"bool"}]},{"iid":8,"type":"8C","hidden":false,"primary":true,"characteristics":[{"aid":1,"iid":9,"type":"23","perms":["pr"],"description":"Name","format":"string","value":"WINDOW_BLIND"},{"aid":1,"iid":10,"type":"6D","perms":["pr","ev"],"ev":false,"description":"Current Position","format":"uint8\x0D\x0A
13:01:00.657 -> >>> client_send_encrypted_: [Client 1073710340] client_send_encrypted_ size=1030
13:01:00.657 -> `

eford321, 2021-02-12:
I did find something of interest looking through the logs. It seems client id's are being reassigned to different devices after disconnects. Would this be a result of caching?
For example, iPad-A is assigned Client 1073703004 upon first connecting and later disconnects. it connects again and has the same client Id, then disconnects. HomePod-B connects and is assigned Client 1073703004. iPad-A connects again and is assigned Client 1073695676.  
Shouldn't  the client ID stay with the previous device and not be reused for a new one?

Yurik72, 2021-02-13:
The client ID is actually pointer to socket (WifiClient*) in case of ESP8266 , therefore it can be repeated

eford321, 2021-02-13:
Ok. Thanks.

Yurik72, 2021-02-14:
A few minutes ago I have commited new version, It fixes some memory leak

eford321, 2021-02-14:
Loaded newest commit.  Getting long periods of "No Response" when opening the home app on phone or Mac. Happened at 12:00 took until 12:04 for app to show updated value.  Again after a disconnect at 12:08 in this log: both apps were showing updated value at 12:09.
...
11:52:43.382 -> *WM: [2] Connection result: WL_CONNECTED
11:52:43.416 -> *WM: [3] lastconxresult: WL_CONNECTED
11:52:43.416 -> *WM: [1] AutoConnect: SUCCESS 
11:52:43.416 -> *WM: [1] STA IP Address: 192.168.1.12
11:52:43.416 -> *WM: [3] unloading 
11:52:43.416 -> Free heap: 28680
11:52:43.416 -> >>> HomeKit: init_storage_ex size 0x1c1
11:52:43.416 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
11:52:43.416 -> >>> Home Integration: hap_services added chararacteristic  1: 6D 
11:52:43.416 -> >>> Home Integration: hap_services added chararacteristic  2: 7C 
11:52:43.449 -> >>> Home Integration: hap_services added chararacteristic  3: 72 
11:52:43.449 -> >>> Home Integration: homekit_is_paired 1
11:52:43.449 -> >>> HomeKit: Free heap: 26560
11:52:43.449 -> >>> server_new: WiFiServer begin at port: 5556
11:52:43.449 -> 
11:52:43.449 -> >>> HomeKit: Starting server
11:52:43.449 -> >>> HomeKit: storage init 0
11:52:43.449 -> >>> HomeKit: Using existing accessory ID: 2C:C0:9C:5E:D9:A2
11:52:43.449 -> >>> HomeKit: find pairings
11:52:43.449 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup
11:52:43.487 -> >>> HomeKit: Configuring mDNS
11:52:43.487 -> >>> HomeKit: MDNS.begin: DRB1-2CC0, IP: 192.168.1.12
11:52:43.487 -> >>> homekit_mdns_init: Accessory Setup ID = YK72
11:52:43.487 -> >>> HomeKit: Init server over
11:53:35.990 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
11:53:35.990 -> >>> HomeKit: Got new client connection: local 192.168.1.12:5556, remote 192.168.1.13:57872
11:53:35.990 -> >>> HomeKit: [Client 1073696196] Got client content from the cache 0
11:53:36.023 -> >>> HomeKit: [Client 1073696196] new client accepted
11:53:36.023 -> >>> homekit_client_process: [Client 1073696196] Reading 160 bytes
11:53:36.023 -> >>> HomeKit: [Client 1073696196] Got 160 incomming data, encrypted is 
11:53:36.058 -> >>> homekit_server_on_message_complete: http_parser message_complete
11:53:36.058 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
11:53:36.058 -> >>> homekit_server_on_pair_verify: Free heap: 21352
11:53:36.058 -> >>> tlv_debug: Got following TLV values:
11:53:36.058 -> >>> tlv_debug: Type 6 value (1 bytes): \x01
11:53:36.058 -> >>> tlv_debug: Type 3 value (32 bytes): \xA3\x8E\xC1b\xE2\x04\xCB4>\xF9\xC3}\x82\x80\xE0\xA0\xA7\x15\xF8\xE4\xF7\xE1\x80z~\xE0\xF3\xE3\x13\xDE\x0D\x1D
11:53:36.095 -> >>> HomeKit: [Client 1073696196] Pair Verify Step 1/2
11:53:36.095 -> >>> homekit_server_on_pair_verify: [Client 1073696196] Importing device Curve25519 public key
11:53:36.095 -> >>> homekit_server_on_pair_verify: [Client 1073696196] Generating accessory Curve25519 key
11:53:36.231 -> >>> homekit_server_on_pair_verify: [Client 1073696196] Exporting accessory Curve25519 public key
11:53:36.231 -> >>> homekit_server_on_pair_verify: [Client 1073696196] Generating Curve25519 shared secret
11:53:36.366 -> >>> homekit_server_on_pair_verify: [Client 1073696196] Generating signature
11:53:36.433 -> >>> homekit_server_on_pair_verify: [Client 1073696196] Generating proof
11:53:36.433 -> >>> homekit_server_on_pair_verify: [Client 1073696196] Encrypting response
11:53:36.467 -> >>> send_tlv_response: [Client 1073696196] Sending TLV response
11:53:36.467 -> >>> tlv_debug: Got following TLV values:
11:53:36.467 -> >>> tlv_debug: Type 6 value (1 bytes): \x02
11:53:36.467 -> >>> tlv_debug: Type 3 value (32 bytes): \x05L\x91<\xC4#v\x06\x82V\xDC\xA3\xD2YG/>\xCF\xD5\xFE\x1Cd\xDE%\x12\x05\xEF\x0Fc\xEA\xCD\x02
11:53:36.467 -> >>> tlv_debug: Type 5 value (101 bytes): \x948\xFD\x876\xDC\xE2\x9D\xB7\xEC\xB3\xEC\xF3\xD9\x85\xB3R\xFC\x9D\x0C\x1E\xEE\x96\x99\xB0\xB6}\x1Ey\xFEhz}\x00\x85\xFE7\x0C\xA8\xC9:\xF8Cy\xE5A\xB4\xA1\xD3f\x12q\xF6\xA5N\x8B\xB3\x95\xBB\xE3Tc\xF3\xD1+\xE3\xA5\xBFma\x051\x89c\xE8\xBE\xE3\xE9r\xD0\xB8\xBA\xF6\xBB7D\xD1\xF1\x9A'@\xB1\xC5=\xDD9x\xC3*\xE1
11:53:36.500 -> >>> client_send: [Client 1073696196] send data size=244, encrypted=false
11:53:36.500 -> >>> client_send: [Client 1073696196] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 \x05L\x91<\xC4#v\x06\x82V\xDC\xA3\xD2YG/>\xCF\xD5\xFE\x1Cd\xDE%\x12\x05\xEF\x0Fc\xEA\xCD\x02\x05e\x948\xFD\x876\xDC\xE2\x9D\xB7\xEC\xB3\xEC\xF3\xD9\x85\xB3R\xFC\x9D\x0C\x1E\xEE\x96\x99\xB0\xB6}\x1Ey\xFEhz}\x00\x85\xFE7\x0C\xA8\xC9:\xF8Cy\xE5A\xB4\xA1\xD3f\x12q\xF6\xA5N\x8B\xB3\x95\xBB\xE3Tc\xF3\xD1+\xE3\xA5\xBFma\x051\x89c\xE8\xBE\xE3\xE9r\xD0\xB8\xBA\xF6\xBB7D\xD1\xF1\x9A'@\xB1\xC5=\xDD9x\xC3*\xE1
11:53:36.568 -> >>> write: [Client 1073696196] socket.write, data_size=244, write_size=244, availableForWrite=2920, WriteError=0
11:53:36.602 -> >>> HomeKit: Free heap: 19992
11:53:36.602 -> >>> homekit_client_process: [Client 1073696196] Reading 249 bytes
11:53:36.602 -> >>> HomeKit: [Client 1073696196] Got 249 incomming data, encrypted is 
11:53:36.602 -> >>> homekit_server_on_message_complete: http_parser message_complete
11:53:36.662 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
11:53:36.662 -> >>> homekit_server_on_pair_verify: Free heap: 20232
[... cut ...]
12:06:53.981 -> 	"iid":	11,
12:06:53.981 -> 	"ev":	true
12:06:53.981 -> }
12:06:53.981 -> >>> HomeKit: Free heap: 15400
12:06:53.981 -> >>> homekit_server_on_update_characteristics: [Client 1073706180] Processing element {
12:06:53.981 -> 	"aid":	1,
12:06:53.981 -> 	"iid":	12,
12:06:53.981 -> 	"ev":	true
12:06:53.981 -> }
12:06:53.981 -> >>> HomeKit: Free heap: 15384
12:06:53.981 -> >>> homekit_server_on_update_characteristics: [Client 1073706180] There were no processing errors, sending No Content response

eford321, 2021-02-14:
continued...

12:06:54.015 -> >>> client_send: [Client 1073706180] send data size=27, encrypted=true
12:06:54.015 -> >>> client_send: [Client 1073706180] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
12:06:54.015 -> >>> client_send_encrypted_: [Client 1073706180] client_send_encrypted_ size=27
12:06:54.015 -> >>> write: [Client 1073706180] socket.write, data_size=45, write_size=45, availableForWrite=2920, WriteError=0
12:06:54.053 -> >>> HomeKit: WiFiServer receives a new client (current 6, max 8)
12:06:54.053 -> >>> HomeKit: Got new client connection: local 192.168.1.12:5556, remote 192.168.1.15:64985
12:06:54.053 -> >>> HomeKit: [Client 1073710756] Got client content from the cache 6
12:06:54.089 -> >>> HomeKit: [Client 1073710756] new client accepted
12:06:54.089 -> >>> homekit_client_process: [Client 1073710756] Reading 160 bytes
12:06:54.089 -> >>> HomeKit: [Client 1073710756] Got 160 incomming data, encrypted is 
12:06:54.089 -> >>> homekit_server_on_message_complete: http_parser message_complete
12:06:54.089 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
12:06:54.089 -> >>> homekit_server_on_pair_verify: Free heap: 15336
12:06:54.089 -> >>> tlv_debug: Got following TLV values:
12:06:54.089 -> >>> tlv_debug: Type 6 value (1 bytes): \x01
12:06:54.125 -> >>> tlv_debug: Type 3 value (32 bytes): \x90\x18D8\xF9Y\x91\xA7\xBFd\xAB\x0D\xB0\x058\\\x05\x97\xBE\xCC\xEE\xE7\xA4?\xC0\x0F}{\x0A\x07\x82\x14
12:06:54.125 -> >>> HomeKit: [Client 1073710756] Pair Verify Step 1/2
12:06:54.125 -> >>> homekit_server_on_pair_verify: [Client 1073710756] Importing device Curve25519 public key
12:06:54.125 -> >>> homekit_server_on_pair_verify: [Client 1073710756] Generating accessory Curve25519 key
12:06:54.260 -> >>> homekit_server_on_pair_verify: [Client 1073710756] Exporting accessory Curve25519 public key
12:06:54.293 -> >>> homekit_server_on_pair_verify: [Client 1073710756] Generating Curve25519 shared secret
12:06:54.399 -> >>> homekit_server_on_pair_verify: [Client 1073710756] Generating signature
12:06:54.467 -> >>> homekit_server_on_pair_verify: [Client 1073710756] Generating proof
12:06:54.502 -> >>> homekit_server_on_pair_verify: [Client 1073710756] Encrypting response
12:06:54.502 -> >>> send_tlv_response: [Client 1073710756] Sending TLV response
12:06:54.502 -> >>> tlv_debug: Got following TLV values:
12:06:54.502 -> >>> tlv_debug: Type 6 value (1 bytes): \x02
12:06:54.502 -> >>> tlv_debug: Type 3 value (32 bytes): \x94\xC4\xA2v\xD5\xE0?2P'"\xFA\x0C^\x12\x91'\xB7\x1DafI\x10\xA2i\x91\xAAGW\xBA\x82,
12:06:54.502 -> >>> tlv_debug: Type 5 value (101 bytes): \xF7!\xFA\x91U\x14\xC5\xC0m\xEA\xAB@\x84\x1A\xD9\xE0\xAAu\xCB]\x9F\x1F\xCC\x0D/\x941\xC2X[\x9C\xB7j\xE1\xA7\xF8\xC9\xCEt\x8A\xBF\xEBF\x9A{pbG\x0F\x0F3\x94'\xC5\\U\xF8,\xDB\x17\xE3\xDB\x86\x17/\x07\x0E\xD2V0<\x0A\xD7w\x14E\x15\x827xw\xB6P'\x0DS\x1E\xD31\xF9w\x81\x05\xC3k\x1D\xAF\x9B\x8B\xEEp
12:06:54.537 -> >>> client_send: [Client 1073710756] send data size=244, encrypted=false
12:06:54.570 -> >>> client_send: [Client 1073710756] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 \x94\xC4\xA2v\xD5\xE0?2P'"\xFA\x0C^\x12\x91'\xB7\x1DafI\x10\xA2i\x91\xAAGW\xBA\x82,\x05e\xF7!\xFA\x91U\x14\xC5\xC0m\xEA\xAB@\x84\x1A\xD9\xE0\xAAu\xCB]\x9F\x1F\xCC\x0D/\x941\xC2X[\x9C\xB7j\xE1\xA7\xF8\xC9\xCEt\x8A\xBF\xEBF\x9A{pbG\x0F\x0F3\x94'\xC5\\U\xF8,\xDB\x17\xE3\xDB\x86\x17/\x07\x0E\xD2V0<\x0A\xD7w\x14E\x15\x827xw\xB6P'\x0DS\x1E\xD31\xF9w\x81\x05\xC3k\x1D\xAF\x9B\x8B\xEEp
12:06:54.604 -> >>> write: [Client 1073710756] socket.write, data_size=244, write_size=244, availableForWrite=2920, WriteError=0
12:06:54.638 -> >>> HomeKit: Free heap: 15136
12:06:54.638 -> >>> homekit_client_process: [Client 1073704860] Reading 249 bytes
12:06:54.638 -> >>> HomeKit: [Client 1073704860] Got 249 incomming data, encrypted is 
12:06:54.638 -> >>> homekit_server_on_message_complete: http_parser message_complete
12:06:54.638 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
12:06:54.638 -> >>> homekit_server_on_pair_verify: Free heap: 15376
12:06:54.672 -> >>> tlv_debug: Got following TLV values:
12:06:54.672 -> >>> tlv_debug: Type 5 value (120 bytes): \xD3\xABN\xED6\x16|\xCD\xEE\xE4L\xD6\xE3Se\x8E\x88\xD6\xD6\x1D\xCA\x04\xF5\xB6\x03[`\xDFY\xF1\xEFb\xD0XF\xA7\xAFI\xF6\x10\xB0C\xC1\xDA.K\x81\x9D\xF4\x04\x13\x99\xB1\xA7\x05\x91Vc_\xF3\x12\xDB\xD6\x9C\xF7\x04WM\x16\xF2\x0D\xB1\xE6\x10\xE6 k\xD95k\xF4\xA8\xA4\xA1\xA4\xFD\x1C\xDB\x99%oI\x8F^\xBB\x9Cx\xAB\xA7;_\x92\xA4\x90ei+\xD3\xFBA\x17_\x14\x00\x00\xCD\xDF\x81\x8B\xA5
12:06:54.706 -> >>> tlv_debug: Type 6 value (1 bytes): \x03
12:06:54.706 -> >>> HomeKit: [Client 1073704860] Pair Verify Step 2/2
12:06:54.706 -> >>> homekit_server_on_pair_verify: [Client 1073704860] Decrypting payload
12:06:54.706 -> >>> homekit_server_on_pair_verify: [Client 1073704860] Searching pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
12:06:54.706 -> >>> HomeKit: [Client 1073704860] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
12:06:54.741 -> >>> homekit_server_on_pair_verify: [Client 1073704860] Verifying device signature
12:06:55.510 -> >>> send_tlv_response: [Client 1073704860] Sending TLV response
12:06:55.510 -> >>> tlv_debug: Got following TLV values:
12:06:55.510 -> >>> tlv_debug: Type 6 value (1 bytes): \x04
12:06:55.510 -> >>> client_send: [Client 1073704860] send data size=105, encrypted=false
[... cut ...]
12:09:47.771 -> >>> write: [Client 1073705892] socket.write, data_size=103, write_size=103, availableForWrite=2920, WriteError=0
12:09:47.771 -> >>> client_send_chunk: [Client 1073705892] client_send_chunk, size=0, offset=3
12:09:47.771 -> >>> client_send: [Client 1073705892] send data size=5, encrypted=true
12:09:47.771 -> >>> client_send: [Client 1073705892] Sending payload: 0\x0D\x0A\x0D\x0A
12:09:47.771 -> >>> client_send_encrypted_: [Client 1073705892] client_send_encrypted_ size=5
12:09:47.807 -> >>> write: [Client 1073705892] socket.write, data_size=23, write_size=23, availableForWrite=2920, WriteError=0

Yurik72, 2021-02-14:
Got it. I'm playing as well with TCP option, but finally my understands it works after delays.
Any how the main resason of this improves are  stability, decrease memory leaks and fragmentation, as soon that happens can work on TCP options more... Hopefully they are not very depended from iOS version

eford321, 2021-02-15:
improved to about 4 hours with commit ee4c274.  Better response also.  See attached log/decode.
[ee4c274 log-decode.txt.zip](https://github.com/Yurik72/ESPHap/files/5983782/ee4c274.log-decode.txt.zip)

Yurik72, 2021-02-16:
More improves in commit https://github.com/Yurik72/ESPHap/commit/f5d4cb46c90883d97fbc41f085b377b0c668dc53

As well can you compile without debug option,  It's helpfull to debug comminication but in production it's eating memory. For instance I'h fixed some "bad" places

eford321, 2021-02-16:
Will do.

Still running 2 copies of ee4c274. Both are hung showing no response even after disconnects and reconnect.  192.168.1.13 and 192.168.1.15 are the devices shown below. Log attached:  ( I will update to the latest commit. )

[image]
 
[image]

...09:22:23.722 -> >>> HomeKit: WiFiServer receives a new client (current 5, max 8)>>> HomeKit: [Client 1073705244] Disconnected!
09:22:41.464 -> >>> HomeKit: [Client 1073705244] Closing client connection
09:22:41.464 -> >>> homekit_server_close_client: [Client 1073705244] The sockect is stoped
09:22:41.464 -> >>> HomeKit: [Client 0] client content free
09:22:45.632 -> >>> HomeKit: WiFiServer receives a new client (current 6, max 8)
09:22:45.632 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.13:56301
09:22:45.632 -> >>> HomeKit: [Client 1073707348] Got client content from the cache 0
09:22:45.632 -> >>> HomeKit: [Client 1073707348] new client accepted
09:22:45.666 -> >>> homekit_client_process: [Client 1073707348] Reading 160 bytes
09:22:45.666 -> >>> HomeKit: [Client 1073706956] Disconnected!
09:22:45.666 -> >>> HomeKit: [Client 1073706956] Closing client connection
09:22:45.666 -> >>> homekit_server_close_client: [Client 1073706956] The sockect is stoped
09:22:45.666 -> >>> HomeKit: [Client 0] client content free
09:22:51.233 -> >>> HomeKit: [Client 1073707348] Disconnected!
09:22:51.233 -> >>> HomeKit: [Client 1073707348] Closing client connection
09:22:51.233 -> >>> homekit_server_close_client: [Client 1073707348] The sockect is stoped
09:22:51.269 -> >>> HomeKit: [Client 0] client content free
09:23:21.997 -> >>> HomeKit: [Client 1073706348] Disconnected!
09:23:21.997 -> >>> HomeKit: [Client 1073706348] Closing client connection
09:23:21.997 -> >>> homekit_server_close_client: [Client 1073706348] The sockect is stoped
09:23:21.997 -> >>> HomeKit: [Client 0] client content free
09:23:22.030 -> >>> HomeKit: [Client 1073707044] Disconnected!
09:23:22.030 -> >>> HomeKit: [Client 1073707044] Closing client connection
09:23:22.030 -> >>> homekit_server_close_client: [Client 1073707044] The sockect is stoped
09:23:22.030 -> >>> HomeKit: [Client 0] client content free
09:23:22.268 -> >>> HomeKit: [Client 1073706924] Disconnected!
09:23:22.268 -> >>> HomeKit: [Client 1073706924] Closing client connection
09:23:22.268 -> >>> homekit_server_close_client: [Client 1073706924] The sockect is stoped
09:23:22.268 -> >>> HomeKit: [Client 0] client content free
09:23:22.751 -> >>> HomeKit: [Client 1073707076] Disconnected!
09:23:22.751 -> >>> HomeKit: [Client 1073707076] Closing client connection
09:23:22.751 -> >>> homekit_server_close_client: [Client 1073707076] The sockect is stoped
09:23:22.751 -> >>> HomeKit: [Client 0] client content free
09:23:26.023 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
09:23:26.023 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.13:56359
09:23:26.023 -> >>> HomeKit: [Client 1073706348] Got client content from the cache 0
09:23:26.023 -> >>> HomeKit: [Client 1073706348] new client accepted
09:23:26.023 -> >>> homekit_client_process: [Client 1073706348] Reading 160 bytes
09:24:25.996 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
09:24:26.030 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.13:56433
09:24:26.030 -> >>> HomeKit: [Client 1073704692] Got client content from the cache 1
09:24:26.030 -> >>> HomeKit: [Client 1073704692] new client accepted
09:24:26.030 -> >>> homekit_client_process: [Client 1073704692] Reading 160 bytes
09:24:26.030 -> >>> HomeKit: [Client 1073706348] Disconnected!
09:24:26.030 -> >>> HomeKit: [Client 1073706348] Closing client connection
09:24:26.065 -> >>> homekit_server_close_client: [Client 1073706348] The sockect is stoped
09:24:26.065 -> >>> HomeKit: [Client 0] client content free
09:25:27.758 -> >>> HomeKit: [Client 1073704692] Disconnected!
09:25:27.758 -> >>> HomeKit: [Client 1073704692] Closing client connection
09:25:27.758 -> >>> homekit_server_close_client: [Client 1073704692] The sockect is stoped
09:25:27.758 -> >>> HomeKit: [Client 0] client content free
09:27:44.265 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
09:27:44.265 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.15:50886
09:27:44.265 -> >>> HomeKit: [Client 1073706348] Got client content from the cache 0
09:27:44.265 -> >>> HomeKit: [Client 1073706348] new client accepted
09:27:44.265 -> >>> homekit_client_process: [Client 1073706348] Reading 160 bytes
09:27:54.026 -> >>> HomeKit: [Client 1073706348] Disconnected!
09:27:54.026 -> >>> HomeKit: [Client 1073706348] Closing client connection
09:27:54.026 -> >>> homekit_server_close_client: [Client 1073706348] The sockect is stoped
09:27:54.026 -> >>> HomeKit: [Client 0] client content free
09:28:13.505 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
09:28:13.505 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.15:50888
[... 9 lines cut ...]
09:29:13.686 -> >>> HomeKit: [Client 1073706348] Closing client connection
09:29:13.686 -> >>> homekit_server_close_client: [Client 1073706348] The sockect is stoped
09:29:13.686 -> >>> HomeKit: [Client 0] client content free
09:29:13.686 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
[... cut ...]
09:30:13.867 -> >>> HomeKit: [Client 1073706348] Closing client connection
09:30:13.867 -> >>> homekit_server_close_client: [Client 1073706348] The sockect is stoped
09:30:13.867 -> >>> HomeKit: [Client 0] client content free
09:30:27.077 -> >>> HomeKit: [Client 1073703508] Disconnected!
09:30:27.077 -> >>> HomeKit: [Client 1073703508] Closing client connection
09:30:27.077 -> >>> homekit_server_close_client: [Client 1073703508] The sockect is stoped
09:30:27.077 -> >>> HomeKit: [Client 0] client content free

Yurik72, 2021-02-16:
Ops, very strange, ESP immediately  disconnects them after connection established. It happens first time , I suppose there is new issue, potentially regression

In my understanding not happens before on the same commit.. I do not see memory consumption from this log

What can happen if we about to limit to memory, we have a problem with TCP stack …….

Thanks for good job and deep testing, still working, please check latest commit

 

 <https://user-images.githubusercontent.com/5996455/108076770-c83fc000-7039-11eb-8e6f-f29d58275c73.png> 

 <https://user-images.githubusercontent.com/5996455/108076181-27e99b80-7039-11eb-81e7-c129e6860f25.png>

eford321, 2021-02-16:
testing f5d4cb4 now.  Accidentally deleted log will restart capture. Heap
was at 12000~ when my Mac showed No Response. Serial output stopped, but
ESP was still functioning from the Home app on my phone.

eford321, 2021-02-16:
Here's a log with 192.168.1.13 (Mac) showing no response. Killing the home application and restarting does not help. Actually seeing no response from wife's iPad, wife's iPhone and two HomePods. Tried from these devices after 13:57 on this log. Log shows no activity.
Connections work from 192.168.1.15 (my iPhone), but the phone was away from home for about 1 hour.
It seems like the other apple devices think they are still connected, but the ESP timed out the connections and ignores their requests. 
The phone that was away reconnected 13:41:33 on this log.
...
11:43:36.454 -> *WM: [2] Connection result: WL_CONNECTED
11:43:36.454 -> *WM: [3] lastconxresult: WL_CONNECTED
11:43:36.454 -> *WM: [1] AutoConnect: SUCCESS 
11:43:36.487 -> *WM: [1] STA IP Address: 192.168.1.11
11:43:36.487 -> *WM: [3] unloading 
11:43:36.487 -> Free heap: 28968
11:43:36.487 -> >>> HomeKit: init_storage_ex size 0x1c1
11:43:36.487 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
11:43:36.487 -> >>> Home Integration: hap_services added chararacteristic  1: 6D 
11:43:36.487 -> >>> Home Integration: hap_services added chararacteristic  2: 7C 
11:43:36.487 -> >>> Home Integration: hap_services added chararacteristic  3: 72 
11:43:36.524 -> >>> Home Integration: homekit_is_paired 1
11:43:36.524 -> >>> HomeKit: Free heap: 26848
11:43:36.524 -> >>> HomeKit: Starting server
11:43:36.524 -> >>> HomeKit: storage init 0
11:43:36.524 -> >>> HomeKit: Using existing accessory ID: 4D:96:96:18:EF:E7
11:43:36.524 -> >>> HomeKit: find pairings
11:43:36.524 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup
11:43:36.524 -> >>> HomeKit: Configuring mDNS
11:43:36.524 -> >>> HomeKit: MDNS.begin: DRB2-4D96, IP: 192.168.1.11
11:43:36.560 -> >>> HomeKit: Init server over
11:43:38.702 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
11:43:38.702 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.28:49653
11:43:38.702 -> >>> HomeKit: [Client 1073703236] Got client content from the cache 0
11:43:38.702 -> >>> HomeKit: [Client 1073703236] new client accepted
11:43:38.739 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
11:43:38.739 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.37:55390
11:43:38.739 -> >>> HomeKit: [Client 1073703964] Got client content from the cache 1
11:43:38.739 -> >>> HomeKit: [Client 1073703964] new client accepted
11:43:38.739 -> >>> HomeKit: [Client 1073703964] Pair Verify Step 1/2
11:43:39.126 -> >>> HomeKit: Free heap: 20208
11:43:39.126 -> >>> HomeKit: [Client 1073703236] Pair Verify Step 1/2
11:43:39.441 -> >>> HomeKit: Free heap: 19800
11:43:39.477 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
11:43:39.477 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.33:50513
11:43:39.477 -> >>> HomeKit: [Client 1073695652] Got client content from the cache 2
11:43:39.477 -> >>> HomeKit: [Client 1073695652] new client accepted
11:43:39.477 -> >>> HomeKit: [Client 1073695652] Pair Verify Step 1/2
11:43:39.930 -> >>> HomeKit: Free heap: 19176
11:43:39.930 -> >>> HomeKit: [Client 1073703964] Pair Verify Step 2/2
11:43:39.930 -> >>> HomeKit: [Client 1073703964] Found pairing with 0F8927F8-0F84-44E7-8181-8714ED73BD51
11:43:40.732 -> >>> HomeKit: [Client 1073703964] Verification successful, secure session established
11:43:40.732 -> >>> HomeKit: Free heap: 19112
11:43:40.732 -> >>> HomeKit: [Client 1073703236] Pair Verify Step 2/2
11:43:40.732 -> >>> HomeKit: [Client 1073703236] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
11:43:41.526 -> >>> HomeKit: [Client 1073703236] Verification successful, secure session established
11:43:41.526 -> >>> HomeKit: Free heap: 18968
11:43:41.561 -> >>> HomeKit: WiFiServer receives a new client (current 3, max 8)
11:43:41.561 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.13:60356
11:43:41.561 -> >>> HomeKit: [Client 1073703636] Got client content from the cache 3
11:43:41.561 -> >>> HomeKit: [Client 1073703636] new client accepted
11:43:41.561 -> >>> HomeKit: [Client 1073703636] Pair Verify Step 1/2
11:43:42.004 -> >>> HomeKit: Free heap: 19240
11:43:42.004 -> >>> HomeKit: [Client 1073695652] Pair Verify Step 2/2
11:43:42.004 -> >>> HomeKit: [Client 1073695652] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
11:43:42.820 -> >>> HomeKit: [Client 1073695652] Verification successful, secure session established
11:43:42.820 -> >>> HomeKit: Free heap: 19176
11:43:42.820 -> >>> HomeKit: [Client 1073703964] Get Accessories
11:43:42.890 -> >>> HomeKit: [Client 1073703236] Get Accessories
11:43:42.927 -> >>> HomeKit: WiFiServer receives a new client (current 4, max 8)
11:43:42.927 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.21:50107
11:43:42.927 -> >>> HomeKit: [Client 1073706668] Got client content from the cache 4
11:43:42.960 -> >>> HomeKit: [Client 1073706668] new client accepted
11:43:42.960 -> >>> HomeKit: [Client 1073706668] Pair Verify Step 1/2
11:43:43.345 -> >>> HomeKit: Free heap: 18880
[... 384 lines cut ...]
13:53:57.725 -> >>> HomeKit: Free heap: 16784
13:53:57.725 -> >>> HomeKit: [Client 1073708844] Pair Verify Step 2/2
13:53:57.725 -> >>> HomeKit: [Client 1073708844] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
[... cut ...]
14:10:27.851 -> >>> HomeKit: Free heap: 16376
14:10:27.851 -> >>> HomeKit: [Client 1073708844] Get Characteristics
14:10:27.851 -> >>> HomeKit: [Client 1073708844] Update Characteristics
14:10:27.851 -> >>> HomeKit: Free heap: 16504
14:10:27.851 -> >>> HomeKit: [Client 1073708844] Updating characteristic 1.11 with integer 31
14:10:27.851 -> >>> HomeKit: [Client 1073708844] Update Characteristics
14:10:27.851 -> >>> HomeKit: Free heap: 16616
14:10:27.851 -> >>> HomeKit: [Client 1073708844] Updating characteristic 1.11 with integer 30

eford321, 2021-02-16:
It appears apple eventually times out the connection and reconnects.  I noticed the Mac home app had the current status, but when I tried the connection, it had already timed out.  Log shows it reconnected at 14:33.  I tried at 14:39 and got no response.  phone working at 14:42.

continuation of log from above:
...
14:16:54.701 -> >>> HomeKit: [Client 1073708844] Disconnected!
14:16:54.701 -> >>> HomeKit: [Client 1073708844] Closing client connection
14:16:54.701 -> >>> HomeKit: [Client 0] client content free
14:17:30.790 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
14:17:30.790 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.15:51232
14:17:30.790 -> >>> HomeKit: [Client 1073708844] Got client content from the cache 0
14:17:30.790 -> >>> HomeKit: [Client 1073708844] new client accepted
14:17:30.790 -> >>> HomeKit: [Client 1073708844] Pair Verify Step 1/2
14:17:30.790 -> >>> HomeKit: Free heap: 16248
14:17:30.790 -> >>> HomeKit: [Client 1073708844] Pair Verify Step 2/2
14:17:30.790 -> >>> HomeKit: [Client 1073708844] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:17:30.790 -> >>> HomeKit: [Client 1073708844] Verification successful, secure session established
14:17:30.790 -> >>> HomeKit: Free heap: 16296
14:17:30.790 -> >>> HomeKit: [Client 1073708844] Get Accessories
14:20:12.560 -> >>> HomeKit: [Client 1073708844] Disconnected!
14:20:12.560 -> >>> HomeKit: [Client 1073708844] Closing client connection
14:20:12.560 -> >>> HomeKit: [Client 0] client content free
14:23:05.454 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
14:23:05.454 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.15:51233
14:23:05.454 -> >>> HomeKit: [Client 1073708844] Got client content from the cache 0
14:23:05.454 -> >>> HomeKit: [Client 1073708844] new client accepted
14:23:05.454 -> >>> HomeKit: [Client 1073708844] Pair Verify Step 1/2
14:23:05.454 -> >>> HomeKit: Free heap: 16184
14:23:05.454 -> >>> HomeKit: [Client 1073708844] Pair Verify Step 2/2
14:23:05.454 -> >>> HomeKit: [Client 1073708844] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:23:05.454 -> >>> HomeKit: [Client 1073708844] Verification successful, secure session established
14:23:05.454 -> >>> HomeKit: Free heap: 16032
14:23:05.454 -> >>> HomeKit: [Client 1073708844] Get Accessories
14:33:13.576 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
14:33:13.576 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.28:49664
14:33:13.576 -> >>> HomeKit: [Client 1073706228] Got client content from the cache 1
14:33:13.576 -> >>> HomeKit: [Client 1073706228] new client accepted
14:33:13.576 -> >>> HomeKit: [Client 1073706228] Pair Verify Step 1/2
14:33:13.576 -> >>> HomeKit: Free heap: 15040
14:33:13.576 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
14:33:13.576 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.21:50119
14:33:13.576 -> >>> HomeKit: [Client 1073694756] Got client content from the cache 2
14:33:13.576 -> >>> HomeKit: [Client 1073694756] new client accepted
14:33:13.576 -> >>> HomeKit: [Client 1073694756] Pair Verify Step 1/2
14:33:13.576 -> >>> HomeKit: Free heap: 14296
14:33:13.576 -> >>> HomeKit: [Client 1073706228] Pair Verify Step 2/2
14:33:13.576 -> >>> HomeKit: [Client 1073706228] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:33:13.576 -> >>> HomeKit: [Client 1073706228] Verification successful, secure session established
14:33:13.576 -> >>> HomeKit: Free heap: 13968
14:33:13.576 -> >>> HomeKit: WiFiServer receives a new client (current 3, max 8)
14:33:14.475 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.37:55430
14:33:14.475 -> >>> HomeKit: [Client 1073706164] Got client content from the cache 3
14:33:14.475 -> >>> HomeKit: [Client 1073706164] new client accepted
14:33:14.475 -> >>> HomeKit: [Client 1073706164] Pair Verify Step 1/2
14:33:14.475 -> >>> HomeKit: Free heap: 12576
14:33:14.475 -> >>> HomeKit: [Client 1073694756] Pair Verify Step 2/2
14:33:14.475 -> >>> HomeKit: [Client 1073694756] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:33:14.475 -> >>> HomeKit: [Client 1073694756] Verification successful, secure session established
14:33:14.475 -> >>> HomeKit: Free heap: 13536
14:33:14.475 -> >>> HomeKit: [Client 1073706228] Get Accessories
14:33:14.475 -> >>> HomeKit: WiFiServer receives a new client (current 4, max 8)
14:33:14.475 -> >>> HomeKit: Got new client connection: local 192.168.1.11:5556, remote 192.168.1.13:54846
14:33:14.475 -> >>> HomeKit: [Client 1073707276] Got client content from the cache 4
14:33:14.475 -> >>> HomeKit: [Client 1073707276] new client accepted
14:33:14.475 -> >>> HomeKit: [Client 1073707276] Pair Verify Step 1/2
14:33:14.475 -> >>> HomeKit: Free heap: 13152
14:33:14.475 -> >>> HomeKit: [Client 1073706164] Pair Verify Step 2/2
14:33:15.578 -> >>> HomeKit: [Client 1073706164] Found pairing with 0F8927F8-0F84-44E7-8181-8714ED73BD51
14:33:15.578 -> >>> HomeKit: [Client 1073706164] Verification successful, secure session established
14:33:15.578 -> >>> HomeKit: Free heap: 13088
14:33:15.578 -> >>> HomeKit: [Client 1073694756] Get Accessories
[... 54 lines cut ...]
14:39:42.299 -> >>> HomeKit: [Client 1073708844] Closing client connection
14:39:42.299 -> >>> HomeKit: [Client 0] client content free
[... cut ...]
14:42:34.185 -> >>> HomeKit: Free heap: 15136
14:42:34.185 -> >>> HomeKit: [Client 1073708844] Update Characteristics
14:42:34.185 -> >>> HomeKit: Free heap: 15120
14:42:34.185 -> >>> HomeKit: Free heap: 15120
14:42:34.185 -> >>> HomeKit: Free heap: 15120
14:42:34.185 -> >>> HomeKit: [Client 1073708844] Get Characteristics
14:42:34.895 -> >>> HomeKit: [Client 1073708844] Update Characteristics
14:42:34.895 -> >>> HomeKit: Free heap: 15336
14:42:34.895 -> >>> HomeKit: [Client 1073708844] Updating characteristic 1.11 with integer 49

[22 more replies]