Yurik72/ESPHap #42: example for PM10 / PM2.5 device

bfaliszek, 2021-01-28:
I could ask for example with a PM10 or PM2.5 air pollution measuring device. 
I tried to do it based on to the EspHap_AirQuality_MQ135 example but it keeps showing me Carbon Dioxide Level(ppm) instead of PM10 level(ug/m3) on my iPhone.

Yurik72, 2021-01-28:
Hi , you are right, i have implemented in example only this characteristic "Carbon Dioxide Level" which is measured by ppm
You mentioned about VOC (ug/mm3). I will think how to combine. Or you can do by yourself by adding those characteristic in
homeintegration.c  
I will apprecciate a pull request

bfaliszek, 2021-01-29:
I wonder how to best solve it, because in the case of air pollution measurements, we measure various parameters (CO2, PM2.5, PM10, VOC etc.). Each of them has its own standards of acceptable values(not to mention the fact that are different acceptable values provided by WHO, US, EU and China). 
Additionally, not every device performs all of these measurements. For example, I have two of my projects, one measures all of the 4 parameters mentioned above, and the other measures only PM2.5 and PM10.

Yurik72, 2021-01-29:
In my understanding,
You should get EspHap_AirQuality_MQ135  as example and depending on capabilities you should add additional Characheristic (or replace Carbon Dioxide Level) , which are supported by Apple
For instance 
CHARACTERISTIC_PM25_DENSITY
CHARACTERISTIC_PM10_DENSITY
CHARACTERISTIC_VOC_DENSITY

From library points of view it's very easy, see impementation of 
hap_new_air_quality_service  ( homeintegration.c )
You can add or replace additional characheristic.
Then behaviour in the code the same as in the given example EspHap_AirQuality_MQ135

Yurik72, 2021-01-29:
Shure we can add additional function to homeintegration.c with parameters, which characheristic will be supported
On the main sketch behaviour very simple, 
For instance following can be used
homekit_service_characteristic_by_type(airqualitylevelservice, HOMEKIT_CHARACTERISTIC_CARBON_DIOXIDE_LEVEL);

If it returns NULL, characheristic is not supported