Yurik72/ESPHap #43: Switch returns to off without input

headblockhead, 2021-01-31:
I ran the EspHapLed example and got it working fine, but the switch always returns to the off position in HomeKit. Is this expected behavior? I expected the switch to "stick" in the On position.

https://user-images.githubusercontent.com/26520767/106383105-d21bbf00-63bb-11eb-9b87-96d03c4bfdd1.mp4

Here's the serial monitor output from the same time, showing that the messages were received.

```
11:54:22.188 -> >>> HomeKit: [Client 1073694364] Got 188 incomming data, encrypted is 
11:54:22.188 -> >>> HomeKit: [Client 1073694364] Update Characteristics
11:54:22.235 -> led_callback
11:54:22.235 -> set_led
11:54:22.235 -> notify hap
11:54:22.235 -> found characteristic
11:54:23.727 -> >>> HomeKit: [Client 1073694364] Got 188 incomming data, encrypted is 
11:54:23.727 -> >>> HomeKit: [Client 1073694364] Update Characteristics
11:54:23.727 -> led_callback
11:54:23.774 -> set_led
11:54:23.774 -> notify hap
11:54:23.774 -> found characteristic
11:54:25.261 -> >>> HomeKit: [Client 1073694364] Got 188 incomming data, encrypted is 
11:54:25.261 -> >>> HomeKit: [Client 1073694364] Update Characteristics
11:54:25.261 -> led_callback
11:54:25.308 -> set_led
11:54:25.308 -> notify hap
11:54:25.308 -> found characteristic
11:54:26.714 -> >>> HomeKit: [Client 1073694364] Got 188 incomming data, encrypted is 
11:54:26.714 -> >>> HomeKit: [Client 1073694364] Update Characteristics
11:54:26.714 -> led_callback
11:54:26.714 -> set_led
11:54:26.714 -> notify hap
11:54:26.714 -> found characteristic
11:55:33.698 -> >>> HomeKit: [Client 1073694364] Got 189 incomming data, encrypted is 
11:55:33.698 -> >>> HomeKit: [Client 1073694364] Update Characteristics
```

Yurik72, 2021-01-31:
Hi, definitely something wrong, I heard this first time, at least for this basic simple sketch, attach to the issue sketch code which you are using, I will check

headblockhead, 2021-02-04:
Sorry for the late reply, the email got filtered to spam 📬. I'm just using the default example (with the only change being the WiFi SSID and password) I'll attach the code in a comment later today as I'm writing this from my phone. Thanks!

Yurik72, 2021-02-04:
Ok, I really met first time instead of huge working device and feedbacks
You will help me, if it's still such behaviour, please comment a couple of lines in example sketch

`        if(ch->value.bool_value!=val){  //wil notify only if different
          ch->value.bool_value=val;
          homekit_characteristic_notify(ch,ch->value);
        }
`

you need always inform apple about actual state, so comment validation if value is changed

`    //    if(ch->value.bool_value!=val){  //wil notify only if different
          ch->value.bool_value=val;
          homekit_characteristic_notify(ch,ch->value);
     //   }
`

Than give me feedback and which version of iOS you are using

Yurik72, 2021-02-15:
Closing issue, due to not reproducable in any known enviropment

closed 2021-02-15 by Yurik72