Yurik72/ESPHap #48: Initial value bool error with  INIT_CHARACHTERISTIC_VAL(bool,ch,false);

chrwh, 2021-02-07:
Hi,

When I try to implement initial value from the library API description, I get and bool and ch not declared error?

Optinally you can set a default (initial) value to be informed Apple about initial state

 homekit_characteristic_t * ch= homekit_service_characteristic_by_type(hapservice, HOMEKIT_CHARACTERISTIC_ON);
 INIT_CHARACHTERISTIC_VAL(bool,ch,false);   //  will inform apple that lights is OFF

I get this error:

EspHapSwitch8266_Double_Web:146:27: error: expected primary-expression before 'bool'
  INIT_CHARACHTERISTIC_VAL(bool,ch,false);   //  will inform apple that lights is OFF
                           ^
EspHapSwitch8266_Double_Web:146:32: error: 'ch' was not declared in this scope
  INIT_CHARACHTERISTIC_VAL(bool,ch,false);   //  will inform apple that lights is OFF
                                ^
EspHapSwitch8266_Double_Web:146:40: error: 'INIT_CHARACHTERISTIC_VAL' was not declared in this scope
  INIT_CHARACHTERISTIC_VAL(bool,ch,false);   //  will inform apple that lights is OFF

Best regards,
Christian

Yurik72, 2021-02-08:
1.Use latest version or version tagged v1.0.7
2. Check what is ch in your code. It should be valid pointer to (homekit_characteristic_t *)

Any other question, please attach full sketch

closed 2021-02-08 by Yurik72

chrwh, 2021-02-08:
Hi,

I'm using v.1.0.7

I now also see and unprotected wifi ESP-E4061A, when finished pairing a setting up a new device, that give access to 192.168.4.1 ?? and evrybody can lok into that?

Here is the hole sketch look for //test

/// BASIC CONFIGURATION 

#define ENABLE_WIFI_MANAGER  // if we want to have built-in wifi configuration
                             // Otherwise direct connect ssid and pwd will be used
                             // for Wifi manager need extra library //https://github.com/tzapu/WiFiManager

#define ENABLE_WEB_SERVER    //if we want to have built in web server /site
#define ENABLE_OTA  //if Over the air update need  , ENABLE_WEB_SERVER must be defined first
#include <Arduino.h>

#ifdef ESP32
#include <SPIFFS.h>
#endif

#ifdef ESP8266
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include "coredecls.h"
#endif

#ifdef ENABLE_WEB_SERVER
#ifdef ESP8266
#include <ESP8266WebServer.h>
 ESP8266WebServer server(80);
#endif

#ifdef ESP32
#include <WebServer.h>
WebServer server(80);
#endif
#endif 

#if defined(ESP32) && defined(ENABLE_OTA)
#include <Update.h>
#endif

#ifdef ENABLE_WEB_SERVER
#include "spiffs_webserver.h"
bool isWebserver_started=false;
#endif 

const int relay_gpio=2;
const int relay1_gpio=0;

#ifdef ENABLE_WIFI_MANAGER
#include <WiFiManager.h>        //https://github.com/tzapu/WiFiManager
#endif

const char* HOSTNAME="my_thing";
const char* ssid     = "ssid";
const char* password = "pass";

extern "C"{
#include "homeintegration.h"
}
#ifdef ESP8266
#include "homekitintegrationcpp.h"
#endif

homekit_service_t* hapservice={0};
homekit_service_t* hapservice_CD={0};

[... 222 lines cut ...]
    homekit_characteristic_t * ch= homekit_service_characteristic_by_type(hapservice_CD, HOMEKIT_CHARACTERISTIC_ON);
    if(ch){
        
        if(ch->value.bool_value!=val){  //wil notify only if different
          ch->value.bool_value=val;
          homekit_characteristic_notify(ch,ch->value);
        }
    }
  }
}

void switch1_callback(homekit_characteristic_t *ch, homekit_value_t value, void *context) {
    Serial.println("switch1_callback");
 set_switch1(ch->value.bool_value);
}
#ifdef ENABLE_WIFI_MANAGER
void startwifimanager() {
  WiFiManager wifiManager;

  if (!wifiManager.autoConnect(HOSTNAME, NULL)) {
      ESP.restart();
      delay(1000);
   }
}
#endif

Yurik72, 2021-02-09:
Hi, About 
I now also see and unprotected wifi ESP-E4061A, when finished pairing a setting up a new device, that give access to 192.168.4.1 ?? and evrybody can lok into that?

This is not related to this library and standard behaviour of WiFi manager, It's properly documented
Definetelly thgs can happen when ESP not able to connect to your Wifi network and i't started in configuration mode.  
So, problem could be
1. Device can't properly connect to your Wifi network , weak signal , what ever
2.  You have upload a new scetch with option clear ESP. Means all data lost

chrwh, 2021-02-16:
Hi,
Can you provide any further help on how to use the below, I can't make it work?
homekit_characteristic_t * ch= homekit_service_characteristic_by_type(hapservice, HOMEKIT_CHARACTERISTIC_ON);
INIT_CHARACHTERISTIC_VAL(bool,ch,false); // will inform apple that lights is OFF
Best regards,
Christian

Yurik72, 2021-02-17:
Yes, but could you explain what is not working, compilation, behaviour... 
As well please attach a log

chrwh, 2021-02-17:
I get this error, if you try to compile the code about (right now the lines are masked out //)

I get this error:

EspHapSwitch8266_Double_Web:146:27: error: expected primary-expression before 'bool'
INIT_CHARACHTERISTIC_VAL(bool,ch,false); // will inform apple that lights is OFF
^
EspHapSwitch8266_Double_Web:146:32: error: 'ch' was not declared in this scope
INIT_CHARACHTERISTIC_VAL(bool,ch,false); // will inform apple that lights is OFF
^
EspHapSwitch8266_Double_Web:146:40: error: 'INIT_CHARACHTERISTIC_VAL' was not declared in this scope
INIT_CHARACHTERISTIC_VAL(bool,ch,false); // will inform apple that lights is OFF

What I want, is the outputs to be OFF when power is initial turned on

Yurik72, 2021-02-17:
Ok, 

1.	Send me a full sketch
2.	Which version you are using, please check . INIT_CHARACHTERISTIC_VAL implemented not so far

chrwh, 2021-02-17:
Hi,

 

I use V.1.0.7 (if I look in the library.properties file it says 1.0.3) but I downloaded from here https://github.com/Yurik72/ESPHap/releases/tag/v1.0.7

 

Atteched the full sketch, look around line 157, they are masked out // at the moment

 

Thanks and best regards,

Christian

Yurik72, 2021-02-17:
I do not see any attachments,

Check your Library folder ESPHap If file homeintegration.h contains

#define INIT_CHARACHTERISTIC_VAL

chrwh, 2021-02-17:
Argh I forgot to attach it. You can see the code above in the thread on GitHub.

Best regards
 
Christian

chrwh, 2021-02-20:
Just checked, the "#define INIT_CHARACHTERISTIC_VAL" is not in the homeintegration.h file, So the link you refere to here as being V.1.0.7 is not it's V.1.0.3 https://github.com/Yurik72/ESPHap/releases/tag/v1.0.7, I now donwloaded from the from page, and it can now compile, but with  many warining. When trying to access the webserver it crashes
[EspHapSwitch8266_Double_Web.zip](https://github.com/Yurik72/ESPHap/files/6013820/EspHapSwitch8266_Double_Web.zip)
 and reboots. This is run in a ESP8266-01 1mb.
File is now attached

chrwh, 2021-02-20:
Updtate: Completely wiped the ESP-01, and it seems to work now. Still getting a lot of compile warnings in regards to Wolfssl?

Yurik72, 2021-02-20:
Ok , good news, warning appears in wolfssl you are right, but you do not see any impact

chrwh, 2021-02-21:
So far no impact with the Wolfssl errors