Yurik72/ESPHap #52: Accessory 'No Response'

karl-fischer, 2021-02-13:
Hello :)

I am using the EspHap_Switch example. I have set the ssid and password to match my network. _#define ENABLE_WIFI_MANAGER_ is commented out.

I have successfully connected the ESP to my home WiFi and paried the switch accessory with the Home app on my iPhone.

The iPad is set to act as a Home Hub, with the intention that I will be able to access the accessories while I am away from home.

When I turn off WiFi on my iPhoneX, to simulate being away from home, and then place my iPhoneX on the desk with the Home app open, the switch accessory continues to show healthy for some minutes, but then shows No Response in the Home app. Note, I haven't pressed the button, just left the Home app open on the Home or Garage room. I'm looking at my iPhoneX and it's saying it right now. It shows an exclamation mark on the accessory in the Home app, and has the state ("Off") in red text. It also says, "Test Switch No Response" (I named the accessory _Test Switch_ when I added it to the Home app.

If I minimise the Home app on my iPhoneX and then open it again, the Test Switch state changes to "Updating" and then after a few seconds, shows "Off", which is the correct status and the "No Response" message goes away.

The iPad is plugged in to power, but the screen has gone to sleep (I don't want to leave the iPad screen on 24/7 with no sleep).

I am not sure how to proceed to make it reliably stay available in the Home app, without it changing to "No Response". Are you able to help with this please? I am keen to do whatever debugging can help with the investigation :)

Details:
- Arduino 1.8.13
- iPhoneX iOS 14.4 (192.168.1.101)
- iPad 8th gen, iPadOS 14.4 (192.168.1.100)
- ESPHap: v1.0.7
- ESP8266 v2.7.4 (in Boards Manager)
- Board: NodeMCU 1.0 (ESP-12E Module) - Note, this is a NodeMCU v1.0 Rev2.0 (192.168.1.155)
- CPU Freq: 120MHz
- Flash Size: 4MB (FS:2MB oTA:~1019KB)
- Debug port: Disabled
- Debug level: None
- IwIP Variant: v2 Lower Memory
- VTables: Flash
- Exceptions: Legacy (new can return nullptr)
- Erase Flash: Only Sketch
- SSL Support: Basic SSL ciphers (lower ROM usage) - Note, I am using WolfSSL 3.13.0 as per Readme instructions.
- Port COM3 - Note, I have the NodeMCU connected via USB.

Thank you :)

Yurik72, 2021-02-14:
Hello :)
Thanks for interesting of this project. Last few days I'm working on library stability, therefore version v1.07 is recomended.
However theare are big improvement on this library.  

1. Take the latest version (just latest commit not labeled yet) , you can see #41 discussion and the latest acvievemt, I have met such behaviour and there are two things done, which really improves and I can't reproduce this more. 
2.  Take a look #49, based on your setting you should compile with option v2 Higher Bandwidth , instead v2 Lower Memory

If you want to support and improve stability try to reproduce this behaviour again and very helpfull will be a CORE LOG (specially i need mDNS logs) at the moment when device is not responding

Then. In theory when it's stable reproducing you can try MDNS.announce() for instance every 5-10 sec in the loop. I did't implement this in the core, becasue it will kill network by garbage and I need an optimal way

Will appreciate with the comments or any pull request of sources or readme, to help to other guys, due to the big interest in the last months

Yurik72, 2021-02-14:
see latest commit https://github.com/Yurik72/ESPHap/commit/148660a97d0128aa96dfd76c4849d7af0b9fc9f1

karl-fischer, 2021-02-14:
Ok, I continued to do more monitoring with the iPad and iPhoneX on the same iCloud account (iPad on WiFi and iPhoneX not on WiFi) and v1.0.7 of ESPHap. I kept the iPhoneX Home app open and watched the accessory for at least an hour, the accessory worked fine the whole time and did not display the 'No Response' message. So, it seems to work fine when they are both logged in to the same iCloud account. 

I then removed the accessory, logged out of iCloud on the iPad, and then logged back in to the iPad using a different iCloud account. This is so the iPad can stay in the house when I sell the house and move out. I invited the iPhoneX to share the home (in Home app), and added the accessory again. After just a couple of minutes, the accessory on the iPhoneX Home app showed, 'No Response'. After a minute or two, the 'No Response' message cleared.

So, now to the testing you requested. I chanaged _IwIP Variant_ from _v2 Lower Memory_, to _v2 Higher Bandwidth_, downloaded commit 148660a, compiled and downloaded to the NodeMCU. I also have the iPad and iPhoneX back on the same iCloud account and the accessory is added again. I watched the iPhoneX for a few minutes, and it is showing the 'No Response' message. This is the first time I have seen this message when the iPad and iPhoneX were logged in to the same iCloud account.

So, with 148660a it appears to be showing 'No Response' even when they iPad and iPhoneX are both on the same iCloud account (remember, I always have the iPad on WiFi in the home and the iPhoneX WiFi is turned off to simulate it being away and trying to control the accessories remotely). It has showed 'No Response' twice now, in the past 5 minutes. However, as I continue to monitor the iPhoneX, the accessory is staying healthy for the past several minutes.

I'd like to get the logs you need, but I don't know how. What is a CORE LOG and how do I get it and the mDNS logs for you? 

Karl :)

Yurik72, 2021-02-15:
Ok, thanks for investigation i will back some TCP option, you can see changes in the commit.
CORE Log you can produce if you select Option on your Arduino:
Debug Port : Serial
Debug Level: CORE+WIFI+.....

karl-fischer, 2021-02-15:
Ok, in the Arduino IDE, I have set:

- Debug Port: Serial
- Debug Level: CORE+WIFI+HTTP_UPDATE+UPDATER+OTA+OOM+MDNS

I will test and report back with a copy of the text from my Serial Monitor window.

Karl :)

eford321, 2021-02-19:
I have tested and found this issue get worse from 148660a.  @karl-fischer can you try running e59cab2 to see if helps you?

I have been trying different timer settings, but am still getting disconnects.  It is acting like the socket keepalive isn't actually enabled.  Looking to find a good place to test with iskeepaliveenabled option. @Yurik72 thoughts?

Yurik72, 2021-02-19:
> Looking to find a good place to test with iskeepaliveenabled option. @Yurik72 thoughts?

Actually I need investigate, because keep Alive work to me, I have played with delays and it works, even if I put phone to fly mode and than back in the given tima-out

@karl i will investigate your logs. thanks

eford321, 2021-02-19:
Do you ever see the disconnects that I have in my logs on your devices? 

My Wifi network is a Google Wifi mesh. I had compatibility  problems with google/HomeKit a couple of years ago. I was able to adjust my network topology to get around previous issues. I'm interested to see if you see similar communications.

Yurik72, 2021-02-20:
Yes, I saw your log, and my understanding you have changed socket options to more fast reproduce and test memory leaks 😊

Now I understand that is your standard behavior….

Definitely  I had different picture and connections are stables with keep Alive, based on setting which on the github latest commit

As well I’m using this on 3 different environment  and picture always the same. Environment is very simple two of them is quite old WiFi routers…

One of them Asus mesh system.  So your network environment helps to find memory leak 😊

eford321, 2021-02-20:
Interesting I can run the EspHap_Switch and EspHapLed8266 examples without disconnects (no response) allowing HOMEKIT_MAX_CLIENTS = 8.  
My WindowBlind sketch is totally idle, just running the hap_homekit_loop(); and has disconnects.  I can't find anything I am doing with HAP setup that the examples are not doing.  Can you look to see if I missed something?  This sketch should run without a button, stepper motor or FRAM chip. 

BTW, the heap problem appears to be completely fixed.
[eford321_WindowCover.ino.zip](https://github.com/Yurik72/ESPHap/files/6016499/eford321_WindowCover.ino.zip)

karl-fischer, 2021-02-20:
Hi People :)

Have you tested with a remote client, via a Home Hub? That's the issue that I'm facing... the remote client shows the accessories as "No Response".

Yurik72, did you see anything helpful in my logs?

Karl :)

eford321, 2021-02-21:
I did not try remotely. I will try that. 

I do know older iPads aren’t good home hubs. That might be the problem.

karl-fischer, 2021-02-21:
I am using an iPad 8th generation as my Home Hub. This is the latest model and was released in 2020 :)

eford321, 2021-02-21:
Interesting findings.   I am running the home app on my phone via LTE and via Wifi on my Mac (phone remote/Mac local).  If I toggle the ESP-Switch from my phone, both the Mac and the phone show the change of icon. If I toggle from my Mac, the phone does not get updated. 
When I test my  window blind sketch it works properly in the same scenario. 

Here's the log of ESP-Switch. Client 1073698796 is my Mac, Client 1073698836 is my remote iPhone.
...
10:36:03.952 -> >>> Home Integration: homekit_is_paired 1
10:36:03.952 -> >>> HomeKit: Free heap: 31512
10:36:03.952 -> >>> HomeKit: Starting server
10:36:03.989 -> >>> HomeKit: storage init 0
10:36:03.989 -> >>> HomeKit: Using existing accessory ID: 3E:3A:0A:5E:12:9D
10:36:03.989 -> >>> HomeKit: find pairings
10:36:03.989 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup
10:36:03.989 -> >>> HomeKit: Configuring mDNS
10:36:03.989 -> >>> HomeKit: MDNS.begin: host-3E3A, IP: 192.168.1.56
10:36:03.989 -> >>> HomeKit: Init server over
10:36:03.989 -> Setting web server
10:36:04.156 -> Web site http://192.168.1.56
10:36:04.156 -> File system http://192.168.1.56/browse
10:36:04.191 -> Update http://192.168.1.56/update
10:36:13.493 -> pm open,type:2 0
10:36:29.043 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
10:36:29.043 -> >>> HomeKit: Got new client connection: local 192.168.1.56:5556, remote 192.168.1.13:50885
10:36:29.043 -> >>> HomeKit: [Client 1073698796] Got client content from the cache 0
10:36:29.112 -> >>> HomeKit: [Client 1073698796] new client accepted
10:36:29.112 -> >>> HomeKit: [Client 1073698796] Pair Verify Step 1/2
10:36:29.393 -> >>> HomeKit: Free heap: 22024
10:36:29.393 -> >>> HomeKit: [Client 1073698796] Pair Verify Step 2/2
10:36:29.393 -> >>> HomeKit: [Client 1073698796] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
10:36:30.187 -> >>> HomeKit: [Client 1073698796] Verification successful, secure session established
10:36:30.187 -> >>> HomeKit: Free heap: 22176
10:36:30.220 -> >>> HomeKit: [Client 1073698796] Get Accessories
10:36:30.220 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:36:30.259 -> >>> HomeKit: Free heap: 22224
10:36:30.259 -> switch_callback
10:36:30.259 -> set_switch:False
10:36:30.259 -> notify hap
10:36:30.259 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:36:30.259 -> >>> HomeKit: Free heap: 22224
10:36:43.160 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:36:43.160 -> >>> HomeKit: Free heap: 22416
10:36:43.160 -> switch_callback
10:36:43.257 -> set_switch:True
10:36:43.257 -> notify hap
10:36:44.311 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:36:44.311 -> >>> HomeKit: Free heap: 22416
10:36:44.311 -> switch_callback
10:36:44.345 -> set_switch:False
10:36:44.345 -> notify hap
10:36:48.224 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:36:48.224 -> >>> HomeKit: Free heap: 22416
10:36:48.224 -> switch_callback
10:36:48.224 -> set_switch:True
10:36:48.224 -> notify hap
10:36:49.262 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:36:49.262 -> >>> HomeKit: Free heap: 21920
10:36:49.262 -> switch_callback
10:36:49.262 -> set_switch:False
10:36:49.262 -> notify hap
10:36:52.805 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:36:52.839 -> >>> HomeKit: Free heap: 22416
10:36:52.839 -> switch_callback
10:36:52.839 -> set_switch:True
10:36:52.839 -> notify hap
10:36:54.057 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:36:54.057 -> >>> HomeKit: Free heap: 22416
10:36:54.057 -> switch_callback
10:36:54.057 -> set_switch:False
10:36:54.057 -> notify hap
10:37:09.130 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
10:37:09.130 -> >>> HomeKit: Got new client connection: local 192.168.1.56:5556, remote 192.168.1.28:49861
[... 62 lines cut ...]
10:45:09.073 -> >>> HomeKit: [Client 1073698836] Update Characteristics
10:45:09.073 -> >>> HomeKit: Free heap: 21928
10:45:09.073 -> switch_callback
10:45:09.073 -> set_switch:True
10:45:09.073 -> notify hap
10:45:12.511 -> >>> HomeKit: [Client 1073698836] Update Characteristics
10:45:12.511 -> >>> HomeKit: Free heap: 22096
10:45:12.511 -> switch_callback
10:45:12.511 -> set_switch:False
10:45:12.511 -> notify hap
10:46:49.801 -> >>> HomeKit: [Client 1073698836] Update Characteristics
10:46:49.801 -> >>> HomeKit: Free heap: 22096
10:46:49.801 -> switch_callback
10:46:49.801 -> set_switch:True
10:46:49.801 -> notify hap
10:46:55.880 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:46:55.880 -> >>> HomeKit: Free heap: 22096
10:46:55.880 -> switch_callback
10:46:55.880 -> set_switch:False
10:46:55.880 -> notify hap
10:49:56.419 -> >>> HomeKit: [Client 1073698796] Update Characteristics
10:49:56.419 -> >>> HomeKit: Free heap: 21904
10:49:56.419 -> switch_callback
10:49:56.419 -> set_switch:True
10:49:56.419 -> notify hap

Yurik72, 2021-02-21:
Interesting, I will be able to check next few days, but this gives an answer of disconnection, I suppose not a problem to fix

Yurik72, 2021-02-21:
I did’t find nothing special in the logs, mDNS works well

Yurik72, 2021-02-21:
One question which is not clear completely, if memory leak problems still appear on the latest commit

eford321, 2021-02-21:
Tried again after waiting more than 1 hour and it is now working normally.  strange. Appears to have been an apple issue.

Yurik72, 2021-02-21:
iOS 14.4 ?

eford321, 2021-02-21:
yes iOS 14.4 and Mac OS 11.2.1.

eford321, 2021-02-23:
> Yes, I saw your log, and my understanding you have changed socket options to more fast reproduce and test memory leaks 😊 Now I understand that is your standard behavior…. 

FYI, My blind sketch runs fine on a ESP32 I don't see any disconnects and I don't see "No Response" 

Here's a 1 hour log for comparison ...
ap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
14:11:31.737 -> >>> Home Integration: hap_services added chararacteristic  1: 0000006D-0000-1000-8000-0026BB765291 
14:11:31.737 -> >>> Home Integration: hap_services added chararacteristic  2: 0000007C-0000-1000-8000-0026BB765291 
14:11:31.737 -> >>> Home Integration: hap_services added chararacteristic  3: 00000072-0000-1000-8000-0026BB765291 
14:11:31.737 -> >>> Home Integration: homekit_is_paired 0
14:11:31.737 -> >>> HomeKit: Starting server
14:11:31.737 -> >>> HomeKit: Using existing accessory ID: FA:75:F6:63:03:43
14:11:31.737 -> >>> HomeKit: Configuring mDNS
14:11:53.547 -> >>> HomeKit: Got new client connection: 56
14:11:53.547 -> >>> HomeKit: [Client 56] Pair Setup 
14:11:53.547 -> >>> HomeKit: [Client 56] Pair Setup Step 1/3
14:11:53.547 -> >>> HomeKit: [Client 56] new pairing content
14:11:53.547 -> >>> HomeKit: started crypto
14:11:53.580 -> >>> HomeKit: Free heap: 170644
14:11:55.010 -> >>> HomeKit: Calculating public key len=384
14:11:55.769 -> >>> HomeKit: Calculating public key result 0
14:11:55.769 -> >>> HomeKit: [Client 56] send_tlv_response
14:11:55.769 -> >>> HomeKit: [Client 56] send_tlv_response done
14:11:59.309 -> >>> HomeKit: [Client 56] Pair Setup 
14:11:59.309 -> >>> HomeKit: [Client 56] Pair Setup Step 2/3
14:12:01.805 -> >>> HomeKit: [Client 56] Pair Setup 
14:12:01.805 -> >>> HomeKit: [Client 56] Pair Setup Step 3/3
14:12:02.083 -> >>> HomeKit: Added pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:12:02.116 -> >>> HomeKit: Configuring mDNS
14:12:02.116 -> >>> HomeKit: [Client 56] Successfully paired
14:12:02.192 -> >>> HomeKit: [Client 56] Closing client connection
14:12:03.413 -> >>> HomeKit: Got new client connection: 57
14:12:03.413 -> >>> HomeKit: [Client 57] Pair Verify Step 1/2
14:12:03.479 -> >>> HomeKit: [Client 57] Pair Verify Step 2/2
14:12:03.479 -> >>> HomeKit: [Client 57] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:12:03.514 -> >>> HomeKit: [Client 57] Verification successful, secure session established
14:12:03.514 -> >>> HomeKit: [Client 57] Get Accessories
14:12:03.689 -> >>> HomeKit: [Client 57] Update Characteristics
14:12:03.723 -> >>> HomeKit: [Client 57] Update Characteristics
14:12:03.723 -> >>> HomeKit: [Client 57] Update Characteristics
14:12:03.824 -> >>> HomeKit: [Client 57] Add Pairing
14:12:04.104 -> >>> HomeKit: Added pairing with 0F8927F8-0F84-44E7-8181-8714ED73BD51
14:12:05.336 -> >>> HomeKit: Got new client connection: 58
14:12:05.336 -> >>> HomeKit: [Client 58] Pair Verify Step 1/2
14:12:05.369 -> >>> HomeKit: Got new client connection: 59
14:12:05.369 -> >>> HomeKit: [Client 59] Pair Verify Step 1/2
14:12:05.435 -> >>> HomeKit: [Client 58] Pair Verify Step 2/2
14:12:05.435 -> >>> HomeKit: [Client 58] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:12:05.435 -> >>> HomeKit: [Client 58] Verification successful, secure session established
14:12:05.469 -> >>> HomeKit: Got new client connection: 60
14:12:05.469 -> >>> HomeKit: [Client 59] Pair Verify Step 2/2
14:12:05.469 -> >>> HomeKit: [Client 59] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:12:05.469 -> >>> HomeKit: [Client 59] Verification successful, secure session established
14:12:05.502 -> >>> HomeKit: Got new client connection: 61
14:12:05.502 -> >>> HomeKit: [Client 60] Pair Verify Step 1/2
14:12:05.535 -> >>> HomeKit: [Client 58] Get Accessories
14:12:05.535 -> >>> HomeKit: [Client 61] Pair Verify Step 1/2
14:12:05.601 -> >>> HomeKit: [Client 59] Get Accessories
14:12:05.601 -> >>> HomeKit: [Client 60] Pair Verify Step 2/2
14:12:05.601 -> >>> HomeKit: [Client 60] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:12:05.639 -> >>> HomeKit: [Client 60] Verification successful, secure session established
14:12:05.639 -> >>> HomeKit: [Client 61] Pair Verify Step 2/2
14:12:05.639 -> >>> HomeKit: [Client 61] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:12:05.672 -> >>> HomeKit: [Client 61] Verification successful, secure session established
14:12:05.672 -> >>> HomeKit: [Client 60] Get Accessories
14:12:05.672 -> >>> HomeKit: [Client 61] Get Accessories
14:12:05.706 -> >>> HomeKit: [Client 60] Update Characteristics
14:12:05.706 -> >>> HomeKit: [Client 60] Get Characteristics
14:12:05.807 -> >>> HomeKit: [Client 59] Update Characteristics
14:12:05.807 -> >>> HomeKit: [Client 58] Get Characteristics
[... 43 lines cut ...]
14:51:24.616 -> >>> HomeKit: [Client 56] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:51:24.616 -> >>> HomeKit: [Client 56] Verification successful, secure session established
14:51:24.616 -> >>> HomeKit: [Client 56] Get Accessories
14:51:24.616 -> >>> HomeKit: [Client 56] Update Characteristics
15:04:49.234 -> >>> HomeKit: [Client 56] Update Characteristics
15:10:43.333 -> >>> HomeKit: [Client 63] Closing client connection
15:10:47.475 -> >>> HomeKit: Got new client connection: 57
15:10:47.475 -> >>> HomeKit: [Client 57] Pair Verify Step 1/2
15:10:47.581 -> >>> HomeKit: [Client 57] Pair Verify Step 2/2
15:10:47.581 -> >>> HomeKit: [Client 57] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
[... cut ...]
15:11:28.096 -> >>> HomeKit: [Client 60] Verification successful, secure session established
15:11:28.096 -> >>> HomeKit: [Client 60] Get Accessories
15:11:28.149 -> >>> HomeKit: [Client 60] Update Characteristics
15:11:58.930 -> >>> HomeKit: [Client 56] Update Characteristics
15:11:59.219 -> >>> HomeKit: [Client 56] Update Characteristics
15:12:16.613 -> >>> HomeKit: [Client 57] Update Characteristics
15:12:16.682 -> >>> HomeKit: [Client 57] Get Characteristics
15:12:33.789 -> >>> HomeKit: [Client 60] Update Characteristics

karl-fischer, 2021-02-24:
So, do you think it is an ESP8266 issue? And that it works fine on the ESP32?

Did you see any "No Response" when running your iPhone on LTE?

eford321, 2021-02-24:
> So, do you think it is an ESP8266 issue? And that it works fine on the ESP32?
> 
> Did you see any "No Response" when running your iPhone on LTE?

Hard to tell so far. There is definitely an issue with the code on an ESP8266, but I don't see it with a smaller sketch like the switch.  Could also be the iPad. There are several reports on reddit saying the iPad's wifi sleep causes issues.  Do you have an appletv or HomePod you can try as a hub?

eford321, 2021-02-24:
ESP-32 went into no response after 2+ hours.  Debug isn't a detailed, but here's the log.  Worked until 17:19 in the attached log
[ESP32 log.txt.zip](https://github.com/Yurik72/ESPHap/files/6032613/ESP32.log.txt.zip)
...

karl-fischer, 2021-02-24:
Hmm, originally I was using an iPad Air, and it was several years old so I thought maybe that was the issue. So I bought a brand new iPad 8th gen, but still causes the issue (as discussed).

My preference is to use the iPad as the Home Hub, because it is fixed to the wall and designed to remain in the house when I sell, but yes, I do have an AppleTV 4k. I will set that up as the Home Hub and test.

I'll do that now...

karl-fischer, 2021-02-24:
Ok, I have set up my Apple TV 4k as a Home Hub, re-added the Accessory and tried to operate the accessory from my iPhone with WiFi turned off.

The accessory still shows as "No Response". When I click on the switch to turn it on/off, it does work, but it still shows as "No Response" the whole time.

I closed the Home app on my iPhone and opened it again, the Accessory is now showing as healthy. I'll sit and watch it for a few minutes.... 

Has been going for 30mins and seems to be working ok. The accessory on the iPhone has not said, "No Response".

Yurik72, 2021-02-24:
Hi for Esp32 I have to port several bug fixes which is found on esp8266, I will keep you informed.
Do you have any other Apple  devices an what is their behavior . For instance I’m comparing with Eve elgato and would say that behavior the same ( reboot router, phone to fly mode, etc)

eford321, 2021-02-24:
No worries on the ESP32. I am still focused on ESP8266 as I have 8 blinds running on them that I want to convert to ESP-HAP from MQTT/Homebridge.  I loaded my sketch on a ESP32 to see if there was any difference. 
Are you asking about my devices or @karl-fischer ?  I have 29 homekit devices in my network. All have been very reliable.

Yurik72, 2021-02-24:
Asked, both. Ok clear seams some bug still present, however I do not see ‘no response’ on esp32 in my home managed by Apple TV, they always healthy.

eford321, 2021-02-24:
The ESP32 definitely performs better than the 8266, but it still fails
after some time.  I am not able to compile with debug on the ESP32 for more
information. See #52

Do you have an ESP8266 to test with?

Yurik72, 2021-02-24:
Yes , sure I’m testing both ESP8266, and ESP32 in parallel, there are my initial target to have one sketch running similar on both platform, I will fix issue with compiling and deeply check no response, in the next few days, today and tomorrow not able due to other things, but target is to finish that and get it running well...

eford321, 2021-02-24:
Great. I appreciate your work on this code!!

It would be interesting to understand the reason for the "Disconnected!"
message on the ESP8266. I have not been able to understand that from
reading the code.

karl-fischer, 2021-02-24:
Hello Yurik72 & eford321 :)

I checked mine periodically over night and this morning when I was legitimately out of the house, the accessory was fine the whole time, it did not show as "No Response".

Perhaps my issue was the fact that I was using an iPad as a Home Hub. I have only found a few random comments around the place saying that iPads don't make good Hubs, but I have not read this consistently. It is really hard to find comment on this.

I wrote my code based on a different library (HomeKit-ESP8266 (by Mixiaoxiao)), unfortunately when I add the accessory back in to my Home app using that library and set my iPhone to LTE, it was showing 'No Response' and also it missed status updates from the accessory, so I'm going to start porting my code to ESPHap. Up until this point, I have just been using the examples provided directly. I have 2 ESP8266's (NodeMCU), which I will program, one is a switch to open the front gate, and the other is a switch to open the garage door, with a sensor to report when the door is open/closed.

I will likely need more than 4 clients able to connect to an Accessory, and it looks like this library is limited to 4 clients for the ESP8266; can this be increased? I understand there may be a memory limitation; I will disable the wifi manager and webserver, because I do not need them, will this help?

Karl :)

eford321, 2021-02-25:
Karl,

Good to hear your connections are stable. Hopefully Apple will fix the ipad hub issue. 

You can increase the number of sessions in arduino_homekit_server.cpp by changing these values. 
#define HOMEKIT_MAX_CLIENTS  4
#define CLIENT_CONTEXT_CACHE_SIZE 4

The Apple spec calls for supporting 8 clients, but it uses more heap for each session.  I regularly see 7 connections active in my network. 

Evan

karl-fischer, 2021-02-25:
Thanks Evan :)

Do you set yours to 8? Have you had any memory issues on the ESP8266?

Karl :)

eford321, 2021-02-25:
I have been using 8, but have been having problems with disconnects. Yurik72 has made great progress on memory leaks and is looking at the disconnect problem now. Hopefully he can figure it out soon. It’s not useable for me yet.

Yurik72, 2021-02-25:
Hi Karl , in addition to Eford123 answer, yes you may use 8 clients and memory is enough for them, my understanding that you do not have disconnection problem, this I will discover and my understanding is specific for his sketch.

karl-fischer, 2021-02-25:
Ok, thanks :)

Do you have an example with this ESPHap library, which has 2 accessories? For my garage door, I'm not using the "garage door" example, I was going to use a switch to open/close the door and a status for door open/closed status.

Karl :)

eford321, 2021-02-26:
>, this I will discover and my understanding is specific for his sketch.
@Yurik72 
I think I may have figured it out.  I believe it is my button logic causing enough delay that HAP keepalives fail when I have several Homekit clients connected to my ESP.  I removed button support and have been running reliably for more than 2.5 hours. I will work to optimize the button function.  I am out of digital IO so I have to work with the analog input on the Wemos D1 mini.

Also, I am not seeing any memory leak.

Yurik72, 2021-02-27:
> Do you have an example with this ESPHap library, which has 2 accessories? For my garage door, I'm not using the "garage door" example, I was going to use a switch to open/close the door and a status for door open/closed status.
> 
> Karl :)

Hi Karl,  
Yes we can do what we want and acceptable for apple:) So, I think you can get garage door as basis, at least contains Current Door State Charachteristic and add additional accessories, in your request Switch

This can be easilly done by calling: 
```c
hap_new_switch_service_as_accessory(const char* szname, hap_callback cb, void* context); 
```

function just before hap_init_homekit_server();

Check if this is OK, otherwise we can find other scenario , which you want

Yurik72, 2021-02-27:
> @Yurik72
> I think I may have figured it out. I believe it is my button logic causing enough delay that HAP keepalives fail when I have several Homekit clients connected to my ESP. I removed button [...]

Hi, good news, anyhow it's interesting to me therefore please send me the latest your sketch. As weel i can help you with button handling , i have several implementation of that including queue handling (like PC keyboard) which is not usual for any arduino library

eford321, 2021-02-27:
I think I have it working after several steps. 

First I moved some pins around to recover a usable digital pin for my button. That eliminated analog. Unfortunately, I was still getting disconnects. Note: It doesn't matter if the button is used or not.

Next, I tried with stepper motor code and library removed with no change. It still failed, so it's not the motor.

Then, I commented out my button function and the function call and it worked for 18 hours with no disconnects.  Odd thing is, if I only commented out the function call, it would still fail. Made me think the code was causing the compiler to do something odd.

Finally, I think I might have figured it out.  I had several timing variables using the millis() function and had them defined as long. I changed them to unsigned long and it has been running for >60 minutes with the motor and button working as expected. 

Also good news, heap is very solid around 20000.

I'll monitor and let you know.

Yurik72, 2021-02-28:
Ok, will be interesting. Any how I’d like to see your sketch, could be interesting.

eford321, 2021-02-28:
Sketch ran for 24 hours on 4 separate window blinds without disconnects!  Also heap is still above 20000.

Unfortunately, I found another problem. I will open a separate issue for that after I gather more data.

Here is my working sketch.  Thanks again for your help!
[WindowCover_ESPHap.ino.zip](https://github.com/Yurik72/ESPHap/files/6057544/WindowCover_ESPHap.ino.zip)

Yurik72, 2021-02-28:
Ok, I will check new issue and your sketch