Yurik72/ESPHap #54: Device not compatible

EIBHomeControl, 2021-02-18:
I've tested the lastest version. When I try to add the new device to Homekit it won't show up but gives me an error. Device not compatible.

Commit c8242dc

Log:
5
>>> HomeKit: WiFiServer receives a new client (current 0, max 4)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.191:59027
>>> HomeKit: [Client 1073705188] Got client content from the cache 0
>>> HomeKit: [Client 1073705188] new client accepted
>>> HomeKit: [Client 1073705188] Pair Setup Step 1/3
>>> HomeKit: [Client 1073705188] Pair Setup Step 2/3
>>> HomeKit: [Client 1073705188] Computing SRP shared secret
>>> HomeKit: [Client 1073705188] Verifying peer's proof
>>> HomeKit: [Client 1073705188] Generating own proof
>>> HomeKit: [Client 1073705188] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 09913711-047C-421A-ABCD-6E69DA74037B
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 1073705188] Successfully paired
>>> HomeKit: [Client 1073705188] Disconnected!
>>> HomeKit: [Client 1073705188] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: WiFiServer receives a new client (current 0, max 4)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.191:59029
>>> HomeKit: [Client 1073706532] Got client content from the cache 0
>>> HomeKit: [Client 1073706532] new client accepted
>>> HomeKit: [Client 1073706532] Pair Verify Step 1/2
>>> HomeKit: Free heap: 21224
>>> HomeKit: [Client 1073706532] Pair Verify Step 2/2
>>> HomeKit: [Client 1073706532] Found pairing with 09913711-047C-421A-ABCD-6E69DA74037B
>>> HomeKit: [Client 1073706532] Verification successful, secure session established
>>> HomeKit: Free heap: 21160
>>> HomeKit: [Client 1073706532] Get Accessories
>>> HomeKit: [Client 1073706532] Disconnected!
>>> HomeKit: [Client 1073706532] Closing client connection
>>> HomeKit: [Client 0] client content free

Seems to me that the paring is successful, but the device is not in Homekit accepted.

v2 Higer bandwidth is set.

If you need more info, pls let me know.

Yurik72, 2021-02-18:
First of all please always mention wich device you are using and which sketch...
I have  met with such situation and 99.9% know the reason
This always happens when during the pairing ESP send wrong characheristic values, for instance
You defined Temperathure charachteristic, it has by default  minimum value=0.0 maximum =100.0, during the pairing if you send to apple value wich out of this range you will get this message. If it happen after pairing, Apple just ignore, but pairing is continued

You can use DEFINE macros to avoid this or change minimum or maximum values
define macros used like this  

HAP_NOTIFY_CHANGES_WITHCONSTRAIN(float, ch_temp, temp_value, 0)  
where ch_temp ->Tempherature charachteristic
temp_value  -> themperature

Give me a feedback pls

EIBHomeControl, 2021-02-18:
Thanks for the fast answer. You're right. If sent a wrong value for den garage state. It's working now.

How can set max. 8 connections? At the moment it says only 4. My sketch is based on the garage door example.

Yurik72, 2021-02-18:
> How can set max. 8 connections? At the moment it says only 4.

There is under testing. Currently ESP8266 tested more less stable with 6-8 client
you can change 
#define HOMEKIT_MAX_CLIENTS  4
#define CLIENT_CONTEXT_CACHE_SIZE 4

in arduino_homekit_server.cpp

EIBHomeControl, 2021-02-18:
> > How can set max. 8 connections? At the moment it says only 4.
> 
> There is under testing. Currently ESP8266 tested more less stable with 6-8 client
> you can change
> #define HOMEKIT_MAX_CLIENTS 4
> #define CLIENT_CONTEXT_CACHE_SIZE 4
> 
> in arduino_homekit_server.cpp

thx.

closed 2021-02-18 by EIBHomeControl