Yurik72/ESPHap #55: QUESTION: Best way to convert html files to PROGMEM string?

Ceda, 2021-02-19:
Hi,
i am looking for tool to convert html files to PROGMEM string. Which are you use for create index_html.h file?
I found https://github.com/SukkoPera/Webbino/blob/master/tools/html2h.py which prepare compressed content.

Thx for info

Yurik72, 2021-02-19:
I have used https://www.buildmystring.com/build.php

 

But this is not work since month ago, I suppose you can easily ask google there are a lot of online

Try https://tomeko.net/online_tools/cpp_text_escape.php?lang=en   , seems it’s works correct

Ceda, 2021-02-19:
Looks nice https://dinobusca.com/stringbuilder/ and found repo BuildMyString.com https://github.com/JeffreyBane/BuildMyString

closed 2021-02-19 by Ceda