Yurik72/ESPHap #56: Only 7 services?

judepereira, 2021-02-21:
In homeintegration.c, the following is defined:
`#define MAX_HAP_SERVICES 7 `

However, at the top of the file, `MAX_SERVICES` is defined as 20.

What’s the difference between these two?

Yurik72, 2021-02-21:
I will check and provide answers in next few days , currently not have access to my PC, as well please describe your wondering an targets

judepereira, 2021-02-21:
My use case is building a hub to which other ESPs will connect to, via ESP NOW. I’ve got it all working (battery powered motion sensors, temp/humidity sensors) to my main hub, but wondering whether I’ll need to use more than one hub to support 10 services (each sensor is one service).

Cheers,
Jude

Yurik72, 2021-02-27:
Ok, 
MAX_SERVICES is outdated declaration  I will  remove thjem from the code
MAX_HAP_SERVICES  is actual and used, 7 is not limited,  you can change and use 10 as well, nex version i will make it more configurable or dynamic

closed 2021-02-27 by Yurik72

judepereira, 2021-02-27:
So on the ESP8266, running over 8 accessories with 4 services in each accessory leads to a mem issue. I'm going to test more than 20 accessories on the ESP32 (it's on its way).

Yurik72, 2021-02-27:
Yes, could be memory issue on ESP8266

Any way you can use latest commits, there are a lot of memory optimization and could be better

On ESP32 I need one commit more after full test within 8266 there are some memory leaks, I know how to fix and will be done in the next few days

However for initial test should be OK, there is impact on the long usage and reliabilty

judepereira, 2021-02-27:
Got it. I opened #62 btw. I couldn't find a contribution notes anywhere, so just opened a PR :)