Yurik72/ESPHap #61: Question on notifying HAP

eford321, 2021-02-25:
I see 2 different functions that appear to notify HAP.: (homekit_characteristic_notify & HAP_NOTIFY_CHANGES).

Is there a preferred use for each of these? Thanks.

Yurik72, 2021-02-27:
homekit_characteristic_notify   : is a function which allows to notify HAP at any  time with any values...
HAP_NOTIFY_CHANGES(name,home_characteristic,val,tollerance) : is a macros,  which finally calls the same function, but macros is able to check (compare ) new and old value with Tollerance(latest param) and in case of changing will call notify.  For instance for the Thermometer (in my opinion ) not neccessary to call every time notification and Tollerance can be for instance 0.5

HAP_NOTIFY_CHANGES_WITHCONSTRAIN  : In addition to previous macros allows to check min and max value for charachtertistic and prevents to send value which is out of range

So, any notification can be used, depends on your sketch

closed 2021-02-27 by Yurik72

eford321, 2021-02-27:
Great. Thanks for the explanation.