Yurik72/ESPHap #64: ESPHap unable to process client connections after restarting HomeBridge instance in network

eford321, 2021-02-28:
Running v.1.0.7 on ESP8266 with multiple example sketches. Devices function properly until a [Homebridge](https://github.com/homebridge/homebridge) is restarted. Afterward all client connections from any device are accepted and then immediately disconnected.  All ESPHap devices go unresponsive until the are rebooted.  This does not effect other native Homekit devices in the network.    I am attempting to log events from the homebridge log, but the log is huge.

Here is the ESPHap device log running sketch EspHapLed8266.  Device is paired and functional. 
Notes: Homebridge is 192.168.1.213.  Homebridge restarted after 14:11:21.444 timestamp. Message from homebridge processed at 14:11:51.555 breaks ESPHap. 
...
14:10:58.887 -> IP address: 
14:10:58.887 -> 192.168.1.23
14:10:58.887 -> Free heap: 34056
14:10:58.887 -> init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
14:10:58.923 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
14:10:58.923 -> >>> Home Integration: hap_services added chararacteristic  1: 25 
14:10:58.923 -> >>> Home Integration: homekit_is_paired 1
14:10:58.923 -> >>> HomeKit: Free heap: 32072
14:10:58.923 -> >>> server_new: WiFiServer begin at port: 5556
14:10:58.923 -> 
14:10:58.923 -> >>> HomeKit: Starting server
14:10:58.923 -> >>> HomeKit: storage init 0
14:10:58.923 -> >>> HomeKit: Using existing accessory ID: B1:E8:4D:32:98:71
14:10:58.923 -> >>> HomeKit: find pairings
14:10:58.957 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup
14:10:58.957 -> >>> HomeKit: Configuring mDNS
14:10:58.957 -> >>> HomeKit: MDNS.begin: ES-B1E8, IP: 192.168.1.23
14:10:58.957 -> >>> homekit_mdns_init: Accessory Setup ID = YK72
14:10:58.957 -> >>> HomeKit: Init server over
14:11:06.422 -> pm open,type:2 0
14:11:12.872 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
14:11:12.906 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:58362
14:11:12.906 -> >>> HomeKit: [Client 1073699420] Got client content from the cache 0
14:11:12.906 -> >>> HomeKit: [Client 1073699420] new client accepted
14:11:12.906 -> >>> homekit_client_process: [Client 1073699420] Reading 158 bytes
14:11:12.906 -> >>> homekit_server_on_message_complete: http_parser message_complete
14:11:12.940 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
14:11:12.940 -> >>> homekit_server_on_pair_verify: Free heap: 24320
14:11:12.940 -> >>> tlv_debug: Got following TLV values:
14:11:12.940 -> >>> tlv_debug: Type 6 value (1 bytes): \x01
14:11:12.940 -> >>> tlv_debug: Type 3 value (32 bytes): \x8E\x16\x11a\xB7\xD9\xD5\x1E3\xE5<\x98\x15+\x88Z\xA4\x00~\xEEU\x86\xD3E\xC7\xA2\xB5\xF8\xDC\x9Ad~
14:11:12.940 -> >>> HomeKit: [Client 1073699420] Pair Verify Step 1/2
14:11:12.974 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Importing device Curve25519 public key
14:11:12.974 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Generating accessory Curve25519 key
14:11:13.085 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Exporting accessory Curve25519 public key
14:11:13.120 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Generating Curve25519 shared secret
14:11:13.230 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Generating signature
14:11:13.298 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Generating proof
14:11:13.335 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Encrypting response
14:11:13.335 -> >>> send_tlv_response: [Client 1073699420] Sending TLV response
14:11:13.335 -> >>> tlv_debug: Got following TLV values:
14:11:13.335 -> >>> tlv_debug: Type 6 value (1 bytes): \x02
14:11:13.335 -> >>> tlv_debug: Type 3 value (32 bytes): \xF3\x12O\x83\x1D\x08\x920\xAF\xB1#O\x84\xB2\x16\x99\xA2\x98#\x18\x1E\xE86'D\xBB\x0A\xDA\xDA\x0B\xB5k
14:11:13.369 -> >>> tlv_debug: Type 5 value (101 bytes): \xEA\x183+\xC2g\x83\x15\x98\x81>\x02E\x02z;{\x0A\xEAq\x18\x10O\x03\xFE\x0A\xCD\x83\xEDGeC\x085W\xB4\xD1\xE9\xBE\x19\x98W\x95\xB0\x9C-\xB1\xD5\xF4\xA3\x1F\xF0\xD0\x9E\xAE\xCB\xA6@\xBB\x87\x1Bf\xA7\xD2\x89\x1A\xE0=@\xFA\xBE\xBCw8B\x12uZ\x10ys*\xE7\xFA#\xE6\x1Cj\x95R 0+ppz\x91\x1C V
14:11:13.369 -> >>> client_send: [Client 1073699420] send data size=244, encrypted=false
14:11:13.406 -> >>> client_send: [Client 1073699420] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 \xF3\x12O\x83\x1D\x08\x920\xAF\xB1#O\x84\xB2\x16\x99\xA2\x98#\x18\x1E\xE86'D\xBB\x0A\xDA\xDA\x0B\xB5k\x05e\xEA\x183+\xC2g\x83\x15\x98\x81>\x02E\x02z;{\x0A\xEAq\x18\x10O\x03\xFE\x0A\xCD\x83\xEDGeC\x085W\xB4\xD1\xE9\xBE\x19\x98W\x95\xB0\x9C-\xB1\xD5\xF4\xA3\x1F\xF0\xD0\x9E\xAE\xCB\xA6@\xBB\x87\x1Bf\xA7\xD2\x89\x1A\xE0=@\xFA\xBE\xBCw8B\x12uZ\x10ys*\xE7\xFA#\xE6\x1Cj\x95R 0+ppz\x91\x1C V
14:11:13.444 -> >>> HomeKit: Free heap: 24080
14:11:13.444 -> >>> homekit_client_process: [Client 1073699420] Reading 247 bytes
14:11:13.444 -> >>> homekit_server_on_message_complete: http_parser message_complete
14:11:13.511 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
14:11:13.511 -> >>> homekit_server_on_pair_verify: Free heap: 23992
14:11:13.511 -> >>> tlv_debug: Got following TLV values:
14:11:13.511 -> >>> tlv_debug: Type 5 value (120 bytes): $Rn\x99\xFA\x8D\x98\x8A\xF4o\x9Dc\xA3\x97J\x98\xB8\x09\xE6v\x1F\x90\xDA1U<.F\xBF&\x14\x14\xD4<\x91\x80\x80e\\\xDA1\xBA\xD7+\x0F\xFA\xA4\xC9\xED\x98\x8Ap\x1B\x9E\x89\xD7\xBB\xFBi+\x08N\x0D\x0F\xD5?\xAA[\xB4-\xA08L\xB5\x05\x0CM\x9B\xC0\xD5\xEC\x83\x83\x8A\xA4\xD9I\x08\xE9\xE4Q\xEF\x8CL\xB1\x89)R\x9F\xE9\x19O\xF7\x91g\xEE\xE5="&X\xBD\x16[\xF3\xAFk\x8D\xFF
14:11:13.511 -> >>> tlv_debug: Type 6 value (1 bytes): \x03
14:11:13.511 -> >>> HomeKit: [Client 1073699420] Pair Verify Step 2/2
14:11:13.511 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Decrypting payload
14:11:13.511 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Searching pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:11:13.554 -> >>> HomeKit: [Client 1073699420] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
14:11:13.554 -> >>> homekit_server_on_pair_verify: [Client 1073699420] Verifying device signature
14:11:14.322 -> >>> send_tlv_response: [Client 1073699420] Sending TLV response
14:11:14.322 -> >>> tlv_debug: Got following TLV values:
14:11:14.322 -> >>> tlv_debug: Type 6 value (1 bytes): \x04
14:11:14.322 -> >>> client_send: [Client 1073699420] send data size=105, encrypted=false
14:11:14.356 -> >>> client_send: [Client 1073699420] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 3\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x04
14:11:14.356 -> >>> HomeKit: [Client 1073699420] Verification successful, secure session established
[... 93 lines cut ...]
14:12:21.322 -> >>> homekit_client_process: [Client 1073699420] Reading 188 bytes
14:12:21.322 -> >>> homekit_client_process: [Client 1073699420] Decrypted 170 bytes, available 0
14:12:21.322 -> Decrypted data (170 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: ES-B1E8._hap._tcp.local\x0D\x0AContent-Length: 50\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":0}]}"
14:12:31.793 -> >>> HomeKit: [Client 1073699420] Disconnected!
14:12:36.805 -> >>> HomeKit: [Client 1073699420] Closing client connection
14:12:36.805 -> >>> homekit_server_close_client: [Client 1073699420] The sockect is stoped
14:12:36.805 -> >>> HomeKit: [Client 0] client content free
14:12:58.156 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
14:12:58.156 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:58495
14:12:58.231 -> >>> HomeKit: [Client 1073700644] Got client content from the cache 0
14:12:58.231 -> >>> HomeKit: [Client 1073700644] new client accepted
14:12:58.231 -> >>> homekit_client_process: [Client 1073700644] Reading 158 bytes
14:13:59.555 -> >>> HomeKit: [Client 1073700644] Disconnected!
14:13:59.555 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
14:13:59.555 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:58568
14:13:59.555 -> >>> HomeKit: [Client 1073699356] Got client content from the cache 2
14:13:59.555 -> >>> HomeKit: [Client 1073699356] new client accepted
[... cut ...]
14:14:03.191 -> >>> HomeKit: [Client 1073700644] Closing client connection
14:14:03.191 -> >>> homekit_server_close_client: [Client 1073700644] The sockect is stoped
14:14:03.191 -> >>> HomeKit: [Client 0] client content free
14:14:58.197 -> >>> HomeKit: [Client 1073699356] Disconnected!
14:15:03.198 -> >>> HomeKit: [Client 1073699356] Closing client connection
14:15:03.198 -> >>> homekit_server_close_client: [Client 1073699356] The sockect is stoped
14:15:03.198 -> >>> HomeKit: [Client 0] client content free

Yurik72, 2021-02-28:
Hi, and thanks for deep testing, one more clarification question,

You mentioned about version v.1.0.7 ? However you know there are a lot of significant changes on the latest commits.

Any how I will try to reproduce as well

eford321, 2021-02-28:
Sorry, I meant to check which commit.  The log was from c822ae8.  I just recompiled with 1f3a5cb
 and the behavior is the same.

Yurik72, 2021-02-28:
Ok, I will check

EIBHomeControl, 2021-03-01:
Same behavior here. Tested with the lastet version.

Yurik72, 2021-03-01:
I'm trust you however can't reproduce and need your explanation
1. I can't understnad at all why is <https://github.com/homebridge/homebridge> impacted on esp8266. You have been restarted this one, means  it lost and establish connection again  to your Apple Home (apple hub or apple tv) but, why is related to connection between esp and apple tv. Because i did't use a HomeBridge, maybe short explanation will be helpfull

2. What i see in the logs
14:13:59.555 -> >>> homekit_client_process: [Client 1073699356] Reading 158 bytes
14:14:03.191 -> >>> HomeKit: [Client 1073700644] Closing client connection

ESP trying to read 158 bytes from the socket, and ( due to not other records) after 3.5 secs it was unsuccessfull WiFiClient.connected()==false.  Meand TCP state !=established.  I can imagine that  ESP get a packet from remote host
WiFiClient.available() returns 158 bytes. 
Next line is 
WiFiClient.read() 
means ESP TCP socket/buffer must be readed and ACK must be sent to remote host ,

What happens in the reality i don't know without network sniffers....

I found very similar issues, but they are not the same, howere advice is to put yield(), to break loop for background processes,
I did this, therefore you can try latest commit.
As weel any ideas appreciated

EIBHomeControl, 2021-03-01:
here another log

it seems that after the restart from the homebridge a secure reconnect is not possible. Connections are coming in 

>>> HomeKit: WiFiServer receives a new client (current 2, max 8)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.188:59997
>>> HomeKit: [Client 1073704148] Got client content from the cache 1
>>> HomeKit: [Client 1073704148] new client accepted
>>> HomeKit: [Client 1073704148] Disconnected!

and after some time it gets disconnected. Normally it should start with Pair Verify.

If you need more logs or sketch pls let me know.

Init states
Target door state: 1 - closed
Current door state: 1 - closed
set obstruction to 0
hap_callback
Pin Open open
Pin Close closed
set current state to 1
set target state to 1
Target door state: 1 - closed
Current door state: 1 - closed
Setting web server
Web site http://192.168.2.145
File system http://192.168.2.145/browse
Update http://192.168.2.145/update
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.191:63678
>>> HomeKit: [Client 1073704780] Got client content from the cache 0
>>> HomeKit: [Client 1073704780] new client accepted
>>> HomeKit: [Client 1073704780] Pair Verify Step 1/2
>>> HomeKit: Free heap: 19064
>>> HomeKit: [Client 1073704780] Pair Verify Step 2/2
>>> HomeKit: [Client 1073704780] Found pairing with 09913711-047C-421A-ABCD-6E69DA74037B
>>> HomeKit: [Client 1073704780] Verification successful, secure session established
>>> HomeKit: Free heap: 19176
>>> HomeKit: [Client 1073704780] Get Accessories
>>> HomeKit: [Client 1073704780] Update Characteristics
>>> HomeKit: Free heap: 18984
>>> HomeKit: Free heap: 18968
>>> HomeKit: WiFiServer receives a new client (current 1, max 8)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.188:59880
>>> HomeKit: [Client 1073704148] Got client content from the cache 1
>>> HomeKit: [Client 1073704148] new client accepted
>>> HomeKit: [Client 1073704148] Pair Verify Step 1/2
>>> HomeKit: Free heap: 18248
>>> HomeKit: [Client 1073704148] Pair Verify Step 2/2
>>> HomeKit: [Client 1073704148] Found pairing with 09913711-047C-421A-ABCD-6E69DA74037B
>>> HomeKit: [Client 1073704148] Verification successful, secure session established
>>> HomeKit: Free heap: 18360
>>> HomeKit: [Client 1073704148] Get Accessories
>>> HomeKit: [Client 1073704148] Get Characteristics
>>> HomeKit: [Client 1073704148] Update Characteristics
>>> HomeKit: Free heap: 18360
>>> HomeKit: Free heap: 18344
>>> HomeKit: Free heap: 18328
>>> HomeKit: WiFiServer receives a new client (current 2, max 8)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.132:55278
>>> HomeKit: [Client 1073706484] Got client content from the cache 2
>>> HomeKit: [Client 1073706484] new client accepted
>>> HomeKit: [Client 1073706484] Pair Verify Step 1/2
>>> HomeKit: Free heap: 18240
>>> HomeKit: [Client 1073706484] Pair Verify Step 2/2
>>> HomeKit: [Client 1073706484] Found pairing with 2F343092-864B-4032-8956-F64F9BF836CF
>>> HomeKit: [Client 1073706484] Verification successful, secure session established
>>> HomeKit: Free heap: 17936
>>> HomeKit: [Client 1073706484] Get Accessories
[... 63 lines cut ...]
>>> HomeKit: [Client 1073704148] new client accepted
>>> HomeKit: [Client 1073707260] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: WiFiServer receives a new client (current 4, max 8)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.20:44086
>>> HomeKit: [Client 1073696740] Got client content from the cache 3
>>> HomeKit: [Client 1073696740] new client accepted
>>> HomeKit: [Client 1073696740] Disconnected!
>>> HomeKit: [Client 1073696740] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: [Client 1073707484] Disconnected!
>>> HomeKit: [Client 1073707484] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: [Client 1073704148] Disconnected!
>>> HomeKit: [Client 1073704148] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: WiFiServer receives a new client (current 2, max 8)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.188:59997
>>> HomeKit: [Client 1073704148] Got client content from the cache 1
>>> HomeKit: [Client 1073704148] new client accepted
>>> HomeKit: [Client 1073704148] Disconnected!
>>> HomeKit: WiFiServer receives a new client (current 3, max 8)
>>> HomeKit: Got new client connection: local 192.168.2.145:5556, remote 192.168.2.188:60011
>>> HomeKit: [Client 1073696740] Got client content from the cache 3
>>> HomeKit: [Client 1073696740] new client accepted

eford321, 2021-03-01:
Same result with 4ff7540.

I am trying to gather more information from my homebridge. I have to be careful not to mess up my configuration and lose 33 devices running on the bridge.

> I can't understnad at all why is https://github.com/homebridge/homebridge impacted on esp8266. You have been restarted this one, means it lost and establish connection again to your Apple Home [...]

My guess is the bridge connects to ESPHap as a client to allow communications from bridged devices. 

I wonder why ESPHap is losing connections with all clients after homebridge tries to reconnect. 

Here is the reconnection:
> 09:42:32.006 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
09:42:32.006 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:43492
09:42:32.006 -> >>> HomeKit: [Client 1073701172] Got client content from the cache 2
09:42:32.006 -> >>> HomeKit: [Client 1073701172] new client accepted
09:42:32.121 -> >>> homekit_client_process: [Client 1073701172] Reading 142 bytes
09:42:32.121 -> >>> homekit_server_on_message_complete: http_parser message_complete

>09:42:32.121 -> >>> **homekit_server_on_message_complete: Unknown endpoint**

>09:42:32.121 -> >>> client_send: [Client 1073701172] send data size=26, encrypted=false
09:42:32.121 -> >>> client_send: [Client 1073701172] Sending payload: HTTP/1.1 404 Not Found\x0D\x0A\x0D\x0A
09:42:46.162 -> >>> HomeKit: [Client 1073701172] Disconnected!
09:42:51.198 -> >>> HomeKit: [Client 1073701172] Closing client connection
09:42:51.198 -> >>> homekit_server_close_client: [Client 1073701172] The sockect is stoped
09:42:51.198 -> >>> HomeKit: [Client 0] client content free
09:43:02.814 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
09:43:02.814 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:43774
09:43:02.814 -> >>> HomeKit: [Client 1073700708] Got client content from the cache 2
09:43:02.934 -> >>> HomeKit: [Client 1073700708] new client accepted
09:43:02.934 -> >>> homekit_client_process: [Client 1073700708] Reading 142 bytes
09:43:12.851 -> >>> HomeKit: [Client 1073700708] Disconnected!

Then I try to turn the LED off from the home app on my Mac, and it fails with "no response"
>09:43:14.793 -> >>> homekit_client_process: [Client 1073699812] Reading 188 bytes
09:43:14.793 -> >>> homekit_client_process: [Client 1073699812] Decrypted 170 bytes, available 0
09:43:14.793 -> Decrypted data (170 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: ES-B1E8._hap._tcp.local\x0D\x0AContent-Length: 50\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":0}]}"
09:43:17.868 -> >>> HomeKit: [Client 1073700708] Closing client connection
09:43:17.868 -> >>> homekit_server_close_client: [Client 1073700708] The sockect is stoped
09:43:17.868 -> >>> HomeKit: [Client 0] client content free
09:43:24.830 -> >>> HomeKit: [Client 1073699812] Disconnected!
09:43:29.831 -> >>> HomeKit: [Client 1073699812] Closing client connection
09:43:29.831 -> >>> homekit_server_close_client: [Client 1073699812] The sockect is stoped
09:43:29.831 -> >>> HomeKit: [Client 0] client content free

Yurik72, 2021-03-01:
Thanks, but this is definitely wrong assumption, secure connection is response of this library, but socket is not simple reads and packet lost, I will continue investigation

Yurik72, 2021-03-01:
About bridge connects to esphap, I think you are not right, there are not any communication supposed between bridge and esp, bridge acts as end device(s) the same as esp, and esp interacts only with Apple devices, therefore I want to understand an impact, you can easily see IP address of the device which connects to esp.
Any other ideas ?

eford321, 2021-03-01:
Homebridge definitely connects as a client to my ESPHap devices.  In my
logs, anytime you see the static ip 192.168.1.213 is homebridge.

Not sure why homebridge needs a client connection to ESPHap, but there are
many plugins available for homebridge that act beyond the typical homekit
bridge service.

It seems that some value in restarted homebridge doesn't match the cached
value in ESPHap?

Why does this affect all clients sessions?

eford321, 2021-03-01:
More info.  It appears homebridge does some type of status requests of each HAP device discovered via bonjour.  Homebridge does keep active sessions with the homekit hub and alternate hubs. It does not keep sessions connected to any other homekit devices.

When it statuses ESPHap, 
>09:42:32.121 -> >>> homekit_client_process: [Client 1073701172] Reading 142 bytes
09:42:32.121 -> >>> homekit_server_on_message_complete: http_parser message_complete
09:42:32.121 -> >>> homekit_server_on_message_complete: Unknown endpoint

ESPHap returns 404 and disconnects
>09:42:32.121 -> >>> client_send: [Client 1073701172] send data size=26, encrypted=false
09:42:32.121 -> >>> client_send: [Client 1073701172] Sending payload: HTTP/1.1 404 Not Found\x0D\x0A\x0D\x0A
09:42:46.162 -> >>> HomeKit: [Client 1073701172] Disconnected!

Something breaks and all reconnects from homebridge or other clients fail and aren't parsed?
>09:43:02.814 -> >>> HomeKit: WiFiServer receives a new client (current 2, max 8)
09:43:02.814 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:43774
09:43:02.814 -> >>> HomeKit: [Client 1073700708] Got client content from the cache 2
09:43:02.934 -> >>> HomeKit: [Client 1073700708] new client accepted
09:43:02.934 -> >>> homekit_client_process: [Client 1073700708] Reading 142 bytes
09:43:12.851 -> >>> HomeKit: [Client 1073700708] Disconnected!
09:43:14.793 -> >>> homekit_client_process: [Client 1073699812] Reading 188 bytes
09:43:14.793 -> >>> homekit_client_process: [Client 1073699812] Decrypted 170 bytes, available 0
09:43:14.793 -> Decrypted data (170 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: ES-B1E8._hap._tcp.local\x0D\x0AContent-Length: 50\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":0}]}"
09:43:17.868 -> >>> HomeKit: [Client 1073700708] Closing client connection
09:43:17.868 -> >>> homekit_server_close_client: [Client 1073700708] The sockect is stoped
09:43:17.868 -> >>> HomeKit: [Client 0] client content free
09:43:24.830 -> >>> HomeKit: [Client 1073699812] Disconnected!
09:43:29.831 -> >>> HomeKit: [Client 1073699812] Closing client connection
09:43:29.831 -> >>> homekit_server_close_client: [Client 1073699812] The sockect is stoped
09:43:29.831 -> >>> HomeKit: [Client 0] client content free

eford321, 2021-03-01:
Is there a way to capture/debug the http request coming from homebridge?

Yurik72, 2021-03-01:
Yes , you can setup any network sniffer, for instance I preferred WireShark , https://www.wireshark.org/

Yurik72, 2021-03-01:
Aaa , forgot because not familiar with the bgidge, bridge can’t connect to esp, because session is secure and open/ closed keys are known only for Apple and Esp...

eford321, 2021-03-01:
From looking at Wireshark captures, Homebridge does not connect to ESP. It appears homebridge sends an MDNS query when it is restarted. ESPHap doesn't respond to this query, does say the following:
>15:20:20.365 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:37292
15:20:20.402 -> >>> HomeKit: [Client 1073699420] Got client content from the cache 0
15:20:20.402 -> >>> HomeKit: [Client 1073699420] new client accepted

ESP doesn't respond, but my other homekit devices do. Here is the packet:
[image]

eford321, 2021-03-01:
Captured again and do so see ESP responding.  This is the only communications I see between ESP and homebridge.....
[image]

Yurik72, 2021-03-02:
Hi All, 
I hope i see a light at the end of the tunnel :)
Thanks fior your help and logs. Now i want to explain my understanding what is happening.

When your home bridge is restarted it always send http request to the ESP. HTTP request is wrong (from Apple communication point of view). But maybe I'm wrong and ESP should reply some correct answers
Currently reply is HTPP 404....

So, what i did. Due to I do not have proper instance of HomeBridge with all plugins, I'm trying to simulate this from normal browser by putting URL http://192.168.88.14:5556/blablabla, where 192.168.88.14 is IP address of my ESP (/blablabla  is wrong query). And defintelly by killing ESP by such request my ESP some times is not available (No response) for normal Apple devices.. Hovewer is not always....

What i did. In case of wrong request, now I still send HTTP404 response , than immediatelly disconnect. As well if new request comming at this moment from the same IP address, i simple do not accept them

Simulating this from normal browser shows much improves and my ESP is always reliable

So, you can try latest commits.

If situation will be reproduced again, you can help me by explanation what kind of request is send by HomeBridge and probably what kind of response should be instead of 404, probably normal Apple devices sending something other

As well as the next attempts, we can try to ignore any request from HomeBridge if previous doesn't help
It can be easilly done by checking IP address, i have prepared and commented a few lines for that
See arduino_homekit_server.cpp line  3622  
If this will help I will try to find proper solution

Yurik72, 2021-03-02:
Small additions to previous. Official docs says about allowed command and responses
[image]

eford321, 2021-03-02:
I will test and see what happens. I can also let you know how other devices are responding to the same MDNS query.

eford321, 2021-03-02:
Still failing with 1583e16.  Here is the trace showing the response from ESP and also a Leviton homekit switch for comparison. Note the Leviton also sends IPv6, since it can't be configured to only use IPv4. 
[image]

Detailed ESP response:
[image]

Detailed Leviton response:
[image]

Yurik72, 2021-03-02:
Hi seems is not related to mDNS, because both devices properly respond, but definitely  I will check differences (for instance cache flush)

Any now  I would like to finish within my assumptions… 

Could you put your HomeBridge into the black list. I want to be 100%  sure that is impacted or not. And then investigation to mDNS

 

 <https://user-images.githubusercontent.com/5996455/109663420-1379c880-7b3a-11eb-8621-a1d694784f15.png> 

 <https://user-images.githubusercontent.com/5996455/109663777-779c8c80-7b3a-11eb-9fae-2236858e244d.png> 

 <https://user-images.githubusercontent.com/5996455/109663820-82efb800-7b3a-11eb-98e1-73202ebcedd9.png>

eford321, 2021-03-02:
black list? not sure what you mean.

Any idea why ESP goes unresponsive for all clients after the homebridge QM query?

EIBHomeControl, 2021-03-02:
Just set my homebridge to the blscklist. Seems to work. After a restart of the homebridge the device is still running.

@eford321 see line 3622 at arduino_homekit_server.cpp

eford321, 2021-03-02:
gotcha!  @EIBHomeControl 

What format are you using for the IP address for xxxx?  I can't get it to compile.

Yurik72, 2021-03-02:
Ahh , Ok sorry. This is internal format stored in 4 bytes integer

But doesn’t matter

I have several ideas and you can help me, I can’t reproduce this issue

1.	Get latest commit , I did it recently
2.	Try with or without “black list” I have prepared piece of code for human readable IP address line 3625
3.	Get a me a log, from ESP when Bridge restarted I have added additional log to see what happens
4.	I did additional  mdns announce line 4336

 

So if this complex will help , we need exactly understand what is helped. Otherwise I will continue investigation

Thanks for your patient, really I can’t reproduce a bug

eford321, 2021-03-02:
Compile problem is with client info line..
Using:
>	String yourIP = "192.168.1.213"; // IP Bridge
  if(wifiClient->remoteIP().toString() ==yourIP)  // this is a bridge
	{
		CLIENT_INFO(ct, "Home bridge IGNORED");
		wifiClient->stop();
		delete wifiClient;
		return NULL;

	}
Getting the following compilation errors:
>Arduino: 1.8.13 (Mac OS X), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), Basic SSL ciphers (lower ROM use), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Serial, None, Only Sketch, 921600"

In file included from /Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp:30:0:
/Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp: In function 'client_context_t* homekit_server_accept_client(homekit_server_t*)':
/Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp:3628:15: error: 'ct' was not declared in this scope
   CLIENT_INFO(ct, "Home bridge IGNORED");
               ^
/Users/evan/Documents/Arduino/libraries/ESPHap/debug.h:27:75: note: in definition of macro 'INFO'
 #define INFO(message, ...) printf_P(PSTR(">>> HomeKit: " message "\n"), ##__VA_ARGS__)
                                                                           ^
/Users/evan/Documents/Arduino/libraries/ESPHap/arduino_homekit_server.cpp:3628:3: note: in expansion of macro 'CLIENT_INFO'
   CLIENT_INFO(ct, "Home bridge IGNORED");
   ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

eford321, 2021-03-02:
Changed to 
>	String yourIP = "192.168.1.213"; // IP Bridge
  if(wifiClient->remoteIP().toString() ==yourIP)  // this is a bridge
	{
		//CLIENT_INFO(ct, "Home bridge IGNORED");
		Serial.println("Home bridge IGNORED");
    wifiClient->stop();
		delete wifiClient;
		return NULL;
	}

Ran with the following log. ESP does continue functioning after Query from Homebridge.

>3:02:35.001 -> init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
13:02:35.001 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
13:02:35.001 -> >>> Home Integration: hap_services added chararacteristic  1: 25 
13:02:35.001 -> >>> Home Integration: homekit_is_paired 1
13:02:35.001 -> >>> HomeKit: Free heap: 31960
13:02:35.001 -> >>> server_new: WiFiServer begin at port: 5556
13:02:35.001 -> 
13:02:35.001 -> >>> HomeKit: Starting server
13:02:35.034 -> >>> HomeKit: storage init 0
13:02:35.034 -> >>> HomeKit: Using existing accessory ID: B1:E8:4D:32:98:71
13:02:35.034 -> >>> HomeKit: find pairings
13:02:35.034 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup
13:02:35.034 -> >>> HomeKit: Configuring mDNS
13:02:35.034 -> >>> HomeKit: MDNS.begin: ES-B1E8, IP: 192.168.1.23
13:02:35.034 -> >>> homekit_mdns_init: Accessory Setup ID = YK72
13:02:35.068 -> >>> HomeKit: Init server over
13:02:42.521 -> pm open,type:2 0
13:03:09.605 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
13:03:09.605 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:37760
13:03:09.605 -> Home bridge IGNORED
13:03:16.631 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
13:03:16.631 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:64000
13:03:16.631 -> >>> HomeKit: [Client 1073699020] Got client content from the cache 0
13:03:16.668 -> >>> HomeKit: [Client 1073699020] new client accepted
13:03:16.668 -> >>> homekit_client_process: [Client 1073699020] Reading 158 bytes
13:03:16.668 -> >>> homekit_server_on_message_complete: http_parser message_complete
13:03:16.668 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
13:03:16.668 -> >>> homekit_server_on_pair_verify: Free heap: 24536
13:03:16.668 -> >>> tlv_debug: Got following TLV values:
13:03:16.668 -> >>> tlv_debug: Type 6 value (1 bytes): \x01
13:03:16.701 -> >>> tlv_debug: Type 3 value (32 bytes): =+t**UX\xD3\xE1\xA6\xCE\x93\x90\xE9\x9F\x80Y\xC4<"a\xDC)\x85\xB2\x01\xE5\xF7\x8B\x82\xCEl
13:03:16.701 -> >>> HomeKit: [Client 1073699020] Pair Verify Step 1/2
13:03:16.701 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Importing device Curve25519 public key
13:03:16.701 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Generating accessory Curve25519 key
13:03:16.840 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Exporting accessory Curve25519 public key
13:03:16.840 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Generating Curve25519 shared secret
13:03:16.980 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Generating signature
13:03:17.054 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Generating proof
13:03:17.054 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Encrypting response
13:03:17.091 -> >>> send_tlv_response: [Client 1073699020] Sending TLV response
13:03:17.091 -> >>> tlv_debug: Got following TLV values:
13:03:17.091 -> >>> tlv_debug: Type 6 value (1 bytes): \x02
13:03:17.091 -> >>> tlv_debug: Type 3 value (32 bytes): \x84\x0D\x0C\xF4\x95\xEA\xD7g\xA3\xFE\xD2\xEA+\x96\xCB \x81Y\xE2V`\x91\x08\x93bDE\x99-\xB2\x8B(
13:03:17.091 -> >>> tlv_debug: Type 5 value (101 bytes): y\xB2\x0D\x13E@\x0B\x9Ep\x86<\xBE]\xE4\xCD\xEE\xAF?~\x91\xD4HS`\xD4\x8E\xFA\xCBN\x1A\x91\xA3\x1C\x08\x84I\xEAZar\xDDq\xEE\xB2Ad\xA8a\xF1\xAAx\x10v\xDA\xC5$hq\x18k1\xD1s\xE5\xC8\x1DI5\x0D,\xE0\x01\xB6\xD0G*4\xF3\x87\x95\x15\x9Aun\x8E\xFFecV\xDAsp\xEA(\xC3\xE2\xB3%\x15{>
13:03:17.124 -> >>> client_send: [Client 1073699020] send data size=244, encrypted=false
13:03:17.124 -> >>> client_send: [Client 1073699020] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 \x84\x0D\x0C\xF4\x95\xEA\xD7g\xA3\xFE\xD2\xEA+\x96\xCB \x81Y\xE2V`\x91\x08\x93bDE\x99-\xB2\x8B(\x05ey\xB2\x0D\x13E@\x0B\x9Ep\x86<\xBE]\xE4\xCD\xEE\xAF?~\x91\xD4HS`\xD4\x8E\xFA\xCBN\x1A\x91\xA3\x1C\x08\x84I\xEAZar\xDDq\xEE\xB2Ad\xA8a\xF1\xAAx\x10v\xDA\xC5$hq\x18k1\xD1s\xE5\xC8\x1DI5\x0D,\xE0\x01\xB6\xD0G*4\xF3\x87\x95\x15\x9Aun\x8E\xFFecV\xDAsp\xEA(\xC3\xE2\xB3%\x15{>
13:03:17.246 -> >>> HomeKit: Free heap: 24296
13:03:17.246 -> >>> homekit_client_process: [Client 1073699020] Reading 247 bytes
13:03:17.246 -> >>> homekit_server_on_message_complete: http_parser message_complete
13:03:17.246 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
13:03:17.246 -> >>> homekit_server_on_pair_verify: Free heap: 24208
13:03:17.246 -> >>> tlv_debug: Got following TLV values:
13:03:17.246 -> >>> tlv_debug: Type 5 value (120 bytes): \xB0\x02\xB4\x8EW\xF4\xA2\x8FNs\xD9\xC2\x9E?$n3\xEB\xFC\x1BDu\xCA\x8Bf\xEC/$\xA2\x06a2\xC04\x00\xC0\xD0\x94\xD05\x13\xB6&\x1D\xD2m\x14{-K\x9C\x92b\x0Ah\xBE\x15\xC4\xF4P\x9E\xFB\x0Fk3h\x00\xC3-\xF8qL\x8A\xD3\xD8\x85\xFE\xF6\xB9\xD89\xFC{\x82hX\xF7\xAB\xFF\xE1\x85\xFC\x92\xC1s\xD1\xCFl\x90gat\xED\xBE\x80\xEE\xA1\xD7\xF3\x9C\x8E\x03\x1Cb9m^\xBC\x84
[... cut ...]
13:03:17.301 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Decrypting payload
13:03:17.301 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Searching pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
13:03:17.301 -> >>> HomeKit: [Client 1073699020] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
13:03:17.301 -> >>> homekit_server_on_pair_verify: [Client 1073699020] Verifying device signature
13:03:18.054 -> >>> send_tlv_response: [Client 1073699020] Sending TLV response
13:03:18.054 -> >>> tlv_debug: Got following TLV values:

eford321, 2021-03-02:
One query response from ESP...
[image]

Yurik72, 2021-03-02:
Oh sorry,

I have committed correct line

INFO( "Home bridge IGNORED");

Instead of 

CLIENT_INFO(ct, "Home bridge IGNORED");

And would like to correct, compile and run this sketch without those lines (without black list) ,

Give me a short logs from ESP, I would like to see http request

 

Then uncomment those lines and run again

 

Waiting for result

EIBHomeControl, 2021-03-02:
@eford321 

little modification needed. -:)

if(wifiClient->remoteIP() == IPAddress(192,168,2,20))  // this is a bridge
	{
	    INFO( "Home bridge IGNORED");
		wifiClient->stop();
		delete wifiClient;
		return NULL;
	//
	}

Yurik72, 2021-03-02:
No problem, this is just test, I need result with that and without to find solution and reach a target 😊

Yurik72, 2021-03-02:
We need to understand root cause, after that code will be fine, don’t worry

EIBHomeControl, 2021-03-02:
so with the lastest commit (a9ca4c5) and the modification for the Homebridge ip I'll get the following log.

pls see the marked sections when the Homebridge get restarted

[CoolTerm Capture 2021-03-02 21-05-34.txt](https://github.com/Yurik72/ESPHap/files/6071505/CoolTerm.Capture.2021-03-02.21-05-34.txt)

eford321, 2021-03-02:
This with with the Homebridge ignored:
>15:14:07.014 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
15:14:07.050 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:43322
15:14:07.050 -> >>> HomeKit: Home bridge IGNORED
15:14:31.330 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
15:14:31.330 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:50373
15:14:31.363 -> >>> HomeKit: [Client 1073699860] Got client content from the cache 0
15:14:31.363 -> >>> HomeKit: [Client 1073699860] new client accepted
15:14:31.363 -> >>> homekit_client_process: [Client 1073699860] Reading 158 bytes
15:14:31.363 -> >>> homekit_server_on_message_complete: http_parser message_complete
15:14:31.363 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
15:14:31.397 -> >>> homekit_server_on_pair_verify: Free heap: 24552
15:14:31.397 -> >>> tlv_debug: Got following TLV values:
15:14:31.397 -> >>> tlv_debug: Type 6 value (1 bytes): \x01
15:14:31.397 -> >>> tlv_debug: Type 3 value (32 bytes): \xE4 \xC8\xEA(b\xE2\xEAf\x15\xAA/b\xB5W\xFA\xCF\x8E\x03\xA7\x13\xB6\xD6$\xA9\xF5\x91\xF3\xC7Q
15:14:31.397 -> >>> HomeKit: [Client 1073699860] Pair Verify Step 1/2
15:14:31.397 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Importing device Curve25519 public key
15:14:31.435 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Generating accessory Curve25519 key
15:14:31.541 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Exporting accessory Curve25519 public key
15:14:31.575 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Generating Curve25519 shared secret
15:14:31.679 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Generating signature
15:14:31.750 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Generating proof
15:14:31.784 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Encrypting response
15:14:31.784 -> >>> send_tlv_response: [Client 1073699860] Sending TLV response
15:14:31.784 -> >>> tlv_debug: Got following TLV values:
15:14:31.784 -> >>> tlv_debug: Type 6 value (1 bytes): \x02
15:14:31.784 -> >>> tlv_debug: Type 3 value (32 bytes): \xC0\x05\xF4\xB8\xE0\x96\x03\xAE\x93\x9FE\xF1b\x18\xCB2\x10x3\xA9[sS=\x13\x8E\xDD\xC3\x97Z}s
15:14:31.818 -> >>> tlv_debug: Type 5 value (101 bytes): \xE9\xE3_\xE6u\xAE\xE0\xD7w{\x02\xA4VS\x07\x19L\xE9\xEDjEPP\xF4\xD2\x18\x04\xCE%\x8F\xDC5\xFDT\x11\xDC\xBA\x11\xB0\x8E\x1B\xCF\x9AN\xFB\xF0\x87\x8F\xF0\xC8\xF8\x9C`\xED\x8CP\x89\x99x4\x12\x86m|\xDC\xD6\x04-,\xBE\xB5\xED\xE4\x1D3/\x1E\xF7\xAA\xAF\xD6\x83\xDA\x92\x8A\xF0\x95Z{\xAF\x0E\xE6:\xA8\x9E\xE1D\x0C\x8D\x83\x8C
15:14:31.818 -> >>> client_send: [Client 1073699860] send data size=244, encrypted=false
15:14:31.856 -> >>> client_send: [Client 1073699860] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 \xC0\x05\xF4\xB8\xE0\x96\x03\xAE\x93\x9FE\xF1b\x18\xCB2\x10x3\xA9[sS=\x13\x8E\xDD\xC3\x97Z}s\x05e\xE9\xE3_\xE6u\xAE\xE0\xD7w{\x02\xA4VS\x07\x19L\xE9\xEDjEPP\xF4\xD2\x18\x04\xCE%\x8F\xDC5\xFDT\x11\xDC\xBA\x11\xB0\x8E\x1B\xCF\x9AN\xFB\xF0\x87\x8F\xF0\xC8\xF8\x9C`\xED\x8CP\x89\x99x4\x12\x86m|\xDC\xD6\x04-,\xBE\xB5\xED\xE4\x1D3/\x1E\xF7\xAA\xAF\xD6\x83\xDA\x92\x8A\xF0\x95Z{\xAF\x0E\xE6:\xA8\x9E\xE1D\x0C\x8D\x83\x8C
15:14:31.890 -> >>> HomeKit: Free heap: 24312
15:14:31.890 -> >>> homekit_client_process: [Client 1073699860] Reading 247 bytes
15:14:31.924 -> >>> homekit_server_on_message_complete: http_parser message_complete
15:14:31.924 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
15:14:31.924 -> >>> homekit_server_on_pair_verify: Free heap: 24224
15:14:31.924 -> >>> tlv_debug: Got following TLV values:
15:14:31.924 -> >>> tlv_debug: Type 5 value (120 bytes): \xB1\x81\xE0\x81\xF8\x04\xB9\x161\x04\xAE\x15\xC2k\xDF:\xF5\xE1\x9B\xBE\x0F{|\xEF\x8B\xA5\x0D\x15\xC1'\x88'\xA3\x1D\x9F\xCF\x8C\xF4\x1C\x8E\x82M\x12\xC1^\xFFAO\x18\xBE\xFD^J\xCE\xF8\x86\xA1N\xECri\xCCA\x19\xD0\xC6v[\x9AR'hz\xFB\xFF\x08\xC2\x97\xE91\xEC\xF0\x80T\x97JN\xB1\x92\x07\xF9g\xD2\xC4\xD9\x82\xF0~\xA1\x9F\xA7+\x91\xB9\xE7\xE2\xED\xAE\xD7tb\xB2\x06\xE1\x9EA4\xB0U?
15:14:31.958 -> >>> tlv_debug: Type 6 value (1 bytes): \x03
15:14:31.958 -> >>> HomeKit: [Client 1073699860] Pair Verify Step 2/2
15:14:31.992 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Decrypting payload
15:14:31.992 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Searching pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
15:14:31.992 -> >>> HomeKit: [Client 1073699860] Found pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239
15:14:31.992 -> >>> homekit_server_on_pair_verify: [Client 1073699860] Verifying device signature
15:14:32.768 -> >>> send_tlv_response: [Client 1073699860] Sending TLV response
15:14:32.802 -> >>> tlv_debug: Got following TLV values:
15:14:32.802 -> >>> tlv_debug: Type 6 value (1 bytes): \x04
15:14:32.802 -> >>> client_send: [Client 1073699860] send data size=105, encrypted=false
15:14:32.802 -> >>> client_send: [Client 1073699860] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 3\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x04
15:14:32.836 -> >>> HomeKit: [Client 1073699860] Verification successful, secure session established
15:14:32.836 -> >>> HomeKit: Free heap: 24464
[... cut ...]
15:14:33.256 -> >>> homekit_server_on_update_characteristics: Free heap: 24592
15:14:33.256 -> >>> HomeKit: Free heap: 24568
15:14:33.256 -> >>> homekit_server_on_update_characteristics: [Client 1073699860] There were no processing errors, sending No Content response
15:14:33.256 -> >>> client_send: [Client 1073699860] send data size=27, encrypted=true
15:14:33.256 -> >>> client_send: [Client 1073699860] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
15:14:33.292 -> >>> client_send_encrypted_: [Client 1073699860] client_send_encrypted_ size=27

eford321, 2021-03-02:
Without Homebridge ignored:

>15:17:20.941 -> pm open,type:2 0
15:17:30.420 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
15:17:30.420 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:43820
15:17:30.420 -> >>> HomeKit: [Client 1073698052] Got client content from the cache 0
15:17:30.420 -> >>> HomeKit: [Client 1073698052] new client accepted
15:17:30.420 -> >>> homekit_client_process: [Client 1073698052] Reading 142 bytes
15:17:30.499 -> >>> homekit_server_on_message_complete: http_parser message_complete
15:17:30.499 -> !!! HomeKit: Unknown endpoint
15:17:30.499 -> >>> client_send: [Client 1073698052] send data size=26, encrypted=false
15:17:30.499 -> >>> client_send: [Client 1073698052] Sending payload: HTTP/1.1 404 Not Found\x0D\x0A\x0D\x0A
15:17:35.467 -> >>> HomeKit: [Client 1073698052] Closing client connection
15:17:35.467 -> >>> homekit_server_close_client: [Client 1073698052] The sockect is stoped
15:17:35.467 -> >>> HomeKit: [Client 0] client content free
15:17:41.817 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
15:17:41.817 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:50587
15:17:41.817 -> >>> HomeKit: [Client 1073697988] Got client content from the cache 0
15:17:41.851 -> >>> HomeKit: [Client 1073697988] new client accepted
15:17:41.851 -> >>> homekit_client_process: [Client 1073697988] Reading 158 bytes

Yurik72, 2021-03-02:
Ok, clear home bridge send empty messages , I will double check code, but if any success result in all combinations ?

eford321, 2021-03-02:
FYI, You don't have to restart Homebridge to cause the Query Multicast, opening a new browser session with homebridge causes it to do the same, send a Pointer record service query looking for all HAP devices(_hap._tcp.local).

Yurik72, 2021-03-03:
Ok, you give me a hint. 

I have simulated by small software huge number of request (mDNS PTR) the same as you have attached from the WireShark

Definitely sometimes I do not see a response, however other device respond (apple device)

And this give me again a hint, I made already changes for ESP32  #34 a month ago and forgot about ESP8266

 

By default ESP8266 has sleep level LIGHT SLEEP , means modem wake up every DTIM. LIGHT SLEEP has 100ms. 

AND mDNS is UDP, means some of this packets can be lost.

Currently I have implemented SLEEP_NONE. In my test environment is much better. However again, I can’t reproduce initial issue

eford321, 2021-03-03:
Why does it stop responding to all at her clients after the mdns ptr?

Yurik72, 2021-03-03:
This exactly my question as well, I can’t reproduce, however the same I commented for Esp32 on the official lib, after that I have changed on my lib Esp32, currently we have to try on esp8266

Yurik72, 2021-03-03:
Hi, thanks, based on this logs. I proper behavior and reliability, device working properly

Yurik72, 2021-03-03:
As well see

https://github.com/espressif/esp-apple-homekit-adk/issues/18

 

I’d say I met with this problem on ESP32, after changes my Home works perfect, I do not see any unresponsive.

Why it’s related on Home Bridge restarted and mDNS  PTR, good question, but I think we are in progress every day

eford321, 2021-03-03:
Interesting! Is this change in bf5de4d?  I tried this on ESP32 and it still when unresponsive after HomeBridge restart.

Yurik72, 2021-03-03:
No, for Esp32 is changes accordingly #34 months ago or more

eford321, 2021-03-03:
> No, for Esp32 is changes accordingly #34 months ago or more
> […](#)

This does not work for me.  ESP32 is behaving the same as ESP8266.  Let me know if you need logs or Wireshark traces.

Yurik72, 2021-03-03:
Yes, will be helpful to understand communication between Esp and others,
Really very interesting bug,
As I mentioned I wrote small soft to kill my esp by mDNS ptr queries, thousands request with delay 100 ms, even at this time esp still working, I’m using esphapadvancedled to switch on/off lamp

eford321, 2021-03-03:
Here is the trace of ESP32 running EspHapLed.  192.168.1.51 is the ESP32. You can see other Hap devices responding also.

[image]

Yurik72, 2021-03-03:
Thanks, I see some differences in response, comparing with other devices.... maybe it a reason, ok will investigate

eford321, 2021-03-03:
Here's the serial log from that same session. I was able to communicate from Mac and iPhone prior to reboot.  I marked the reboot.
[ESP32 log.txt](https://github.com/Yurik72/ESPHap/files/6077262/ESP32.log.txt)

Yurik72, 2021-03-04:
Thanks,

What I see at the log, after rebooting some  Apple clients still accessible and characteristic is updated, However ESP got several unknown request,

My questions are

1.	If device still unresponsive even after 5-10 min ? As well after Reopen Apple Home Application ?
2.	What kind of http request is sends by Hopmebridge to ESP, can you get it captured from WireShark ?

Yurik72, 2021-03-04:
And forgot ask you, @EIBHomeControl  mentioned
-----
Just set my homebridge to the blscklist. Seems to work. After a restart of the homebridge the device is still running.
-----
So, what is happenig on your enviropment with black list for HomeBridge? Maybe we need back to investigation my firts idea

EIBHomeControl, 2021-03-04:
> ## And forgot ask you, @EIBHomeControl mentioned
> ## Just set my homebridge to the blscklist. Seems to work. After a restart of the homebridge the device is still running.
> So, what is happenig on your enviropment with black list for HomeBridge? Maybe we need back to investigation my firts idea

So at the moment it's running with the black list enabled. After a restart of the homebridge the device is still running.
Let's see tomorrow, then we have more the 20 hours.

eford321, 2021-03-04:
>So, what is happenig on your enviropment with black list for HomeBridge? Maybe we need back to investigation my firts idea

Sorry, yes the blacklist works for me.  This log from earlier shows the device responding after homebridge reboot.

> This with with the Homebridge ignored:

eford321, 2021-03-04:
>My questions are
>1.	If device still unresponsive even after 5-10 min ? As well after Reopen Apple Home Application ?

Yes the device remains unresponsive after 5-10 min. Reopening home app on Mac or iOS shows updating then no response.
Here is a 11+ minute log of console string with Homebridge PTR QM. (192.168.1.213 is homebridge, others are Mac and iOS devices.

>09:29:50.890 -> >>> HomeKit: WiFiServer receives a new client (current 6, max 8)
09:29:50.890 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:60746
09:29:50.890 -> >>> HomeKit: [Client 1073701388] Got client content from the cache 6
09:29:50.890 -> >>> HomeKit: [Client 1073701388] new client accepted
09:29:50.926 -> >>> homekit_client_process: [Client 1073701388] Reading 142 bytes
09:29:50.926 -> >>> homekit_server_on_message_complete: http_parser message_complete
09:29:50.926 -> !!! HomeKit: Unknown endpoint
09:29:50.926 -> >>> client_send: [Client 1073701388] send data size=26, encrypted=false
09:29:50.926 -> >>> client_send: [Client 1073701388] Sending payload: HTTP/1.1 404 Not Found\x0D\x0A\x0D\x0A
09:29:55.930 -> >>> HomeKit: [Client 1073701388] Closing client connection
09:29:55.967 -> >>> homekit_server_close_client: [Client 1073701388] The sockect is stoped
09:29:55.967 -> >>> HomeKit: [Client 0] client content free
09:29:58.467 -> >>> homekit_client_process: [Client 1073700212] Reading 188 bytes
09:29:58.467 -> >>> homekit_client_process: [Client 1073700212] Decrypted 170 bytes, available 0
09:29:58.533 -> Decrypted data (170 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: ES-B1E8._hap._tcp.local\x0D\x0AContent-Length: 50\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":0}]}"
09:30:08.488 -> >>> HomeKit: [Client 1073700212] Disconnected!
09:30:11.555 -> >>> HomeKit: WiFiServer receives a new client (current 6, max 8)
09:30:11.555 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:59180
09:30:11.555 -> >>> HomeKit: [Client 1073698084] Got client content from the cache 6
09:30:11.555 -> >>> HomeKit: [Client 1073698084] new client accepted
09:30:11.555 -> >>> homekit_client_process: [Client 1073698084] Reading 158 bytes
09:30:13.493 -> >>> HomeKit: [Client 1073700212] Closing client connection
09:30:13.493 -> >>> homekit_server_close_client: [Client 1073700212] The sockect is stoped
09:30:13.493 -> >>> HomeKit: [Client 0] client content free
09:31:15.036 -> >>> HomeKit: [Client 1073698084] Disconnected!
09:31:15.036 -> >>> HomeKit: WiFiServer receives a new client (current 6, max 8)
09:31:15.036 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:59252
09:31:15.036 -> >>> HomeKit: [Client 1073700212] Got client content from the cache 4
09:31:15.036 -> >>> HomeKit: [Client 1073700212] new client accepted
09:31:15.036 -> >>> homekit_client_process: [Client 1073700212] Reading 158 bytes
09:31:16.556 -> >>> HomeKit: [Client 1073698084] Closing client connection
09:31:16.556 -> >>> homekit_server_close_client: [Client 1073698084] The sockect is stoped
09:31:16.556 -> >>> HomeKit: [Client 0] client content free
09:31:30.244 -> >>> homekit_client_process: [Client 1073700948] Reading 188 bytes
09:31:30.281 -> >>> homekit_client_process: [Client 1073700948] Decrypted 170 bytes, available 0
09:31:30.281 -> Decrypted data (170 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: ES-B1E8._hap._tcp.local\x0D\x0AContent-Length: 50\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"ev":true}]}"
09:31:40.619 -> >>> HomeKit: [Client 1073700948] Disconnected!
09:31:45.625 -> >>> HomeKit: [Client 1073700948] Closing client connection
09:31:45.625 -> >>> homekit_server_close_client: [Client 1073700948] The sockect is stoped
09:31:45.671 -> >>> HomeKit: [Client 0] client content free
09:32:01.957 -> >>> homekit_client_process: [Client 1073698988] Reading 188 bytes
09:32:01.957 -> >>> homekit_client_process: [Client 1073698988] Decrypted 170 bytes, available 0
09:32:01.957 -> Decrypted data (170 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: ES-B1E8._hap._tcp.local\x0D\x0AContent-Length: 50\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"ev":true}]}"
09:32:03.990 -> >>> HomeKit: WiFiServer receives a new client (current 5, max 8)
09:32:03.990 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.15:60474
09:32:03.990 -> >>> HomeKit: [Client 1073699844] Got client content from the cache 5
09:32:03.990 -> >>> HomeKit: [Client 1073699844] new client accepted
09:32:03.990 -> >>> homekit_client_process: [Client 1073699844] Reading 158 bytes
09:32:11.576 -> >>> HomeKit: [Client 1073700212] Disconnected!
09:32:12.173 -> >>> HomeKit: [Client 1073698988] Disconnected!
09:32:15.651 -> >>> HomeKit: WiFiServer receives a new client (current 6, max 8)
09:32:15.651 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.213:33142
09:32:15.651 -> >>> HomeKit: [Client 1073698084] Got client content from the cache 6
09:32:15.691 -> >>> HomeKit: [Client 1073698084] new client accepted
09:32:15.691 -> >>> homekit_client_process: [Client 1073698084] Reading 142 bytes
09:32:16.589 -> >>> HomeKit: [Client 1073700212] Closing client connection
09:32:16.589 -> >>> homekit_server_close_client: [Client 1073700212] The sockect is stoped
[... cut ...]
09:43:57.830 -> >>> HomeKit: [Client 1073700212] Got client content from the cache 0
09:43:57.830 -> >>> HomeKit: [Client 1073700212] new client accepted
09:43:57.830 -> >>> homekit_client_process: [Client 1073700212] Reading 142 bytes
09:44:07.809 -> >>> HomeKit: [Client 1073700212] Disconnected!
09:44:12.798 -> >>> HomeKit: [Client 1073700212] Closing client connection
09:44:12.798 -> >>> homekit_server_close_client: [Client 1073700212] The sockect is stoped
09:44:12.798 -> >>> HomeKit: [Client 0] client content free

eford321, 2021-03-04:
>My questions are
>2.	What kind of http request is sends by Hopmebridge to ESP, can you get it captured from WireShark ?

I don't see http from Homebridge to ESP, only MDNS QM.  There is no unicast traffic to/from homebridge & ESP. Only the multicast QM message.  Is ESP interpreting this as http?

Here is the Wireshark capture of the QM request:
[image]

Yurik72, 2021-03-04:
But this is definitely good news and my first assumption that HTTP request from Homebridge to ESP killing ESP.  Maybe is not HTTP but request came to Homekit port

Otherwise I do not see any other problem for instance mDNS. You can help me by capturing exact HTTP request by wireshark, it will  save my time. Is that possible ? 

As well I recently committed additional traces

 

Please uncomment them line 3507 and send me a result after HomeBridge restarting , it will capture all requests coming to _hap port

Thanks in advance

Yurik72, 2021-03-04:
I send previous e-mail without reading that

---Only the multicast QM message. Is ESP interpreting this as http?

No.. no.. any messages is just set of bytes, if they sent to mdns port should be interpreted as mdns, if they sent to Apple port should be interpreted as http

Device just listening both ports

 

Apple port : 5556

mDNS port 5353

 

 <https://user-images.githubusercontent.com/5996455/109983114-85841600-7cd0-11eb-95b2-0446d05015c1.png>

eford321, 2021-03-04:
> But this is definitely good news and my first assumption that HTTP request from Homebridge to ESP killing ESP. Maybe is not HTTP but request came to Homekit port Otherwise I do not see any other [...]

I assume it is being passed to homekit port, because Homebridge is asking for _hap._tcp.local devices. 
Do you want the raw frame? Sorry I clipped it earlier.
[image]

Here is the trace with new commit.  line 3507 in arduino_homekit_server.cpp is uncommented.

>12:20:09.819 -> init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
12:20:09.855 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
12:20:09.855 -> >>> Home Integration: hap_services added chararacteristic  1: 25 
12:20:09.855 -> >>> Home Integration: homekit_is_paired 1
12:20:09.855 -> >>> HomeKit: Free heap: 32576
12:20:09.855 -> >>> server_new: WiFiServer begin at port: 5556
12:20:09.855 -> 
12:20:09.855 -> >>> HomeKit: Starting server
12:20:09.855 -> >>> HomeKit: storage init 0
12:20:09.855 -> >>> HomeKit: Using existing accessory ID: B1:E8:4D:32:98:71
12:20:09.855 -> >>> HomeKit: find pairings
12:20:09.893 -> >>> HomeKit: Found admin pairing with 6FDD5BE0-78E1-4A4C-B5D7-98277EEFA239, disabling pair setup
12:20:09.893 -> >>> HomeKit: Configuring mDNS
12:20:09.893 -> >>> HomeKit: Wifi Sleep type:2
12:20:09.893 -> >>> HomeKit: Set Wifi Sleep type :0
12:20:09.893 -> >>> HomeKit: MDNS.begin: ES-B1E8, IP: 192.168.1.23
12:20:09.893 -> >>> homekit_mdns_init: Accessory Setup ID = YK72
12:20:09.893 -> >>> HomeKit: Init server over
12:20:12.068 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
12:20:12.068 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.13:57300
12:20:12.068 -> >>> HomeKit: [Client 1073697532] Got client content from the cache 0
12:20:12.102 -> >>> HomeKit: [Client 1073697532] new client accepted
12:20:12.102 -> >>> homekit_client_process: [Client 1073697532] Reading 158 bytes
12:20:12.102 -> >>> homekit_server_on_message_complete: http_parser message_complete
12:20:12.102 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
12:20:12.102 -> >>> homekit_server_on_pair_verify: Free heap: 26288
12:20:12.138 -> >>> tlv_debug: Got following TLV values:
12:20:12.138 -> >>> tlv_debug: Type 6 value (1 bytes): \x01
12:20:12.138 -> >>> tlv_debug: Type 3 value (32 bytes): \xE4:\x0AY\x8ED"\xD7\xF8\xDF\x17\xC9a\x1C6\x0D\xC518\xF8\xE3\xF4\x94]H\x80\x9D\xC3\x81\xA8\xECy
12:20:12.138 -> >>> HomeKit: [Client 1073697532] Pair Verify Step 1/2
12:20:12.138 -> >>> homekit_server_on_pair_verify: [Client 1073697532] Importing device Curve25519 public key
12:20:12.138 -> >>> homekit_server_on_pair_verify: [Client 1073697532] Generating accessory Curve25519 key
12:20:12.282 -> >>> homekit_server_on_pair_verify: [Client 1073697532] Exporting accessory Curve25519 public key
12:20:12.282 -> >>> homekit_server_on_pair_verify: [Client 1073697532] Generating Curve25519 shared secret
12:20:12.421 -> >>> homekit_server_on_pair_verify: [Client 1073697532] Generating signature
12:20:12.495 -> >>> homekit_server_on_pair_verify: [Client 1073697532] Generating proof
12:20:12.495 -> >>> homekit_server_on_pair_verify: [Client 1073697532] Encrypting response
12:20:12.495 -> >>> send_tlv_response: [Client 1073697532] Sending TLV response
12:20:12.532 -> >>> tlv_debug: Got following TLV values:
12:20:12.532 -> >>> tlv_debug: Type 6 value (1 bytes): \x02
12:20:12.532 -> >>> tlv_debug: Type 3 value (32 bytes): \xC3fC\xFC\xDBeS\xF9_\xAB\xA7\xBA \x06 \x97\xEFU\x91\x0C\xFB\xF7\x98\x99E0\xE5\xE0\x95g\xCE/
12:20:12.532 -> >>> tlv_debug: Type 5 value (101 bytes): \x8Ca\xEF#Q\xA3\xD68V>V\x0D6\xF5M>\xC5m\x11\xE8\x1E\xF2&\xCA\x92\x1B\x8D\x9E\x0B8\xBF\xB5!T`hh*\xC6\xDE\xD6\xD2\xC9\xF3]p\xB3\x82\xECu\x93\x1AJM\x07\xE6\xB1k>L\xCB^K\xD9\xA4\x82\xA7\x18\xBD/aYt\xD1\x11b\x94\x08\x17\x17\x13?\xD8\xBF\xE8\xBAdp\xC0PF\x00\x95k\x12[ I#]I
12:20:12.569 -> >>> client_send: [Client 1073697532] send data size=244, encrypted=false
12:20:12.569 -> >>> client_send: [Client 1073697532] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02\x03 \xC3fC\xFC\xDBeS\xF9_\xAB\xA7\xBA \x06 \x97\xEFU\x91\x0C\xFB\xF7\x98\x99E0\xE5\xE0\x95g\xCE/\x05e\x8Ca\xEF#Q\xA3\xD68V>V\x0D6\xF5M>\xC5m\x11\xE8\x1E\xF2&\xCA\x92\x1B\x8D\x9E\x0B8\xBF\xB5!T`hh*\xC6\xDE\xD6\xD2\xC9\xF3]p\xB3\x82\xECu\x93\x1AJM\x07\xE6\xB1k>L\xCB^K\xD9\xA4\x82\xA7\x18\xBD/aYt\xD1\x11b\x94\x08\x17\x17\x13?\xD8\xBF\xE8\xBAdp\xC0PF\x00\x95k\x12[ I#]I
12:20:12.640 -> >>> HomeKit: Free heap: 25944
12:20:12.640 -> >>> HomeKit: WiFiServer receives a new client (current 1, max 8)
12:20:12.640 -> >>> HomeKit: Got new client connection: local 192.168.1.23:5556, remote 192.168.1.37:54753
12:20:12.640 -> >>> HomeKit: [Client 1073697892] Got client content from the cache 1
12:20:12.671 -> >>> HomeKit: [Client 1073697892] new client accepted
12:20:12.671 -> >>> homekit_client_process: [Client 1073697892] Reading 158 bytes
12:20:12.671 -> >>> homekit_server_on_message_complete: http_parser message_complete
12:20:12.671 -> >>> homekit_server_on_pair_verify: HomeKit Pair Verify
12:20:12.671 -> >>> homekit_server_on_pair_verify: Free heap: 25816
12:20:12.671 -> >>> tlv_debug: Got following TLV values:
12:20:12.705 -> >>> tlv_debug: Type 6 value (1 bytes): \x01
[... cut ...]
12:21:28.717 -> Decrypted data (170 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: ES-B1E8._hap._tcp.local\x0D\x0AContent-Length: 50\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":1}]}"
12:21:39.214 -> >>> HomeKit: [Client 1073697532] Disconnected!
12:21:44.215 -> >>> HomeKit: [Client 1073697532] Closing client connection
12:21:44.215 -> >>> homekit_server_close_client: [Client 1073697532] The sockect is stoped
12:21:44.410 -> >>> HomeKit: [Client 0] client content free

eford321, 2021-03-04:
Also, ESP responds to mDNS from Homebridge _hap._tcp.local QM. (.213 is homebridge, .23 is ESP)
[image]

But ESP ignores mdns QM requests looking for _arduino._tcp.local coming from other devices when I don't have OTA in sketch. This packet is ignored:
[image]

Yurik72, 2021-03-04:
OK, THANKS 

Will analyze. I have prepared log for ESP8266. But I see you tested with ESp32… 

Do you need log for ESP32 or you can use ESP8266 ?

 

About ->Also, ESP responds to mDNS from Homebridge _hap._tcp.local QM. (.213 is homebridge, .23 is ESP)

ESP is respond and advertise broadcast query (224.0.0.251 is broadcast).I do not see any direct communication  between 23 and 213

 

 <https://user-images.githubusercontent.com/5996455/110004667-b15dc680-7ce5-11eb-9c03-2011bfa6f6f4.png> 

 <https://user-images.githubusercontent.com/5996455/110005019-0b5e8c00-7ce6-11eb-8a43-e153c33efd2a.png>

[22 more replies]

closed 2021-03-05 by eford321