Yurik72/ESPHap #65: ESP32 and cJSON compile/linking errors

judepereira, 2021-02-28:
It appears that ESPHap bundles cJSON as well as the ESP32 framework. When I compile my project, I see many errors (included just 2 here, since they're the same):
```
.pio/build/local_32/libb41/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_ReplaceItemInObject+0x0): first defined here
/Users/jude/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libjson.a(cJSON.o): In function `cJSON_ReplaceItemInObjectCaseSensitive':
cJSON.c:(.text.cJSON_ReplaceItemInObjectCaseSensitive+0x0): multiple definition of `cJSON_ReplaceItemInObjectCaseSensitive'
.pio/build/local_32/libb41/ESPHap/cJSON.c.o:cJSON.c:(.text.cJSON_ReplaceItemInObjectCaseSensitive+0x0): first defined here
```

Is this a well known issue?

Yurik72, 2021-02-28:
Hi, I’m already notified that cJSON is part of ESP32 library 1.0.5, issued recently

This is not a problem for previous version 1.0.4, which I’m using, therefore I need properly adapt, until that you can use 1.0.4

judepereira, 2021-02-28:
Oh, I see. Thanks! I'll downgrade the lib for now.

judepereira, 2021-02-28:
Downgrading works for now. Thanks!

closed 2021-02-28 by judepereira