Yurik72/ESPHap #66: ESP32 - invalid flash format address?

judepereira, 2021-02-28:
I've got ESPHap running on my ESP32, however, when it attempts to format the flash storage in storage.c, the ESP crashes immediate:
```
>>> HomeKit: Formatting flash at 0x200000
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
```

Looking for what `InstrFetchProhibited` means, it turns out that if the code accesses an invalid address, this error is thrown: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#instrfetchprohibited

Something seems to be happening in `homekit_storage_init()` in `storage.c`, which causes it to crash immediately.

Is this a known issue?

judepereira, 2021-02-28:
Furthermore, would it be a good idea to use SPIFFS here too?

Yurik72, 2021-02-28:
Exactly , spiffs are used there is a target, if you will look a readme. Issue is unknown, I know about thousands devices running on Esp32 , including my home, send me a sketch, I will check tomorrow.

judepereira, 2021-02-28:
The sketch is unfortunately a massive project :(

However, I've made some headway in debugging this:
```
BT-1: 0x400db7c6 is in on_storage_change (/Users/jude/Documents/Arduino/libraries/ESPHap/storage.c:82).
82			callbackstorage();
BT-2: 0x400db81b is in homekit_storage_init (/Users/jude/Documents/Arduino/libraries/ESPHap/storage.c:140).
140			on_storage_change();
BT-3: 0x400db399 is in homekit_server_task (/Users/jude/Documents/Arduino/libraries/ESPHap/server.c:3517).
3517	    int r = homekit_storage_init();
BT-4: 0x40088f4d is in vPortTaskWrapper (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:143).
```

Looks like `on_storage_change` is an invalid pointer. Let me confirm that.

judepereira, 2021-02-28:
Okay, so I commented out the call to `callbackstorage()`, and I was able to initiate the setup with HomeKit this time. Everything worked until the pairing completed, and then the ESP32 crashed with:
```
CORRUPT HEAP: Bad head at 0x3ffdc0e0. Expected 0xabba1234 got 0x3ffdc298
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4013fa6b on core 0

Backtrace: 0x4008c990:0x3ffda6c0 0x4008cbc1:0x3ffda6e0 0x4013fa6b:0x3ffda700 0x4008c595:0x3ffda730 0x40084ee2:0x3ffda750 0x400852e9:0x3ffda770 0x4000bec7:0x3ffda790 0x400dbcb9:0x3ffda7b0 0x400d99a9:0x3ffda7d0 0x400db5f5:0x3ffda890 0x400d6bad:0x3ffda8b0 0x400d8e39:0x3ffda910 0x400da33b:0x3ffda940 0x400db41f:0x3ffda990 0x40088f4d:0x3ffda9b0

```

judepereira, 2021-02-28:
Here's the decoded stacktrace:
```
PC: 0x4013fa6b is in __assert_func (../../../.././newlib/libc/stdlib/assert.c:63).
BT-0: 0x4008c990 is in invoke_abort (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:155).
BT-1: 0x4008cbc1 is in abort (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:170).
BT-2: 0x4013fa6b is in __assert_func (../../../.././newlib/libc/stdlib/assert.c:63).
BT-3: 0x4008c595 is in multi_heap_free (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:214).
BT-4: 0x40084ee2 is in heap_caps_free (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:268).
BT-5: 0x400852e9 is in _free_r (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42).
BT-7: 0x400dbcb9 is in homekit_storage_pairing_free (/Users/jude/Documents/Arduino/libraries/ESPHap/storage.c:453).
453			free(pairing->device_id);
BT-8: 0x400d99a9 is in homekit_server_on_pair_verify (/Users/jude/Documents/Arduino/libraries/ESPHap/server.c:2029).
2029				homekit_storage_pairing_free(pairing);
BT-9: 0x400db5f5 is in homekit_server_on_message_complete (/Users/jude/Documents/Arduino/libraries/ESPHap/server.c:3047).
3047	            homekit_server_on_pair_verify(context, (const byte *)context->body, context->body_length);
BT-10: 0x400d6bad is in http_parser_execute (/Users/jude/Documents/Arduino/libraries/ESPHap/http_parser.c:1927).
1927	        CALLBACK_NOTIFY(message_complete);
BT-11: 0x400d8e39 is in homekit_client_process (/Users/jude/Documents/Arduino/libraries/ESPHap/server.c:3166).
3166			http_parser_execute(
BT-12: 0x400da33b is in homekit_run_server (/Users/jude/Documents/Arduino/libraries/ESPHap/server.c:3385).
3385	                    homekit_client_process(context);
BT-13: 0x400db41f is in homekit_server_task (/Users/jude/Documents/Arduino/libraries/ESPHap/server.c:3554).
3554	    homekit_run_server(server);
BT-14: 0x40088f4d is in vPortTaskWrapper (/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:143).

```

judepereira, 2021-02-28:
Looks like this is related to https://github.com/espressif/arduino-esp32/issues/3948

Yurik72, 2021-03-01:
Hi, there are two things

1.	 about the crash, during pairing verification. This is my fault on the latest commit :(. Currently I fix that and double check. It works now, just get latest commit, sorry for inconvinient. 
2.	You have a problem with  storage and callback, for 99% i know the reason. therefore asked you about sketch....
        You should double check how you how many accesories and service you add and if they less 
#define MAX_HAP_SERVICES 7
#define MAX_HAP_ACCESSORIES 7

Do not forget that in that definition max services is 6. Latest [7] is zero terminated

So if you add more, you simply destroyed callback pointer... Just try one service first and you will see

Yurik72, 2021-03-02:
Hi, any update on that ?

judepereira, 2021-03-02:
Just got around to running this. It paired just right! Thank you :)

closed 2021-03-02 by judepereira