Yurik72/ESPHap #67: Heater implementation

vladtsit, 2021-03-02:
Hi, is it possible to implement Heater accessory?
May I just correct Thermostat example?
Thanks!

Yurik72, 2021-03-02:
Examples folder contains https://github.com/Yurik72/ESPHap/tree/master/examples/EspHap_ThermostatControl
This is without final logic what is pin wiil be used and how PWM or just level. I think you can do it easily

vladtsit, 2021-03-02:
Thank you. EspHap_ThermostatControl work seamlessly, but I need only Heater, without Cooling features. I believe that appropriate device exists as HEATER_COOLER, but not supported by you. Do you have some time to add it?  I promise some good drink to you :)

vladtsit, 2021-03-02:
I even modified you code and added heater_cooler, but seems something goes wrong - device appears in Homekit but only showing temperature, hot showing controls.

homeintegration.c:
//heater service
homekit_service_t* hap_add_heater_service(const char* szname, hap_callback cb, void* context) {

	homekit_service_t*service = hap_new_heater_service(szname, cb, context);

	return hap_add_service(service);
}
homekit_service_t* hap_add_heater_service_as_accessory(int acctype, const char* szname, hap_callback cb, void* context) 
{
	homekit_service_t* baseservice = hap_new_homekit_accessory_service(szname, "0");
	homekit_service_t* thermoservice = hap_new_heater_service(szname, cb, context); 
	homekit_service_t* svc[3];
	svc[0] = baseservice;
	svc[1] = thermoservice;
	svc[2] = NULL;
	hap_accessories[hap_mainaccesories_current] = NEW_HOMEKIT_ACCESSORY(
		.category = (homekit_accessory_category_t)acctype,
		.services = svc
	);

	hap_mainaccesories_current++;
	hap_accessories[hap_mainaccesories_current] = NULL;
	//return thermoservice;
return thermoservice;
}
homekit_service_t* hap_new_heater_service(const char* szname, hap_callback cb, void* context) {
	return NEW_HOMEKIT_SERVICE(HEATER_COOLER, .characteristics=(homekit_characteristic_t*[]) {
			NEW_HOMEKIT_CHARACTERISTIC(
				ACTIVE, true,
				.callback = HOMEKIT_CHARACTERISTIC_CALLBACK(
					cb, .context = context
				)
			),
			NEW_HOMEKIT_CHARACTERISTIC(CURRENT_TEMPERATURE, 0),
			NEW_HOMEKIT_CHARACTERISTIC(CURRENT_HEATING_COOLING_STATE, 0),
			NEW_HOMEKIT_CHARACTERISTIC(TARGET_HEATING_COOLING_STATE, 0,
				.callback = HOMEKIT_CHARACTERISTIC_CALLBACK(
					cb, .context = context
				)),
			NEW_HOMEKIT_CHARACTERISTIC(NAME, szname),
			NEW_HOMEKIT_CHARACTERISTIC(TEMPERATURE_DISPLAY_UNITS, 0),

			NEW_HOMEKIT_CHARACTERISTIC(HEATING_THRESHOLD_TEMPERATURE, 0,
				.callback = HOMEKIT_CHARACTERISTIC_CALLBACK(
					cb, .context = context
				)),
			
			NULL
	});

}

Yurik72, 2021-03-02:
Very strange , those code are tested and allows control, I didn’t push examples without testing,

Send me your sketch

Yurik72, 2021-03-02:
I will check apple specification, currently do no remember

vladtsit, 2021-03-02:
your sample is working perfectly, but I need different assessory to be implemented which is Heater, not thermostat

vladtsit, 2021-03-02:
```
/**
 Defines that the accessory contains a heater and/or cooler.
 
 Accessory Category:
 - 20: Heaters
 - 21: Air Conditioners
 
 Required Characteristics:
 - ACTIVE
 - CURRENT_TEMPERATURE
 - CURRENT_HEATER_COOLER_STATE
 - TARGET_HEATER_COOLER_STATE
 
 Optional Characteristics:
 - NAME
 - ROTATION_SPEED
 - TEMPERATURE_DISPLAY_UNITS
 - SWING_MODE
 - COOLING_THRESHOLD_TEMPERATURE
 - HEATING_THRESHOLD_TEMPERATURE
 - LOCK_PHYSICAL_CONTROLS
 */
#define HOMEKIT_SERVICE_HEATER_COOLER HOMEKIT_APPLE_UUID2("BC")
```

vladtsit, 2021-03-02:
seems I did it, above *HEATING_COOLING* needs to changes to *HEATER_COOLER*

Yurik72, 2021-03-03:
The better way to add new function(s) for Heater not changing exisitng and then providing to me your example :) within pull request
I think it will be helpfull to other guys and I can review

vladtsit, 2021-03-03:
Thanks for recommendation, finally I found simpler way to reach required result by modifying library and thermostat sample. What I need - to restrict values of CURRENT_HEATING_COOLING_STATE and TARGET_HEATING_COOLING_STATE to 0 and 1. Remove treshholds and set low/high range of target temperature. I was able to do it by changing homeintegration.c and characteristics.h. Is there more accurate way to do it from arduino code without changing library code.

Yurik72, 2021-03-03:
Yes in arduino code you have pointer to characheristic and can set min/max/threshold, however you right probably I need to define a macros to easily do that

vladtsit, 2021-03-03:
thank you! Also, if possible, please allow to remove not required characteristics. For example, in my scenario I do not need thresholds.

Yurik72, 2021-03-04:
Check
https://github.com/Yurik72/ESPHap/tree/master/examples/EspHap_HeaterControl

The sketch is not finished how to proper handle devices. However is proper shows on the apple
Please make a pull request after your changes