Yurik72/ESPHap #70: Adding examples from @maximkulkin ESP-RTOS repository

eford321, 2021-03-05:
Have you converted any of the RTOS examples from @maximkulkin's repository to run on ESPHap?  

I was planning to experiment with the magic home example:
https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/magic_home_strip

Any recommendations from your previous work?

Thanks.

Yurik72, 2021-03-06:
Hi I didn’t’ check all examples from MaximKulkin , as you see 

 

https://github.com/Yurik72/ESPHap/tree/master/examples

 

contains examples for most of used devices….

 

About magic_home.  Due to my preference to play with WS2812 led strip I has implemented https://github.com/Yurik72/ESPHap/blob/master/examples/EspHap_RGB_Motion/EspHap_RGB_Motion.ino

 

Magic_home is little different and contains just 3 channel of color.

 

Using WS2812 give more effects including any firing effect ang usage as Text matrix….

 

 

In my home I have used devices which installed https://github.com/Yurik72/ESPHomeController  library

Due time limitation is not properly documented and user interface is not finished.  

But library contains possibility to control WS2812 and 3 channel RGB

 

I can make implementation of Magic Home or can help you😊 

 

https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/magic_home_strip

eford321, 2021-03-06:
Thanks for the feedback.   There's really nothing special about the Magic Home controller unless you have the RF controller.  I see your examples already support most of this, I will work to combine parts of the examples EspHap_RGB_Motion and EspHapAdvancedDimmableLed into an RGB strip sketch.

I still have 5 more blinds to make and 3 Sonoff switches to convert to ESPHap before I look at LEDs.😊

I agree on the WS2812. I will use them for new installs. I still have 3 large RGB strips installed that I will convert from MQTT to ESPHap.

closed 2021-03-06 by eford321

Yurik72, 2021-03-06:
As well please have a look esphomecontroler, I have started with that, but due to absence any good HomeKit library switched to esphap, at least try one device with the one device, it supports major basics

eford321, 2021-03-06:
I will take a look. Thanks.