Yurik72/ESPHap #71: CameraHap

xrust83, 2021-03-06:
C:\Users\xrust\w1\ESPHap\examples\ESPHAPCamera\ESPHAPCamera.ino: In function 'void setup()':
ESPHAPCamera:171:22: error: 'hap_set_max_pairing' was not declared in this scope
 hap_set_max_pairing(2);
                      ^
sketch\Motion.cpp: In function 'uint8_t send_email(uint8_t*, size_t)':
Motion.cpp:304:21: error: 'hap_setstopflag' was not declared in this scope
     hap_setstopflag();
                     ^
Несколько библиотек найдено для "FreeRTOS.h"
 Используется: C:\Users\xrust\Documents\Arduino\libraries\ESP32_BLE_Arduino
Не используется: C:\Users\xrust\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\BLE
Несколько библиотек найдено для "wolfssl.h"
 Используется: C:\Users\xrust\Documents\Arduino\libraries\wolfssl
Не используется: C:\Program Files (x86)\Arduino\libraries\wolfssl
Несколько библиотек найдено для "WiFi.h"
 Используется: C:\Users\xrust\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\WiFi
Не используется: C:\Program Files (x86)\Arduino\libraries\WiFi
Не используется: C:\Users\xrust\Documents\Arduino\libraries\WiFi
exit status 1
'hap_set_max_pairing' was not declared in this scope

Yurik72, 2021-03-06:
hap_set_max_pairing is clearly defined in homeintagration.h
Fo succeess compilation you need proper installation of ESPHap library,( check readme). Or if you don't want to use integration with Apple comment line 14 in config.h
```c
#define ENABLE_HAP   /// Define if device should support Apple Home Kit connection
```

closed 2021-03-06 by Yurik72

xrust83, 2021-03-06:
I need a Homekit
I read the description as best I could. I uninstalled version 1.0.5 and installed 1.0.4 for esp32
Copied the wolfssl 3.13.0 library from the archive, but the error does not disappear
Вы понимаете русский?

xrust83, 2021-03-06:
[image]

Yurik72, 2021-03-06:
Ah, ok Library manager doesn’t see the latest version.  You cane get latest commit manually from github , without arduino library manager

 

 <https://user-images.githubusercontent.com/58448955/110205049-06811000-7e7f-11eb-8b5b-255e5bc8fda8.png>

Yurik72, 2021-03-06:
Да

xrust83, 2021-03-06:
Я понизил версию для эксперимента. уже вернул обратно 1,0,3

Yurik72, 2021-03-06:
Надо брать 1.0.7

xrust83, 2021-03-06:
1.0.7 это для esp8266, а я пытаюсь прошить камеру плата esp32-Cam в описании сказано что 1,0,5 не работает и проверено с 1,0,4.  До прошивки еще не доходит. Базовая прошивка камеры из ардуино CameraWebServer прошивается

xrust83, 2021-03-06:
Может я с платами напутал?

// Select camera model
//#define CAMERA_MODEL_WROVER_KIT_YK
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
#define CAMERA_MODEL_AI_THINKER

камера AI THINKER ESP32 CAM

Yurik72, 2021-03-06:
Оно бы прошивалfсь но не работало бы.

Просто возьми вручную последню версию кодов библитотеки ESPHap, после этого все скомпилируеться.

xrust83, 2021-03-06:
Спасибо, получилось!!! Снес полностью Arduino IDE установил только esp32 1.0.4 и ESPHap 1.0.7. Скопировал репозиторий (git clone https://github.com/Yurik72/ESPHap/) и вручную повторно переписал файлы ESPHap. так же в директорий c:\Program Files (x86)\Arduino\libraries\ скопировал папку wolfSSL из архива, в ручную заменил файлы settings.h и user_settings.h Прошивка скомпилировалась и камера прошилась. Homekit находит охранное устройство. Какой код сопряжения или где его нужно было ввести в исходнике?

xrust83, 2021-03-06:
111-11-111 no work :-(

Yurik72, 2021-03-06:
Код сопряжения 1111 1111 ( 8 единиц)

Yurik72, 2021-03-06:
Странно , он зашит в прошивке там никаких магий, должен сопрягаться без всяких вопросов

Yurik72, 2021-03-06:
Возможно проблема wolfssl, возьми полностью мою директорию и замени, потом восстановишь , у меня уже больше 1000 пользователей, так что все должно работать