Yurik72/ESPHap #74: Momentary Button/Switch possible?

chrwh, 2021-03-07:
Hi,
Is it possible to make a momentary button? I need this to increment a value in the other end :-) and since I can't read the current value out of the thing I'm controlling, I just need to increment it, and then have another momentary button to decrement it.
Best regards,
Christian

Note. I just tested the latest release, and it seems to be much more stable now - THANK you SO MUCH :-)

Yurik72, 2021-03-08:
Hi, thanks for feedback. You mentioned more stabel. means some instability still exists ?

About your question and momentary button, I do not completelly understand your task generally.   Maybe you can explain and it will be easy to propose some solution. Any how library is limmited for Apple possibilities. All accessories and charachteristic you can see there https://developer.apple.com/support/homekit-accessory-protocol/  . Means everything which is supported by Apple can be implemented

chrwh, 2021-03-09:
It has been running for a couple of days now. Sometimes when you open the phone or ipad, is says no connection, but after a few seconds it will be ok again.
What I'm looking for is like a tactile switch but on the screen, so not an on/off switch, but a push and release switch. I have somewhat solved it, by creating an automation in the home app, so when the switch is turned on, it will be turn off again right away. By doing that I can use it to increment a variable in another system.