Yurik72/ESPHap #79: Linking error

lyuq-a, 2021-03-20:
Hello!

After update of the ESPHap library and Espressif 32 platform. Code stopped to build at the linking stage.

Here is errors from linker:

Indexing .pio/build/heltec_wifi_lora_32/lib7ea/libWiFiManager.a
Linking .pio/build/heltec_wifi_lora_32/firmware.elf
/Users/lyuq/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libwpa_supplicant.a(base64.o): In function `base64_encode':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/wpa2/utils/base64.c:31: multiple definition of `base64_encode
'
.pio/build/heltec_wifi_lora_32/libbee/ESPHap/esphap_base64.c.o:/Users/lyuq/OneDrive/Arduino/libraries/ESPHap/esphap_base64.c:39: first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/heltec_wifi_lora_32/firmware.elf] Error 1

Yurik72, 2021-03-20:
This this is well known issue. Which already fixed on the commit https://github.com/Yurik72/ESPHap/commit/785ecee43263e9fbe0746ef5ff49dcbf5c6fbebc

However I didn’t release a new version, please just latest commit and let me know

Yurik72, 2021-03-22:
fixed https://github.com/Yurik72/ESPHap/releases/tag/v1.1

closed 2021-03-22 by Yurik72

bfaliszek, 2021-03-30:
I have the same issue under ESP32 core 1.0.5 and 1.0.6 and ESPHap 1.1
`/Users/bfaliszek/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libwpa_supplicant.a(base64.o): In function `base64_encode':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/wpa2/utils/base64.c:31: multiple definition of `base64_encode'
Z`

Yurik72, 2021-03-30:
Get latest commit, I have fixed, If Yes please confirm

bfaliszek, 2021-03-30:
@Yurik72 thanks! With last commit it's work. I've tested it on Arduino esp32 core 1.0.6