Yurik72/ESPHap #80: Cannot connect to accessory from iphone

BbIKTOP, 2021-03-21:
Hi,

Found your library today and cannot get example working. Using EspHapLed8266.ino with esp12f, have following error when connecting from ios14.4.1:
```
........
IP address: 
172.16.5.249
Free heap: 31984
init_hap_storageFailed to read pair.dat
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Free heap: 30240
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x200000
>>> HomeKit: storage init 1
>>> HomeKit: if r
>>> HomeKit: generating id
>>> HomeKit: Generated new accessory ID: 98:F0:04:0B:77:CB
>>> HomeKit: Generated new accessory ID: 1073699004
>>> HomeKit:  key generated
>>> HomeKit:  key saved
>>> HomeKit: Generated new accessory key
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 23896
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: ES-CC53, IP: 172.16.5.249
>>> HomeKit: Init server over
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61130
>>> HomeKit: [Client 1073701396] Got client content from the cache 0
>>> HomeKit: [Client 1073701396] new client accepted
>>> HomeKit: [Client 1073701396] Pair Setup Step 1/3
>>> HomeKit: [Client 1073701396] Disconnected!
>>> HomeKit: [Client 1073701396] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 22272
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61131
>>> HomeKit: [Client 1073700452] Got client content from the cache 0
>>> HomeKit: [Client 1073700452] new client accepted
>>> HomeKit: [Client 1073700452] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700452] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073700452] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21864
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61132
>>> HomeKit: [Client 1073702980] Got client content from the cache 0
>>> HomeKit: [Client 1073702980] new client accepted
>>> HomeKit: [Client 1073702980] Pair Setup Step 1/3
>>> HomeKit: [Client 1073702980] Pair Setup Step 2/3
>>> HomeKit: [Client 1073702980] Computing SRP shared secret
>>> HomeKit: [Client 1073702980] Verifying peer's proof
>>> HomeKit: [Client 1073702980] Generating own proof
>>> HomeKit: [Client 1073702980] Disconnected!
>>> HomeKit: [Client 1073702980] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21736
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFi connected, ip: 172.16.5.249, mask: 255.255.0.0, gw: 172.16.1.1
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61134
>>> HomeKit: [Client 1073700452] Got client content from the cache 0
>>> HomeKit: [Client 1073700452] new client accepted
>>> HomeKit: [Client 1073700452] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700452] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073700452] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21800
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61135
>>> HomeKit: [Client 1073700452] Got client content from the cache 0
>>> HomeKit: [Client 1073700452] new client accepted
>>> HomeKit: [Client 1073700452] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700452] Pair Setup Step 2/3
>>> HomeKit: [Client 1073700452] Computing SRP shared secret
>>> HomeKit: [Client 1073700452] Verifying peer's proof
>>> HomeKit: [Client 1073700452] Generating own proof
>>> HomeKit: [Client 1073700452] Disconnected!
>>> HomeKit: [Client 1073700452] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21496
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61136
>>> HomeKit: [Client 1073699156] Got client content from the cache 0
>>> HomeKit: [Client 1073699156] new client accepted
>>> HomeKit: [Client 1073699156] Pair Setup Step 1/3
>>> HomeKit: WiFiServer receives a new client (current 1, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61137
>>> HomeKit: [Client 1073700692] Got client content from the cache 1
>>> HomeKit: [Client 1073700692] new client accepted
>>> HomeKit: [Client 1073700692] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700692] Refusing to pair: another pairing in progress
>>> HomeKit: [Client 1073699156] Disconnected!
>>> HomeKit: [Client 1073700692] Disconnected!
>>> HomeKit: [Client 1073699156] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21408
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: [Client 1073700692] Closing client connection
>>> HomeKit: [Client 0] client content free
```
It lasts for a few minutes, then home.app on iphone says "Unable to Add Accessory Accessory is not reachable". What am I doing wrong?
Btw, is it ok that 
```
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
```
Repeats each time?

BbIKTOP, 2021-03-21:
Another pairing try, restarted esp, this time "/pair.dat" exists, as it has been created at the first run:
```
.........
IP address: 
172.16.5.249
Free heap: 31984
init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Free heap: 30240
>>> HomeKit: Starting server
>>> HomeKit: storage init 0
>>> HomeKit: Using existing accessory ID: CC:53:3D:E8:D6:85
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 23896
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: ES-CC53, IP: 172.16.5.249
>>> HomeKit: Init server over
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61140
>>> HomeKit: [Client 1073701396] Got client content from the cache 0
>>> HomeKit: [Client 1073701396] new client accepted
>>> HomeKit: [Client 1073701396] Pair Setup Step 1/3
>>> HomeKit: [Client 1073701396] Disconnected!
>>> HomeKit: [Client 1073701396] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 22144
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFi connected, ip: 172.16.5.249, mask: 255.255.0.0, gw: 172.16.1.1
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61141
>>> HomeKit: [Client 1073703244] Got client content from the cache 0
>>> HomeKit: [Client 1073703244] new client accepted
>>> HomeKit: [Client 1073703244] Pair Setup Step 1/3
>>> HomeKit: [Client 1073703244] Pair Setup Step 2/3
>>> HomeKit: [Client 1073703244] Computing SRP shared secret
>>> HomeKit: [Client 1073703244] Verifying peer's proof
>>> HomeKit: [Client 1073703244] Generating own proof
>>> HomeKit: [Client 1073703244] Socket not available to write size:172,available:0
>>> HomeKit: [Client 1073703244] Attempt :0, available:0
>>> HomeKit: [Client 1073703244] Attempt :1, available:0
>>> HomeKit: [Client 1073703244] Attempt :2, available:0
>>> HomeKit: [Client 1073703244] Attempt :3, available:0
>>> HomeKit: [Client 1073703244] Attempt :4, available:0
!!! HomeKit: [Client 1073703244] socket.write, data_size=172, write_size=0
>>> HomeKit: [Client 1073703244] Disconnected!
>>> HomeKit: WiFi connected, ip: 172.16.5.249, mask: 255.255.0.0, gw: 172.16.1.1
>>> HomeKit: WiFiServer receives a new client (current 1, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61142
>>> HomeKit: [Client 1073700788] Got client content from the cache 1
>>> HomeKit: [Client 1073700788] new client accepted
>>> HomeKit: [Client 1073700788] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700788] Refusing to pair: another pairing in progress
>>> HomeKit: [Client 1073700788] Disconnected!
>>> HomeKit: [Client 1073703244] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 22464
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: [Client 1073700788] Closing client connection
>>> HomeKit: [Client 0] client content free
```

Yurik72, 2021-03-21:
Hi  I will check, but this is see first time.., hundred devices works without that

BbIKTOP, 2021-03-21:
Maybe I did something wrong, but I just copied an example, downloaded lib and followed instructions. Could you advice me please any steps to check/fix? have no idea where to start tbh

BbIKTOP, 2021-03-21:
Here, as my first try was using PIO, I made another one with freshly installed  Arduino 1.8.13:
```
..ip:172.16.5.249,mask:255.255.0.0,gw:172.16.1.1
.
IP address: 
172.16.5.249
Free heap: 34040
init_hap_storageFailed to read pair.dat
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Free heap: 32056
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x200000
>>> HomeKit: storage init 1
>>> HomeKit: if r
>>> HomeKit: generating id
>>> HomeKit: Generated new accessory ID: 01:57:E3:E4:30:5E
>>> HomeKit: Generated new accessory ID: 1073697436
>>> HomeKit:  key generated
>>> HomeKit:  key saved
>>> HomeKit: Generated new accessory key
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 25376
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
pm open,type:2 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
sleep disable
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: ES-0516, IP: 172.16.5.249
>>> HomeKit: Init server over
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61245
>>> HomeKit: [Client 1073700012] Got client content from the cache 0
>>> HomeKit: [Client 1073700012] new client accepted
>>> HomeKit: [Client 1073700012] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700012] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073700012] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 22696
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61246
>>> HomeKit: [Client 1073701508] Got client content from the cache 0
>>> HomeKit: [Client 1073701508] new client accepted
>>> HomeKit: [Client 1073701508] Pair Setup Step 1/3
>>> HomeKit: [Client 1073701508] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073701508] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 22424
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61247
>>> HomeKit: [Client 1073697588] Got client content from the cache 0
>>> HomeKit: [Client 1073697588] new client accepted
>>> HomeKit: [Client 1073697588] Pair Setup Step 1/3
>>> HomeKit: [Client 1073697588] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073697588] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 22136
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61248
>>> HomeKit: [Client 1073699204] Got client content from the cache 0
>>> HomeKit: [Client 1073699204] new client accepted
>>> HomeKit: [Client 1073699204] Pair Setup Step 1/3
>>> HomeKit: [Client 1073699204] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073699204] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21904
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61249
>>> HomeKit: [Client 1073698428] Got client content from the cache 0
>>> HomeKit: [Client 1073698428] new client accepted
>>> HomeKit: [Client 1073698428] Pair Setup Step 1/3
>>> HomeKit: [Client 1073698428] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073698428] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21720
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
```

BbIKTOP, 2021-03-21:
Looks like I found the reason:

```
>>> HomeKit: Got new client connection: local 172.16.5.249:5556, remote 172.16.5.134:61845
>>> HomeKit: [Client 1073701540] Got client content from the cache 0
>>> HomeKit: [Client 1073701540] new client accepted
>>> HomeKit: [Client 1073701540] Pair Setup Step 1/3
>>> HomeKit: [Client 1073701540] Pair Setup Step 2/3
>>> HomeKit: [Client 1073701540] Computing SRP shared secret
>>> HomeKit: [Client 1073701540] Verifying peer's proof
>>> HomeKit: [Client 1073701540] Generating own proof
state: 5 -> 2 (7c0)
rm 0
pm close 7
>>> HomeKit: [Client 1073701540] Disconnected!
reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with etogo.net, channel 6
dhcp client start...
ip:172.16.5.249,mask:255.255.0.0,gw:172.16.1.1
>>> HomeKit: WiFi connected, ip: 172.16.5.249, mask: 255.255.0.0, gw: 172.16.1.1
>>> HomeKit: [Client 1073701540] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 23240
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
pm open,type:0 0
>>> HomeKit: arduino_homekit_preinit success
```
Looks like ESP12 Wifi disconnects on heavy load. Is it possible to fix, either on my side or in the lib, like adding "delay" to make wifi stack working, or something?
Thank you!

Yurik72, 2021-03-21:
Is not possible to do something wrong, usually it works fine immediately  , analyzing your log I remember that met one time with that, problem solved after restarting router, this is definitely strange and not usual

BbIKTOP, 2021-03-21:
I've got Unifi APs setup, did restart for AP that has this ESP connected to. The same ESP12 works fine with Espressif RTOS Homekit example right now.

Yurik72, 2021-03-21:
I will check, as mentioned I met this second time from thousands

BbIKTOP, 2021-03-21:
Thank you! Please tell me, if you'd need any assistance, like testing and so on.

BbIKTOP, 2021-03-22:
Found it. Типа.
I suppose, the problem is that:
Arduino Core for ESP8266 is based on nonos sdk, so it is single threaded. During long calculations in the s_mp_exptmod, wifi timeout occurs and wifi restarts after that. The solution is to call yield(), which will sort wifi things out, I suppose. After a brief investigation, have added yield() in the integer.c line 3532:
```
  bitbuf = 0;

  for (;;) {
    yield();
    /* grab next digit as required */

```
```
$ diff integer.c integer.c.orig
3532d3531
<     yield();
```
And everything works so far. Every single connect works fine, I tried it about 20+ times. I cannot make a PR as wolfSSL is not in the repo. Maybe it will be a good idea to wrap it in the some `#ifdef` to make it called for ESP8266 only, as ESP32 is multithreaded RTOS based and does not have such a problem.

Можно ли добавить такой пачь в репу? Спасибо огромное!

BbIKTOP, 2021-03-22:
Or like that, to reduce load:
```
  bitbuf = 0;

  unsigned long yieldTime=millis();
  int cnt=0;
  for (;;) {
    if(cnt++>20) {
      unsigned long yieldNow=millis();
      if(yieldNow-yieldTime>500) {
        yieldTime=yieldNow;
        yield();
      }
      cnt=0;
    }

    /* grab next digit as required */
    if (--bitcnt == 0) {
```

Yurik72, 2021-03-22:
Ok, thanks for investigation, I defintelly not met with that before. 
I have used yield definetelly as much as needed but not in S_MP_EXPTMOD
Your fix already integrated and commited

closed 2021-03-22 by Yurik72

BbIKTOP, 2021-03-22:
`if (yieldNow - yieldTime > 500)` is too much, change to `if (yieldNow - yieldTime > 20)` please

Yurik72, 2021-03-22:
Ok, but I need to check. I have checked previous and it works....

BbIKTOP, 2021-03-22:
Конечно. Я проверил - иногда работает, иногда нет. Чаще работает, но иногда сбоит, наверное от ключа зависит. Я подумал и вообще убрал проверку времени, показалось, что от нее вреда больше, поскольку каждый раз millis() вызывается, создавая ненужную нагрузку. Оставил только if(cnt++>20), пока что "полет нормальный" (ну, не считая протечек из соседней проблемы)

mgx0, 2021-03-24:
Hello,

where can do a change? I'm experiencing the same problem.

the line `Connection: close received, client will be disconnected` comes when I press "Continue" after entering the pairing code. I have tried on NodeMCU clones, ESP-12E/F and also on [this](https://storage.googleapis.com/stateless-www-faranux-com/2016/12/e9363b86-witty-cloud-board-esp8266.jpg)

I'm currently using latest version of the library

```
IP address: 
10.0.0.235
Free heap: 34400
init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Update the CPU to run at 160MHz
>>> HomeKit: Free heap: 32632
>>> HomeKit: Starting server
>>> HomeKit: storage init 0
>>> HomeKit: Using existing accessory ID: 11:2A:28:A6:CE:A5
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 26288
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: SHT-112A, IP: 10.0.0.235
>>> HomeKit: Init server over
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 10.0.0.235:5556, remote 10.0.0.50:55386
>>> HomeKit: [Client 1073700084] Got client content from the cache 0
>>> HomeKit: [Client 1073700084] new client accepted
>>> HomeKit: [Client 1073700084] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700084] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073700084] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 23944
```

mgx0, 2021-03-24:
I'll probably open a new issue for that