Yurik72/ESPHap #81: Memory leak?

BbIKTOP, 2021-03-22:
Looks like there's a memory leak. Did some connects/disconnects to the ihone home.app, observe heap size is getting smaller. After a night working, could not pair, SRP computing failed. Did pair/remove a 5-10 times. The log is a bit large, but you could observe as free heap size is getting lower with each new pairing:
```
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 23784
>>> HomeKit: Calculating public key len=384
>>> HomeKit: .Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: ES-F184, IP: 172.16.5.158
>>> HomeKit: Init server over
Started at 160 MHz
--------------------------
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.134:53196
>>> HomeKit: [Client 1073702588] Got client content from the cache 0
>>> HomeKit: [Client 1073702588] new client accepted
>>> HomeKit: [Client 1073702588] Pair Setup Step 1/3
>>> HomeKit: [Client 1073702588] Disconnected!
>>> HomeKit: [Client 1073702588] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 22048
>>> HomeKit: Calculating public key len=384
>>> HomeKit: .Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.134:53197
>>> HomeKit: [Client 1073703276] Got client content from the cache 0
>>> HomeKit: [Client 1073703276] new client accepted
>>> HomeKit: [Client 1073703276] Pair Setup Step 1/3
>>> HomeKit: [Client 1073703276] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073703276] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21320
>>> HomeKit: Calculating public key len=384
>>> HomeKit: .Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.134:53198
>>> HomeKit: [Client 1073699268] Got client content from the cache 0
>>> HomeKit: [Client 1073699268] new client accepted
>>> HomeKit: [Client 1073699268] Pair Setup Step 1/3
>>> HomeKit: [Client 1073699268] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073699268] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21560
>>> HomeKit: Calculating public key len=384
>>> HomeKit: .Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
Exception in thread rx:

--- exit ---
Traceback (most recent call last):
  File "/Users/viktor/.platformio/penv/lib/python3.9/site-packages/serial/serialposix.py", line 575, in read
    buf = os.read(self.fd, size - len(read))
[... 934 lines cut ...]
>>> HomeKit: [Client 1073700708] Remove Pairing
>>> HomeKit: Removed pairing with BFCC8C69-4800-4E19-B084-EBE75F3F2C83
>>> HomeKit: Last admin pairing was removed, enabling pair setup
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 15176
>>> HomeKit: Calculating public key len=384
>>> HomeKit: .Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: [Client 1073703764] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: [Client 1073701356] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: [Client 1073702540] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: [Client 1073700708] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.201:59316
>>> HomeKit: [Client 1073700708] Got client content from the cache 0
>>> HomeKit: [Client 1073700708] new client accepted
>>> HomeKit: [Client 1073700708] Pair Verify Step 1/2
>>> HomeKit: Free heap: 14480
>>> HomeKit: [Client 1073700708] Disconnected!
```

Yurik72, 2021-03-22:
Yes, I should fix , this is only related to pairing, which usually once, in production , paired mode tested well and no memory leak, be sure as well that you are using latest version

BbIKTOP, 2021-03-22:
And here comes the error:
```
>>> HomeKit: [Client 1073700708] Pair Verify Step 1/2
>>> HomeKit: Free heap: 14480
>>> HomeKit: [Client 1073700708] Disconnected!
>>> HomeKit: [Client 1073700708] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.134:53332
>>> HomeKit: [Client 1073700708] Got client content from the cache 0
>>> HomeKit: [Client 1073700708] new client accepted
>>> HomeKit: [Client 1073700708] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700708] Pair Setup Step 2/3
>>> HomeKit: [Client 1073700708] Computing SRP shared secret
>>> HomeKit: [Client 1073700708] Verifying peer's proof
>>> HomeKit: [Client 1073700708] Generating own proof
>>> HomeKit: [Client 1073700708] Pair Setup Step 3/3
>>> HomeKit: Added pairing with BFCC8C69-4800-4E19-B084-EBE75F3F2C83
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 1073700708] Successfully paired
>>> HomeKit: [Client 1073700708] Disconnected!
>>> HomeKit: WiFiServer receives a new client (current 1, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.134:53333
>>> HomeKit: [Client 1073701356] Got client content from the cache 1
>>> HomeKit: [Client 1073701356] new client accepted
>>> HomeKit: [Client 1073701356] Pair Verify Step 1/2
>>> HomeKit: Free heap: 17680
>>> HomeKit: [Client 1073701356] Pair Verify Step 2/2
>>> HomeKit: [Client 1073701356] Found pairing with BFCC8C69-4800-4E19-B084-EBE75F3F2C83
>>> HomeKit: [Client 1073701356] Verification successful, secure session established
>>> HomeKit: Free heap: 17624
>>> HomeKit: [Client 1073701356] Get Accessories
>>> HomeKit: [Client 1073701356] Update Characteristics
>>> HomeKit: Free heap: 17584
>>> HomeKit: [Client 1073701356] Update Characteristics
>>> HomeKit: Free heap: 17680
>>> HomeKit: [Client 1073701356] Update Characteristics
>>> HomeKit: Free heap: 17680
>>> HomeKit: [Client 1073701356] Get Characteristics
>>> HomeKit: WiFiServer receives a new client (current 2, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.1.35:60916
>>> HomeKit: [Client 1073703764] Got client content from the cache 2
>>> HomeKit: [Client 1073703764] new client accepted
>>> HomeKit: [Client 1073703764] Pair Verify Step 1/2
>>> HomeKit: Free heap: 17064
>>> HomeKit: [Client 1073703764] Pair Verify Step 2/2
>>> HomeKit: [Client 1073703764] Found pairing with BFCC8C69-4800-4E19-B084-EBE75F3F2C83
>>> HomeKit: [Client 1073703764] Verification successful, secure session established
>>> HomeKit: Free heap: 16992
>>> HomeKit: [Client 1073703764] Get Accessories
>>> HomeKit: [Client 1073703764] Get Characteristics
>>> HomeKit: [Client 1073700708] Closing client connection
>>> HomeKit: [Client 0] client content free
>>> HomeKit: WiFiServer receives a new client (current 2, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.201:59321
>>> HomeKit: [Client 1073700708] Got client content from the cache 0
>>> HomeKit: [Client 1073700708] new client accepted
>>> HomeKit: [Client 1073700708] Pair Verify Step 1/2
>>> HomeKit: Free heap: 16880
>>> HomeKit: [Client 1073700708] Pair Verify Step 2/2
>>> HomeKit: [Client 1073700708] Found pairing with BFCC8C69-4800-4E19-B084-EBE75F3F2C83
>>> HomeKit: [Client 1073700708] Verification successful, secure session established
>>> HomeKit: Free heap: 17008
>>> HomeKit: WiFiServer receives a new client (current 3, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.136:54316
>>> HomeKit: [Client 1073702540] Got client content from the cache 3
>>> HomeKit: [Client 1073702540] new client accepted
>>> HomeKit: [Client 1073702540] Pair Verify Step 1/2
>>> HomeKit: Free heap: 16800
>>> HomeKit: [Client 1073700708] Get Accessories
[... 339 lines cut ...]
>>> HomeKit: Free heap: 11704
>>> HomeKit: Calculating public key len=384
>>> HomeKit: .Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 172.16.5.158:5556, remote 172.16.5.134:53545
>>> HomeKit: [Client 1073700988] Got client content from the cache 0
>>> HomeKit: [Client 1073700988] new client accepted
>>> HomeKit: [Client 1073700988] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700988] Pair Setup Step 2/3
>>> HomeKit: [Client 1073700988] Computing SRP shared secret
!!! HomeKit: [Client 1073700988] Failed to compute SRP shared secret (code -2)
>>> HomeKit: [Client 1073700988] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700988] Disconnected!
>>> HomeKit: [Client 1073700988] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 12264
>>> HomeKit: Calculating public key len=384
>>> HomeKit: .Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
```

BbIKTOP, 2021-03-22:
Cloned latest from this repo yesterday

Yurik72, 2021-03-22:
Yes I see , but this is during pairing/removing process, my last focus is to stabilize production, so, if you see any issues after pairing ? Pairing process I will improve.

BbIKTOP, 2021-03-22:
Dunno really, left it working for whole night and couldn’t reconnect at the morning, but it might be due to the leaked memory before that. I’d like to help you to find a bug but need your assistance as i am not familiar with the code. Do you need assistance? I could help. Я правильно понимаю твое имя здесь, ты говоришь по-русски?

Yurik72, 2021-03-22:
Да по русски говорю :) 
Можно общаться :) дело в том что с одним американцем вычистили недавно он реально жестко тестировал смотри #41 , и у меня больше не повторялось , плюс один итальянец два дня тестировал esp32 , 40 устроуйств на одной esp, и тоже график памяти стабильный, поэтому хотел бы понять какая последовательность и сценарий , конечно хочу довести это ,

Yurik72, 2021-03-22:
Ну и если по ходу что то увидишь сообщай, как понимаешь это хобби, но у меня уже дома до 10 устройств работает, месяц, с последнего фикса, поэтому если видишь давай найдём

Yurik72, 2021-03-22:
У меня такое ощущение что старая версия, возьми просто последнюю , прямо с гитхаб

BbIKTOP, 2021-03-22:
Я прямо с гитхапа и брал вчера, я же говорю, так что код у меня последний который ты выкладывал здесь. Хобби - у меня тоже, даже скорее, как сейчас модно говорить, прокрастинация))
Я просто товарищу пообещал сделать устройство, ему нужно мерить уровень воды в баке, включат. Насосы и еще по мелочи, начал искать интеграцию с хомякитом, пока твоя либа лучшая, еспресиф на ртосе, лень переписывать, поэтому у меня выхода нет - готов помочь с поиском ошибок. Я начал смотреть код, не очень понимаю некоторые моменты, если бы ты мне помог объяснениями в виде ответов на мои вопросы, дело пошло бы быстрее. Я вижу у тебя встречаются #ifdef, они реально используются, или это какие-то остатки? С тобой есть какие-то каналы связи помимо переписки тут? Я хочу вставить отладочный менеджер памяти и увидеть где протечка для начала

BbIKTOP, 2021-03-22:
Или вот еще вчера общался со старшими пацанами, авторами esp8266 arduino core, они спрашивают, а почему волк, а не  медведь? Медведь, говорят, в комплекте, уже есть в arduino core. Волк действительно шустрее?

Yurik72, 2021-03-22:
Да можно Вайбер или телеграмм , но не могу понять где и какая последовательность в утечке памяти , какой пример ? Делал точно тысячи коннектор дисконектов и память стабильная , поэтому давай просто повтор  И напиши посл действий там решим

BbIKTOP, 2021-03-22:
Дай пожалуйста твой вибер или телеграм, чтобы тут спам не разводить?
Последовательность моих действий была простая: скачал `git clone https://github.com/Yurik72/ESPHap/` в библиотеки pio, взял твой пример EspHapLed8266, откомпилировал и запустил. После этого начал соединяться и отсоединяться с телефона, иногда помигивая светодиодом с того же телефона. Выше полный лог, правда, он большой, но я не знаю что важно, а что нет. В конце лога видно, что память закончилась и SRP не отрабатывает, потому что crypto_srp_compute_key отваливается с -2. Я пытаюсь разобраться, но вижу много непонятного, например вот два server.c: просто server.c и serverA8266.c. Они оба используются?

Я, собственно, что хотел сделать - поотлаживать выделения памяти, как я это делал когда-то в детстве, когда для этого не было модных либ и тулзей, например заменить все выделения памяти на макросы:
 
```
#define MEMDEBUG

#ifdef MEMDEBUG

#define DBG_MALLOC(size) debugMalloc(__FILE__, __LINE__, size)
#define DBG_FREE(ptr) debugFree(__FILE__, __LINE__, ptr)

struct memChunk
{
    char *file;
    int line;
    int size;
    void *ptr;
    unsigned long time;
} memChunks[1024];

int memChunksSize = 0;

void *debugMalloc(char *file, int line, int size)
{
    void *ptr = malloc(size);
    if (ptr == NULL)
    {
        // Allocation error
        return (NULL);
    }
    memChunks[memChunksSize].file = file;
    memChunks[memChunksSize].line = line;
    memChunks[memChunksSize].size = size;
    memChunks[memChunksSize].ptr = ptr;
    memChunks[memChunksSize].time = millis();
    memChunksSize++;

    return (ptr);
}

void debugFree(char *file, int line, void *ptr)
{
    int i;
    for (i = 0; i < memChunksSize && memChunks[i].ptr != ptr; i++);
    if (i >= memChunksSize)
    {
        // Not found, an attempt to free non allocated memory detected
        // Raise error here
        free(ptr);
        return;
    }

    // Everything is ok, found allocated chunk
    // Remove freed entry
    memcpy(&memChunks[i], &memChunks[memChunksSize - 1], sizeof(memChunks[i]));
    memChunksSize--;
    free(ptr);
}

void debugDumpAllocatedMemory(void)
{
    Serial.printf("Allocated memory dump start:\n");
    for (int i = 0; i < memChunksSize; i++)
    {
        Serial.printf("%d) Allocated at %lu, file [%s], line [%d], size %lu (0x%lX)\n",
                      i,
                      memChunks[i].time,
                      memChunks[i].file,
                      memChunks[i].line,
                      memChunks[i].size,
                      memChunks[i].ptr);
    }
    Serial.printf("Allocated memory dump end\n");
}

#else

#define DBG_MALLOC(size) malloc(size)
#define DBG_FREE(ptr) free(ptr)

#endif
```

Я не проверял, просто накидал как пример, но идея, думаю, понятна - меняешь все malloc и free в коде на DBG_MALLOC и DBG_FREE, для отладки делаешь `#define MEMDEBUG`, а для прода `#undef MEMDEBUG`. В loop() делаешь вызов `debugDumpAllocatedMemory`, например по кнопке прерыванием ставишь флаг и емли в лупе он взведен, вызываешь `debugDumpAllocatedMemory`. Ну,или просто вызываешь `debugDumpAllocatedMemory` каждые 5 минут. В результате в табличке увидишь, какая память давно висит и не освободилась, где выделена, когда и сколько. Если нужно, аналогичным образом делаешь debugCalloc, debugRealloc и все остальное, что там еще есть на эту тему.

Так вот, я хотел это сделать, и обнаружил, что у тебя есть какие-то твои аллокаторы, например tlv_malloc, потому хотел проконсультироваться что у тебя делает что, да и вообще предложить помощь почистить код, все же коллективно делать работу хоть немного да легче - вдруг чем-то смогу помочь?

Yurik72, 2021-03-23:
Вайбер или телеграм +380503121075

Я просто в твои  логах увидел, что то типа

>>> HomeKit: [Client 1073700708] Pair Setup Step 1/3

>>> HomeKit: [Client 1073700708] Pair Setup Step 2/3

>>> HomeKit: [Client 1073700708] Computing SRP shared secret

!!! HomeKit: [Client 1073700708] Failed to compute SRP shared secret (code -2)

>>> HomeKit: [Client 1073700708] Pair Setup Step 1/3

 

 

Это означает идет процесс нового Pairing. Возможно там остались leak, но это не важно т.к. операция одноразовая

А вот после него все достаточно хорошо вычещено, я тебе присылал ссылку на  #41 (ESP8266) 

 

А вот один итальянец тестировал. Сразу 40 устройств. Поєтому вначале я хотел понять цепочку, когда начинает уходить память

 

 

 <https://user-images.githubusercontent.com/747087/109708494-c4fd1600-7b9b-11eb-8c9d-30ed5082cd62.png>

BbIKTOP, 2021-03-23:
написал в вибер, удали тут, чтоб не светить номер, спасибо!

Yurik72, 2021-03-23:
I will keep issue opened, Memory leaks are observed only in pairing mode and doesn't affects production

rmaksim, 2021-11-22:
ребята привет!
вы решили данную ошибку?
```
.platformio/penv/lib/python3.9/site-packages/serial/serialposix.py", line 575, in read
    buf = os.read(self.fd, size - len(read))
```

**если вам как-то поможет** - то у меня тоже появилась данная ошибка...
пришло пару плат - решил проверить как работают - залил мелкую прошивку (небольшой webserver-ок)
и...
на первой же плате все время пропадал wifi и перегружался esp8266 (nodemcu v3)
взял другую плату - все работает как часы!!!