Yurik72/ESPHap #82: collect2: error: ld returned 1 exit status

mgx0, 2021-03-23:
Hi,

having issues compiling EspHapLed.ino on ESP32, error is:

```
Building in release mode
Linking .pio/build/esp32doit-devkit-v1/firmware.elf
/Users/mgtn/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libwpa_supplicant.a(base64.o): In function `base64_encode':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/wpa2/utils/base64.c:31: multiple definition of `base64_encode'
.pio/build/esp32doit-devkit-v1/libddc/ESPHap/esphap_base64.c.o:/Users/mgtn/Library/Mobile Documents/com~apple~CloudDocs/Documents/IoT/PlatformIO/Projects/HomeKitTest4_32/lib/ESPHap/esphap_base64.c:39: first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32doit-devkit-v1/firmware.elf] Error 1
```

library latest as today: https://github.com/Yurik72/ESPHap/releases/tag/v1.1
MacOS 10.14.6, VSCode + PlatformIO

Espressif32 latest as today, **pio update** output below:
```
Updating platformio/contrib-piohome           3.3.4 @ ~3.3.4                     [Up-to-date]
Updating platformio/contrib-pysite            2.38.201019 @ ~2.38.0              [Incompatible 2.39.201019]
Updating platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]
Updating platformio/tool-scons                4.40100.2 @ ~4.40100.2             [Up-to-date]

Platform Manager
================
Platform atmelavr
--------
Updating platformio/atmelavr                  3.2.0                              [Up-to-date]
Updating platformio/toolchain-atmelavr        1.70300.191015 @ ~1.70300.0        [Up-to-date]
Updating platformio/framework-arduino-avr     5.1.0 @ ~5.1.0                     [Up-to-date]
Updating platformio/tool-avrdude              1.60300.200527 @ ~1.60300.0        [Up-to-date]

Platform espressif32
--------
Updating platformio/espressif32               3.1.1                              [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.97 @ ~2.50200.0            [Up-to-date]
Updating platformio/framework-arduinoespressif32 3.10005.210308 @ ~3.10005.0        [Up-to-date]
Updating platformio/tool-esptoolpy            1.30000.201119 @ ~1.30000.0        [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]

Platform espressif8266
--------
Updating platformio/espressif8266             2.6.3                              [Up-to-date]
Updating platformio/toolchain-xtensa          2.40802.200502 @ ~2.40802.0        [Up-to-date]
Updating platformio/framework-arduinoespressif8266 3.20704.0 @ ~3.20704.0             [Up-to-date]
Updating platformio/framework-esp8266-rtos-sdk 1.5.0-beta.5 @ >=1.5.0-beta        [Up-to-date]
Updating platformio/tool-esptool              1.413.0 @ <2                       [Up-to-date]
Updating platformio/tool-esptoolpy            1.30000.201119 @ ~1.30000.0        [Up-to-date]
Updating platformio/tool-mkspiffs             1.200.0 @ ~1.200.0                 [Up-to-date]
Updating platformio/tool-mklittlefs           1.203.210203 @ ~1.203.0            [Up-to-date]
```

is there anything I can do to troubleshoot this? I've spent hours ...

funny is that I have the same sketch running on an ESP32 on the table, compiled yesterday

Yurik72, 2021-03-23:
I have fixed, please check with latest commit. If yes, please close an issue

mgx0, 2021-03-23:
that's what I call THE RESPONSE! thanks a lot!

closed 2021-03-23 by mgx0