Yurik72/ESPHap #83: Cannot pair with esp8266

mgx0, 2021-03-24:
Hello,

where can do a change? I'm experiencing pairing problem on 8266, EspHapLed example

the line `Connection: close received, client will be disconnected` comes when right after I press "Continue" after entering the pairing code. I have tried on NodeMCU clones, ESP-12E/F

I'm currently using latest version of the library. is there any debug I can enable to provide more information?

thanks

```
....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with S2, channel 11
dhcp client start...
....ip:10.0.0.235,mask:255.255.255.0,gw:10.0.0.1
.
IP address: 
10.0.0.235
Free heap: 34144
init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Update the CPU to run at 160MHz
>>> HomeKit: Free heap: 32376
>>> server_new: WiFiServer begin at port: 5556

>>> HomeKit: Starting server
>>> HomeKit: storage init 0
>>> HomeKit: Using existing accessory ID: 11:2A:28:A6:CE:A5
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> crypto_srp_new: crypto_srp_new
>>> crypto_srp_new: Initializing SRP
>>> arduino_homekit_preinit: Free heap: 27744
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 26032
>>> HomeKit: Calculating public key len=384
pm open,type:2 0
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
sleep disable
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: SHT-112A, IP: 10.0.0.235
>>> homekit_mdns_init: Accessory Setup ID = YK72
>>> HomeKit: Init server over
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 10.0.0.235:5556, remote 10.0.0.50:55617
>>> HomeKit: [Client 1073700340] Got client content from the cache 0
>>> HomeKit: [Client 1073700340] new client accepted
>>> homekit_client_process: [Client 1073700340] Reading 126 bytes
>>> homekit_server_on_message_complete: http_parser message_complete
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 23400
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1073700340] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 23328
>>> send_tlv_response: [Client 1073700340] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \xBE\xC6\xD5<\x16KR'1\xF4\x09.\x01\xF1\x8B/pl\xE4\x88[\x9F\x85\x80\x06\xE0\x18k\x92/\xE6h-\xCE|\xF2\xD1\x9F\xB2f\xB8\x9B{\xF9$\xDC\xA3>R\xA7\x9Efn\x14\x99_\x1F\x90\xB3\x13o\x05\x97\xEDxv\xE6Ew\xB4\x17\xAA|\xB8Z\xA3!\x0E\xE0y\x886\x95\xB3m\x00)\x96\xFB\x83tf\xCF\x8AQ\x8AJZ\x14\xFFU6U3\xDB\xC9\xA8\x0A\xDCN\x1F\x1B\x09>SY\xF2\x9D\xE1#J\x82c\x9EJ\xC3m4\xD3k\x92}tPa/\xC3z.\xF0r\xE3Ao2`\x187\xAB\xBD\xCE\xE6\xDC<\xF1xG\xFBr`\xAF6\xC5\xEE\xED\x90Q\x96x)wV\xC7\x92)\xEFm%\xAA\x90\xF8[\x9F\xD2:\xDD\xCC\xC6!\x80\xF5\x1B\xAC\x1F/}\xB6\x81\x0F\xAB'>\xC5\x9AH\x0BO\xE8\xCE)L\x92\xC4\x0B\xCC\xEFk\xA0\xFE:\xCC\xB0\x09R\x89\xF4\x8C\x84e[y<\xC3\x99\xA2\xBFM\xCE7XGO\xA3\x81\xB0\xAFx\xD6\xFB\xA2\x0F1n\xA6\xE5\x88\xCA\\%\xA25\xE9\x98\xE0b}\xA9\x13\xD7\x0E\xC6\xE7<0Y\xD3\xBF\x04.(\xD4\xD3\xB1U\x10\xEAm\xC8u\xCF\x92\x11:\xE2\xF2\xE0\xA5\x97\x15p\xBE\xB2\x95$\x85M\xB4\x980\x18P\x84"\x1C\xA6\x98\xE8,\xCB\x0FU\xCE\xC2};\xBBx>\x84k\xBDM\xD3j\x1C\xCB)s\x8Cr\x91\xFC\x80Yyq\xA0\x19\xC5%\x83[A\x12\x0Fr/\x9EOk\xB3\x19$}\xF6\xF7'\xD1X\xDD4\xDA7W-Ggq\x8C\x87\x06\xB1\xBC\x98
>>> tlv_debug: Type 2 value (16 bytes): e\xAD\xB0Kh\xF9\x98\x1C\x9B\x13\xE8_\xE9A\xE6\xB4
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 1073700340] send data size=513, encrypted=false
>>> client_send: [Client 1073700340] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\xBE\xC6\xD5<\x16KR'1\xF4\x09.\x01\xF1\x8B/pl\xE4\x88[\x9F\x85\x80\x06\xE0\x18k\x92/\xE6h-\xCE|\xF2\xD1\x9F\xB2f\xB8\x9B{\xF9$\xDC\xA3>R\xA7\x9Efn\x14\x99_\x1F\x90\xB3\x13o\x05\x97\xEDxv\xE6Ew\xB4\x17\xAA|\xB8Z\xA3!\x0E\xE0y\x886\x95\xB3m\x00)\x96\xFB\x83tf\xCF\x8AQ\x8AJZ\x14\xFFU6U3\xDB\xC9\xA8\x0A\xDCN\x1F\x1B\x09>SY\xF2\x9D\xE1#J\x82c\x9EJ\xC3m4\xD3k\x92}tPa/\xC3z.\xF0r\xE3Ao2`\x187\xAB\xBD\xCE\xE6\xDC<\xF1xG\xFBr`\xAF6\xC5\xEE\xED\x90Q\x96x)wV\xC7\x92)\xEFm%\xAA\x90\xF8[\x9F\xD2:\xDD\xCC\xC6!\x80\xF5\x1B\xAC\x1F/}\xB6\x81\x0F\xAB'>\xC5\x9AH\x0BO\xE8\xCE)L\x92\xC4\x0B\xCC\xEFk\xA0\xFE:\xCC\xB0\x09R\x89\xF4\x8C\x84e[y<\xC3\x99\xA2\xBFM\xCE7XGO\xA3\x81\xB0\xAFx\xD6\xFB\xA2\x0F1n\xA6\xE5\x03\x81\x88\xCA\\%\xA25\xE9\x98\xE0b}\xA9\x13\xD7\x0E\xC6\xE7<0Y\xD3\xBF\x04.(\xD4\xD3\xB1U\x10\xEAm\xC8u\xCF\x92\x11:\xE2\xF2\xE0\xA5\x97\x15p\xBE\xB2\x95$\x85M\xB4\x980\x18P\x84"\x1C\xA6\x98\xE8,\xCB\x0FU\xCE\xC2};\xBBx>\x84k\xBDM\xD3j\x1C\xCB)s\x8Cr\x91\xFC\x80Yyq\xA0\x19\xC5%\x83[A\x12\x0Fr/\x9EOk\xB3\x19$}\xF6\xF7'\xD1X\xDD4\xDA7W-Ggq\x8C\x87\x06\xB1\xBC\x98\x02\x10e\xAD\xB0Kh\xF9\x98\x1C\x9B\x13\xE8_\xE9A\xE6\xB4\x06\x01\x02
>>> homekit_server_on_pair_setup: [Client 1073700340] Server on pair setup end
>>> homekit_client_process: [Client 1073700340] Reading 536 bytes
>>> homekit_client_process: [Client 1073700340] Reading 43 bytes
>>> HomeKit: [Client 1073700340] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073700340] Closing client connection
>>> homekit_server_close_client: [Client 1073700340] The sockect is stoped
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> crypto_srp_new: crypto_srp_new
>>> crypto_srp_new: Initializing SRP
>>> arduino_homekit_preinit: Free heap: 25400
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 23688
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
```

Yurik72, 2021-03-24:
Lets's try to update wolfssl , with latest version from archive, than let me know

mgx0, 2021-03-24:
you mean zip archive from ESPHap library? there are two zip files
or do you mean download newest from wolfssl github?

Yurik72, 2021-03-24:
Yes zip archive

Yurik72, 2021-03-24:
Use https://github.com/Yurik72/ESPHap/blob/master/wolfssl/wolfSSL_3_13_0.rar

mgx0, 2021-03-24:
nope, still doesn't pair, connection is closed as soon I confirm the code

```scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with S2, channel 11
dhcp client start...
...ip:10.0.0.235,mask:255.255.255.0,gw:10.0.0.1
.
IP address: 
10.0.0.235
Free heap: 34144
init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Update the CPU to run at 160MHz
>>> HomeKit: Free heap: 32376
>>> server_new: WiFiServer begin at port: 5556

>>> HomeKit: Starting server
>>> HomeKit: storage init 0
>>> HomeKit: Using existing accessory ID: 11:2A:28:A6:CE:A5
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> crypto_srp_new: crypto_srp_new
>>> crypto_srp_new: Initializing SRP
>>> arduino_homekit_preinit: Free heap: 27744
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 26032
>>> HomeKit: Calculating public key len=384
pm open,type:2 0
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
sleep disable
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: SHT-112A, IP: 10.0.0.235
>>> homekit_mdns_init: Accessory Setup ID = YK72
>>> HomeKit: Init server over
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 10.0.0.235:5556, remote 10.0.0.213:56222
>>> HomeKit: [Client 1073700148] Got client content from the cache 0
>>> HomeKit: [Client 1073700148] new client accepted
>>> homekit_client_process: [Client 1073700148] Reading 126 bytes
>>> homekit_server_on_message_complete: http_parser message_complete
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 23776
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1073700148] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 23704
>>> send_tlv_response: [Client 1073700148] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): x\xF3\xB6\x1FPT6\xFA␡$\x89\xD3\x91\x95k\xCF\xED\xD3^N\x09` \xBD\x10\xB4\x0As\xAA|\xBB2r\xFD\x02\xCEtC\x9AK\x83\xB0\xCF\xE1\x91_\xB1\x06u_\xC7H\x0F\xFDo\x93R\x112\x01\x81`\xA8V\x0C\x8A_Y\x02v\x97V\xDB\x194o\xA8\xF0\xF3\x08\xB0\\\x9BV\xC0\xC2}\xE0\x8F\x1D\xE6\x84\xBB\xAF\xAB3e#\xDC\x8D&\xC4i<\xD1\x94|\xF8Z\xF62;\xE7`+\xD6\x1F\x01\xF6/+\x9B\xEB[\xDA\xAA\xB1~\xDFD(9\x01\xB6J\xA3\xB6m5A\xC1(\xD7\xE5\xEE\xF9\xBAc\xF2\x1D+\xFF\xD6\x98\xC5\xFE\x8A\xFD\xF5\x05\xB4\xDB[a\x0B\xD5\x86Z]\xD9\x9B\x10\xEB;\x88\x9FT\x95p\xC9\x92\x142\xD1\x85\xF1\xBF\x1BM'4\\e\xDEeJ\x01\x04%\xAC(\x16\x1F\xBC\xC00{\xAE\x00\x8C\xB8\x8B\x95\xC9\x17\x8A\x04Ykx\xEC(7\xC8\xB4Rz\xA8\xAE\xAE\xEA\xB8\xA7\xA7N\xB6&\xAA^[\xA3\xC6?\x17\xEAc_,\xA83\x0C\xC1\x95^q\xEB\xA6\x88\x0E\x8E'~\xE1\x8F\x9B\xFB~cx*#&\xF2\xDF\xD5tp\x05\xB9G\xBBj\xA3qq\xDB\xDA\x13\xFF\xAD\x06\x875\xD0\xD7P\xAB@o\x1F\x16M\xB2\x087\xDF\x90-\x1F}\x01\xF7\xB0\xEC\xEB\xC6lq\xA7\\\xA7=Fk\x0C\x09\x9Bs\x96\x93N\xA3\x09\xBB\xAD\xF8b\x95s\xD5\x18\xD80\\\xCE\xE1\xA5-}\xF6\xD5u\xA76\xEEmq=\xA8\xB7\x08&\xAA\xB4[\x037\xE0\x08\xA8 .\x14j\x99\xC7:]~@\xBA\x97R]
>>> tlv_debug: Type 2 value (16 bytes): \xB0kz\xFE=y2ET\x94\x85-\x18\x9DO\xE0
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 1073700148] send data size=513, encrypted=false
>>> client_send: [Client 1073700148] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFFx\xF3\xB6\x1FPT6\xFA␡$\x89\xD3\x91\x95k\xCF\xED\xD3^N\x09` \xBD\x10\xB4\x0As\xAA|\xBB2r\xFD\x02\xCEtC\x9AK\x83\xB0\xCF\xE1\x91_\xB1\x06u_\xC7H\x0F\xFDo\x93R\x112\x01\x81`\xA8V\x0C\x8A_Y\x02v\x97V\xDB\x194o\xA8\xF0\xF3\x08\xB0\\\x9BV\xC0\xC2}\xE0\x8F\x1D\xE6\x84\xBB\xAF\xAB3e#\xDC\x8D&\xC4i<\xD1\x94|\xF8Z\xF62;\xE7`+\xD6\x1F\x01\xF6/+\x9B\xEB[\xDA\xAA\xB1~\xDFD(9\x01\xB6J\xA3\xB6m5A\xC1(\xD7\xE5\xEE\xF9\xBAc\xF2\x1D+\xFF\xD6\x98\xC5\xFE\x8A\xFD\xF5\x05\xB4\xDB[a\x0B\xD5\x86Z]\xD9\x9B\x10\xEB;\x88\x9FT\x95p\xC9\x92\x142\xD1\x85\xF1\xBF\x1BM'4\\e\xDEeJ\x01\x04%\xAC(\x16\x1F\xBC\xC00{\xAE\x00\x8C\xB8\x8B\x95\xC9\x17\x8A\x04Ykx\xEC(7\xC8\xB4Rz\xA8\xAE\xAE\xEA\xB8\xA7\xA7N\xB6&\xAA^[\xA3\xC6?\x17\xEAc_,\xA83\x0C\xC1\x95^q\x03\x81\xEB\xA6\x88\x0E\x8E'~\xE1\x8F\x9B\xFB~cx*#&\xF2\xDF\xD5tp\x05\xB9G\xBBj\xA3qq\xDB\xDA\x13\xFF\xAD\x06\x875\xD0\xD7P\xAB@o\x1F\x16M\xB2\x087\xDF\x90-\x1F}\x01\xF7\xB0\xEC\xEB\xC6lq\xA7\\\xA7=Fk\x0C\x09\x9Bs\x96\x93N\xA3\x09\xBB\xAD\xF8b\x95s\xD5\x18\xD80\\\xCE\xE1\xA5-}\xF6\xD5u\xA76\xEEmq=\xA8\xB7\x08&\xAA\xB4[\x037\xE0\x08\xA8 .\x14j\x99\xC7:]~@\xBA\x97R]\x02\x10\xB0kz\xFE=y2ET\x94\x85-\x18\x9DO\xE0\x06\x01\x02
>>> homekit_server_on_pair_setup: [Client 1073700148] Server on pair setup end
>>> homekit_client_process: [Client 1073700148] Reading 536 bytes
>>> homekit_client_process: [Client 1073700148] Reading 43 bytes
>>> HomeKit: [Client 1073700148] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073700148] Closing client connection
>>> homekit_server_close_client: [Client 1073700148] The sockect is stoped
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> crypto_srp_new: crypto_srp_new
>>> crypto_srp_new: Initializing SRP
>>> arduino_homekit_preinit: Free heap: 25400
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 23688
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
```

Yurik72, 2021-03-24:
Yes I see a log will try to investigate, this is first time I’m met

 

>>> HomeKit: [Client 1073700148] Connection: close received, client will be disconnected

 

 

Apple sends response with Connection:Close header. However it should continue to asking available accessories….

BbIKTOP, 2021-03-24:
Just wait a bit more, phone will try to connect again later. Sometimes it connects on the first attempt, sometimes on the second or third. Also, please make sure you have esp set to frequency 160 and have updated wolfssl

Yurik72, 2021-03-24:
Frequency is set 160 by library.  

Maybe yes try wait for the second attempts. Which IOS you are using

mgx0, 2021-03-24:
14.4.1 but it was like this on okder versions too. I'll keep trying

Yurik72, 2021-03-24:
Ok, suggestion from Victor.

Restart HomeApp on phone

BbIKTOP, 2021-03-24:
I have first connect _always_ failed like that. Then, there's a second one, which is about 90% successful. Sometimes 3rd or 4th, on 1% of attempts, I have to relaunch home.app and then it works.

mgx0, 2021-03-24:
and for ESP8266 I should use wolfSSL_3_13_0.rar which is included in the library, is that correct?

Yurik72, 2021-03-24:
Yes wolfSSL_3_13_0.rar works for both ESP8266/ESP32

mgx0, 2021-03-24:
good, perfect. awesome job man. And Viktor thank you too 😁 this is the future!

Yurik72, 2021-03-31:
Closed, since is not library problem

closed 2021-03-31 by Yurik72

alisadeghy, 2021-09-05:
I'm encountering the same problem and I did whatever you've discussed about in this issue but it did not fixed.
I should try like a million times to pair one time! Help! :(

Yurik72, 2021-09-05:
Hi , first of all, please be 100% sure , that you are using latest version, then attach logs, 
Problem is solved 1 year ago, therefore I do not see any obstacles

alisadeghy, 2021-09-05:
I'm pretty much sure about the libraries.
ESPHap: 1.1.0
ESP8266: 3.0.2

Log:
.......
IP address: 
192.168.1.100
Free heap: 35424
init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Update the CPU to run at 160MHz
>>> HomeKit: Free heap: 33680
>>> HomeKit: Starting server
>>> HomeKit: storage init 0
>>> HomeKit: Using existing accessory ID: E8:FF:B5:39:9A:E4
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 27336
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: ES-E8FF, IP: 192.168.1.100
>>> HomeKit: Init server over
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 192.168.1.100:5556, remote 192.168.1.101:53203
>>> HomeKit: [Client 1073699100] Got client content from the cache 0
>>> HomeKit: [Client 1073699100] new client accepted
>>> HomeKit: [Client 1073699100] Pair Setup Step 1/3
>>> HomeKit: [Client 1073699100] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073699100] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 25552
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 192.168.1.100:5556, remote 192.168.1.101:53204
>>> HomeKit: [Client 1073700100] Got client content from the cache 0
>>> HomeKit: [Client 1073700100] new client accepted
>>> HomeKit: [Client 1073700100] Pair Setup Step 1/3
>>> HomeKit: [Client 1073700100] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073700100] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 25064

and over and over again

alisadeghy, 2021-09-05:
I also updated the CPU to run at 160 MHz as well but no luck.

Yurik72, 2021-09-05:
Hi you don’t, need to update CPU clock, it’s done automatically insight library, ok
Could you please copy latest version 1.1 from the master, without updating from arduino, since latest version  more then 6 month not any blame about pairing on esp8266, 
So delete library manually from your library folder and copy directly from the master

alisadeghy, 2021-09-05:
Copied from master, but still the same log.

Yurik72, 2021-09-05:
Ok , looks like not version issue Apple closes connection after first step, could you send me your sketch

alisadeghy, 2021-09-05:
I use sketches from library example, didn't change anything except SSID and password.

`/*
  ESPHap example EspHapLed for ESP8266

  This example for ESPHap shows how to control a LED
  with Apple Home app. It implements accessory type "light bulb".

  This example code is part of the public domain
*/

#include <Arduino.h>

#ifdef ESP32
#include <SPIFFS.h>
#endif
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

#include "coredecls.h"
const char* ssid     = "*******";
const char* password = "***********";

const int identity_led=2;
const int led_gpio = 4;

extern "C"{
#include "homeintegration.h"
}
#include "homekitintegrationcpp.h"

homekit_service_t* hapservice={0};
String pair_file_name="/pair.dat";
 //esp8266::MDNSImplementation::MDNSResponder* MDNS1;   

LOCAL os_timer_t callbacktimer;

void setup() {
  disable_extra4k_at_link_time();
  Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network
#ifdef ESP32
     if (!SPIFFS.begin(true)) {
      // Serial.print("SPIFFS Mount failed");
     }
#endif
#ifdef ESP8266
     if (!SPIFFS.begin()) {
      Serial.print("SPIFFS Mount failed");
     }
#endif

    Serial.println(ssid);
#ifdef ESP8266
  WiFi.mode(WIFI_STA);
    WiFi.begin((char*)ssid, (char*)password);
#else
    WiFi.begin(ssid, password);
#endif
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
   // Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

   pinMode(led_gpio,OUTPUT);
///setup identity gpio
   // hap_set_identity_gpio(identity_led);    //identity_led=2 will blink on identity

/// now will setup homekit device

    //this is for custom storaage usage
    // In given example we are using \pair.dat   file in our spiffs system
    //see implementation below
    Serial.print("Free heap: ");
    Serial.println(system_get_free_heap_size());

    init_hap_storage();
  
    set_callback_storage_change(storage_changed);

    /// We will use for this example only one accessory (possible to use a several on the same esp)
    //Our accessory type is light bulb, Apple interface will proper show that
    hap_setbase_accessorytype(homekit_accessory_category_lightbulb);

    /// init base properties
    hap_initbase_accessory_service("ES","Yurik72","0","EspHapLed","1.0");

   //we will add only one light bulb service and keep pointer for nest using
    hapservice= hap_add_lightbulb_service("Led",led_callback,(void*)&led_gpio);
   
   //and finally init HAP
   
    hap_init_homekit_server();
}

void loop() {
 
  hap_homekit_loop();
  //delay(2);
  return;
}

void init_hap_storage(){
  Serial.print("init_hap_storage");
 
    
  File fsDAT=SPIFFS.open(pair_file_name, "r");
 if(!fsDAT){
   Serial.println("Failed to read pair.dat");
   return;
 }
  int size=hap_get_storage_size_ex();
  char* buf=new char[size];
  memset(buf,0xff,size);
 int readed=fsDAT.readBytes(buf,size);
 // Serial.print("Readed bytes ->");
//  Serial.println(readed);
  hap_init_storage_ex(buf,size);
  fsDAT.close();
  delete []buf;
}

void storage_changed(char * szstorage,int bufsize){

  SPIFFS.remove(pair_file_name);
  File fsDAT=SPIFFS.open(pair_file_name, "w+");
  if(!fsDAT){
    Serial.println("Failed to open pair.dat");
    return;
  }
  fsDAT.write((uint8_t*)szstorage, bufsize);

  fsDAT.close();
}
//can be used for any logic, it will automatically inform Apple about state changes
void set_led(bool val){
   Serial.println("set_led"); 
  digitalWrite(led_gpio, val?HIGH:LOW);
  //we need notify apple about changes
  
  if(hapservice){
    Serial.println("notify hap"); 
    //getting on/off characteristic
    homekit_characteristic_t * ch= homekit_service_characteristic_by_type(hapservice, HOMEKIT_CHARACTERISTIC_ON);
    if(ch){
        Serial.println("found characteristic"); 
        if(ch->value.bool_value!=val){  //wil notify only if different
          ch->value.bool_value=val;
          homekit_characteristic_notify(ch,ch->value);
        }
    }
  }
}

void led_callback(homekit_characteristic_t *ch, homekit_value_t value, void *context) {
    Serial.println("led_callback");
 set_led(ch->value.bool_value);
}`

alisadeghy, 2021-09-05:
One question, does this whole thing need internet connection to access anything from any server, like Apple or something?

BbIKTOP, 2021-09-05:
> One question, does this whole thing need internet connection to access anything from any server, like Apple or something?

Sorry for my intrusion, but yes, sure, as hap works thru the apple servers, internet connection is definitely required. Or, what do you think, how could your accessory update its state in your home.app then?

alisadeghy, 2021-09-05:
I don't know, maybe in a local communication between accessory and Apple device! I'm not that much into these Home Kit BTW, so it might be the problem, cause I live in Iran and we are banned from so many things over the internet!

alisadeghy, 2021-09-05:
To be clear I could pair a couple of times with exact same ESP8266 and internet connection and everything but suddenly this "Connection Close" thing started to show up!

Yurik72, 2021-09-05:
You don’t need internet connection for pairing, everything runs locally and not any communication to ‘Apple’ server , it just phone and esp building handshake….

alisadeghy, 2021-09-05:
Ok then it's not the issue 🤷🏻‍♂️

Yurik72, 2021-09-05:
Pairing itself not requires any internet connection, as well as normal communication  !!!l I think issue could be related to network environment , but exact answer we can find only sniffering (for instance using ware shark), communication broken exactly after 1 stage, means we need understand who is doing that,  if device replying yes or not.. any other logical validation are one the second an third stages

Yurik72, 2021-09-05:
See previous answer , maybe other things are changed , port blocking .. etc…

Sent from my iPhone

> On 5 Sep 2021, at 23:16, alisadeghy ***@***.***> wrote:
> 
> issue

alisadeghy, 2021-09-05:
I completely changed the network environment, made a hotspot via my laptop and connected ESP and phone to that, no luck.
Same thing happened.

alisadeghy, 2021-09-05:
Also started over on a fresh computer with fresh install of IDE and libraries, thought it might be a file corruption somewhere in other libraries or an interference, still doesn't work. I will get another ESP6288 and an ESP32, hopefully the problem will be solved. Thanks for the time and if any ring bells please let me know. 🙏🏻

alisadeghy, 2021-09-26:
Alright, I tried 3 more ESP8266s and they work just fine, I don't know what happened to this board but it won't work with this library anymore, although it works with another Homekit library and every other sketch!
Any tip to make it work? Like flashing with new firmware or something?