Yurik72/ESPHap #85: /setup doesn’t appear to fully work

Vweston, 2021-03-26:
I compiled the AdvancedLED example. When I go to /setup from my web browser I get the following response:

{"setup_uri":"X-HM://005277SG7YK72","pin":"111-11-111"}

Instead of a qrcode.

Yurik72, 2021-03-27:
Could you provide me a log from ESP starting from rebooting until this moment

Thanks

Vweston, 2021-03-27:
Here is a log from re-booting the 8266 device.

`#$`⸮c⸮⸮ls⸮`⸮⸮o⸮*WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
<form action='/info'    method='get'><button>Info</button></form><br/>
<form action='/exit'    method='get'><button>Exit</button></form><br/>
 
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: B******N***
*WM: [3] Using Password: L*****S****
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.59
*WM: [3] unloading 
>>> HomeKit: init_storage_ex size 0x1c1
>>> Home Integration: hap_services added chararacteristic  0: 23 
>>> Home Integration: hap_services added chararacteristic  1: 25 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Free heap: 28280
>>> HomeKit: Starting server
>>> HomeKit: storage init 0
>>> HomeKit: Using existing accessory ID: 7F:E9:13:47:9C:FC
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 21936
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: EspLed-7FE9, IP: 192.168.1.59
>>> HomeKit: Init server over `

I also noted that if I use /browse it shows the file setup.html with 0 (zero) bytes. 

I have not tried this on a ESP32 device yet, but if you need me I can.

thanks for this project. It is starting to really come together.

Vweston, 2021-03-27:
I did try compiling it for a nodeMCU-32s, but was unable to get it compiled with the following error

`

In file included from /var/folders/_3/vxnqz7t138z2bkfd27wb91gh0000gq/T/arduino_modified_sketch_200835/EspHapAdvancedLed.ino:17:0:
/Users/vweston/Documents/Arduino/libraries/ESPHap/hapweb/hap_webserver.hpp: In function 'void saveFileBrowse()':
/Users/vweston/Documents/Arduino/libraries/ESPHap/hapweb/hap_webserver.hpp:433:59: error: cannot convert 'const char*' to 'const __FlashStringHelper*' for argument '1' to 'void saveFile(const __FlashStringHelper*, const __FlashStringHelper*)'
  saveFile(FPSTR(FILE_BROWSE_FILE), FPSTR(FILE_BROWSE_HTML));
                                                           ^
/Users/vweston/Documents/Arduino/libraries/ESPHap/hapweb/hap_webserver.hpp: In function 'void saveSetup()':
/Users/vweston/Documents/Arduino/libraries/ESPHap/hapweb/hap_webserver.hpp:446:47: error: cannot convert 'const char*' to 'const __FlashStringHelper*' for argument '1' to 'void saveFile(const __FlashStringHelper*, const __FlashStringHelper*)'
  saveFile(FPSTR(FILE_SETUP), FPSTR(SETUP_HTML));
                                               ^
/Users/vweston/Documents/Arduino/libraries/ESPHap/hapweb/hap_webserver.hpp: In function 'void saveIndex()':
/Users/vweston/Documents/Arduino/libraries/ESPHap/hapweb/hap_webserver.hpp:450:57: error: cannot convert 'const char*' to 'const __FlashStringHelper*' for argument '1' to 'void saveFile(const __FlashStringHelper*, const __FlashStringHelper*)'
   saveFile(FPSTR(FILE_INDEX), FPSTR(INDEX_HTML_INTERNAL));
                                                         ^
/var/folders/_3/vxnqz7t138z2bkfd27wb91gh0000gq/T/arduino_modified_sketch_200835/EspHapAdvancedLed.ino: In function 'void setup()':
EspHapAdvancedLed:86:35: error: cannot convert 'const char*' to 'const __FlashStringHelper*' for argument '1' to 'void set_indexhml(const __FlashStringHelper*)'
     set_indexhml(FPSTR(INDEX_HTML));
                                   ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/vweston/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
cannot convert 'const char*' to 'const __FlashStringHelper*' for argument '1' to 'void set_indexhml(const __FlashStringHelper*)'
`

hope this helps.

-vw

Yurik72, 2021-03-28:
On ESP32 i do not have compilation problem.
So, please check 
1. You are using latest version of ESPHap library v1.1 or better latest commit
2. You are using ESP32 board library 1.0.5

As well please read suggestions on isssue #51  .
I would say there is not a problem with that. SPIFFS system are used for this purpose, therefore  right partition should be selected. SPIFFS should be formatted. It coud happens with same device with previous firmware before. 

So, 
1. Completelly clean (ERASE) ESP
2. Upload any data to SPIFFS (It will be formatted) 
3. /browse should works and you should see your ESP file system

Vweston, 2021-03-29:
I gave up compiling for ESP32 after installing the Arduino IDE 1.8.13 on a clean install of MacOS Catalina. I used the Board Manager to install ESP8266 (v 2.7.4) & ESP32 (v 1.0.5 & even v 1.0.6). I installed just the necessary libraries (WifiManager v 0.16.0) (EspHap v 1.1.0) No matter what specific ESP32 board I select I get a compiler terminate trying to load ESP8266Wifi libraries. I think the ESP32 problem is an Arduino IDE issue where it isn’t recognizing the compiler directives. I’ll deal with the ESP32 later, for Now I’’m working on ESP8266 D1 Mini boards.

As for the ESP8266, that compiles correctly, but setup is still not displaying anything but what I shared earlier. I also flashed it to a D1 Mini with a Erase Flash option. Still no luck. The setup.html file is created, but is zero bytes according to /browse option. Otherwise it appears to work great.

-- VW

Vweston, 2021-03-30:
after playing around with this, I was able to get a reliable result. Since setup.html was always a zero byte file I started there. I revised saveFile() in hap_webserver.hpp. here is the code I revised in hap_webserver.hpp

`//void saveFile(const __FlashStringHelper* name, const __FlashStringHelper* content) {
//	String data = FPSTR(content);
//	String filename = name;
//	if (!filename.startsWith("/")) filename = "/" + filename;
//	//INFO("Save file %s", filename.c_str());
//	//INFO("Save file content %s", data.substring(1,90).c_str());
//	File fb = SPIFFS.open(filename, "w+");
//	if (fb)
//		fb.print(data);
//	else
//		ERROR("Save file failed %s", filename.c_str());
//	if (fb)
//		fb.close();
//}
    
void saveFile(const __FlashStringHelper* name, const __FlashStringHelper* content) {
    //String data = FPSTR(content);
    String filename = name;
    if (!filename.startsWith("/")) filename = "/" + filename;
    //INFO("Save file %s", filename.c_str());
    //INFO("Save file content %s", data.substring(1,90).c_str());
    File fb = SPIFFS.open(filename, "w+");
    if (!fb)
        ERROR("Save file failed %s",filename.c_str());
    else {
        fb.print(content);
        //Serial.println(fb.size());  //read the file size here
        fb.close();
        //INFO("%s File Saved", filename.c_str());
    }
}`

what I found was that data was always empty with the saveSetup() call therefore making setup.html always empty. by eliminating ` //String data = FPSTR(content);` and just printing content directly it appears to always work. I think because `__FlashStringHelper*` has already occurred. Not sure on that.

here is my saveSetup()
`void saveSetup() {
	saveFile(FPSTR(FILE_SETUP), FPSTR(SETUP_HTML));
};`

It seams to work perfectly for index.html, setup.htm and file browse.html at least for ESP8266 boards. It should be ok for ESP32 but not tested.

/setup.html displays correctly with the appropriate qrcode & scans & adds the accessory properly now.

as for `/setup` it appears to be doing what is says.

`{"setup_uri":"X-HM://00527Y91QYK72","pin":"111-11-111"}`

just like it was told to do.

`  server.on(PSTR("/setup"),[]() {server.send(200,  FPSTR(TEXT_PLAIN), get_setup_json());});  \
`

it returns correctly. I just don't know how to get /setup to return /setup.html so that the pairing qrcode displays.

Hope this helps

Yurik72, 2021-03-30:
Ok, I found an error, this is happens due to low memory on ESP.   
File setup is 26k. If free heap is less is not written... :)  This is my mistake. currently I have implemented proper way to save file
However is not finish with final logic. When file should be overwritted. Always , If not exists.....
You can check and help me with pull request

Vweston, 2021-03-31:
I've updated my EspHap library with your recent changes and am currently using them in my tests. So far its working great. I'll keep testing but haven't seen any issues so far. Great work!! I'll let you know if I find any issues.

closed 2021-03-31 by Yurik72