Yurik72/ESPHap #86: ESP reboot on pairing

lyuq-a, 2021-03-30:
ESP rebooting during pairing when I try to set more than 6 accessories on the one ESP32.

Here is pairing log:
init_hap_storageReaded bytes ->1409
>>> HomeKit: init_storage_ex size 0x581
>>> Home Integration: hap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
>>> Home Integration: hap_services added chararacteristic  1: 00000011-0000-1000-8000-0026BB765291 
>>> Home Integration: add hap_new_switch_service_as_accessory as accessory
>>> Home Integration: add_hum as accessory , next accessory 4
>>> Home Integration: add_temp as accessory , next accessory 5
>>> Home Integration: add_hum as accessory , next accessory 6
>>> Home Integration: add_temp as accessory , next accessory 7
>>> Home Integration: add_hum as accessory , next accessory 10
>>> Home Integration: add_temp as accessory , next accessory 11
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 39:E5:5F:05:07:8C
>>> HomeKit: Configuring mDNS
>>> HomeKit: Got new client connection: 55
>>> HomeKit: [Client 55] Pair Setup 
>>> HomeKit: [Client 55] Pair Setup Step 1/3
>>> HomeKit: [Client 55] new pairing content
>>> HomeKit: started crypto
>>> HomeKit: Free heap: 157628
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: [Client 55] send_tlv_response
>>> HomeKit: [Client 55] send_tlv_response done
>>> HomeKit: [Client 55] Pair Setup 
>>> HomeKit: [Client 55] Pair Setup Step 2/3
>>> HomeKit: [Client 55] Pair Setup 
>>> HomeKit: [Client 55] Pair Setup Step 3/3
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x3ffd1ad4  PS      : 0x00060a30  A0      : 0x800d8edd  A1      : 0x3ffd6110  
A2      : 0x3ffd1ad4  A3      : 0x3ffbdd8a  A4      : 0x3ffd612c  A5      : 0x00000000  
A6      : 0xffffffff  A7      : 0xffffffff  A8      : 0x800d2770  A9      : 0x3ffd60f0  
A10     : 0x3ffbdd8a  A11     : 0x00000581  A12     : 0x0000001d  A13     : 0x3ffd61b4  
A14     : 0x00000014  A15     : 0x14ba342d  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x3ffd1ad4  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x3ffd1ad4:0x3ffd6110 0x400d8eda:0x3ffd6130 0x400d928f:0x3ffd6150 0x400d81c1:0x3ffd61e0 0x400d88a5:0x3ffd6350 0x400d4b93:0x3ffd6370 0x400d8bdb:0x3ffd63c0 0x40089f22:0x3ffd6420

Backtrace decoding gives the next:
Decoding stack results
0x400d8eda: on_storage_change at /Users/lyuq/OneDrive/Arduino/libraries/ESPHap/storage.c line 82
0x400d928f: homekit_storage_add_pairing at /Users/lyuq/OneDrive/Arduino/libraries/ESPHap/storage.c line 365
0x400d81c1: homekit_server_on_pair_setup at /Users/lyuq/OneDrive/Arduino/libraries/ESPHap/server.c line 1427
0x400d88a5: homekit_server_on_message_complete at /Users/lyuq/OneDrive/Arduino/libraries/ESPHap/server.c line 3044
0x400d4b93: http_parser_execute at /Users/lyuq/OneDrive/Arduino/libraries/ESPHap/http_parser.c line 1927
0x400d8bdb: homekit_server_task at /Users/lyuq/OneDrive/Arduino/libraries/ESPHap/server.c line 3167
0x40089f22: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Here is a part of my code from "setup()" which leads to ESP32 reboot:

`
  init_hap_storage();
  set_callback_storage_change(storage_changed);

  hap_setbase_accessorytype(homekit_accessory_category_bridge);
  hap_set_device_setupId((char*)"TR01");

  hap_initbase_accessory_service("TechRoom", "test", "0.3", "TechRoom", "0.0.3");

  techBridge = hap_add_temperature_service("TechBridge");
  bathVent = hap_new_switch_service_as_accessory("BathFan", fan_callback, (void*)&bathVentPin);
  stairsLight = hap_add_relaydim_service_as_accessory(homekit_accessory_category_lightbulb,
                                                          "StairsLight", led_callback, (void*)&stairsLightPin);
  outHum = hap_add_hum_as_accessory(homekit_accessory_category_thermostat,"OutHumidity");
  outTemp = hap_add_temp_as_accessory(homekit_accessory_category_thermostat,"OutTemp");
  hallHum = hap_add_hum_as_accessory(homekit_accessory_category_thermostat,"HallHumidity");
  hallTemp = hap_add_temp_as_accessory(homekit_accessory_category_thermostat,"HallTemterature");
  outLit = hap_add_light_service_as_accessory(homekit_accessory_category_sensor,"OutLight", NULL, NULL);
  outPres = hap_add_light_service_as_accessory(homekit_accessory_category_sensor,"Pressure", NULL, NULL);
  if (homekit_is_paired())
  {
    homekit_characteristic_t * ch = homekit_service_characteristic_by_type(stairsLight, HOMEKIT_CHARACTERISTIC_ON);
    ch = homekit_service_characteristic_by_type(stairsLight, HOMEKIT_CHARACTERISTIC_BRIGHTNESS);
    INIT_CHARACHTERISTIC_VAL(int,ch,25);
    led_callback(ch, value1, NULL);
    ch = homekit_service_characteristic_by_type(bathVent, HOMEKIT_CHARACTERISTIC_ON);
    INIT_CHARACHTERISTIC_VAL(bool,ch,false);
    fan_callback(ch, value1, NULL);
  }

  hap_init_homekit_server();
`

If I comment lines starting with "hallHum" and "hallTemp" everything work fine.

Yurik72, 2021-03-31:
I have increased  max accessories, due to the comunity test
Currently ESP8266 allows up to 8 and ESP32 up to 45 , accesories on the same board
Just get latest commit

closed 2021-03-31 by Yurik72

lyuq-a, 2021-04-05:
Hello! Almost everything works well.
But I see strange system behavior in 8th accessory. The eighth (in the list) accessory in a few minutes after pairing becomes unsupported. It does not depend on the type of the accessory. I've tried to set temperature, humidity and light sensors as 8th accessory.

Yurik72, 2021-04-05:
Hi, 

I need two things 

1.	ESP32 or ESP8266
2.	Log, when this is happened

lyuq-a, 2021-04-05:
1. ESP32
2. Logs I'll make a little later.

lyuq-a, 2021-04-06:
First log. Humidity sensor is 8 accessory:

 ⸮:⸮⸮⸮⸮A ⸮.....WiFi connected
IP address: 
172.16.1.6
init_hap_storageReaded bytes ->3729
>>> HomeKit: init_storage_ex size 0xe91
>>> Home Integration: hap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
>>> Home Integration: hap_services added chararacteristic  1: 00000011-0000-1000-8000-0026BB765291 
>>> Home Integration: add hap_new_switch_service_as_accessory as accessory
>>> Home Integration: add_hum as accessory , next accessory 4
>>> Home Integration: add_temp as accessory , next accessory 5
>>> Home Integration: add_hum as accessory , next accessory 8
>>> Home Integration: add_temp as accessory , next accessory 9
>>> Home Integration: add_hum as accessory , next accessory 10
>>> Home Integration: add_temp as accessory , next accessory 11
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 0F:FD:49:22:B6:F0
>>> HomeKit: Configuring mDNS
LoRa reciever starting task...
  LoRa HALL sensors data recieved, Vcc = 3761
>>> HomeKit: Got new client connection: 55
>>> HomeKit: [Client 55] Pair Setup 
>>> HomeKit: [Client 55] Pair Setup Step 1/3
>>> HomeKit: [Client 55] new pairing content
>>> HomeKit: started crypto
>>> HomeKit: Free heap: 150880
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: [Client 55] send_tlv_response
>>> HomeKit: [Client 55] send_tlv_response done
>>> HomeKit: [Client 55] Pair Setup 
>>> HomeKit: [Client 55] Pair Setup Step 2/3
>>> HomeKit: [Client 55] Pair Setup 
>>> HomeKit: [Client 55] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 39A6D7E9-2114-4CEC-99D9-7051B24C395F
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 55] Successfully paired
>>> HomeKit: [Client 55] Closing client connection
>>> HomeKit: Got new client connection: 56
>>> HomeKit: [Client 56] Pair Verify Step 1/2
>>> HomeKit: [Client 56] Pair Verify Step 2/2
>>> HomeKit: [Client 56] Found pairing with 39A6D7E9-2114-4CEC-99D9-7051B24C395F
>>> HomeKit: [Client 56] Verification successful, secure session established
>>> HomeKit: [Client 56] Get Accessories
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit : [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Update Characteristics
>>> HomeKit: [Client 56] Get Characteristics
[... 115 lines cut ...]
>>> HomeKit: [Client 57] Verification successful, secure session established
>>> HomeKit: [Client 57] Get Accessories
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
>>> HomeKit: [Client 56] Get Characteristics
set_fan
notify hap
found characteristic
  LoRa HALL sensors data recieved, Vcc = 3761
set_fan
notify hap
found characteristic
>>> HomeKit: [Client 56] Remove Pairing
>>> HomeKit: Removed pairing with 39A6D7E9-2114-4CEC-99D9-7051B24C395F
>>> HomeKit: Last admin pairing was removed, enabling pair setup
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 56] Closing client connection
  LoRa HALL sensors data recieved, Vcc = 3761