Yurik72/ESPHap #87: Failed to compute SRP shared secret (code -2)

IT-freak-Jake, 2021-04-05:
Hi, I have a problem with connecting to Home Kit. I'm using ESP-12E. As you can see in logs ESP can't compute SRP secret.

17:02:35.468 -> init_hap_storage>>> HomeKit: init_storage_ex size 0x1c1
17:02:35.468 -> >>> Home Integration: hap_services added chararacteristic  0: 23 
17:02:35.468 -> >>> Home Integration: hap_services added chararacteristic  1: 25 
17:02:35.468 -> >>> Home Integration: homekit_is_paired 0
17:02:35.468 -> >>> HomeKit: Update the CPU to run at 160MHz
17:02:35.468 -> >>> HomeKit: Free heap: 18872
17:02:35.515 -> >>> HomeKit: Starting server
17:02:35.515 -> >>> HomeKit: storage init 0
17:02:35.515 -> >>> HomeKit: Using existing accessory ID: AD:B5:A6:88:9E:0D
17:02:35.515 -> >>> HomeKit: find pairings
17:02:35.515 -> >>> HomeKit: Preinit pairing context
17:02:35.515 -> >>> HomeKit: Using user-specified password: 111-11-111
17:02:35.515 -> >>> HomeKit: Free heap: 12528
17:02:41.586 -> >>> HomeKit: Calculating public key len=384
17:02:44.697 -> >>> HomeKit: Calculating public key result 0
17:02:44.697 -> >>> HomeKit: arduino_homekit_preinit success
17:02:44.697 -> >>> HomeKit: Configuring mDNS
17:02:44.743 -> >>> HomeKit: Wifi Sleep type:2
17:02:44.743 -> >>> HomeKit: Set Wifi Sleep type :0
17:02:44.743 -> >>> HomeKit: MDNS.begin: TestLight-ADB5, IP: 192.168.2.187
17:02:44.743 -> >>> HomeKit: Init server over
17:02:50.580 -> >>> HomeKit: WiFiServer receives a new client (current 0, max 8)
17:02:50.580 -> >>> HomeKit: Got new client connection: local 192.168.2.187:5556, remote 192.168.2.209:49772
17:02:50.580 -> >>> HomeKit: [Client 1073712788] Got client content from the cache 0
17:02:50.580 -> >>> HomeKit: [Client 1073712788] new client accepted
17:02:50.580 -> >>> HomeKit: [Client 1073712788] Pair Setup Step 1/3
17:02:51.648 -> >>> HomeKit: [Client 1073712788] Pair Setup Step 2/3
17:02:51.648 -> >>> HomeKit: [Client 1073712788] Computing SRP shared secret
17:03:00.652 -> !!! HomeKit: [Client 1073712788] Failed to compute SRP shared secret (code -2)
17:03:00.652 -> >>> HomeKit: [Client 1073712788] Pair Setup Step 1/3
17:04:03.671 -> >>> HomeKit: [Client 1073712788] Disconnected!
17:04:08.687 -> >>> HomeKit: [Client 1073712788] Closing client connection
17:04:08.687 -> >>> HomeKit: Free saved_preinit_pairing_context
17:04:08.687 -> >>> HomeKit: [Client 0] Clear the pairing context
17:04:08.687 -> >>> HomeKit: [Client 0] client content free
17:04:08.687 -> >>> HomeKit: Preinit pairing context
17:04:08.687 -> >>> HomeKit: Using user-specified password: 111-11-111
17:04:08.734 -> >>> HomeKit: Free heap: 10704
17:04:20.847 -> >>> HomeKit: Calculating public key len=384
17:04:27.132 -> >>> HomeKit: Calculating public key result 0
17:04:27.132 -> >>> HomeKit: arduino_homekit_preinit success

Yurik72, 2021-04-05:
There is a problem of available memory (free heap)

As you can see all my examples has ~28k when started. In your case 

17:02:35.468 -> >>> HomeKit: Free heap: 18872

 

This is not enough, check your sketch…

IT-freak-Jake, 2021-04-06:
Ok, it is a problem. Thanks for the help

closed 2021-04-06 by IT-freak-Jake