Yurik72/ESPHap #88: After removing accessory from Apple Home, cannot add it again

sw1970, 2021-04-15:
Hi, thank you for an excellent build. The code examples and pairing work fine. But if I then remove the accessory from Apple Home, I cannot add it to Apple Home again without re-flashing the ESP. After removing the accessory from Apple Home, when I try to add the accessory again, the app stays on "Select an Accessory to add..." and the device does not appear, even after power downs and Home app restarts. Is there a way to pair the accessory again without having to re-flash? I appreciate that I might need to arrange some way of externally triggering a deletion of the pairing data (e.g. using a momentary switch as a reset) - if this is the case, I would be grateful for your advice on the code changes to achieve the wiping of the pairing data. Many thanks.

Yurik72, 2021-04-16:
Hi, i suppose that is not a bug, Apple allows to remove devices where their offline. At this moment device keep pairing and consider that is paired therefore is not appears for new pairing.  Pairing information are stored in the  pair.dat file under SPIFFS system
Completelly removing this file means, device lost pairing and will be ready for new one. You can remove this file, by using internal file browser it's present in most examples. Or you can implement your own "reset" function, for instance when some button is long pressed. You just need to remove this file and restart device.

Yurik72, 2021-04-17:
I'm closing issue due to the previous explanation

sw1970, 2021-04-17:
Hi, thanks - this idea worked well. Here's what I did:

In the setup:

const int switch_gpio = 4;
pinMode(switch_gpio, INPUT); // This GPIO is normally low

And in the loop:
if (digitalRead(switch_gpio) == 1) {SPIFFS.remove(pair_file_name); } // If button pressed, delete HomeKit config file

If the user removes from HomeKit, the device is then "ghosted", can use this button connected to a GPIO to do a "reset" and can then pair again. Needs a power cycle as well. Also tried using the flash button on GPIO0 and works just as well, and doesn't use up a valuable GPIO.

closed 2021-04-17 by sw1970