Yurik72/ESPHap #89: Compilation error function base64_encode

joaquinpolom, 2021-04-16:
macOS Big Sur , Arduino 1.8.13, All libraries updated. Board ESP32 version 1.0.6.
Compiling the example EspHapLed I get the following error
Deleting the files esphap_base64.h and esphap_base64.c don't solve the problem
with the version 1.0.5 of ESP32 there are another errors about cJON

/Users/../Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libwpa_supplicant.a(base64.o): In function `base64_encode':
Multiple libraries were found for "WiFi.h"
 Used: /Users/..../Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /Users/../Arduino/libraries/WiFiNINA
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/wpa2/utils/base64.c:31: multiple definition of `base64_encode'
libraries/ESPHap/esphap_base64.c.o:/Users/../Arduino/libraries/ESPHap/esphap_base64.c:39: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.

joaquinpolom, 2021-04-16:
Deleting the function base64_encode in the file esphap_base64.c should solve the problem.

Yurik72, 2021-04-17:
This is already solved #79 , just get latest commit

closed 2021-04-17 by Yurik72