Yurik72/ESPHap #90: Platformio compilation

beckmx, 2021-04-17:
I was trying to migrate a project from arduino to platformio, but I am getting this issue and no matter what I do, looks like the wolfssl or wolfSSL definition cannot be found not in the libs or as a dependency lib, any suggestions where else to look for?

Yurik72, 2021-04-18:
Hi I did migration some of the project, please have a look example  https://github.com/Yurik72/IoTManager

You should have wolfssl under lib folder

closed 2021-04-18 by Yurik72

beckmx, 2021-04-19:
Just as a complementary doc, I needed to include these flags to make it compilable:
```
-DESP32
	-DARDUINO
	-Wno-error=maybe-uninitialized
	-Wno-error=uninitialized
	-Wno-error=implicit-function-declaration
	-Wno-error=misleading-indentation
	-Wno-error=return-type
	-Wno-error=parentheses
	-Wno-error=stringop-overflow
	-Wno-error=comment
```
And modified the include:
`#include <hapfilestorage/hapfilestorage.hpp>`

Looks like it was using the backward slash in some of the examples