Yurik72/ESPHap #91: ESPHAPCamera example

okanduzyel, 2021-04-25:
Hi,

esp32 camera example appears security accessory in homekit and unfortunately, it never passes pairing section, although I enter 11111111 code.

Yurik72, 2021-04-25:
First please use latest version/commit, than attach a log, I did never see a problem in 100 devices,

okanduzyel, 2021-04-25:
- First, is it homekit IP camera?

- I use v1.1.0
- Required that solution #89, otherwise, it doesn't compile. 
- I copied wolfSSL into the library folder and inside the ESPHAP folder.
- Homekit recognizes that it is an alarm.
- When it is at the pairing, the phone infinitely tries to pair but no answer from esp32.
- At this moment, there isn't any log output that appears.

Yurik72, 2021-04-27:
I didn’t catch about the serial log and why they can be collected, if device appears for the pairing , looks like a progress, but log should clear shows the problem during the pairing, for instance camera modules very sensitive for power, you can see that WD burn out is turning off, I found this is a common problem,
So beside of the log , try device with different (100% guaranteed) power source

okanduzyel, 2021-04-27:
> * First, is it homekit IP camera?

??

okanduzyel, 2021-04-27:
Please give more information about that project. 

- First, Is it homekit IP camera? If the answer is yes, Could you say snapshot resolution and streaming resolution?
- Could you share a video or instruction on how to build and flash correctly?

Yurik72, 2021-04-27:
No is not IP HomeKit camera, it’s only motion detection based on the camera, so you on the HomeKit you see motion detection, this is described in the reader