Yurik72/ESPHap #93: ESP32 not discoverable by iPhone

henrixapp, 2021-05-26:
Hi,

I am just trying out this library to build a airquality sensor with an esp32. I started out with the EspHap_AirQuality example and adjusted it to use hapwebserver. It connects to my wifi and the website is working. But I can not discover it via iPhone or iPad.

The log states 

```
IP address: 
xxx.xxx.xxx.xxx
Free heap: 200448
>>> HomeKit: init_storage_ex size 0x581
>>> Home Integration: hap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
>>> Home Integration: hap_services added chararacteristic  1: 00000095-0000-1000-8000-0026BB765291 
>>> Home Integration: hap_services added chararacteristic  2: 00000093-0000-1000-8000-0026BB765291 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 45:2D:6A:FD:42:DE
>>> HomeKit: Configuring mDNS
Setting web server
>>> HomeKit: Save saveFileChunked file /index.html
>>> HomeKit: Save saveFileChunked file /setup.html
Web site http://xxx.xxx.xxx.xxx
File system http://xxx.xxx.xxx.xxx/browse
Update http://xxx.xxx.xxx.xxx/update
``` 

Afterwards it just outputs the PPM, but is not discoverable by the Home App on the devices. I can see it via mdns browser on chrome.

Many thanks in advance for this awesome library!

Best,
Henrik

Yurik72, 2021-05-26:
Hi, is not completelly clear a problem, what you mean not discoverable, 
Based on the piece of logs, provided above, device is not paired yet
>>> Home Integration: homekit_is_paired 0

Means, you need pair first. 

What you see by mdns browser ?

henrixapp, 2021-05-26:
>What you see by mdns browser ?

I can see the entry for the ip of my esp32 in the `_hap._tcp` section.

With "not discoverable", I mean that when I try to pair it, it does not show up.

Thank you for your quick response!

Yurik72, 2021-05-26:
Ok, now more clear.
Looks strange due to mDNS works and shows records.  I suppose you are doing right by selecting "Add device Manually" after Add device pressed on the Home App ?
If your Apple device connected to the same Wifi Router ? With the same IP range ?
As well try to restart Home App on the Apple...
Let me know then

henrixapp, 2021-05-26:
Yes, I use the "Add device Manually". And the devices are connected to the same wifi  router and should be in range (`xxx.xxx.xxx.224` (ESP32) vs `xxx.xxx.xxx.141` (iPhone)). The subnet mask is `255.255.248.0`. Don't know if this is a problem.

henrixapp, 2021-05-26:
So when I run `dns-sd -Z _hap._tcp local.` on my Mac, I get

```
ES-452D._hap._tcp                               TXT     "sh=lWdLYQ==" "ci=10" "sf=1" "ff=0" "s#=1" "c#=1" "id=45:2D:6A:FD:42:DE" "pv=1.0" "md=EspHapAir"
```

according to https://stackoverflow.com/questions/49590546/how-does-homekit-find-nearby-devices `ff` should not be 0.

Yurik72, 2021-05-26:
This is not  case
From HAP specification 
”ff” Pairing Feature flags (e.g. ”0x3” for bits 0 and 1). Required if non-zero. See Table 5-4 (page 49).

More thousands devices works on this library with ff=0. Problem is different... 
Subnet mask should be OK , Briefly checking broadcast address  224.0.0.251 is allowed for this..

Try to restart Home App

henrixapp, 2021-05-26:
hmm ok. I tried to restart the app, the phone, no luck. how can I check if the broadcast address is allowed for this?

Yurik72, 2021-05-26:
I mean that subnet mask 255.255.248.0  allows to use address  224.0.0.251 (which is broadcast) as local addresses and Apple device should get this response via mDNS
In one enviropment i see similar problem, but device finally appears after 5 min..
Do you have other Apple Homekit devices on your Network working properly ?
As well you can use network sniffer to analyze traffic. Seems that ESP32 working properly... Due you able get right mDNS intoduction.   
Try to check another simple example ESPHapLed (for ESP32) 
If you are using latest commit for this library ? please check as well

henrixapp, 2021-05-26:
I don't have another homekit device (it is my first try), but other network discovery services (for printer, airplay etc) work fine. I am using the latest commit from master. The ESPHapLed also does not work. Maybe I try a fresh install on a mac later today(i first did not have esphap via git installed).

Yurik72, 2021-05-26:
Ok, keep me informed.  I didn't meet with the similar problem before.  As mentioned with more thousands devices

henrixapp, 2021-05-27:
Small update: I installed homebridge and it finds the sensor rightaway.... I can also add the homebridge via their qr code, but can not discover it via non scanning a code. The connection between iPhone and sensor still seems broken (I can't read out the airquality on the iphone, only in the homebridge UI, but the sensor is listed and saying that it does not respond).

Yurik72, 2021-05-28:
Looks like you have a problem on your network enviropment.  If you have possibility just try another one. For instance setup any new wifi router. 
Anyhow the HAP communication is quite simple. End devices introduces theyselfs by mDNS (seems is working on your env), 
Means, Apple devices sends multicast query (to port 5353) ,  devices should respond to this query.... 
And second steps, when devices IP is resolved, communication by HAP protocol uses port 5556 (in case of ESPHap), this  port is not fixed and introduced by mDNS. In case of ESPHap you can change that.
So, no other idea, than you have some closed ports on your router.

henrixapp, 2021-05-28:
Can I make esphap more verbose?

henrixapp, 2021-05-28:
So i made it more verbose.... The homebridge installation obviously contacting the esp32:

```
[130801][I][server.c:3236] homekit_server_accept_client(): >>> HomeKit: Got new client connection: 57

[130803][I][server.c:2254] homekit_server_on_update_characteristics(): >>> HomeKit: [Client 57] Update Characteristics

[130819][I][server.c:3236] homekit_server_accept_client(): >>> HomeKit: Got new client connection: 58

[130821][I][server.c:2123] homekit_server_on_get_characteristics(): >>> HomeKit: [Client 58] Get Characteristics

[130831][I][server.c:3176] homekit_client_process(): >>> HomeKit: [Client 58] Connection: close received, client will be disconnected

[130840][I][server.c:3190] homekit_server_close_client(): >>> HomeKit: [Client 58] Closing client connection

[132786][I][server.c:3236] homekit_server_accept_client(): >>> HomeKit: Got new client connection: 58

[132788][I][server.c:2051] homekit_server_on_get_accessories(): >>> HomeKit: [Client 58] Get Accessories

[132798][I][server.c:3176] homekit_client_process(): >>> HomeKit: [Client 58] Connection: close received, client will be disconnected

[132805][I][server.c:3190] homekit_server_close_client(): >>> HomeKit: [Client 58] Closing client connection

[132945][I][server.c:3236] homekit_server_accept_client(): >>> HomeKit: Got new client connection: 58

[132948][I][server.c:2123] homekit_server_on_get_characteristics(): >>> HomeKit: [Client 58] Get Characteristics

[132956][I][server.c:3176] homekit_client_process(): >>> HomeKit: [Client 58] Connection: close received, client will be disconnected

[132966][I][server.c:3190] homekit_server_close_client(): >>> HomeKit: [Client 58] Closing client connection
```

But no entries from my iPhone, even if I scan the code on "/setup.html".

henrixapp, 2021-07-08:
Problem seems to be network related. Works in different environment.

closed 2021-07-08 by henrixapp