Yurik72/ESPHap #95: Will adaptive lighting support be added

hassbian-ABC, 2021-05-27:


Yurik72, 2021-05-28:
I have spent a litle time, and can't find any official doc from Apple how adaptive ligh is supported. There is a new feathure in iOS14
Hovewer I found, that is already implemeted by HAP-NodeJS 
https://github.com/homebridge/HAP-NodeJS/blob/beta/src/accessories/Light-AdaptiveLighting_accessory.ts

Means device should provide ColorTemperature charachteristic....
So, answering to your question , yes will be implemented
When .... not any promises. Need a few experiments...

Yurik72, 2021-05-28:
I have added in the latest commit 3 functions 
	homekit_service_t* hap_add_ambientlight_service(const char* szname, hap_callback cb, void* context);
	homekit_service_t*  hap_add_ambientlight_service_as_accessory(int acctype, const char* szname, hap_callback cb, void* context);
	homekit_service_t* hap_new_ambientlight_service(const char* szname, hap_callback cb, void* context);

But not tested yet, 
They can be used in any RGB examples instead of hap_add_rgb......

So, if you test and give examples, I will appreciate